emsesp/EMS-ESP #80: RC35 allow selection between heating circuit 1 & 2

lobocobra, 2019-03-26:
Referring to closed topic #65

Enable the owner of a RC35 to select between heating circuit 1&2.
Heating circuit 2 is active, if you use a floor heating. 

The Circuit needs only to be selectable in the console, whereas it would make sense to add following commands:
```
rootThermostat[THERMOSTAT_daytemp]  =       _float_to_char(s, EMS_Thermostat.daytemp);
rootThermostat[THERMOSTAT_nighttemp]  =     _float_to_char(s, EMS_Thermostat.nighttemp);
rootThermostat[THERMOSTAT_holidaytemp]  =   _float_to_char(s, EMS_Thermostat.holidaytemp);
rootThermostat[THERMOSTAT_heatingtype]  =   _int_to_char(s, EMS_Thermostat.heatingtype);
rootThermostat[THERMOSTAT_circuitcalctemp]= _int_to_char(s, EMS_Thermostat.circuitcalctemp);
```

While "thermostat temp" is usefull to adjust the current temperature, above MQTT commands can be used in automatic scripts, to optimize the heating settings.

lobocobra, 2019-03-29:
Finally I could upgrade my code to 6b. :-)

I will do some final testing (remove the mqtt raw) and then I make a PR.

lobocobra, 2019-03-29:
Now I realized that the heating SW has a bug.... 
- EMS_Thermostat.setpoint_roomTemp to  "HK1MonitorMessage 0x48! byte 2" is only updated when the automatic mode swiches to the night. In this case 0x47 and 0x48 byte 2 show the right EMS_Thermostat.setpoint_roomTemp 

- When its switching back to day automatically then 0x48 byte 2 is not updated anymore and becomes 0.

10 | 00 | 3E | 7 |   | 1 | 2 | analog | °C | Raumtemperatur Soll
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

`
- But when it becomes day, this value goes to 0. If you then shortly change the mode to night and then to day and then to auto it is set correctly.
=> I did now insert an ugly hack to fix this (), but I will have to follow this up.

lobocobra, 2019-03-29:
I did create the PR now (but forgot to remove MWTT-raw)
=> for me the code works now quite well (day/night)

I do habe a BUDERUS GB125 heating with an RC35/MC10 with an attached floor heating and no boiler.

proddy, 2019-04-03:
I'll take a look, thanks

proddy, 2019-04-06:
merged into dev. Can you test if it works?

lobocobra, 2019-04-07:
Many thanks for the merge. I learned alot, the way you changed the code.
=> I fixed the discussed issues and now its working 100%.

lobocobra, 2019-04-07:
With the lates adjustment by @proddy, all is working now for me. :-)
=> If you have a similar setup, please test the lates DEV code and report to me the issues. I will try to fix them.

closed 2019-04-07 by lobocobra

Frederik-V, 2019-06-05:
Dear @lobocobra.

I'm pretty new on this fora and starting to explore the possibilities of the EMS-ESP.
On top, I'm a Home Assistant (HA) user in order to control my home domotica.

I was very enthusiastic to read on the fora that it is possible to control the 2 heating circuits.
I'm using HC1 for radiators in the bedrooms (level 1), HC2 as floor heating (level 0).

Is there a possibility to control both Heating Circuits at the same time with MQTT?
Meaning, having two thermostat control systems in the Home Assistant GUI.
Now you need to switch all the time between the Heating Circuits in order to read out the temperature settings. 

Thank you in advance for your answer.