emsesp/EMS-ESP #81: Disable boiler messages in console and MQTT, if no boiler is connected.

lobocobra, 2019-03-26:
Referring to closed topic #65

If you do not have a boiler connected to the heating, it would be nice, if the MQTT messages and the console messages for the boiler are not active. 

As proposed by you to unset the Boiler_type (value 0x00) the console and MQTT displaying could be prevented.

proddy, 2019-03-29:
Thinking about this, there must be a controller or something that is the UBA Master. In your case which device is this and what is heating up your floor heating?

lobocobra, 2019-03-29:
My floor-heating is connected to heating-circuit 2. I saw it even written in the manual, that floor-heatings are always HC2. 
=> What I believe and see in the admin menu of the heating,  is that heating-circuit 1 is prepared for a potential boiler, but not activated.

I will check in UBA master if I can read out something usefull, in order to know if a boiler is actually there or not.

lobocobra, 2019-04-14:
@proddy 
I am stuck in the code :(. I need to extract data beyond the 27 bytes, which are read with `thermostat read 47`
=> I  tried to extend the variable in  ems.h as follows: `#define EMS_MAX_TELEGRAM_LENGTH 48` 

Following the code I see no reason why it would not show in the telegram more than 27bytes. 
=> Has this to do with the EMS BUS itself?

proddy, 2019-04-14:
I limit it to 32 bytes as this was the max for my EMS bus. You can experiment by receiving more data by changing first in
  Line 25 in `ems.h`. The `EMS_MAX_TELEGRAM_LENGTH` 

if that doesn't work then we'll need to adjust the UART driver, so line 113 in `emsuart.cpp` change the 31 to something larger (EMS_MAX_TELEGRAM_LENGTH - 1 typically).

lobocobra, 2019-04-14:
Hi @proddy 

Many thanks for your fast response. 

The 2nd hint was exactly what I was looking for, but it seems that my bus is also limited at 32 bytes.
=> I changed both lines of codes but it did not help, but I left the changes as the `EMS_MAX_TELEGRAM_LENGTH - 1 ` makes sense anyway :-)

So in conclusion I do either need to adapt`void _process_RC35Set(uint8_t src, uint8_t * data, uint8_t length)` or I write a new function that responds on read requets of 5 bytes length (that would make work also my MWTT RAW thing ).

If it works all I will ask if, the changes make sense for the community. I extend the heating control in a way that my house optimizes the temp itself. For that I need to read more values from the heating.

proddy, 2019-04-14:
yes the best way is to read it in two 32 byte chunks, by setting the offset of the first to 0 and the 2nd to 31 for example

lobocobra, 2019-04-15:
I have now a fully working version of the code and added following variables (they can be changed by MQTT):
- ausschalthysterese
- einschalthysterese
- antipendelzeit
- kesselpumennachlauf
- maxvorlauf 
- auslegungstemp"
- roomoffset 
- minoutsidetemp
- housetype
- tempaveragebool

While all is working for a RC35, I must admitt. that had to hack the code and proddy will not be able  to accept it, the way I did it. I would be ready to invest time, if someone is really interested (as I was) to have those values. But making it nicely would mean to adapt the code of proddy strongly. 
=> The version I made I can easily update for my usage from time to time.
=> Of course you can also use it if you have a RC35 with heating circuit 2 and you want to be able to change all important values of your heating remotely.

lobocobra, 2019-04-18:
closed as it is solved for me and it seems not to be an issue for the community

closed 2019-04-18 by lobocobra