emsesp/EMS-ESP #84: Cannot communicate with boiler : Unknown command. Use ? for help.

JF002, 2019-03-30:
Hello,

My intallation is composed of a boiler Junkers Cerapur Smart A and a thermostat Junkers FR100.
The connection between the boiler and the thermostat is made with the BUS connector (Heatronic 3).

I ordered the EMS board from bbqkees, and I try to connect it on the EMS bus. I followed the instructions (pin mapping, pay attention to use 3.3V with ESP8266,...) and flashed my ESP8266 with version 1.6 of EMS-ESP.

The soft starts, I can connect to Wifi AP and Telnet. But the board does not seems to see the bus : the led is blinking, the `info` command returns 0 packet received, and no more information is displayed with command `log v`
More, when RX is correctly connected and serial port is disabled (`set serial off`), the following line is repeated very often :

`
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Error! Unable to read from EMS bus. Retrying in 20 seconds...
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
`

As far as I understand the code, this log message should only appear if I type something wrong on the console, but not if something wrong is received from the BUS.

Do you know what's going wrong? 
Is my boiler/thermostat not supported (electrically/protocol/software)?
Did I forget something?

Here are some logs:

Boot:

```
[WIFI] MODE AP --------------------------------------
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   62:01:94:3C:59:87
Enabling serial port output
[UART] Opened Rx/Tx connection
Started scan of EMS bus for known devices
[TELNET] Telnet connection established

* Connected to: EMS-ESP version 1.6.0
* Device is in AP mode with SSID ems-esp
*
* Commands:
*  ?=help, CTRL-D=quit telnet
*  set, system, reboot
*  info                                         show data captured on the EMS bus
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  types                                        list supported EMS telegram type IDs
*  queue                                        show current Tx queue
*  autodetect                                   detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    do a read on all type IDs
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
*  boiler comfort <hot | eco | intelligent>     set boiler warm water comfort setting

System Logging set to None
Error! Unable to read from EMS bus. Retrying in 20 seconds...

```

INFO:

```
EMS-ESP system stats:
  System logging set to Verbose
  LED is on, Silent mode is off
  0 external temperature sensor connected
  Thermostat is disabled, Boiler is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=no, Tx is not active, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=0

Boiler stats:
  Boiler type: <not enabled>
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: ?
  Selected flow temperature: ? C
  Current flow temperature: ? C
  Return temperature: ? C
  Gas: ?
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: ? %
  Burner current power: ? %
  Flame current: ? uA
  System pressure: ? bar
  System service code: ??
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: ? C
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
Error! Unable to read from EMS bus. Retrying in 20 seconds...
system
ESP8266 System stats:

 [APP] EMS-ESP version: 1.6.0
 [APP] MyESP version: 1.1.6
 [APP] Build timestamp: 2019-03-30 16:32:55
 [APP] Uptime: 112 seconds (00:01:52)
 [APP] System Load: 1%
 [WIFI] Device is in AP mode with SSID ems-esp
 [WIFI] WiFi MAC: 60:01:94:3C:59:87
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 3.0.0-dev(c0f7b44)
 [SYSTEM] CPU chip ID: 0x3C5987
 [SYSTEM] Core version: 2_5_0
 [SYSTEM] Boot version: 5
 [SYSTEM] Boot mode: 1
 [FLASH] Flash chip ID: 0x1640C8
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 425248
 [MEM] Max OTA size: 2715648
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24768
 [MEM] Stack: 1920
```

Thanks for your help!

proddy, 2019-03-30:
I'm assuming you're connected up to the wifi (set wifi_ssid, set wifi_password) to get out of the AP mode, and serial is off like you've shown. Also you're using a standard ESP8266 chip like a nodemcuv2 or wemos d1. Do you get the "unknown command" also when in telnet mode?

JF002, 2019-03-30:
@proddy Thanks for your help!
I'm using a Wemos D1 board (v2.3.0). 
The board is still in AP mode, I did not connect it to my home wifi. 
When the board is running, I connect to its AP, then Telnet to 192.168.4.1. 
All the messages from my OP come from the telnet console.

proddy, 2019-03-30:
When in AP mode the EMS bus link is disabled.

JF002, 2019-03-30:
> When in AP mode the EMS bus link is disabled.

That could explain my issue... I somehow missed this information when reading the doc. I'll try again soon by connecting the ESP to my home wifi, then!

JF002, 2019-03-31:
Ok, once EMS-ESP is connected to my home wifi, it receives values, shows them with command `info` and transfert them to my MQTT broker. 
Thanks for your help!

closed 2019-03-31 by JF002