emsesp/EMS-ESP #86: RC35  Set room temperature value and Current room temp value missing

majo551, 2019-04-04:
**Bug description**
*Set and Actual room temperature from RC35 thermostat is not read. The value provided in telnet using info  is "? "*

**Steps to reproduce**
*sending "info" in telnet *

**Expected behavior**
*real values to be seen for set and room temperature*

**Screenshots**

> Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 19:32:19 4/4/2019
  Mode is set to auto

**Additional context**
*Full output*

> info
EMS-ESP system stats:
  System logging set to Verbose
  LED is on, Silent mode is off
  0 external temperature sensor connected
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=350, # Tx telegrams=0, # Crc Errors=55

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline (ProductID:123 Version:04.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 57 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 44.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1420 times
  Warm Water active time: 32 days 19 hours 0 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 22.8 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 65 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Outside temperature: 16.1 C
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 7759 times
  Total burner operating time: 333 days 22 hours 0 minutes
  Total heat operating time: 301 days 3 hours 0 minutes
  Total UBA working time: 1334 days 18 hours 37 minutes

Solar Module stats:
    Collector temperature: 42.8 C
    Bottom temperature: 43.0 C
    Pump modulation: 0 %
    Pump active: off

Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 19:44:21 4/4/2019
  Mode is set to auto

proddy, 2019-04-04:
I assume you're using the 1st heating circuit. Do `set log v` and `thermostat read E3` and see if a telegram comes back.

majo551, 2019-04-04:
Hi
I have 2 circuits physically installed

the test:
thermostat read E3
Requesting type (0xE3) from dest 0x10
(00:16:18.816) Sending read of type 0xE3 to 0x10: telegram: 0B 90 E3 00 20
(CRC=D3), #data=1
(00:16:18.956) Thermostat -> me, type 0xE3 telegram: 10 0B E3 00 (CRC=73)

What is strange that when I set the thermostat to manual mode I receive the
temperature.
When switched back to Auto, the temp is gone

Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: 21.5 C
  Current room temperature: 22.0 C
  Thermostat time is 21:23:35 4/4/2019
  Mode is set to manual

Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 21:24:19 4/4/2019
  Mode is set to auto

Thanks
M.

proddy, 2019-04-04:
the `thermostat read E3` command fetches the temperatures from the RC35. In your case it returns nothing back which is why it shows up as '?'. Can you try two more things

`thermostat read 48` (for HC2)
to see if anything comes back

and the mother of all tests is `log t` and leave it for 1-2 minutes and see what messages are being broadcast by your RC35.

majo551, 2019-04-04:
00:50:47.275) Sending read of type 0x48 to 0x10: telegram: 0B 90 48 00 20
(CRC=4D), #data=1
(00:50:47.474) Thermostat -> me, type 0x48 telegram: 10 0B 48 00 05 03 00
00 DB F0 3C 00 00 00 00 00 00 11 05 00 (CRC=41), #data=16

log t

System Logging set to Thermostat only
(00:51:55.935) Thermostat -> Boiler, type 0x14 telegram: 10 88 14 00 03
(CRC=6E), #data=1
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices...
(00:52:00.999) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 01 00
00 DB 66 05 00 00 00 00 00 00 00 05 00 (CRC=54), #data=16
(00:52:01.524) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 25 2B
25 08 28 00 02 00 05 05 2D 01 01 04 3C 05 3C 01 FF 3C FF 0E 05 05 02 02
(CRC=45), #data=27
(00:52:01.953) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 04 15
04 38 0F 03 01 (CRC=5F), #data=8
(00:52:08.695) Thermostat -> 0x21, type 0xAC telegram: 10 21 AC 00 05 00 03
(CRC=0D), #data=3
(00:52:16.842) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 10 11 11
00 DC 00 E2 00 E1 7D 00 7D 00 (CRC=7A), #data=13
(00:52:17.684) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11
(CRC=30), #data=2
(00:52:31.596) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00
00 00 00 00 00 00 00 00 00 00 02 (CRC=9B), #data=14
(00:52:33.580) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 04 15
04 38 2F 03 01 (CRC=78), #data=8
(00:52:37.605) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 01 00
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=76), #data=16
(00:52:38.448) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00
00 00 (CRC=91), #data=4
(00:52:48.492) Thermostat -> all, type 0x48 telegram: 10 00 48 00 05 03 00
00 DC F0 3C 00 00 00 00 00 00 11 05 00 (CRC=63), #data=16
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices...
(00:53:00.723) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 01 00
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=1E), #data=16
(00:53:01.266) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 25 2B
25 08 28 00 02 00 05 05 2D 01 01 04 3C 05 3C 01 FF 3C FF 0E 05 05 02 02
(CRC=45), #data=27
(00:53:01.709) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 04 15
04 39 0F 03 01 (CRC=57), #data=8
(00:53:08.947) Thermostat -> 0x21, type 0xAC telegram: 10 21 AC 00 05 00 03
(CRC=0D), #data=3
(00:53:17.090) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 10 11 11
00 DC 00 E2 00 E2 7D 00 7D 00 (CRC=4A), #data=13
(00:53:17.940) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11
(CRC=30), #data=2
(00:53:33.827) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 04 15
04 39 2F 03 01 (CRC=70), #data=8
(00:53:37.851) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 01 00
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=76), #data=16
(00:53:38.694) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00
00 00 (CRC=91), #data=4
(00:53:48.806) Thermostat -> Boiler, type 0x10 telegram: 10 88 10 00 1B
(CRC=66), #data=1
(00:53:48.832) Boiler -> Thermostat, type 0x10 telegram: 08 10 10 00 33 4C
00 D6 93 01 0F 17 16 00 00 00 (CRC=9D), #data=12
(00:53:48.935) Thermostat -> Boiler, type 0x10 telegram: 10 88 10 0C 1B
(CRC=7E), #data=1
(00:53:48.974) Boiler -> Thermostat, type 0x10 telegram: 08 10 10 0C 33 4C
00 D6 93 01 11 16 23 00 00 00 (CRC=ED), #data=12
(00:53:49.065) Thermostat -> Boiler, type 0x10 telegram: 10 88 10 18 1B
(CRC=56), #data=1
(00:53:49.113) Boiler -> Thermostat, type 0x10 telegram: 08 10 10 18 33 4C
00 D6 93 01 11 16 22 00 00 00 (CRC=AB), #data=12
(00:53:49.181) Thermostat -> Boiler, type 0x10 telegram: 10 88 10 24 1B
(CRC=2E), #data=1
[... 65 lines cut ...]
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=76), #data=16
Publishing boiler data via MQTT
(00:54:38.809) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00
00 00 (CRC=91), #data=4
(00:54:51.979) Thermostat -> all, type 0x48 telegram: 10 00 48 00 05 03 00
00 DC F0 3C 00 00 00 00 00 00 11 05 00 (CRC=63), #data=16
Publishing boiler data via MQTT
Calling scheduled data refresh from EMS devices...
(00:55:02.596) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 01 00
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=1E), #data=16
(00:55:05.959) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 04 15
04 3B 13 03 01 (CRC=37), #data=8
(00:55:09.687) Thermostat -> 0x21, type 0xAC telegram: 10 21 AC 00 05 00 03
(CRC=0D), #data=3
(00:55:17.592) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 10 11 11
00 DC 00 E2 00 E2 7D 00 7D 00 (CRC=4A), #data=13
(00:55:18.437) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11
(CRC=30), #data=2
(00:55:34.963) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 04 15
04 3B 30 03 01 (CRC=1C), #data=8
(00:55:38.478) Thermostat -> all, type 0x3E telegram: 10 00 3E 00 04 01 00
00 DC 66 05 00 00 00 00 00 00 00 05 00 (CRC=76), #data=16
Publishing boiler data via MQTT
(00:55:39.064) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00
00 00 (CRC=91), #data=4

majo551, 2019-04-04:
one more test:
thermostat mode 1
Setting thermostat mode to 1
Write to 0x10 was successful
(00:58:45.393) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 25 2B
25 08 28 00 01 00 05 05 2D 01 01 04 3C 05 3C 01 FF 3C FF 0E 05 05 02 02
(CRC=4A), #data=27
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:58:45.813) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 1B 64
64 00 (CRC=08), #data=4
(00:58:53.356) Thermostat -> all, type 0x48 telegram: 10 00 48 00 04 01 00
00 DB F0 3C 00 00 00 00 00 00 00 05 00 (CRC=6D), #data=16
(00:58:53.767) Thermostat -> all, type 0x3D telegram: 10 00 3D 00 01 25 2B
25 08 28 00 01 00 05 05 2D 01 01 04 3C 05 3C 01 FF 3C FF 0E 05 05 02 02
(CRC=B4), #data=27
(00:58:54.969) Thermostat -> all, type 0x3D telegram: 10 00 3D 1B 00 02 28
19 0A 01 00 FF 32 2D 00 F4 05 05 02 (CRC=F5), #data=15
info

Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: 18.5 C
  Current room temperature: 22.0 C
  Thermostat time is 22:02:49 4/4/2019
  Mode is set to manual

Calling scheduled data refresh from EMS devices...
(00:59:00.975) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 80 02 2B
00 DB 66 05 00 00 00 00 00 64 00 1B 00 (CRC=0A), #data=16
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
(00:59:01.522) Thermostat -> me, type 0x3D telegram: 10 0B 3D 00 01 25 2B
25 08 28 00 01 00 05 05 2D 01 01 04 3C 05 3C 01 FF 3C FF 0E 05 05 02 02
(CRC=4A), #data=27
(00:59:02.070) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 04 16
04 03 0F 03 01 (CRC=FE), #data=8
Publishing boiler data via MQTT
Publishing boiler data via MQTT

thermostat read E3
(00:59:21.682) Thermostat -> me, type 0xE3 telegram: 10 0B E3 00 (CRC=73)

thermostat read 48
Publishing boiler data via MQTT
(01:02:07.967) Thermostat -> me, type 0x48 telegram: 10 0B 48 00 04 01 00
00 DB F0 3C 00 00 00 00 00 00 00 05 00 (CRC=05), #data=16

For Mode 1 the info comes back though reading E3 does not

št 4. 4. 2019 o 21:59 Majo <majo551@gmail.com> napísal(a):

proddy, 2019-04-05:
sorry, made a typo. It's 3E and not E3. But I can see the values are coming in nicely in the logs you sent.
So right now I'm not sure exactly what the problem is anymore. Are you saying if the thermostat mode is in auto and you do a 'refresh' the temperatures don't get updated? It would help if you make an easy reproducible test so you and I can find out what is not working. And look at the code too and submit a pull request when any necessary changes.

majo551, 2019-04-05:
Hi
yes, my observation is that when the mode is Auto, it does not return
temperatures in info
When switching between modes, the temperature is sent in info.

I found one note regarding this similar issue in other issues:
https://github.com/proddy/EMS-ESP/issues/80#issuecomment-478093587

This might be something to do with the RC35 software bug and the solution
might be switching modes ?(:-) before checking temperatures?

Mode Auto:
thermostat read 3E
(04:47:25.440) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 01 *00
00 D8* 66 3C 00 00 00 00 00 00 00 05 00 (CRC=B3), #data=16

INFO:
Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 11:57:20 5/4/2019
  Mode is set to auto

I wonder why at least the current temp is not shown, as the data payload
says: 00D8 = 216= 21.6 C.  For the set temp it is 00, but current temp
value is available. Info shows "?" despite existing temp

*CHANGING MODE TO 1:*
thermostat read 3E
(04:52:02.562) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 80 02* 2B
00 D8* 66 3C 00 00 00 00 00 64 00 1F 00 (CRC=E5), #data=16

BOTH SET and CURRENT TEMP returned

*INFO:*
*Thermostat stats:*
*  Thermostat type: RC35 (ProductID:86 Version:21.08)*
*  Setpoint room temperature: 21.5 C*
*  Current room temperature: 21.6 C*
*  Thermostat time is 12:00:21 5/4/2019*
*  Mode is set to manual*

*CHANGINGMODE TO 2:*
thermostat read 3E
(04:53:41.566) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 04 01* 00
00 D8* 66 3C 00 00 00 00 00 00 00 1E 00 (CRC=85), #data=16

*Now again , in Auto mode the Set temp is gone, but the current is
available*

*but INFO is ? for both*

*Thermostat stats:*
*  Thermostat type: RC35 (ProductID:86 Version:21.08)*
*  Setpoint room temperature: ? C*
*  Current room temperature: ? C*
*  Thermostat time is 12:03:12 5/4/2019*
*  Mode is set to auto*

lobocobra, 2019-04-05:
If you have 2 HC, then I would assume that your heating is on the 2nd one. 
=> The current firmware does only support HC1 for a RC35

I had the same issues like you and solved them by adapting the firmware (PR pending). Meanwhile you might want to give a try to my changes, which can also be found here....
https://github.com/lobocobra/EMS-ESP-HC2

For me all is now working using my firmware and internal commands "thermostat mode xxx" etc wrork again (and I added some additional commands).

proddy, 2019-04-06:
@lobocobra I've merged your changes into the latest dev. Thanks for the graet work.
@majo551 still not sure why the values aren't coming in. can you try again with the latest dev branch. I don't have an RC35 so cannot check

lobocobra, 2019-04-07:
@proddy many thanks for merging.

I shortly checked the new code and it works partially....
- publish new mqtt messages => woks!  displayed values are correct
- change temp by mqtt           => does not work
- thermostat temp                   => works! on console
- thermostat mode                  => works! on console 
- thermostat stats                    => does not work (don't show any values) 
- thermostat mqtt reg. mess.   => thermostat is not coming all 2 min, only when the values are changed
                                                        this is a problem if you work with OpenHab

I will check which part of the adapted code does the difference.

proddy, 2019-04-07:
thanks for testing. 

 **change temp by mqtt => does not work**  -  I confirm it works for me on a RC20. What do you see in the lof?

 **thermostat stats** - this also works for me and I don't see why it shouldn't for an RC35. What do you get printed exactly?

 **thermostat mqtt reg. mess** - so you want to always publish thermostat temperature values even if the values haven't changed? The logic now only sends if there is a difference since the last measurement.

lobocobra, 2019-04-07:
:-) While I still am searching for the reasons, let me meanwhile answer your questions....

- change temp by mqtt => function CMD_daytemp,nighttemp,holidaytemp look like they were not subscribed. 
- thermostat stats show a "?" and no values. The reason here is, that only 1 of the 2 values have in my setup a value as I have no "current room temparature" sensor.
- thermostat mqtt reg. mess 
   => I fully understand the logic behind your code and its in general good. But if you have with Openhab a webinterface, then you need to have the values confirmed once a while, if not you will get empty values after a Openhab restart until you re-change the temps, which happens maybe all 4 month?

lobocobra, 2019-04-07:
I finally found the problem and will make a new PR. :-)
=> I subscribed the new commands in MQTT
=> found what code sniplets were needed to have the thermostat stats working 
      (if you have a floor - heating guided by external temp (standard setup) then you do not have the temp sensor activated
=> By increasing the periodicity of the thermostat message over MQTT, openhab works again after a restart.

BTW.... I get some good programming lessons here by looking at what and how you corrected my sniplets. Many thanks for that. :-)

lobocobra, 2019-04-07:
@majo551
Can you test the latest DEV code made by proddy and tell us, what is working/not-working?

As you have a similar setup to me, I believe that almost everything should workd

majo551, 2019-04-14:
Hi
I installed the  EMS-ESP version 1.7.0b4
Info provided the temperatures now:
Thermostat stats:
  Thermostat type: RC35 (ProductID:86 Version:21.08)
  Setpoint room temperature: 21.5 C
  Current room temperature: 22.9 C
  Thermostat time is 11:49:26 14/4/2019
  Mode is set to auto

I have to say though that I had to switch my boiler to the factory reset
few days ago . So it might work also due to this reset.

 It seems that playing with telegram type 35 (DHW)  caused unwanted
behavior with my boiler warm water functionality. The boiler did not wan to
prepare the warm water after few telegrams sent. I noticed my boiler and
thermostat use to send a different length telegram than in the
documentation:
https://domoticproject.com/ems-bus-buderus-nefit-boiler/#The_EMS_Bus
if you look at the description of 0x35 and the example, first thing that I
noticed is a typo(?) when in the telegram it says 33 not 35.
Second thing was that this sequence did not start the water warming at all
in  an neither combinations of the  code types  (telegram type 35 or 33)
and values 35/39 for start

I even tried the built in shower functionality, and it seems for my Buderus
GB172  there might be a different telegram needed for this. Anyway after
playing with this telegram and the shower functionality  my water warming
stopped completely which led to my boiler manual  hard reset:-)
>>>>>>>>>>

info
[image: image.png]
>>>>>>>>>>>>
My Thermostat sends something like this (different to the documentation)
when I press DHW once button on RC35
Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 *11 33* (CRC=12),
#data=2
then I can find in the log :
0x30 -> Boiler, type 0x35 telegram: 30 08 35 00 00 00 00 (CRC=CE), #data=3
Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30),
#data=2

wheras the document says 0B 08 *33* 00 *39  (CRC)*

But to confirm the original question, I can see now the thermostat
temperatures in info.

M.

proddy, 2019-04-14:
the 33 vs 35 in that website is just a typo I think. So would it be useful to add this feature to this project too, like activate/deactivate the DHW once? I don't have an RC35 so never considered adding it. If so please create a new enhancement request.

Can we close this issue if you have the temp values coming in correctly now?

majo551, 2019-04-14:
yes, we can close the issue with the temp not showing

closed 2019-04-29 by proddy