emsesp/EMS-ESP #87: Home Assistant API (ESPHOME like)

TimoFriedri, 2019-04-05:
Hi,

do you have any plans to integrate the native Home Asisstant API. This works very nice as one can see with the ESPHome project and the according integration in HA.
For me the best part, you can skip MQTT, which I personally do not like so much.

Some Info:

https://esphome.io/components/api.html
https://www.home-assistant.io/components/esphome/

Greetings
Timo

proddy, 2019-04-06:
that would be a nice addition. Not everyone uses HA hence the generic MQTT support. Would you like to have a go at implementing this yourself?

rknall, 2019-04-18:
I would rather prefer a more generic approach, like Homie for instance: https://homieiot.github.io/implementations/

proddy, 2019-04-18:
I did have a version of EMS-ESP based on Espurna (https://github.com/xoseperez/espurna/wiki) a few releases ago but dropped it and went solo after I found it easier and more lightweight. It is possible to port to homie/tasmota/esphome though if someone wants to try. But it'll take a lot of time keeping up the support. One of the tricky things is the way I re-write the serial UART libraries so all Serial.* code could interfere with the EMS bus.

rknall, 2019-04-18:
Homie is not a distro, but rather a topic/message specification for how to arrange the data. So, although Homie-ESP exists, it could also be done in EMS-ESP. I am currently trying to get the right message definitions for my thermostat, and as soon as I have a PR, I'll send it through. So far it seems to work great, and is very neat to use.

proddy, 2019-06-14:
I do plan to do a ESPHome port at some point. I like what they've done.

proddy, 2020-07-06:
v2 uses HA discovery.

closed 2020-07-06 by proddy