emsesp/EMS-ESP #91: Wishlist: measure cost of gas etc.

mediaDS, 2019-04-07:
Hi proddy,

one topic of your wishlist is:

> Measure amount of gas in m3 per day for the hot water vs the central heating, and convert this into cost.

Do you already have ideas on how to do this? I'm planning on programming something in this direction - we already have:

- gas burner on/off
- gas power while the burner is on
- cost of gas
- we also have the time the gas burner is on
- heating or warm water on/off
- solar power activated

We also have the specification of the gas type in terms of efficiency (we have different types of gas here in Germany with different efficiency) which we can find on the monthly or annual bill. Would be nice if we could throw some ideas together.

have fun,
Dieter

proddy, 2019-04-07:
yes! This has been on my list for a while. Current thinking is:

- First get the maximum usage for the boiler. For example, my Nefit Trendline HRC30/CW5 is rated at 30kW max power and water class 5 (HoogRendementsCombiketel). So at full 100% burn for heating it's 30KWh of energy each hour and tap water is 114%. There are only a few classes of boilers which are typically between CW3 and CW6 and each class is 0.75m3/h different if we compare the specs.

- Then get the power % values and gas usage from the 0x18 broadcast type at each time interval, store them on the ESP in SPIFFS (file system) with a date stamp.

- To convert to m3 use a conversion factor of  0.12307692F (kWh * 0.12307692f = m³) for natural gas, which means a calorific value of 29MJ/m³ (specifically for Dutch users, this value is rather low, and a value of 35MJ/m³ is more realistic. More information can be found on the Gasunie website).

lobocobra, 2019-04-07:
I did the same thing based on a SONOFF POW2 and made some experiences.
"unfortunately it works so accurate, that I would be ready to support, but not be the main developer. 

Below variables might help you thinking it through. Of course you have no tank using gaz. but the temp of incoming gaz, changes for sure how much you use. 

**On the output side, what I would recommend...**
- usage per/h today
- usage per/h yesterday
- used gas/oil TTL yesterday
- used $ per month (is calculated by GRAFANA)
- sec/heating cycle (yesterday / avg / last 24h)
   => this is needed to check if your heating is setup idealy

**On the variables side (for oil heating)**
- BRENNERoeltemp xx.xx	// density of oil is changing depending the temp
- BRENNERoelmaxcapacity     // how much % is left in your tank
- BRENNERoelmincapacity      // when do you need to shut-down the heating to prevent issues
- BRENNERoelstand xxx           // how much oil is left
- SET BRENNERoelprice100L    // price per unit kwh,
- Set BrennerCorrCycleSec       // delay between start and 100% used gaz/oil

mediaDS, 2019-04-07:
Hmm, okay. It's a bit different here but I will try to gather all the neccessary informations about the different gas types. Let's see if I can put this together to a working calculation - but the user must provide some information for the calculations.

The ESP should provide the amount of gas used over a time period and then send it to f.i. ioBroker, where the real calculations take place - I usually use vis for visualisation, Grafana and InfluxDB for the history diagrams.

lobocobra, 2019-04-07:
The gaz types and similar, should be variables kept in the ESP. To have a KISS system (keep it simple and stupid), you do not want to have the logic spread over different systems. 

What I learned is that over time the calculation becomes more accurate. But they change whenever your heating was checked by a specialist. 
=> Means you need to calculate the used m3 or l/oil every time from a value that is valid since the last reset. For me reset occurs, when the heating specialist comes and does change the settings of my heating.

This allows you you to adjust the l/h@100% (or m3 gaz/hour in your case)  regularily and to get very close numbers to the reality. 

My solution does output from the ESP every hour, freshly calculated numbers and sends them to a DB (GRAFANA). Openhab then only displays the values. New variables are then updated over MQTT (which can originate from ioBROKER or similar).

Bonusbartus, 2019-04-07:
Instead of the Energy per hour, I was thinking though it mught me easier to just store/ the amount of gas per hour instead of using calculation factors. I checked the manual for the topline compact hrc 25 and 30, and they specify the max amount of gas per hour for heating (100% burn) and for tapwater (114% burn) and there seems to be a linear relation. So when buring at 50% for an hour it should be exactly 50% m3 of the value specified as the max in the manual.

It could be just a simple parameter to be stored once on the spiffs either using a set command or a settings file.

lobocobra, 2019-04-07:
If you save values in the SPIFF then we need to add SPIFF rotation into the code.
=> If not it will wear out pretty fast. 
Not a complicated thing, but a necessity... in the code of ARENDST it can be seen, how this can be done.

proddy, 2019-04-08:
@lobocobra luckily the code to store values in permanent EEPROM and rotate the sectors is already in the code (see myESP.cpp). I use this to store stack dumps when testing crashes.

mediaDS, 2019-04-09:
To have accurate measurements the ESP8266 needs to have an accurate date/time, maybe via ntp. I looked at the behauviour of my boiler:

It usually starts ignition, the gas power goes to 46%, then down to 28% after 13 seconds and down to 23% after 10 seconds. Then up to 27% after 4 seconds. Then down to 23% again after 20 seconds. Then the selected temperature is reached and the gas burner is switched off.

I would approach this with a (boiler dependent or user set) variable containing the maximum gas throughput at 100% and a counter, sampling the gas power (while the burner is on) ie. every second - multiplying the gas power (divided by 100) with the variable and add it to the counter. This way I have the gas consumption for every second and can then calculate all the following values like gas consumption last hour/1 hour/24 hours and so forth. I wouldn't need much memory, just a counter and a few variables. Becasue the ESP8266 is powerd 24/7 I would even use the internal RAM. Counter/cost/gas consumption reset could be done via telnet and/or MQTT.

proddy, 2019-08-21:
@mediaDS NTP is in version 1.9 and it has better supporting for persisting data to EEPROM and SPIFFs. So theoretically we can implement this now.

mediaDS, 2019-10-06:
Hi Paul,

sorry, was on holiday and then knee-deep into my own projects. Just installed your latest code - very impressive and very well done! I'll have a closer look into it later on and report back.

Dieter

duftstenen, 2020-08-01:
Hi guys. I have just started looking at this repos in order to get some interesting data from my Bosch ZSB14-3. 

I'm quite new on the ESP interface, but if I'm not mistaking, this seems to be the right way in order to pull from data from the unit. My main goal is to get some gas usage reported, which was the reason why I came across this thread. 

Is the functionality to calculate gas usage still possible / or in development?

proddy, 2020-08-01:
I don't believe anything has happened. In theory we can store and send the amount of gas per hour with EMS-ESP as the infrastructure is there but someone needs to document the exact requirements

duftstenen, 2020-08-03:
Hi proddy. Thanks for your swift reply. Sounds good with the infrastructure in place. I would love to help document the exact requirement in order to achieve it. How do we do this best?

What I want to be able to in the end, is to have the gas usage in m3 received (in decimal?) on frequent basis (hourly?) and viewed in Home Assistant, so I can follow the usage in Home Assistant, just like having e.g. the temperature from the Bosch ZSB14-3 showing HA, received by MQTT.

proddy, 2020-08-03:
I'm not an expert in this but the other guys on this post seem to know what they're doing. As @Bonusbartus and @mediaDS mentioned it makes sense to store the amount of gas used per hour. I think there needs to be 2 user settings to define the amount of gas used at a 100% burn for hot tap water and central heating for your specific boiler. Then when the burner is on (we detect this now) we start a counter. On the hour we calculate the consumption, do some magic, total the two numbers and send it out via MQTT. In case of a shutdown, we can persist the information in EEPROM (ESP8266 only).

S-Przybylski, 2020-08-05:
Dear @proddy 
i recommend to have a look at my idea to calculate the effective gas consumption. Due to the fact that the GB172-20 has a non linear heating curve it is vital to reflect that. In addition to that the integration function and the summarization function needs also to be designed in detail. Up to now my calculation is 90% accurate. (error of 10% = main reason: the heating curve is not as accurate as stated in the docs....)
My unfinished (more / less working) procedure: #223

glitter-ball, 2020-10-13:
How linear is the relationship between burner power (30-100% in my case) and gas consumption? I have a 24kW boiler so, can I assume 7.2kW at 30%, 24kW at 100% and linear in between?

If I could fix the burner power at intermediate steps for a period, I could time the gas meter and work out gas consumption / power for them as well and get a more accurate relationship. But... there's no easy way to tell the boiler to fire at 70%, say, for a period. The chimney sweep function lets you fire the burner at 30% and 100% but nothing in between.

The other approach would be to look at temperature rise across the boiler and  flow rate (which would depend on pump speed). That would give boiler output power and you could work that back to gas input. But, that will vary with efficiency, which varies with return temperature and other things.

Might have to settle for videoing the gas meter and the burner power together for 10 minutes and dumping sampled values in a spreadsheet to establish the relationship between them.

Not easy!

hamster65, 2020-10-14:
I'd assume efficiency decreases with more burner power because the heat exchanger surface does not grow, but has to transfer more energy over to the water. Is there something in the data sheets about that?

We have a Buderus GB142. It has 15 kW. Burner power is constant at 100% when warm water is heating up at 4 in the morning. During this time, i measure about 18 kW gas consumption.
Right now, burner power is around 25 % and gas is around a litte bit below 4 kW.

Method of measurement: Reed contact attached to the gas meter. Outputs one pulse for 0,01 m3 of gas. Pulses counted over 4 minutes, times 15 to scale up to 1 hour, times 1.22 kWh per m3, according to information of our gas supplier.

Also note that the amount of energy per m3 changes over time. Our supplier provides the factor each month and I used a weighted average over the last 2 years. Might be something the user has to calibrate, then, depending on the gas quality and boiler model.

is it the same for oil burners?

S-Przybylski, 2020-10-14:
Dear @hamster65 
My GB172-20 uses at maximum 123% for warm water heating. Therefore it takes around 24kW out of the gas then.
This is based on the information i found somewhere: GB172-20 25% = 5,2kW, 100% = 20,6kW and 123% 23,8kW
Could it be that you heating also uses more than 100% for warm water heating?

hamster65, 2020-10-14:
Interesting, I am not aware of that but I'll look into the specs.

Another thought: Would it be interesting to measure the condensation? It could be done by counting pulses of a sensor (e.g. tipping bucket).

Bartekn86, 2020-11-08:
I would be happy to buy a ready-made device that connects to a gas meter and is preferably compatible with the mqtt

glitter-ball, 2020-11-08:
There are solutions out there using esphome with a pulse sensor to pass info to HA.

I want to so something with water. I have a digital water meter outside the house under the pavement.

It’s read by some remote reading device using 866MHz radio which won’t be accessible to customers because frequent reading would quickly exhaust the lithium battery sealed in the meter.

But... I have a water softener which meters most of the water use in the house via a turbine meter with a Hall Effect sensor. So I was thinking about passing these pulses to something to count them and forward to HA over MQTT. 

I know EMS-ESP has a spare analogue input and maybe there’s a digital one too? Maybe there’s an option to count pulses and send a raw count on to HA in the MQTT payload?

Conversion factors could be applied in HA and consumption displayed.

Bartekn86, 2020-11-08:
some link to tindie ?

glitter-ball, 2020-11-08:
https://www.rogerfrost.com/an-electricity-usage-monitor-with-home-assistant/

This has some ideas. You might need a different sensor if your gas meter doesn’t have a flashing LED. Some of them have a pulse contact output but I don’t know if this accessible.

lsw271, 2021-01-24:
> There are solutions out there using esphome with a pulse sensor to pass info to HA.
>
> I want to so something with water. I have a digital water meter outside the house under the pavement.
>
> It’s read [...]

Hi, about measuring water look at this https://github.com/weetmuts/wmbusmeters