emsesp/EMS-ESP #93: Version 1.7 stable

gl3nni3, 2019-04-11:
Hey Proddy,
Do you have any idea when version 1.7 is stable?

gl3nni3, 2019-04-11:
@proddy

proddy, 2019-04-13:
I haven't touched it for a while. If no one reports any issues I'll push it to master. Does 1.7 beta work with your changes?

lobocobra, 2019-04-14:
I can confirm that it works at least solid for me :)

higgers, 2019-04-15:
Seems to be working for me too (apart from the usual problem I have with my boiler not responding to requests but that's a known issue with my boiler.)

SpaceTeddy, 2019-04-16:
for me it workes also fine!

kwertie01, 2019-04-16:
The `set tx_delay` feature is working fine :-)

proddy, 2019-04-16:
thanks for testing guys. One more change for the Solar Module then I'll push it to Master

proddy, 2019-04-29:
@gl3nni3 is 1.7 working for you? I rewrote a lot of the original EMS+ code

closed 2019-05-11 by proddy