emsesp/EMS-ESP #94: add DHW once activate functionality for RC35

majo551, 2019-04-14:
**Is your feature request related to a problem? Please describe.**
add command/function to activate  DHW once  for RC35

**Describe the solution you'd like**
be able to send Telnet / MQTT request to activate and monitor the status of the  **DHW once**  for RC35

proddy, 2019-04-18:
can you
- add the raw commands that will do this (for on and off)
- describe the function name you'd like to see in the telnet
- describe the topic and payload options you'd like to see in the MQTT

majo551, 2019-04-18:
if the condition for DHW is met (temp below 5 (normal operation) or 10
degrees (eco operation)  from desired temp (depending on mode),
the ON command sent by RC35 to my boiler :
Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 23 33 (CRC=76),
#data=2
folowed by Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 11
(CRC=10), #data=2

so the raw command  is  *0B 08 35 00 23 33*  (tested, working)
and, not sure if needed, but followed by
*0B 08 35 00 01 11*

This sequence(*08 35 00 01 11) * is OFTEN   sent by RC35 from time to time
again, so I guess it can be just keeping the circulation running

Off (manual interruption, if DHW running) :
* 0B 08 35 00 03 11*
followed by
*0B 08 35 00 01 11*

If the temp. condition is not met, RC35 starts the circulation onlyWITH:
*ON:  0B 08 35 00 01 33 followed by *
*0B 08 35 00 01 11*

All this is observed from the logs. It does not follow the Wiki
documentation logic(might be outdated, or for different boiler)

Information on DHW is part of the
UBAMonitorWWMessage

function name  in telnet:
DHW_ONCE ON
DHW_ONCE OFF
and probably:
DWH_ONCE STATUS

MQTT:

e.g.
home/ems-esp/boiler_cmd_dhw  for activating  (cmd = ON, OFF)

home/ems-esp/boiler_ dhw_status
content of the 0x34 – UBA Monitor Hot Water
if the document is correct it should be:
[image: image.png]

{"Selected_DHW_Temp":"58", "Current_DHW_Temp":"40.2" ,"Day_Mode_DHW":"No",
"DHW_Once_Activated":"Yes",
"Thermal_Disinf_DHW":"No","Prepare_DHW":"Yes", "DHW_Work_Minutes:"36" ,
"DHW_Start_Counter":"3600"}

proddy, 2019-06-13:
Ok. Just to be clear, 

1) where would the values Selected_DHW_Temp, Current_DHW_Temp, Day_Mode_DHW, DHW_Once_Activated, Thermal_Disinf_DHW, Prepare_DHW, DHW_Work_Minutes, DHW_Start_Counter come from?

2) What should the command DWH_ONCE STATUS do?

3) Do the commands to change DHW need to also detect the operation mode (eco/normal) and the temperature? 

4) Can you try and program this yourself in your own fork with help from others?

vanbogaertetienne, 2019-07-31:
Hello Proddy,
count me as well for this feature, as we do the requests maually at the moment and would like to automate this.
1. This data is contained in the 0x34 UBAMonitorWW telegram, as such:
       Byte 5= DHW_Selected_Temp
       Byte 6&7 = DHW_Current_Temp
       Byte 11 bit 0 = DHW_Day_Mode
       Byte 11 bit 1 = DHW_Once_Activated
       Byte 11 bit 2 = DHW_Thermal_disinf
       Byte 11 bit 3 = DHW_Prepare
       Byte 11 bit 5 = DHW_Temperature_OK
       Byte 15-17 = DHW_Work_Minutes
       Byte 18-21 = DHW_Start_Counter
    It looks like the last two are already available in your current code, since the values of a decoded telegram correspond to the value you give when I type info in the telnet console
2. This should return the value of telegram 0x34, byte 11 bit 1 (DHW__Once_Activated)

3) No, this is done on the boiler (as far as my testing goes)

4) I don't have much experience in coding in C, so if possible, could you implement this? Otherwise where do I start to install the toolchain/debugging tools needed?

proddy, 2019-07-31:
@vanbogaertetienne ok, pretty straight forward then. I'm working on a new web interface for versin 1.9.0 so when thats done I'll start on this next,

vanbogaertetienne, 2019-07-31:
For the 'request hot water once' function, majo551's telegram of 10 08 35 00 23 33 might work if the sender address is changed, but I found better luck to follow dani's values (see https://domoticproject.com/ems-bus-buderus-nefit-boiler/ under 4.7-flags hot water).
This would be a write telegram to the boiler, message 35 with value 39 for enable, and value 3 for disable, e.g.
  0B 08 35 00 39 (enable hot_water_once)
  0B 08 35 00 03 (disable hot_water_once)

Is this enough information for you? I've checked out the git repo and am adding the code for the status information to my local branch, once I've validated that this works I might try to write the code for the enable hot_water_once function, but I am a bit weary of writing something that might mess up the boiler configuration.

vanbogaertetienne, 2019-08-01:
Did some additional testing today. So sending the telegrams as written above works for starting and stopping the hot water generation cycle. To get the blinking status light on the RC35, the additional telegram of 0B 08 35 00 01 11 is needed. I'll keep monitoring this cycle today in order to see if anything is needed to stop the blinking light ;)

proddy, 2019-09-16:
got it. This is next on my todo list.

proddy, 2019-10-06:
@vanbogaertetienne want to help out implementing this next? See #195

proddy, 2019-11-13:
implemented in 1.9.3

closed 2019-11-13 by proddy