emsesp/EMS-ESP #95: Getting water tap flow & solar module

rknall, 2019-04-14:
So, I cannot read out information about the current water flow and for that shower detection is not working obviously. Also, I have an additional - not picked up -  device on the EMS bus, which I would like to get working, if I could find any telegram information about it.

So autodetect shows me the following:

> Started scan of EMS bus for known devices
> Boiler found. Model Nefit Topline Compact with TypeID 0x08, ProductID 115, Version 03.06
> * Setting Boiler type to Model Nefit Topline Compact, TypeID 0x08, ProductID 115, Version 03.06
> Device found. Model RFM20 Receiver with TypeID 0x09, ProductID 68, Version 02.03
> * Setting Thermostat type to Model RC35, TypeID 0x10, ProductID 86, Version 01.15

RFM20 I assume is the solar receiver, but no information get's picked up for it.

Info on the devices show up as such:

> EMS-ESP system stats:
>   System logging set to None
>   LED is on, Silent mode is off
>   0 external temperature sensor connected
>   Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is enabled
> 
> EMS Bus stats:
>   Bus Connected=yes, Tx is active, # Rx telegrams=15457, # Tx telegrams=0, # Crc Errors=5
> 
> Boiler stats:
>   Boiler type: Nefit Topline Compact (ProductID:115 Version:03.06)
>   Hot tap water: off
>   Central heating: off
>   Warm Water activated: on
>   Warm Water circulation pump available: off
>   Warm Water comfort setting: Hot
>   Warm Water selected temperature: 55 C
>   Warm Water desired temperature: 70 C
>   Warm Water current temperature: 56.1 C
>   Warm Water current tap water flow: 0.0 l/min
>   Warm Water # starts: 257 times
>   Warm Water active time: 26 days 4 hours 44 minutes
>   Warm Water 3-way valve: off
>   Selected flow temperature: 7 C
>   Current flow temperature: 37.3 C
>   Return temperature: 38.8 C
>   Gas: off
>   Boiler pump: off
>   Fan: off
>   Ignition: off
>   Circulation pump: off
>   Burner selected max power: 0 %
>   Burner current power: 0 %
>   Flame current: 0.0 uA
>   System pressure: 1.1 bar
>   System service code: 0H (203)
>   Heating temperature setting on the boiler: 90 C
>   Boiler circuit pump modulation max power: 63 %
>   Boiler circuit pump modulation min power: 30 %
>   Outside temperature: 8.4 C
>   Boiler temperature: 41.2 C
>   Pump modulation: 0 %
>   Burner # starts: 13923 times
>   Total burner operating time: 1246 days 16 hours 55 minutes
>   Total heat operating time: 1220 days 12 hours 11 minutes
>   Total UBA working time: 3158 days 6 hours 29 minutes
> 
> Thermostat stats:
>   Thermostat type: RC35 (ProductID:86 Version:01.15)
>   Setpoint room temperature: ? C
>   Current room temperature: ? C
>   Thermostat time is 00:26:58 15/4/2019
>   Mode is set to auto

Thermostat question marks I assume might be solved by 1.7beta, but have not tried that yet. As can be seen, no information about the solar panels. Now, what bothers me is the following entries:

[... 42 lines cut ...]
> (00:40:04.059) Boiler -> me, type 0x14 telegram: 08 0B 14 00 45 65 44 (CRC=91), #data=3
> (00:40:04.200) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 76 00 00 00 00 00 62 80 00 02 31 01 85 00 00 0B 30 48 00 CB 00 00 00 (CRC=50), #data=25
> (00:40:04.373) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17
> (00:40:05.165) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> (00:40:14.121) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 76 00 00 00 00 00 62 80 00 02 31 01 85 00 00 0B 30 48 00 CB 00 00 00 (CRC=50), #data=25
> (00:40:14.337) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17
> (00:40:24.059) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 76 00 00 00 00 00 62 80 00 02 31 01 85 00 00 0B 30 48 00 CB 00 00 00 (CRC=50), #data=25
> (00:40:24.275) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17
> (00:40:26.128) Thermostat -> Boiler, type 0x14 telegram: 10 88 14 00 03 (CRC=6E), #data=1
> (00:40:26.663) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99), #data=14
> (00:40:27.274) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 04 00 0F 1A 17 00 01 (CRC=B5), #data=8
> (00:40:28.760) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 00 (CRC=91), #data=4
> (00:40:33.972) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 76 00 00 00 00 00 62 80 00 02 31 01 85 00 00 0B 30 48 00 CB 00 00 00 (CRC=50), #data=25
> (00:40:34.262) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17
> (00:40:44.010) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 76 00 00 00 00 00 62 80 00 02 31 01 84 00 00 0B 30 48 00 CB 00 00 00 (CRC=34), #data=25
> (00:40:44.225) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17
> (00:40:49.505) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 01 (CRC=00), #data=2
> 
> inf(00:40:53.909) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA), #data=13
> (00:40:54.147) Boiler -> all, type 0x18 telegram: 08 00 18 00 07 01 75 00 00 00 00 00 62 80 00 02 31 01 84 00 00 0B 30 48 00 CB 00 00 00 (CRC=4C), #data=25
> o(00:40:54.373) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 54 01 9C 80 00 00 00 00 00 00 36 63 1B 64 B7 00 00 00 1A D1 5B 00 35 62 00 00 (CRC=E3), #data=27
> (00:40:54.629) Boiler -> all, type 0x1C telegram: 08 00 1C 00 8A 0A 0D 02 0B 00 00 00 00 00 00 (CRC=12), #data=11
> (00:40:54.911) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 31 02 31 20 00 00 03 00 00 93 5C 00 01 01 00 (CRC=C1), #data=17

Any ideas?

proddy, 2019-04-15:
if you look at the code, line 1081 in ems.cpp you'll see the flow is picked up from the 9th byte of the 0x34 telegram. In the log you posted above it's always zero. Why I have no idea.

As for supporting the RFM20, I you have details of the messages we can add it. Otherwise, you'll have to do a lot of trial and error to reverse engineer any responses back from send commands. It's basically what I did 6 months long to write this original code! Feel free to do a PR and share with others.

rknall, 2019-04-15:
> if you look at the code, line 1081 in ems.cpp you'll see the flow is picked up from the 9th byte of the 0x34 telegram. In the log you posted above it's always zero. Why I have no idea.

Yeah, I figured as much, just wanted to make sure that I did not overlook anything

> As for supporting the RFM20, I you have details of the messages we can add it. Otherwise, you'll have to do a lot of trial and error to reverse engineer any responses back from send commands. It's [...]

I'll gladly will do the work, just have no idea where to start. If you have anything already, I would greatly appreciate it. To be quite honest, currently I am not really sure, what device that actually is? I am guessing the temperature sensor from the solar panels, but not quite sure yet (have bought the house including the installation and plans are somewhat "special")

proddy, 2019-04-15:
The RFM20 I think is just a wireless receiver that connects the EMS master (like a boiler) to a wireless thermostat without the need to have the two-wire EMS bus connected.

The autodetect basically scans through all *known* devices in `ems_devices.h`. If an EMS device is attached but not listed in this file (because we don't know the device ID or product ID) then it will not show up during the scan. So you got me thinking - perhaps I'll add a new option to `autodetect` like `autodetect deep` which basically does a brute force scan on all IDs between 0 and 255 and see what pops up. And also a `autodetect list` which shows a list of everything it found because the reads are asynchronous. In your case, this should tell you if you have any solar devices attached that are EMS compatible.

proddy, 2019-04-15:
I added the function `autodetect deep`. can you try that ?

rknall, 2019-04-15:
Yes I can, which branch or PR is it contained in? If you can send me a precompiled version, I am using the Wemos D1 together with bbqkees translation board

proddy, 2019-04-15:
it's in the dev branch. Instructions on how to build are in the README

rknall, 2019-04-16:
I can confirm, that the deep scan is working:

> Starting a deep EMS device scan. This will take about 1 minute. Please wait...
Boiler found. Model Nefit Topline Compact (TypeID:0x08 ProductID:115 Version:03.06)
Device found. Model RFM20 Receiver with TypeID 0x09, ProductID 68, Version 02.03
Finished the deep EMS device scan.
> 
> These 3 EMS devices were detected:
 Nefit Topline Compact (TypeID:0x08 ProductID:115 Version:03.06)
 RC35 (TypeID:0x08 ProductID:86 Version:01.15)
 RFM20 Receiver (TypeID:0x09 ProductID:68 Version:02.03)
> 
> If any are marked as 'unknown', please report as a GitHub issue so we can update the EMS devices database.

But it seems, that my solar panels are connected without any additional items. I am currently trying to get the messages into Wireshark (via extcap) as it would allow for better diagram analysis. So far I get about half of all messages across without changing your code at all (just by using the telnet connection). I'll keep you posted, but this issue can be closed now

closed 2019-04-16 by rknall