emsesp/EMS-ESP #98: could Junkers FR10 get implemented?

sadrov, 2019-04-16:
Hi Proddy,
awesome work, thanks a lot!
I have a question, my boiler is a junkers cerapur comfort TOP 30-3 but is automatically recognized as a Bosch condens 2500. Not a big problem in the end because i can read the values and they seem to be correct. That's great!
My thermostat is not automatically recognized however. It's a junkers FR10. Could you implement it in the list please, when you find the time. Which logs do you need for that?
I'm quite new with this github and i'm not a coder so apologies if i'm not following protocol.
This is an extract of what i get from verbose logging:
System Logging set to Verbose
(00:08:14.231) 0x10 -> all, type 0xFF telegram: 90 00 FF 00 00 6F 03 01 00 BE 00 BE (CRC=85), #data=8
(00:08:14.490) 0x10 -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5), #data=1
(00:08:14.522) Boiler -> 0x10, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86), #data=12
(00:08:14.698) Boiler -> all, type 0x18 telegram: 88 00 18 00 41 02 80 55 3A 09 03 25 C0 01 B2 80 00 01 BE FF FF 0E 00 00 00 00 00 02 18 (CRC=AF), #data=25                                                                                                                 
(00:08:14.938) Boiler -> all, type 0x34 telegram: 88 00 34 00 2B 01 B2 80 00 A1 00 00 01 00 00 1E 7E 00 0E CC 00 (CRC=C9), #data=17
(00:08:15.618) 0x10 -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7), #data=1
(00:08:15.643) Boiler -> 0x10, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29), #data=10
Boiler found. Model Bosch Condens 2500 with TypeID 0x08, ProductID 95, Version 18.13
* Setting Boiler type to Model Bosch Condens 2500, TypeID 0x08, ProductID 95, Version 18.13
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:08:18.616) 0x10 -> Boiler, type 0x23 telegram: 90 08 23 00 42 (CRC=5F), #data=1
(00:08:18.649) 0x10 -> Boiler, type 0x23 telegram: 90 08 23 02 00 (CRC=19), #data=1
(00:08:18.682) 0x10 -> Boiler, type 0x1A telegram: 90 08 1A 00 42 (CRC=BB), #data=1
(00:08:18.706) 0x10 -> Boiler, type 0x1A telegram: 90 08 1A 02 00 (CRC=FD), #data=1
(00:08:24.652) Boiler -> all, type 0x18 telegram: 88 00 18 00 42 02 83 55 3C 09 03 25 C0 01 B3 80 00 01 C0 FF FF 0E 00 00 00 00 00 02 18 (CRC=2C), #data=25                                                                                                                 
(00:08:24.861) Boiler -> all, type 0x34 telegram: 88 00 34 00 2B 01 B3 80 00 A1 00 00 01 00 00 1E 7E 00 0E CC 00 (CRC=DF), #data=17
(00:08:26.452) 0x10 -> Boiler, type 0x23 telegram: 90 08 23 00 41 (CRC=5C), #data=1
(00:08:26.485) 0x10 -> Boiler, type 0x23 telegram: 90 08 23 02 00 (CRC=19), #data=1
(00:08:26.517) 0x10 -> Boiler, type 0x1A telegram: 90 08 1A 00 41 (CRC=B8), #data=1
(00:08:26.542) 0x10 -> Boiler, type 0x1A telegram: 90 08 1A 02 00 (CRC=FD), #data=1

Thanks a lot in advance for having a look.

proddy, 2019-04-16:
could you build the latest build from the dev branch and do an `autodetect deep` which should return back the details of your thermostat. Then we can add it.

sadrov, 2019-04-16:
Many thanks already for the quick reply!
I realise this question is superrookie, but what's the easiest way to do this?
Is it a matter of connecting the wemos to a computer and build the latest build through something like the arduino IDE?

proddy, 2019-04-16:
https://github.com/proddy/EMS-ESP/blob/master/README.md#building-the-firmware

sadrov, 2019-04-16:
Yeah I found it, that's why I deleted the question :) I'm trying to figure this out. It's all new to me.
the dev branch doesn't seem to have a platformio.ini file by default so I renamed the platformio.ini-example. I bought the bbqkees board including the wemos D1 mini so I'm not changing anything in the file itself. You'll hear it when I succeeded to simple step :)

sadrov, 2019-04-16:
this is what I got:

autodetect deep
System Logging set to None
Starting a deep EMS device scan. This will take about 1 minute. Please wait...
Boiler found. Model Bosch Condens 2500 (TypeID:0x08 ProductID:95 Version:18.13)
Finished the deep EMS device scan.

These 1 EMS devices were detected:
 Bosch Condens 2500 (TypeID:0x08 ProductID:95 Version:18.13)

If any are marked as 'unknown', please report as a GitHub issue so we can update the EMS devices database.

So the junkers FR10 is not detected. Whenever I turn the knob or push the button on the thermostat I see it right away in the log.
Do you have any ideas on how to proceed?
Thanks a lot!

proddy, 2019-04-17:
strange that it didn't find the thermostat. Can you do a `log v` and then `send 0B 90 02 00 20` which will request the version data from the thermostat which I think is type ID 0x10.

You could also manually set it using `set thermostat_type 10`

sadrov, 2019-04-19:
finally found a bit of time to test this out. So with autodetect deep, only the boiler is detected.

I manually set the thermostat type to 10 and then whenever i turn the thermostat knob or push the mode button the log reflects it:
in green:
(00:12:13.468) Thermostat -> all, type 0x6F telegram: 90 00 FF 00 00 6F 01 01 00 3C 00 B9 (CRC=F8)
(00:12:13.786) Thermostat -> all, type 0x6F telegram: 90 00 FF 00 00 6F 01 01 00 41 00 B9 (CRC=15)
(00:12:14.065) Thermostat -> all, type 0x6F telegram: 90 00 FF 00 00 6F 01 01 00 46 00 B9 (CRC=09)

in purple:
(00:12:32.502) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 04 03 (CRC=F2)
(00:12:32.535) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 00 49 (CRC=54)
(00:12:32.568) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 02 00 (CRC=19)

not sure if the send command delivered anything useful:
send 0B 90 02 00 20
(00:18:21.291) Boiler -> all, type 0x18 telegram: 88 00 18 00 00 00 B6 55 00 00 02 20 C0 01 00 80 00 00 CB FF FF 0E 00 00 00 00 00 00 00 (CRC=10), #data=25                                                                                                                 
<--- UBAMonitorFast(0x18) received
(00:18:21.500) Boiler -> all, type 0x34 telegram: 88 00 34 00 2B 01 01 80 00 81 00 00 01 00 00 1E 94 00 0E CF 00 (CRC=FB), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:18:21.843) Thermostat -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7)
<--- Version(0x02) received
(00:18:21.867) Boiler -> Thermostat, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29), #data=10
<--- Version(0x02) received
Boiler found. Model Bosch Condens 2500/Junkers Cerapur Comfort (DeviceID:0x08 ProductID:95 Version:18.13)
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:18:31.284) Boiler -> all, type 0x18 telegram: 88 00 18 00 00 00 B7 55 00 00 02 20 C0 01 01 80 00 00 CD FF FF 0E 00 00 00 00 00 00 00 (CRC=98), #data=25                                                                                                                 
<--- UBAMonitorFast(0x18) received
(00:18:31.493) Boiler -> all, type 0x34 telegram: 88 00 34 00 2B 01 00 80 00 81 00 00 01 00 00 1E 94 00 0E CF 00 (CRC=ED), #data=17
<--- UBAMonitorWWMessage(0x34) received
(00:18:31.943) Thermostat -> all, type 0x6F telegram: 90 00 FF 00 00 6F 03 01 00 AA 00 BB (CRC=D0)
(00:18:32.212) Thermostat -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5)
<--- UBAParametersMessage(0x16) received
(00:18:32.237) Boiler -> Thermostat, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86), #data=12
<--- UBAParametersMessage(0x16) received
log n
System Logging set to None

I'm very tempted to order a junkers CR10 which may or may not behave similar as the rc10.
Would you have some more ideas perhaps?
thanks a lot :)

proddy, 2019-04-20:
We should be able to figure this out. Did you pull the latest build from yesterday? I’ve been making changes to how EMS+ is handled

sadrov, 2019-04-20:
I think so, this was on EMS-ESP version 1.7.0b8
thermostat read doens't give any response
I also tried this with following response:
thermostat temp 22 
Write not supported for this model Thermostat

sadrov, 2019-04-25:
Hi @proddy 
do you have any more ideas on how to proceed?

proddy, 2019-04-25:
It’s strange that the thermostat doesn’t respond to a version check with ‘send 0B 90 02 00 20’.  That’s the first thing we need to solve. We could decipher the thermostat calls from the ems+ type 6F. Could you look at the temperature and at the same time record what is broadcast on the 6F and we’ll see if we can work out where the values are stored.

sadrov, 2019-04-25:
I've been logging for a while and changing the setpoint temperature on the thermostat, nothing of type 6F is broadcast but when i turn the knob I do get following messages containing 6F, not sure if that is what you are after, though.
(03:32:05.915) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 9B 00 C4 (CRC=4B), #data=8
(03:32:06.225) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 A0 00 C4 (CRC=A7), #data=8
(03:32:06.536) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 A5 00 C4 (CRC=B3), #data=8
(03:32:08.072) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 9B 00 C4 (CRC=4B), #data=8
(03:32:08.383) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 96 00 C4 (CRC=7F), #data=8
(03:32:09.012) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 A0 00 C4 (CRC=A7), #data=8
(03:32:09.315) Thermostat -> all, type 0xFF telegram: 90 00 FF 00 00 6F 02 01 00 A5 00 C4 (CRC=B3), #data=8

sadrov, 2019-04-25:
when sending  ‘send 0B 90 02 00 20’ in telnet, I get no answer, not even when logging is on

proddy, 2019-04-29:
Seems to be a known issue with the Junkers. We're working on it!

sadrov, 2019-04-30:
amazing, thanks!

proddy, 2019-05-05:
@sadrov we made some changes to the 1.7 code to better handle EMS+ messages. Can you look at the current room temperature and also the setpoint temperature on your FR10 thermostat and post back here any telegrams you see starting with `90 00 FF 00 00 6F`

sadrov, 2019-05-05:
I just flashed the wemos with the latest firmware version 1.7.0b14. Good news is that now the wifi password with special characters is indeed working.
When I do 'info' I get the same info as before, only my boiler is not automatically recognized anymore. So I manually set boiler and thermostat to type 08 and 10
When I turn the knob on the thermostat between 18.5/19/19.5, this is what I get from the log for the setpoint temperature:
`(00:15:52.026) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 B9 00 BF (CRC=98) #data=6
(00:15:54.305) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 BE 00 BF (CRC=84) #data=6
(00:15:55.964) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 B9 00 BF (CRC=98) #data=6
(00:15:56.929) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 BE 00 BF (CRC=84) #data=6
(00:15:57.900) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 B9 00 BF (CRC=98) #data=6
(00:15:58.962) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 BE 00 BF (CRC=84) #data=6
(00:15:59.926) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 C3 00 BF (CRC=69) #data=6
`
when 'info' the data from the thermostat is not recognized `Thermostat stats:
  Thermostat: DeviceID: 0x10 (ProductID:0 Version:?)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to ?
`
not sure about the current room temperature, it currently says 18.5 degrees on the thermostat and this is what I get from log t
`(00:24:01.497) Thermostat -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:24:01.519) Boiler -> Thermostat, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:24:03.311) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 00 00 (CRC=F9) #data=1
(00:24:03.336) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:24:03.368) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 04 03 (CRC=F2) #data=1
(00:24:07.668) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 AA 00 B7 (CRC=DC) #data=6
(00:24:07.960) Thermostat -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:24:07.986) Boiler -> Thermostat, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86) #data=12
(00:24:08.181) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 00 00 (CRC=1D) #data=1
(00:24:08.205) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:25:01.082) Thermostat -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:25:01.104) Boiler -> Thermostat, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:25:06.469) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 AA 00 B7 (CRC=DC) #data=6
(00:25:06.728) Thermostat -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:25:06.764) Boiler -> Thermostat, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86) #data=12
Requesting scheduled EMS device data
(00:26:00.570) Thermostat -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:26:00.596) Boiler -> Thermostat, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:26:00.945) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 00 00 (CRC=F9) #data=1
(00:26:00.978) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:26:01.011) Thermostat -> Boiler, type 0x1A telegram: 90 08 1A 04 03 (CRC=F2) #data=1
(00:26:05.367) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 AA 00 B6 (CRC=DD) #data=6
(00:26:05.659) Thermostat -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:26:05.688) Boiler -> Thermostat, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86) #data=12
(00:26:05.848) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 00 00 (CRC=1D) #data=1
(00:26:05.880) Thermostat -> Boiler, type 0x23 telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:27:00.440) Thermostat -> Boiler, type 0x02 telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:27:00.462) Boiler -> Thermostat, type 0x02 telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:27:04.143) Thermostat -> all, type 0x006F telegram: 90 00 FF 00 00 6F 03 01 00 AA 00 B7 (CRC=DC) #data=6
(00:27:04.403) Thermostat -> Boiler, type 0x16 telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:27:04.436) Boiler -> Thermostat, type 0x16 telegram: 88 10 16 00 FF 58 55 00 00 F7 0F 01 05 64 0A 01 (CRC=86) #data=12
`

sadrov, 2019-05-05:
sorry for the weird formatting. Not sure yet how i should properly copy/paste code in here

proddy, 2019-05-05:
for fomatting large chunks of code use 2 or 3 backquotes `

proddy, 2019-05-05:
if you do an `autodetect deep` is anything found?

proddy, 2019-05-05:
Also does your thermostat show the actual room temperature too? Was it 19.1 degrees by any chance?

sadrov, 2019-05-05:
with and `autodetect deep` this is all I'm getting:
System Logging set to None
Starting a deep EMS device scan. This can take up to 2 minutes. Please wait...
Finished the deep EMS device scan.
System Logging set to None

Yes my thermostat shows the actual room temperature, it rounds of to half a degree and at the time it was jumping between 18.5 and 19 so I think the 19.1 that you found was definitely the actual room temperature measured by the thermostat, seems like you are on the good track :) :) that's great!

proddy, 2019-05-06:
right, I forgot the sending/Tx doesn't work with Junkers yet, which is used to query the devices. There's another issue related to that problem which is waiting for a logic analyzer to see what is blocking the traffic. So it may be a while until someone has figured it out.

proddy, 2019-05-07:
https://github.com/proddy/EMS-ESP/issues/103

proddy, 2019-06-09:
can you try again with the latest dev built and set tx_mode to 2 ?

proddy, 2019-07-27:
closing. The values from an FR10 should be picked up, otherwise please re-open

sadrov, 2019-09-29:
hey proddy, now that heating season started again and i'm home since a long while (just shortly unfortunately), I finally found time to flash 1.8.3 on the wemos and I confirm that the FR10 is indeed recognized now and room termperature and setpoint are being read correctly! That's awesome!! Hopefully you bright minds can figure out a way to change the setpoint.
Thanks again so much for all the work you do!

proddy, 2019-09-30:
sure, I'll send over some commands you can test using telnet to get things going. BTW the current version is 1.9.1 which has some more improvements (and a web interface). There are some reports of issues with Junkers HT3 so I would stick with 1.9.0 or 1.8.3 for now.

sadrov, 2019-09-30:
Of course, happy to help out! 
I tried 1.9 but I couldn't get it connected to my main wifi (neither through telnet or web interface ). I think because of my wifi password, it's long and autogenerated and contains ' and " characters that I think are messing it up. So I tried 1.8.3 instead and that worked :)
I should try changing my wifi password to see if removing the " works, but I wanted to avoid that for now.
I'm using tx_mode 3 in 1.8.3.

proddy, 2019-09-30:
do please try 1.9.x. If there is a problem with strange characters in the wifi_ssid or wifi_password then its a bug I need to fix.

sadrov, 2019-09-30:
the wifi password contains "\ somewhere, these two characters are next to each other within the password.
Let's say for convenience the password is:  ```passw"\ord```
When I set the wifi password in the 1.9.x web interface I'm getting:
```
"network": {
    "ssid": "example",
    "wmode": 0,
    "password": "passw\"\\ord"
  }
```
so I'm really suspecting these 2 characters, or maybe the fact that they're consecutive, is causing the problem.  When I set the password through telnet and restart the wemos, it always boots in access point mode so I assume that doesn't do it either. But it did work on the versions 1.8.x and earlier

proddy, 2019-09-30:
nice, I have an example. I'll work on a fix. thanks for sharing.

sadrov, 2019-09-30:
more thanks to you :D 
it's not my real password so let's hope it's not another character causing it, this one just looked most suspicious to me. Hope you can reproduce it.

proddy, 2019-10-01:
I tested with a new network, using password: 

pass\"word

and it worked. I did this using telnet with the `set wifi_password` command. It must be something else then?

sadrov, 2019-10-01:
ok, that has been solved now!
Based on your info I tried again, it didn't work, I flashed the precompiled 1.9.0 or the latest one over the existing data on the wemos chip and realised some values that I set before reflashing it lingered around. So when I first did a esptool.py erase_flash and then flashed the precompiled 1.9.0, the wifi with all these weird characters in the password worked! 
So that's that! :)
next problem: chip is connected and the led light doesn't blink but is just on. With telnet info I see all the correct values but boiler nor thermostat get detected automatically (and neither with autodetect deep), on the web page I read: ```EMS Bus Connected but Tx is not working.```  and no values are displayed

sadrov, 2019-10-01:
also publishing to mqtt works but only when I do a manual publish command in telnet. It doesn't do it automatically every 120 seconds

proddy, 2019-10-01:
There are more reports of 1.9.0 and 1.9.1 not working well with Junkers HT3. What settings do you have? dump the output of `set`

sadrov, 2019-10-01:
after setting tx_mode to 3 and restarting, mqtt publishing is solid, I think maybe before the restart the firmware crashed.
Now no EMS devices get detected and I get:
```EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=0
  Tx: no signal
```
``set`` output:
```
The following set commands are available:

  set erase
  set <wifi_mode <ap | client>
  set <wifi_ssid | wifi_password> [value]
  set mqtt_enabled <on | off>
  set <mqtt_ip | mqtt_username | mqtt_password> [value]
  set mqtt_heartbeat <on | off>
  set mqtt_base [value]
  set mqtt_port [value]
  set ntp_enabled <on | off>
  set serial <on | off>
  set led <on | off>                 toggle status LED on/off
  set led_gpio <gpio>                set the LED pin. Default is the onboard LED 2. For external D1 use 5
  set dallas_gpio <gpio>             set the external Dallas temperature sensors pin. Default is 14 for D5
  set dallas_parasite <on | off>     set to on if powering Dallas sensors via parasite power
  set listen_mode <on | off>         when set to on all automatic Tx are disabled
  set shower_timer <on | off>        send MQTT notification on all shower durations
  set shower_alert <on | off>        stop hot water to send 3 cold burst warnings after max shower time is exceeded
  set publish_time <seconds>         set frequency for publishing data to MQTT (0=off)
  set heating_circuit <1 | 2>        set the main thermostat HC to work with (if using multiple heating circuits)
  set tx_mode <n>                    changes Tx logic. 1=ems generic, 2=ems+, 3=Junkers HT3

Stored settings:

  wifi_mode=client
  wifi_ssid=mcnulty
  wifi_password=***************************************************
  mqtt_enabled=on
  mqtt_ip=domoticz
  mqtt_username=
  mqtt_password=
  mqtt_base=home
  mqtt_port=1883
  mqtt_heartbeat=off
  serial=off
  ntp_enabled=off
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  heating_circuit=1
  tx_mode=3
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120
```

proddy, 2019-10-02:
`mqttlog` will show the last messages sent out. It will send messages when the data has been changed, i.e. if the payload is the same it ignores it. doing a `publish` will force the mqtt. Maybe this has something to do with the Tx not working which we need to fix first.

Can you grab the latest 1.9.1 dev build and do an `devices` and then if that shows nothing do a `autodetect quick` and then another `devices`. Let's see if it comes back with anything.

sadrov, 2019-10-03:
I just flashed ```version 1.9.1b10```
`EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=1
  Tx: no signal `

It finds devices but doesn't recognize them
```
autodetect quick

[EMS] Detected new EMS Device with ID 0x08
[EMS] Detected new EMS Device with ID 0x09
[EMS] Detected new EMS Device with ID 0x10
devices

These device IDs are on the EMS Bus: 0x08 0x09 0x10
No devices recognized. This could be because Tx is disabled or failing.
```

I'm following  #103 as well. I'm also in tx_mode 3
```
0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
```
the same telegrams as Neonox31 got.

proddy, 2019-10-03:
could to turn logging on and run the autodetect again so I can see the telegrams that its trying to send. Use

```
log v
autodetect quick
```

sadrov, 2019-10-03:
```
autodetect quick

[MQTT] TCP Disconnected
(01:27:50.880) 0x08 -> all, type 0x18, telegram: 88 00 18 00 00 01 1F 32 00 00 02 00 C0 01 4D 80 00 01 31 FF FF 08 00 00 00 00 00 00 00 (CRC=2C) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(01:27:51.089) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 4F 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=E4) #data=17
<--- UBAMonitorWWMessage(0x34)
Starting scheduled query from EMS devices
(01:28:00.909) 0x08 -> all, type 0x18, telegram: 88 00 18 00 00 01 21 32 00 00 02 00 C0 01 4D 80 00 01 31 FF FF 08 00 00 00 00 00 00 00 (CRC=BD) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
[MQTT] Connecting to MQTT...
(01:28:01.150) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 4D 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=C8) #data=17
<--- UBAMonitorWWMessage(0x34)
(01:28:10.866) 0x08 -> all, type 0x18, telegram: 88 00 18 00 00 01 1F 32 00 00 02 00 C0 01 4F 80 00 01 31 FF FF 08 00 00 00 00 00 00 00 (CRC=00) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(01:28:11.075) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 4D 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=C8) #data=17
<--- UBAMonitorWWMessage(0x34)
[MQTT] TCP Disconnected
(01:28:20.847) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
<--- UBADevices(0x07)
[EMS] Detected new EMS Device with ID 0x08
Requesting type Version(0x02) from dest 0x08
[EMS] Detected new EMS Device with ID 0x09
Requesting type Version(0x02) from dest 0x09
[EMS] Detected new EMS Device with ID 0x10
Requesting type Version(0x02) from dest 0x10
```

proddy, 2019-10-03:
and anything in the `queue`?

sadrov, 2019-10-03:
nope, it says tx queue is empty. Also after I do an autodetect quick

sadrov, 2019-10-03:
I scrolled back a bit in this topic and did a ```send 0B 90 02 00 20``` in telnet, now my thermostat is recognized again and showing correct values in telnet ```EMS Device recognized as Thermostat: Junkers FR10 (DeviceID:0x10 ProductID:111 Version:16.06)```
and while doing this the queue looked like 
```
Tx queue (1/50)
 [1] action=read dest=0x10 type=0x06 offset=0 length=6 dataValue=32 comparisonValue=0 type_validate=0x00 comparisonPostRead=0x00 @ (00:10:01)
```
not sure if any of this is helpful or related

proddy, 2019-10-03:
yes, that is helpful. let me think about it and run some tests

proddy, 2019-10-04:
err hang on, if you're on Junkers I would actually expect send `0B 90 02 00 20` not to work as Junkers behaves differently to Buderus and the source address needs the MSB set. Strange that it returned something. What happens if you do a `send 8B 08 02 00 20` - does it recognize the boiler?

When you type `info` does it say the Protocol is Junkers HT3? (on latest 1.9.1 build)

sadrov, 2019-10-04:
It doesn't specifically say the protocal is Junkers. I'm on 1.9.1b10, tx_mode 3
I know my boiler and thermostat are Junkers heatronic 3.
A ``send 8B 08 02 00 20`` doesn't seem to do anything: ``(10:26:08.308) Sending raw: 8B 08 02 00 20 (CRC=10) #data=124
`` and the queue is empty.
Setting the boiler type manually is not possible anymore in 1.9.1, correct?

info
```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=6
  Tx: no signal

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 40.0 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 3912 times
  Warm Water active time: 5 days 21 hours 38 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 68 C
  Current flow temperature: 59.8 C
  Return temperature: 45.4 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 50 %
  Burner current power: 50 %
  Flame current: 0.1 uA
  System pressure: 1.4 bar
  System service code:  (0)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 66680 times
  Total burner operating time: 187 days 8 hours 37 minutes
  Total heat operating time: 181 days 10 hours 58 minutes

Thermostat stats:
  Thermostat: Junkers FR10 (ProductID:111 Version:16.06)
  Thermostat time is ?
  Heating Circuit 1
   Current room temperature: 18.4 C
   Setpoint room temperature: 19.5 C

```

proddy, 2019-10-04:
the protocol is in the latest build from yesterday.

sadrov, 2019-10-04:
oh my bad, I'm always looking for the latest release here https://github.com/proddy/EMS-ESP/releases/tag/1.9.0 and find ``EMS-ESP-1_9_1b10-d1_mini-debug.bin`` as the latest file.
Just realised now I should probably pull the latest dev branch, give me a minute to try. Sorry about this

sadrov, 2019-10-04:
ok, got 1.9.1b11 now and it says this 
```
EMS Bus stats:
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=0, # CRC errors=0
  Tx: no signal
```
Doesn't look like send 8B 08 02 00 20 finds the boiler,
```
send 8B 08 02 00 20

(00:00:56.834) Sending raw: 8B 08 02 00 20 (CRC=10) #data=108
(00:00:59.084) 0x08 -> all, type 0x18, telegram: 88 00 18 00 32 01 E5 32 27 09 03 25 C0 01 6B 80 00 01 86 FF FF 0D 00 00 00 00 00 02 18 (CRC=65) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
Starting scheduled query from EMS devices
(00:01:04.236) 0x08 -> all, type 0x18, telegram: 88 00 18 00 32 01 E7 32 26 09 03 25 C0 01 6C 80 00 01 87 FF FF 0D 00 00 00 00 00 02 18 (CRC=39) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:04.445) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 6B 80 00 A1 00 00 01 00 00 21 32 00 0F 48 00 (CRC=16) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:11.873) 0x10 -> 0x08, type 0x02, telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:01:11.900) 0x08 -> 0x10, type 0x02, telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:01:14.264) 0x08 -> all, type 0x18, telegram: 88 00 18 00 32 01 EC 32 24 09 03 25 C0 01 6B 80 00 01 87 FF FF 0D 00 00 00 00 00 02 18 (CRC=6E) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:14.504) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 6B 80 00 A1 00 00 01 00 00 21 32 00 0F 48 00 (CRC=16) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:24.219) 0x08 -> all, type 0x18, telegram: 88 00 18 00 32 01 F1 32 23 09 03 25 C0 01 6B 80 00 01 87 FF FF 0D 00 00 00 00 00 02 18 (CRC=CC) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:24.429) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 6B 80 00 A1 00 00 01 00 00 21 32 00 0F 48 00 (CRC=16) #data=17
<--- UBAMonitorWWMessage(0x34)
log n
```

The Thermostat is still recognized by sending the other telegram
```
send 0B 90 02 00 20

(00:02:36.564) Sending raw: 0B 90 02 00 20 (CRC=7C)
(00:02:36.622) 0x10 -> me, type 0x02, telegram: 90 0B 02 00 6F 10 06 00 00 00 00 00 00 02 (CRC=E7) #data=10
<--- Version(0x02)
EMS Device recognized as Thermostat: Junkers FR10 (DeviceID:0x10 ProductID:111 Version:16.06)
Requesting type RCTime(0x06) from dest 0x10
(00:02:38.032) Sending read of type 0x06 to 0x10, telegram: 8B 10 06 00 20 (CRC=C0)
(00:02:38.116) 0x08 -> all, type 0x07, telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5D) #data=15
<--- UBADevices(0x07)
```

proddy, 2019-10-04:
ok, good, its recognized as Junkers.

what does `send 0B 88 02 00 20` show? Does it bring back the boiler details?

also with logging, can you send over the output for `autodetect quick`

sadrov, 2019-10-04:
Doesn't look like that is bringen back the boiler 
```
send 0B 88 02 00 20

(00:01:36.718) Sending raw: 0B 88 02 00 20 (CRC=BC)
(00:01:37.456) 0x10 -> 0x08, type 0x02, telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:01:37.483) 0x08 -> 0x10, type 0x02, telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:01:40.092) 0x08 -> all, type 0x18, telegram: 88 00 18 00 32 01 54 32 00 01 03 20 C0 01 56 80 00 01 66 FF FF 0D 00 00 00 00 00 00 00 (CRC=30) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:40.302) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 55 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=01) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:41.925) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 00 33 (CRC=2E) #data=1
(00:01:41.958) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:01:41.983) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 00 33 (CRC=CA) #data=1
(00:01:42.015) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:01:45.342) 0x10 -> all, type 0x006F, telegram: 90 00 FF 00 00 6F 03 01 00 C3 00 C3 (CRC=15) #data=6
<--- JunkersStatusMessage(0x6F)
(00:01:45.642) 0x10 -> 0x08, type 0x16, telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:01:45.667) 0x08 -> 0x10, type 0x16, telegram: 88 10 16 00 FF 4B 32 00 00 F7 0F 01 05 64 0A 01 (CRC=98) #data=12
(00:01:47.899) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 52 32 00 01 03 20 C0 01 55 80 00 01 64 FF FF 0D 00 00 00 00 00 00 00 (CRC=65) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:50.121) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 52 32 00 01 03 20 C0 01 55 80 00 01 64 FF FF 0D 00 00 00 00 00 00 00 (CRC=65) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:50.361) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 55 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=01) #data=17
<--- UBAMonitorWWMessage(0x34)
```
```
autodetect quick

(00:00:28.950) 0x08 -> all, type 0x18, telegram: 88 00 18 00 36 01 4A 32 00 01 03 20 C0 01 50 80 00 01 5A FF FF 0D 00 00 00 00 00 00 00 (CRC=A2) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:29.159) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 50 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=4F) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:36.970) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 00 35 (CRC=28) #data=1
(00:00:36.994) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:00:37.027) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 00 35 (CRC=CC) #data=1
(00:00:37.060) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:00:38.930) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
<--- UBADevices(0x07)
[EMS] Detected new EMS Device with ID 0x08
Requesting type Version(0x02) from dest 0x08
[EMS] Detected new EMS Device with ID 0x09
Requesting type Version(0x02) from dest 0x09
[EMS] Detected new EMS Device with ID 0x10
Requesting type Version(0x02) from dest 0x10
(00:00:39.160) 0x08 -> all, type 0x18, telegram: 88 00 18 00 35 01 48 32 00 01 03 20 C0 01 50 80 00 01 5A FF FF 0D 00 00 00 00 00 00 00 (CRC=FC) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:39.370) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 50 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=4F) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:39.662) 0x08 -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 02 59 01 04 7E 04 1E 0C 00 00 00 03 FC DA 00 F5 36 80 00 (CRC=A5) #data=27                                                                                                             
<--- UBAMonitorSlow(0x19)
(00:00:40.127) Sending read of type 0x02 to 0x08, telegram: 8B 08 02 00 20 (CRC=10)
(00:00:40.215) 0x08 -> all, type 0x07, telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5D) #data=15
<--- UBADevices(0x07)
(00:00:40.407) Sending read of type 0x02 to 0x09, telegram: 8B 09 02 00 20 (CRC=18)
(00:00:40.595) Sending read of type 0x02 to 0x10, telegram: 8B 10 02 00 20 (CRC=D0)
(00:00:41.085) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
<--- UBADevices(0x07)
(00:00:48.970) 0x08 -> all, type 0x18, telegram: 88 00 18 00 35 01 48 32 00 01 03 20 C0 01 4F 80 00 01 59 FF FF 0D 00 00 00 00 00 00 00 (CRC=FB) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:49.211) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 4F 80 00 81 00 00 01 00 00 21 32 00 0F 48 00 (CRC=E4) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:54.103) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 00 35 (CRC=CC) #data=1
(00:00:54.136) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:00:54.169) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 04 03 (CRC=F2) #data=1
(00:00:54.291) 0x10 -> 0x08, type 0x02, telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:00:54.319) 0x08 -> 0x10, type 0x02, telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:00:56.546) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 00 34 (CRC=29) #data=1
(00:00:56.570) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:00:56.603) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 00 34 (CRC=CD) #data=1
```
```
These device IDs are on the EMS Bus: 0x08 0x09 0x10
No devices recognized. This could be because Tx is disabled or failing.
```

proddy, 2019-10-04:
I hate to ask, but does `send 8B 08 02 00 20` work with 1,8.3 and 1.9.0? I've gone through the code and can't figure out why it doesn't it work with 1.9.1

proddy, 2019-10-04:
hang on with re-flashing, I had another brain wave....

proddy, 2019-10-04:
I'm still confused. But have a new build - nothing has changed so don't expect it all to work, but I want to try one more thing.

```
log v
autodetect
```

to see the send messages and responses.

BTW would be better if we used Gitter to chat than GitHub postings?

sadrov, 2019-10-04:
```
autodetect

Started scanning the EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x20
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
Requesting type Version(0x02) from dest 0x38
Requesting type Version(0x02) from dest 0x48
(00:00:25.168) Sending read of type 0x02 to 0x02, telegram: 8B 02 02 00 20 (CRC=40)
(00:00:29.385) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 E5 32 29 09 03 25 C0 01 5D 80 00 01 7C FF FF 0D 00 00 00 00 00 02 18 (CRC=47) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:30.382) Sending read of type 0x02 to 0x08, telegram: 8B 08 02 00 20 (CRC=10)
(00:00:30.466) 0x08 -> all, type 0x07, telegram: 88 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=5D) #data=15
<--- UBADevices(0x07)
(00:00:30.694) Sending read of type 0x02 to 0x09, telegram: 8B 09 02 00 20 (CRC=18)
(00:00:30.850) Sending read of type 0x02 to 0x10, telegram: 8B 10 02 00 20 (CRC=D0)
(00:00:30.975) Sending read of type 0x02 to 0x11, telegram: 8B 11 02 00 20 (CRC=D8)
(00:00:31.435) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 00 33 (CRC=CA) #data=1
(00:00:31.467) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 02 00 (CRC=FD) #data=1
(00:00:31.500) 0x10 -> 0x08, type 0x1A, telegram: 90 08 1A 04 03 (CRC=F2) #data=1
(00:00:31.570) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 E6 32 29 09 03 25 C0 01 5D 80 00 01 7E FF FF 0D 00 00 00 00 00 02 18 (CRC=F7) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:31.780) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 5D 80 00 A1 00 00 01 00 00 21 48 00 0F 48 00 (CRC=06) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:32.037) Sending read of type 0x02 to 0x17, telegram: 8B 17 02 00 20 (CRC=E8)
(00:00:32.652) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
<--- UBADevices(0x07)
(00:00:33.255) Sending read of type 0x02 to 0x18, telegram: 8B 18 02 00 20 (CRC=90)
(00:00:35.824) 0x10 -> all, type 0x006F, telegram: 90 00 FF 00 00 6F 03 01 00 C8 00 C6 (CRC=3C) #data=6
<--- JunkersStatusMessage(0x6F)
(00:00:36.091) 0x10 -> 0x08, type 0x16, telegram: 90 88 16 00 18 (CRC=B5) #data=1
(00:00:36.115) 0x08 -> 0x10, type 0x16, telegram: 88 10 16 00 FF 4B 32 00 00 F7 0F 01 05 64 0A 01 (CRC=98) #data=12
(00:00:36.127) Sending read of type 0x02 to 0x20, telegram: 8B 20 02 00 20 (CRC=49)
(00:00:36.460) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 00 33 (CRC=2E) #data=1
(00:00:36.492) 0x10 -> 0x08, type 0x23, telegram: 90 08 23 02 00 (CRC=19) #data=1
(00:00:37.626) Sending read of type 0x02 to 0x21, telegram: 8B 21 02 00 20 (CRC=41)
(00:00:41.562) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 EC 32 28 09 03 25 C0 01 5D 80 00 01 7E FF FF 0D 00 00 00 00 00 02 18 (CRC=03) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:41.771) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 5D 80 00 A1 00 00 01 00 00 21 48 00 0F 48 00 (CRC=06) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:43.064) Sending read of type 0x02 to 0x30, telegram: 8B 30 02 00 20 (CRC=C9)
(00:00:46.019) 0x10 -> 0x08, type 0x02, telegram: 90 88 02 00 0A (CRC=F7) #data=1
(00:00:46.041) 0x08 -> 0x10, type 0x02, telegram: 88 10 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=29) #data=10
(00:00:46.900) Sending read of type 0x02 to 0x38, telegram: 8B 38 02 00 20 (CRC=89)
(00:00:51.554) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 F3 32 26 09 03 25 C0 01 5D 80 00 01 7E FF FF 0D 00 00 00 00 00 02 18 (CRC=5C) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:00:51.763) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 5D 80 00 A1 00 00 01 00 00 21 48 00 0F 48 00 (CRC=06) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:00:52.333) Sending read of type 0x02 to 0x48, telegram: 8B 48 02 00 20 (CRC=22)
(00:00:53.646) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 F4 32 26 09 03 25 C0 01 5D 80 00 01 7E FF FF 0D 00 00 00 00 00 02 18 (CRC=73) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
Starting scheduled query from EMS devices
(00:01:01.570) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=ED) #data=15
<--- UBADevices(0x07)
(00:01:01.795) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 01 F8 32 25 09 03 25 C0 01 5D 80 00 01 7F FF FF 0D 00 00 00 00 00 02 18 (CRC=F0) #data=25                                                                                                                   
<--- UBAMonitorFast(0x18)
(00:01:02.005) 0x08 -> all, type 0x34, telegram: 88 00 34 00 2B 01 5D 80 00 A1 00 00 01 00 00 21 48 00 0F 48 00 (CRC=06) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:02.266) 0x08 -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 59 01 04 96 04 1E 5F 00 00 00 03 FD 17 00 F5 4E 80 00 (CRC=97) #data=27                                                                                                             
<--- UBAMonitorSlow(0x19)
```
yeah probably not a bad idea...I'm new to this, but I signed in to Gitter with my github account, so I assume my username is the same?

proddy, 2019-10-05:
the gitter channel is https://gitter.im/EMS-ESP/community#

proddy, 2019-10-05:
arrgghh, this is all confusing and hurting my head! Can you try `send 8B 88 02 00 20`

sadrov, 2019-10-05:
Yeah!
```
send 8B 88 02 00 20

(00:01:53.146) Sending raw: 8B 88 02 00 20 (CRC=74)
(00:01:53.193) 0x08 -> me, type 0x02, telegram: 88 0B 02 00 5F 12 0D 00 00 00 00 00 00 00 (CRC=DF) #data=10
<--- Version(0x02)
EMS Device recognized as Boiler: Bosch Condens 2500/Junkers Heatronic 3 (DeviceID:0x08 ProductID:95 Version:18.13)
```

[4 more replies]

closed 2019-10-05 by proddy