emsesp/EMS-ESP #100: RC35 Temperatures show ?

sysrun, 2019-04-17:
I already read the other issues.

Just installed the current dev branch - Temperatures show ?

Switched to manual mode and send "thermostat read 3E"

Reply was:
Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 80 02 2E 7D 00 00 00 00 00 00 00 00 64 11 2C 00 (CRC=58), #data=16

Setpoint Temp for HK1 is 0x2E = so 46/2 = 23 Degree - so the received telegram is correct.

But "info" shows "Setpoint room temperature: ? C" - and its not changing after toggling the modes and waiting several minutes.

proddy, 2019-04-17:
fixed in dev. If either the setpoint or current room temps were 0 or unavailable then both would show a "?". Now in your case it would display the setpoint temp correctly but the current temp would be empty. not sure if this is the desired result?

sysrun, 2019-04-18:
The "current temp" is stated as "7D 00" in my case - so no sensor value, which is correct

Just updated to newest dev - works like expected now :)

  Setpoint room temperature: 23.0 C
  Current room temperature: ? C

closed 2019-04-18 by proddy