emsesp/EMS-ESP #99: nefit proline hrc 24 cw4

jteeuw, 2019-04-17:
Hi , i'am using the bbq Kees board but my nefit proline hrc 24 cw4 is not recognized :

Started scan of EMS bus for known devices
Unrecognized device found. TypeID 0x08, ProductID 122, Version 03.09
Device found. Model BC10 Base Controller with TypeID 0x09, ProductID 114, Version 01.06
* Setting Thermostat type to Model RC20/Nefit Moduline 300, TypeID 0x17, ProductID 77, Version 02.20

is there a way to connect to the bioler ?

proddy, 2019-04-17:
yes, I've added it version 1,7 which is in the dev branch. Do you know how to build and upload the firmware yourself?

jteeuw, 2019-04-18:
That is good news! 
I'am not familiar with the built and upload procedure.

proddy, 2019-04-18:
Have a look at the README which explains how to do this.

jteeuw, 2019-04-24:
Can confirm that the boiler is detected  with de dev version, thank you for this project !

besmit, 2019-04-28:
I have similar boiler (ProLine HRC 24-CW3) connected but 2 values are not available/displayed correctly. @jteeuw can you check if the same happens with your ProLine?

- Warm Water current temperature: ? C => never seen a correct value
- Boiler temperature: 0.0 C => always reported as 0.0

Thanks

proddy, 2019-04-28:
whats the 0x34 telegram showing you in the log? It gets broadcast every 10 seconds

besmit, 2019-04-28:
I got these values in the log:

```
(12:00:35.756) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 80 00 80 00 21 00 00 01 00 01 34 EF 00 BA 14 00 (CRC=0C), #data=17
(12:00:46.441) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 80 00 80 00 21 00 00 01 00 01 34 EF 00 BA 14 00 (CRC=0C), #data=17
(12:00:55.730) Boiler -> all, type 0x34 telegram: 08 00 34 00 37 80 00 80 00 21 00 00 01 00 01 34 EF 00 BA 14 00 (CRC=0C), #data=17
```

proddy, 2019-04-28:
I see the current temperatures has a value of x8000 which means it's not set. Those values most be coming from another device, perhaps a mixer?

besmit, 2019-04-28:
I have a basic installation with the boiler and a moduline 300, no extra's in between. I will check the service display if I can actually read these values maybe they are not present at all.

besmit, 2019-04-28:
Checked the service display and manual, both do not mention values regarding the warm water temperature. They do mention the CV water temp (I guess thats the boiler temp)

If I open that hot water tap the logging goes extremly fast and produces the data as attached.
[hrc24.log](https://github.com/proddy/EMS-ESP/files/3125484/hrc24.log)

proddy, 2019-04-30:
Try to see what temps are available as displayed on the front of the boiler and see if they appear in any of the telegrams

besmit, 2019-04-30:
Would love to try that out but the Wemos crashes after x amount of log lines. So i first need to get the Wemos stable and retain it's configuration as posted in topic #106

proddy, 2019-05-11:
@besmit if you're still seeing crashes after leaving verbose logging on for a long time let me know. It may have a memory leak somewhere in the code. Doing a 'system' will show you available memory.

closed 2019-06-09 by proddy