emsesp/EMS-ESP #101: nefit trendline data not right

scheric, 2019-04-18:
## problem
the data i receive from the EMS bus is not right.
- Boiler temperature `(sometimes it is right picture below)`
- Pump modulation
- Central heating
- all thermostat information

### my setup

`nefit trendline` <====> `EMS-OT converter` <===> `Toon`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; `\/`
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;`wemos d1 mini`
```
EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=10, # Tx telegrams=0, # Crc Errors=202

Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline (ProductID:123 Version:06.07)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 45.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2956 times
  Warm Water active time: 3 days 8 hours 27 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 49 C
  Current flow temperature: 49.1 C
  Return temperature: 40.9 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 85 %
  Burner current power: 23 %
  Flame current: 25.9 uA
  System pressure: 1.1 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 90 %
  Boiler circuit pump modulation min power: 40 %
  Outside temperature: ? C
  Boiler temperature: 2073 C
  Pump modulation: 0 %
  Burner # starts: 0 times
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes
  Total UBA working time: 154 days 17 hours 38 minutes

Thermostat stats:
  Thermostat type: TypeID: 0x02 (ProductID:0 Version:?)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to ?
```
[image]

proddy, 2019-04-18:
The "EMS-OT converter" is not properly implemented. See if you can locate what is happening from the telegrams you're getting and the code that is deciphering. In the meantime, I'll probably remove support for the OT converter

closed 2019-04-18 by proddy

scheric, 2019-04-18:
Okay thanks for the info.
But that does not explain why

- Boiler temperature (sometimes it is right picture below)
- Pump modulation
- Central heating

are not working properly.