emsesp/EMS-ESP #102: MQTT / JSON Format

stbuerger, 2019-04-18:
Hi Proddy and thx a lot for your code.
I've just installed it with the complete kit from bbqkees on a 10y Buderus Logamax Plus GB162-25.
The code worked right on and provides data through the telnet interface and via MQTT.
Just when I look at the MQTT JSON for topic "boiler_data" I have a question of understanding the intended format. To me it looks like all items - except ServiceCodeNumber - are sent quoted as as string. ServiceCodeNumber is unquoted and therefor taken as a float number.
At least that is how the published topic is seen by the Mosquitto broker:

{"wWSelTemp":"47","selFlowTemp":"37","outdoorTemp":"19.6","wWActivated":"on","wWComfort":"Hot","wWCurTmp":"43.0","wWCurFlow":"0.0","wWHeat":"off","curFlowTemp":"34.9","retTemp":"34.7","burnGas":"off","heatPmp":"on","fanWork":"off","ignWork":"off","wWCirc":"off","selBurnPow":"100","curBurnPow":"0","sysPress":"1.4","boilTemp":"35.8","pumpMod":"78","ServiceCode":"0Y","ServiceCodeNumber":204}

Given this format it will need quite some effort to feed this to influxdb through the telegraf inputs.mqtt_consumer plugin. Before I start on this I'd like to know ..
- am I doing something wrong
- is this "works-as-designed"
- will this stay as is - or going to be changed in future 

Thx a lot
-stefan

proddy, 2019-04-18:
Wow, I never considered that. You're absolutely right that these values should not be sent as strings in the JSON payload. I'll make the change and push to the version 1.7.0 in under the dev branch. Thanks for pointing this out.

proddy, 2019-04-18:
fixed in 1.7

stbuerger, 2019-04-19:
Thx - that's fast and I fear I cannot keep up. OR - I find out how to update OTA from a couple of 100km away ;-)

closed 2019-04-19 by stbuerger