emsesp/EMS-ESP #104: Buderus GB162 with RC300

moustic999, 2019-04-26:
Hello,

I have ordered board with wemos from bbqkees. It is preinstalled with 1.6.

Here is the setup I have :
Buderus GB162 with RC300 and Web Gateway KM200

from KM200 and main screen, I got following info 

0x08 : MC10 UBA3.5 productId 115 version 05.09
0x09 : BC10 productId 68 version 02.03
0x10 : RC300 productId 158 version NF11.08

I don't know how to find the KM200 Id and productId.

I have written a python api to query/send data via KM200.
now I want to help you to support EMS+ on EMS-ESP.

I don't know yet how to upgrade the wemos, but I will try to read the howto. 

from telnet, nothing has been autodiscovered, I forced boiler and thermostat, here is the info i get 

EMS-ESP system stats:
  System logging set to Basic
  LED is on, Silent mode is off
  0 external temperature sensor connected
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus Connected=yes, Tx is active, # Rx telegrams=0, # Tx telegrams=0, # Crc Errors=4

Boiler stats:
  Boiler type: TypeID: 0x08 (ProductID:0 Version:?)
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 59.6 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2222 times
  Warm Water active time: 28 days 5 hours 4 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 31 C
  Current flow temperature: 25.9 C
  Return temperature: 25.9 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.3 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 15.8 C
  Boiler temperature: 25.9 C
  Pump modulation: 63 %
  Burner # starts: 67097 times
  Total burner operating time: 274 days 20 hours 2 minutes
  Total heat operating time: 246 days 14 hours 58 minutes

Thermostat stats:
  Thermostat type: TypeID: 0x10 (ProductID:0 Version:?)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 20:15:21 26/4/2019
  Mode is set to ?

hope this helps

please tell me how I can help.

moustic999, 2019-04-26:
the link to my Python api for KM200 
https://github.com/moustic999/bosch-thermostat-http-client-python/

moustic999, 2019-04-26:
in log I see lot of line from/to 0x48

`(02:51:59.580) 0x48 -> Boiler, type 0x05 telegram: 48 88 05 2C 01 (CRC=8D), #data=1
(02:51:59.588) Boiler -> 0x48, type 0x05 telegram: 08 48 05 2C (CRC=5F)
Calling scheduled data refresh from EMS devices...
Read operations not yet supported for this model thermostat
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(02:52:00.258) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 51 02 51 21 00 00 03 00 00 9E B0 00 08 AE 00 80 00 (CRC=5B), #data=19
(02:52:00.539) Boiler -> all, type 0x18 telegram: 08 00 18 00 1F 01 04 64 00 01 01 20 60 80 00 02 51 01 04 00 00 0D 30 59 00 CC 00 00 00 (CRC=22), #data=25
(02:52:00.904) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1D 01 01 A6 (CRC=3F), #data=3
(02:52:00.929) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1D 01 A6 (CRC=B1), #data=2
(02:52:00.953) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1D 01 01 A7 (CRC=3E), #data=3
(02:52:00.977) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1D 01 A7 (CRC=B0), #data=2
(02:52:01.001) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1D 01 01 A8 (CRC=31), #data=3
(02:52:01.025) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1D 01 A8 (CRC=BF), #data=2
(02:52:01.280) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1E 01 01 A6 (CRC=27), #data=3
(02:52:01.305) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1E 01 A6 (CRC=BD), #data=2
(02:52:01.528) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1E 01 01 A7 (CRC=26), #data=3
(02:52:01.553) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1E 01 A7 (CRC=BC), #data=2
(02:52:01.832) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1E 01 01 A8 (CRC=29), #data=3
(02:52:01.857) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1E 01 A8 (CRC=B3), #data=2
(02:52:02.328) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1B 02 01 A6 (CRC=03), #data=3
(02:52:02.353) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1B 01 A6 (CRC=A9), #data=2
(02:52:02.560) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1B 02 01 A7 (CRC=02), #data=3
(02:52:02.584) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1B 01 A7 (CRC=A8), #data=2
(02:52:02.634) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4), #data=1
(02:52:03.128) 0x48 -> Thermostat, type 0xFF telegram: 48 90 FF 1B 02 01 A8 (CRC=0D), #data=3
(02:52:03.152) Thermostat -> 0x48, type 0xFF telegram: 10 48 FF 1B 01 A8 (CRC=A7), #data=2
(02:52:05.191) Thermostat -> all, type 0xFF telegram: 10 00 FF 0D 01 A5 00 5C 03 BE (CRC=A8), #data=6
(02:52:05.509) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0), #data=1
(02:52:08.628) 0x48 -> Thermostat, type 0x02 telegram: 48 90 02 00 01 (CRC=09), #data=1
(02:52:08.651) Thermostat -> 0x48, type 0x02 telegram: 10 48 02 00 9E (CRC=FD), #data=1
(02:52:08.876) 0x48 -> Thermostat, type 0x02 telegram: 48 90 02 00 01 (CRC=09), #data=1
(02:52:08.899) Thermostat -> 0x48, type 0x02 telegram: 10 48 02 00 9E (CRC=FD), #data=1
(02:52:09.156) 0x48 -> Thermostat, type 0x02 telegram: 48 90 02 00 01 (CRC=09), #data=1
(02:52:09.179) Thermostat -> 0x48, type 0x02 telegram: 10 48 02 00 9E (CRC=FD), #data=1
(02:52:09.363) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 (CRC=CA), #data=5
(02:52:09.580) 0x48 -> Thermostat, type 0x02 telegram: 48 90 02 00 01 (CRC=09), #data=1
(02:52:09.603) Thermostat -> 0x48, type 0x02 telegram: 10 48 02 00 9E (CRC=FD), #data=1
(02:52:10.341) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 9C 01 03 80 00 00 00 00 3F 01 06 1A 06 09 F3 00 00 00 05 6B 43 00 FD 6C 00 00 (CRC=97), #data=27
(02:52:10.589) Boiler -> all, type 0x1C telegram: 08 00 1C 00 92 06 15 11 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F0), #data=25
(02:52:10.808) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 51 02 51 21 00 00 03 00 00 9E B0 00 08 AE 00 80 00 (CRC=5B), #data=19
(02:52:11.051) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 01 00 00 00 00 (CRC=CA), #data=13
(02:52:11.364) Boiler -> all, type 0x18 telegram: 08 00 18 00 1F 01 04 64 00 01 01 20 60 80 00 02 51 01 04 00 00 0D 30 59 00 CC 00 00 00 (CRC=22), #data=25
(02:52:11.634) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78), #data=1
(02:52:15.413) 0x48 -> Thermostat, type 0xF9 telegram: 48 90 F9 00 11 FF 01 69 08 (CRC=3C), #data=5
(02:52:15.481) Thermostat -> 0x48, type 0xF9 telegram: 10 48 F9 00 FF 01 69 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03 00 02 (CRC=B5), #data=21
(02:52:16.437) 0x48 -> Thermostat, type 0xF9 telegram: 48 90 F9 00 11 FF 01 6A 08 (CRC=3A), #data=5
(02:52:16.505) Thermostat -> 0x48, type 0xF9 telegram: 10 48 F9 00 FF 01 6A 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03 00 02 (CRC=3E), #data=21
(02:52:16.859) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC), #data=1
(02:52:17.437) 0x48 -> Thermostat, type 0xF9 telegram: 48 90 F9 00 11 FF 01 6B 08 (CRC=38), #data=5
(02:52:17.504) Thermostat -> 0x48, type 0xF9 telegram: 10 48 F9 00 FF 01 6B 08 4F 00 00 00 02 00 00 00 02 00 03 00 03 00 03 00 02 (CRC=47), #data=21
(02:52:18.460) 0x48 -> Thermostat, type 0xF9 telegram: 48 90 F9 00 11 FF 01 6C 08 (CRC=36), #data=5
[... cut ...]
(02:52:23.380) 0x48 -> Boiler, type 0xE4 telegram: 48 88 E4 17 02 (CRC=57), #data=1
(02:52:23.388) Boiler -> 0x48, type 0xE4 telegram: 08 48 E4 17 (CRC=BF)
(02:52:28.629) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 04 14 1A 1C 18 04 01 10 FF 00 (CRC=C5), #data=11
(02:52:30.308) Boiler -> all, type 0x34 telegram: 08 00 34 00 3C 02 51 02 51 21 00 00 03 00 00 9E B0 00 08 AE 00 80 00 (CRC=5B), #data=19
(02:52:30.514) Boiler -> all, type 0x18 telegram: 08 00 18 00 1F 01 04 64 00 01 01 20 60 80 00 02 51 01 04 00 00 0D 30 59 00 CC 00 00 00 (CRC=22), #data=25
`

proddy, 2019-04-26:
I will add these devices to the dev branch 1.7. When you’ve reflashed please do an ‘autodetect deep’ to check whats om the EMS bus.

proddy, 2019-04-28:
The 0x48 must be the Web gateway

moustic999, 2019-04-29:
I updated to dev version : here is the result of the autodetect deep : 
Starting a deep EMS device scan. This will take about 1 minute. Please wait...
Finished the deep EMS device scan.

These 2 EMS devices were detected:
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 unknown? (DeviceID:0x48 ProductID:189 Version:04.06)

it is strange that the RC300 has not been detected.
what alse info Can I tey to get ?

moustic999, 2019-04-29:
(00:05:54.743) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9E 08 (CRC=CB)
(00:05:54.806) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9E 08 03 00 00 00 1E 00 00 00 4B 00 00 00 55 00 00 00 4B (CRC=2B), #data=18
(00:05:55.239) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9E 07 (CRC=C4)
(00:05:55.301) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9E 07 03 00 00 00 1E 00 00 00 30 00 00 00 3C 00 00 00 30 (CRC=27), #data=18
(00:05:55.767) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9E 00 (CRC=C3)
(00:05:55.829) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9E 00 4B 00 00 00 01 00 00 00 01 00 1B 00 06 00 1B 00 01 (CRC=0A), #data=18
(00:05:56.090) Sending read of type 0x02 to 0x7B: telegram: 0B FB 02 00 20 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FA 6E 05 00 68 ED FF 3F 0
(00:05:57.466) Gateway -> Thermostat, type 0x0103 telegram: 48 90 FF 00 01 03 74 (CRC=01)
(00:05:57.492) Thermostat -> 0x48, type 0x0374 telegram: 10 48 FF 00 03 74 (CRC=13)
(00:05:57.564) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 44 02 44 20 00 00 03 00 00 9F 14 00 08 B4 00 80 00 (CRC=46), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:05:57.795) Boiler -> all, type 0x18 telegram: 08 00 18 00 1D 01 6A 64 00 01 01 20 60 80 00 02 44 01 67 00 00 0D 30 59 00 CC 00 00 00 (CRC=4A), #data=25
<--- UBAMonitorFast(0x18) received
(00:05:58.090) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 25 01 01 A8 (CRC=E8)
(00:05:58.116) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 25 01 A8 (CRC=5F)
(00:05:58.386) Gateway -> Thermostat, type 0x0103 telegram: 48 90 FF 05 01 03 74 (CRC=29)
(00:05:58.412) Thermostat -> 0x48, type 0x0374 telegram: 10 48 FF 05 03 74 (CRC=07)
(00:05:59.240) Sending read of type 0x02 to 0x7C: telegram: 0B FC 02 00 20 37 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 48 7B 05 00 68 ED FF 3F 0
(00:06:07.589) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 44 02 44 20 00 00 03 00 00 9F 14 00 08 B4 00 80 00 (CRC=46), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:06:07.795) Boiler -> all, type 0x18 telegram: 08 00 18 00 1D 01 6A 64 00 01 01 20 60 80 00 02 44 01 66 00 00 0D 30 59 00 CC 00 00 00 (CRC=2E), #data=25
<--- UBAMonitorFast(0x18) received
(00:06:09.390) Sending read of type 0x02 to 0x7D: telegram: 0B FD 02 00 20 3F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 EE A2 05 00 68 ED FF 3F 0
(00:06:17.589) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 44 02 44 20 00 00 03 00 00 9F 14 00 08 B4 00 80 00 (CRC=46), #data=19
<--- UBAMonitorWWMessage(0x34) received
(00:06:17.795) Boiler -> all, type 0x18 telegram: 08 00 18 00 1D 01 69 64 00 01 01 20 60 80 00 02 44 01 66 00 00 0D 30 59 00 CC 00 00 00 (CRC=56), #data=25
<--- UBAMonitorFast(0x18) received
(00:06:19.490) Sending read of type 0x02 to 0x7E: telegram: 0B FE 02 00 20 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 62 CA 05 00 68 ED FF 3F 0
(00:06:22.262) Gateway -> Boiler, type 0x05 telegram: 48 88 05 2C 01 (CRC=8D)
(00:06:22.269) Boiler -> 0x48, type 0x05 telegram: 08 48 05 2C (CRC=5F)
(00:06:22.986) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A6 (CRC=3F)
(00:06:23.014) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1D 01 A6 (CRC=B1)
(00:06:23.242) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A7 (CRC=3E)
(00:06:23.270) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1D 01 A7 (CRC=B0)
(00:06:23.538) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A8 (CRC=31)
(00:06:23.567) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1D 01 A8 (CRC=BF)
(00:06:24.042) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A6 (CRC=27)
(00:06:24.070) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1E 01 A6 (CRC=BD)
(00:06:24.266) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A7 (CRC=26)
(00:06:24.294) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1E 01 A7 (CRC=BC)
(00:06:24.610) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A8 (CRC=29)
(00:06:24.637) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1E 01 A8 (CRC=B3)
(00:06:24.840) Sending read of type 0x02 to 0x7F: telegram: 0B FF 02 00 20 2F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 48 DF 05 00 68 ED FF 3F 0
(00:06:25.114) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A6 (CRC=03)
(00:06:25.141) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1B 01 A6 (CRC=A9)
(00:06:25.362) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A7 (CRC=02)
(00:06:25.389) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1B 01 A7 (CRC=A8)
(00:06:25.666) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A8 (CRC=0D)
(00:06:25.693) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1B 01 A8 (CRC=A7)
se(00:06:27.572) Boiler -> all, type 0x19 telegram: 08 00 19 00 00 70 01 66 80 00 00 00 03 42 01 06 7C 06 0F 25 00 00 00 05 70 11 00 FD C8 00 00 (CRC=A2), #data=27
<--- UBAMonitorSlow(0x19) received
(00:06:27.815) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 6C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E7 EA 05 00 68 ED FF 3F 0
(00:06:28.120) Boiler -> all, type 0x1C telegram: 08 00 1C 00 92 06 15 11 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F0), #data=25

(00:06:28.439) Boiler -> all, type 0x34 telegram: 08 00 34 00 0A 02 44 02 44 20 00 00 03 00 00 9F 14 00 08 B4 00 80 00 (CRC=46), #data=19

moustic999, 2019-04-29:
strange also that 0x48 is not always considered as the gateway ? when sending data to it ?

proddy, 2019-04-29:
the 0x48 is just a line I missed when coding the debug messages. I'll add it later.
So it seems because you have a gateway you can't talk the RC300 directly which kinda makes sense as it takes over as the master controller. The problem now is figuring out which telegram types corresponds to which function. Start with setting the temperature on the thermostat and seeing what is sent on 0xFF01

moustic999, 2019-04-29:
note also that RC300 is within boiler, it is not set up as room controller.

I can disconnect the Web gateway and check what happens.

moustic999, 2019-04-29:
I disconnected the gateway an autodetect have found properly boiler and thermostat types : 
EMS-ESP system stats:
  System logging set to Basic
  LED is on, Silent mode is off
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: Poll=15 ms, # Rx telegrams read=0, # CRC errors=4
  Tx: available, Tx delay is off, # Tx telegrams sent=0

Boiler stats:
  Boiler: Nefit Topline Compact/Buderus GB162 (ProductID:115 Version:05.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 53.8 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2228 times
  Warm Water active time: 28 days 6 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 38 C
  Current flow temperature: 36.2 C
  Return temperature: 35.9 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.3 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 16.8 C
  Boiler temperature: 36.2 C
  Pump modulation: 69 %
  Burner # starts: 67204 times
  Total burner operating time: 275 days 18 hours 29 minutes
  Total heat operating time: 247 days 12 hours 29 minutes

Thermostat stats:
  Thermostat: RC300/RC310 (ProductID:158 Version:11.08)
  Setpoint room temperature: 0.5 C
  Current room temperature: ? C
  Thermostat time is 15:25:48 29/4/2019
  Mode is set to ?

moustic999, 2019-04-29:
once I connect back the Gateway, nothing is discovered anymore.

moustic999, 2019-04-29:
I made some trials, 
For info I have only 1 Heating Circuit.
To change the mode ( auto or Manual ) 

set to auto : 
(00:02:24.523) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 00 01 B9 FF (CRC=6E)
then this comes
(00:02:26.155) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 08 01 B9 FF (CRC=C3)
(00:02:26.347) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 15 01 A5 03 (CRC=EF)

set to manual : 
(00:02:30.651) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 00 01 B9 00 (CRC=91)
then this comes
(00:02:31.272) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 02 (CRC=16)

in MANUAL mode : 
set manualRoomsetpoint : 

set to 20°C
(00:19:38.092) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 0A 01 B9 28 (CRC=E9)

set to 20.5°C
(00:22:27.249) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 0A 01 B9 29 (CRC=E8)
(00:22:28.251) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 29 (CRC=75)

set to 21°C
same kind of message but last byte = 2A 

I also checked about changing temp when heating is in auto, but didn't find anything special in logs.
KM200 gateway <> rc300 is always sending message on bus. 
it is hard to correlate values easily.

after GW-> Thermostat 01B9, there is always a thermostat ->All 01A5 triggered

proddy, 2019-04-29:
that's it, type 0x01B9 
 offset 0A, the 1st byte will show the temps. Hex 28 = 40  = 20 c (divided by 2). And Hex 29 = 41 = 20.5 c
 offset 08 is the mode, 1st byte is auto=FF and manual=00

moustic999, 2019-04-29:
So then in the Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 29 (CRC=75)
type 01A5 
offset 03 
show temp also

proddy, 2019-04-29:
yes, how often are 0x01B9 and 0x01A5 broadcasted? Is it regular or only when the thermostat is manually adjusted by hand? We need to decide where to the correct values from.

moustic999, 2019-04-29:
from what I can see, none of them are broadcasted on a regular basis. 
0x01B9 is fired only when I change the temp or the mode from the gateway

0x01A5 is triggered when a change happen in the thermostat, then it send this message with new values 

there is tons of exchange between gateway and thermostat, also between gateway and boiler.
I can get a big log of these if this is usefull.

moustic999, 2019-04-30:
if needed, I can launch request command from the gateway to try to capture the message on the bus, but tell me what I can try, that will be the most useful for you !

proddy, 2019-04-30:
We'll stick with monitoring 0x01A5 then. The problem is how to read the results adhoc. I haven't figured out how to send Read commands to the EMS+ yet (mainly because I don't have any EMS+ at home!)

moustic999, 2019-04-30:
I think that 0x01A5 is 1 of "send status" command
look at this log : 
00:06:22.986) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A6 (CRC=3F)
(00:06:23.014) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1D 01 A6 (CRC=B1)
(00:06:23.242) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A7 (CRC=3E)
(00:06:23.270) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1D 01 A7 (CRC=B0)
(00:06:23.538) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1D 01 01 A8 (CRC=31)
(00:06:23.567) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1D 01 A8 (CRC=BF)
(00:06:24.042) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A6 (CRC=27)
(00:06:24.070) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1E 01 A6 (CRC=BD)
(00:06:24.266) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A7 (CRC=26)
(00:06:24.294) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1E 01 A7 (CRC=BC)
(00:06:24.610) Gateway -> Thermostat, type 0x0101 telegram: 48 90 FF 1E 01 01 A8 (CRC=29)
(00:06:24.637) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1E 01 A8 (CRC=B3)
(00:06:24.840) Sending read of type 0x02 to 0x7F: telegram: 0B FF 02 00 20 2F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 48 DF 05 00 68 ED FF 3F 0
(00:06:25.114) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A6 (CRC=03)
(00:06:25.141) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1B 01 A6 (CRC=A9)
(00:06:25.362) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A7 (CRC=02)
(00:06:25.389) Thermostat -> 0x48, type 0x01A7 telegram: 10 48 FF 1B 01 A7 (CRC=A8)
(00:06:25.666) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A8 (CRC=0D)
(00:06:25.693) Thermostat -> 0x48, type 0x01A8 telegram: 10 48 FF 1B 01 A8 (CRC=A7)

the gateway send 0x0101, offset 1D last byte A6, thermostat answer 0x01A6 offset 1D with no value
then message is repeated for next value ( A7, A8 ) 
then gateway begins again to 0x0201

I think this is the read command 0x0101 or 0x0102, maybe this could also indicate which heating circuit to query ?
based on this, can you help me to build the command i would need to send to the thermostat so that it could answer with  0105 answer, but which offset to try ?

moustic999, 2019-04-30:
@proddy can you explain this :
(00:06:25.114) Gateway -> Thermostat, type 0x0201 telegram: 48 90 FF 1B 02 01 A6 (CRC=03)
(00:06:25.141) Thermostat -> 0x48, type 0x01A6 telegram: 10 48 FF 1B 01 A6 (CRC=A9)
gateway is 0x48, RC300 is 0x10
What is 0x90 ? 
in the first telegram gateway send to 90 ??? known as thermostat ?? next line is thermostat to Gateway but then thermostat is 0x10 
strange or normal ?

proddy, 2019-04-30:
It's explained in my documentation. 90 is the 0x10 but with the 8th bit set meaning its a Read command, or a request to fetch data from the thermostat. But...I've been looking at those ems+ messages and I think I have the logic wrong. I need some time to figure it out.

moustic999, 2019-04-30:
got it ! I re-read carefully
for the EMS+ 0xFF or 0xF9, do you have any idea why we can have different value ?

(00:21:43.446) Gateway -> Thermostat, type 0x0103 telegram: 48 90 FF 01 01 03 71 (CRC=0C)
(00:21:43.475) Thermostat -> 0x48, type 0x0371 telegram: 10 48 FF 01 03 71 (CRC=12)
(00:21:44.171) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 AF 08 (CRC=A9)
(00:21:44.234) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 AF 08 0B 00 00 00 14 00 00 00 19 00 00 00 4D 00 00 00 19 (CRC=99), #data=18

I think the typeId is not correct in 0x11FF  that always occurs when there is F9 and in these case we got multiple data. I suspect these data are temperatures for something, I will try to correlate them later

proddy, 2019-04-30:
correct, I'm thinking we need to interpret the messages differently and read up until the first 0xFF. Then its the offset followed by the 2-byte type ID and lastly the data.

So `48 90 F9 00 11 FF 01 AF 08` is actually a request from the Gateway (x48) to read data from the Thermostat (x10/x90) for 17 bytes of data (x11) from type 0xAF08. I'm not sure what the x01 is yet after the xFF, i.e.
 **<gateway 48> <thermostat 90> 0xF900 <17-bytes 11> 0xFF <offset 01> <type 0xAF08>**

Then the Thermostat will respond with `10 48 F9 00 FF 01 AF 08 0B 00 00 00 14 00 00 00 19 00 00 00 4D 00 00 00 19` which can be broken down into:
 **<thermostat 10> <gateway 48> 0xF900 0xFF <offset x01> <type 0xAF08> <the 17 bytes of data>**

what do you think?

moustic999, 2019-04-30:
here are a bunch of message in such format
you could be right ! that would mean that there is no offset in this mode ?
(04:37:34.626) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9C 07 (CRC=C0)
(04:37:34.695) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9C 07 03 00 00 00 1E 00 00 00 30 00 00 00 3C 00 00 00 30 (CRC=D5), #data=18
(04:37:35.154) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9C 00 (CRC=C7)
(04:37:35.223) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9C 00 4B 00 00 00 01 00 00 00 01 00 1B 00 06 00 1B 00 01 (CRC=F8), #data=18
(04:37:43.954) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9D 08 (CRC=CD)
(04:37:44.018) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9D 08 03 00 00 00 1E 00 00 00 4B 00 00 00 55 00 00 00 4B (CRC=A0), #data=18
(04:37:44.674) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9D 07 (CRC=C2)
(04:37:44.738) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9D 07 03 00 00 00 1E 00 00 00 30 00 00 00 3C 00 00 00 30 (CRC=AC), #data=18
(04:37:45.274) Gateway -> Thermostat, type 0x11FF telegram: 48 90 F9 00 11 FF 01 9D 00 (CRC=C5)
(04:37:45.337) Thermostat -> 0x48, type 0xFF01 telegram: 10 48 F9 00 FF 01 9D 00 4B 00 00 00 01 00 00 00 01 00 1B 00 06 00 1B 00 01 (CRC=81), #data=18

also while playing with mode I captured this line :
(04:43:57.003) Thermostat -> all, type 0xFF01 telegram: 10 00 F7 00 FF 01 B9 35 19 (CRC=A4)
that time it is an F7

moustic999, 2019-04-30:
an finally here is the result from playing this evening : 

in AUTO Mode, setting temporary temp to 21.5 == 0x2B
(03:43:45.965) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 08 01 B9 2B (CRC=FA)
(03:43:46.445) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 08 01 B9 2B (CRC=17)

(03:43:49.547) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 2B (CRC=77)
<--- RCPLUSStatusMessage(0x1A5) received
(03:43:49.739) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 06 01 A5 2B (CRC=5F)
<--- RCPLUSStatusMessage(0x1A5) received

SET to Manual mode ( manual temp is 20.5 ) == 0x29
(03:49:44.575) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 00 01 B9 00 (CRC=91)
(03:49:45.825) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 02 (CRC=16)
(03:49:51.726) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 29 (CRC=75)
<--- RCPLUSStatusMessage(0x1A5) received
 (03:49:51.918) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 06 01 A5 29 (CRC=5D)
<--- RCPLUSStatusMessage(0x1A5) received

MANUAL Mode, change temporary temp to 23  == 0x2E
(04:00:15.126) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 08 01 B9 2E (CRC=FF)
(04:00:18.483) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 08 01 B9 2E (CRC=12)
nothing more : normal because temporary temp is used only in auto mode

MANUAL Mode, change manual temp to 18.5 == 0x25
(04:04:05.003) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 0A 01 B9 25 (CRC=E4)
(04:04:05.258) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 0A 01 B9 25 (CRC=09)
(04:04:06.608) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 01 (CRC=00)
(04:04:08.712) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 25 (CRC=79)
<--- RCPLUSStatusMessage(0x1A5) received
(04:04:08.906) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 06 01 A5 25 25 (CRC=87)
<--- RCPLUSStatusMessage(0x1A5) received

MANUAL Mode, change eco level to 17 == 0x22
(04:14:11.034) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 04 01 B9 22 (CRC=93)
(04:14:14.466) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 04 01 B9 22 (CRC=7E)
again as currently in manual nothing more happen...

MANUAL Mode, change confort2 level to 22 == 0x2C
(04:19:52.085) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 02 01 B9 2C (CRC=AD)
(04:19:54.027) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 02 01 B9 2C (CRC=40)

Set to AUTO mode ( at that time should be in ECO --> 16°C) ( confort2 is set to 22 )
(04:22:11.563) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 00 01 B9 FF (CRC=6E)
(04:22:13.749) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 03 01 A5 20 (CRC=7C)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:13.943) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 06 01 A5 20 2C (CRC=84)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:14.972) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 08 01 B9 FF (CRC=C3)
(04:22:15.164) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 15 01 A5 04 (CRC=E8)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:16.347) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 (CRC=88)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:17.123) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 00 01 B9 FF (CRC=83)
(04:22:18.194) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 12 01 B9 00 (CRC=EC)
(04:22:18.388) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 01 99 (CRC=93)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:18.874) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 01 (CRC=15)
<--- RCPLUSStatusMessage(0x1A5) received
(04:22:25.694) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 04 01 A5 16 (CRC=72)
<--- RCPLUSStatusMessage(0x1A5) received

proddy, 2019-04-30:
I have a theory about the F7 too. I think its best if I create a wiki page in use that to capture all findings

proddy, 2019-05-01:
I could use your brain power to help figure out https://github.com/proddy/EMS-ESP/wiki/Deciphering-EMS-Plus

moustic999, 2019-05-02:
with the help of this post, we could progress 
https://www.mikrocontroller.net/topic/309075?goto=3928825#3895776

example : 
10 48 F9 00  FF 019E 00 4B 00 00 00 01 00 00 00 01 00 1B 00 06 00 1B 00 01

sender : 10
receiver : 48
F9 00
FF
type ? 019E
??? : 004B
data1 : 00 00 00 01
data2 : 00 00 00 01
data3 : 00 1B 00 06
data4 : 00 1B 00 01

moustic999, 2019-05-02:
check this : https://github.com/norberts1/hometop_HT3/blob/master/HT3/sw/lib/ht_discode.py
starting line 200, it seems they have some proper decoding for EMS2

moustic999, 2019-05-02:
I read the link in my previous answer ! 
very intereisting, almost everything is decoded ! 
Heatronic 3 is EMS2 ;-)

proddy, 2019-05-02:
yes! Reading it now...it's interesting. I need to google translate the German to understand what the values mean. Does it support thermostats too?

moustic999, 2019-05-02:
supporting these deviceIds :     
deviceaddress_white_list = [0x08, 0x0a, 0x0b, 0x0c, 0x0d, 0x10, 0x18, 0x19, 0x1a,
                                0x20, 0x21, 0x22, 0x23, 0x28, 0x29,
                                0x30, 0x31, 0x48]
and refers 0x48 as a Modem later in the code !
So It looks, very promising ;-)
I will deep into tonight and tomorrow. 
@proddy, Do you have a Discord to discuss ? It will be better than comments in an issue ;-)

proddy, 2019-05-02:
Discord username is Proddy#0739

closed 2019-06-09 by proddy