emsesp/EMS-ESP #106: Unable to read config file

besmit, 2019-04-28:
Compiled the dev branch 1.7b10 from GIT for Wemos D1 mini V2. Everytime the Wemos is rebooted or power cycled the settings are lost which requires me to re-enter all the mqtt parameters.

When logging in to the telnet session I get the following information:

[FS] Failed to read config file

Any suggestions why it doesn't seem to save the config file or is unable to open it?

proddy, 2019-04-28:
could be many things, most likely the SPIFFS segments are not reserved. Are you using PlatformIO and have all the latest libraries installed? To be safe try 'platformio run -t erase' which basically formats the Wemos D1 so you can flash it again.

besmit, 2019-04-29:
From what I can see all libraries are up to date:

```Library Storage: c:\Users\jobexcel\EMS-ESP\.piolibdeps
Updating ArduinoJson                     @ 6.10.1         [Up-to-date]
Updating AsyncMqttClient                 @ 0.8.2          [Up-to-date]
Updating AsyncTCP                        @ 1.0.3          [Up-to-date]
Updating CRC32                           @ 2.0.0          [Up-to-date]
Updating CircularBuffer                  @ 1.3.0          [Up-to-date]
Updating EEPROM_Rotate                   @ 0.9.2          [Up-to-date]
Updating ESPAsyncTCP                     @ 1.2.0          [Up-to-date]
Updating JustWifi                        @ 2.0.2          [Up-to-date]
Updating OneWire                         @ 2.3.4          [Up-to-date]
```

I also issued the 'pio run -t erase' command which completes successful 
```pio run -t erase
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.10.1
|-- <OneWire> 2.3.4
|-- <MyESP>
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM_Rotate> 0.9.2
|   |   |-- <EEPROM> 1.0
|   |-- <ArduinoJson> 6.10.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
Looking for serial port...
Auto-detected: COM6
Ready for erasing
============================================================================================== [SUCCESS] Took 3.68 seconds ==============================================================================================
```

Afterwards I run the 'pio run -t upload' which is also successful. 
```pio run -t upload
Processing d1_mini (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <CRC32> 2.0.0
|-- <CircularBuffer> 1.3.0
|-- <JustWifi> 2.0.2
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.10.1
|-- <OneWire> 2.3.4
|-- <MyESP>
|   |-- <JustWifi> 2.0.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <ESPAsyncTCP> 1.2.0
|   |   |   |-- <ESP8266WiFi> 1.0
|   |-- <EEPROM_Rotate> 0.9.2
|   |   |-- <EEPROM> 1.0
|   |-- <ArduinoJson> 6.10.1
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <DNSServer> 1.1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS> 1.2
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <TelnetSpy>
|   |   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
Retrieving maximum program size .pioenvs\d1_mini\firmware.elf
Checking size .pioenvs\d1_mini\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [======    ]  60.6% (used 49652 bytes from 81920 bytes)
PROGRAM: [====      ]  41.1% (used 429356 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pioenvs\d1_mini\firmware.bin
Uploading 433504 bytes from .pioenvs\d1_mini\firmware.bin to flash at 0x00000000
................................................................................ [ 18% ]
................................................................................ [ 37% ]
................................................................................ [ 56% ]
................................................................................ [ 75% ]
................................................................................ [ 94% ]
........................                                                         [ 100% ]
============================================================================================= [SUCCESS] Took 21.21 seconds =============================================================================================

```

However after setting the mqtt settings and issuing a reboot results in
```reboot
* R�
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
[FS] Failed to read config file
```

proddy, 2019-04-29:
that is very odd. Everything looks fine with your build & upload. Its not a 8MB Wemos Pro is it by any chance? I'll take a look tonight to see if I can reproduce.

Also, did you ever get it to build before with the 1.7 dev version or is this the first time?

besmit, 2019-04-29:
Nope it's a Wemos D1 mini (got it from bbqkees as a kit) 

It was originally installed with the 1.6 stable version via bbqkees which worked except it didn't recognize the boiler so that was what made me move to the 1.7 dev branch.

I tried the 'erase' option with verbose option and I do get something about AsyncTCP

```Installing dependencies
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Platform incompatible library C:\Users\xxx\EMS-ESP\.piolibdeps\AsyncTCP_ID1826
CMollected 36 compatible librariesore details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode

Scanning dependencies...
Skip platform incompatible dependency {u'platforms': [u'espressif32'], u'version': u'^1.0.0', u'name': u'AsyncTCP'}
```

proddy, 2019-04-30:
is it still crashing? Does `platformio update` show you're using espressif8266 version 2.0.4? Did you try `platformio.exe run --target clean` first? You can ignore the library incompatibility warning as its only for the ESP32 version of my MyESP library which you arn't using.

proddy, 2019-04-30:
also when connected to Telnet type `system` and let us know what it shows you

besmit, 2019-04-30:
For 'pio update' i get the following
```
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating contrib-piohome                 @ 2.0.2          [Up-to-date]
Updating tool-pioplus                    @ 2.1.4          [Up-to-date]

Platform Manager
================
Platform Espressif 8266
--------
Updating espressif8266                   @ 2.0.4          [Up-to-date]
Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
Updating tool-esptool                    @ 1.413.0        [Up-to-date]
Updating tool-espotapy                   @ 1.1.3          [Up-to-date]
Updating framework-arduinoespressif8266  @ 2.20500.190223 [Up-to-date]
Updating toolchain-xtensa                @ 1.40802.0      [Up-to-date]
```
and the system command outputs the following

```ESP8266 System stats:

 [APP] EMS-ESP version: 1.7.0b10
 [APP] MyESP version: 1.1.7
 [APP] Build timestamp: 2019-04-30 18:56:12
 [APP] Uptime: 0 days, 0 hours, 1 minutes, 17 seconds
 [APP] System Load: 0%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.xxx.xxx
 [WIFI] WiFi signal strength: 60%
 [WIFI] WiFi MAC: 84:0D:xx:xx:xx:x
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 3.0.0-dev(c0f7b44)
 [SYSTEM] CPU chip ID: 0xABFAF7
 [SYSTEM] Core version: 2_5_0
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [FLASH] Flash chip ID: 0x164020
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 433504
 [MEM] Max OTA size: 2707456
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 20968
```

proddy, 2019-04-30:
that's strange as everything looks fine here. I'll run some more tests. No one else has reported back anything strange so far, but that doesn't rule out a bug.

besmit, 2019-04-30:
The strange thing is that I can set all the parameters and recall them with the 'set' command but after I issue the 'reboot' the first line after loading the kernel I see on the serial console is ```[FS] Failed to read config file```, looking at the code this would mean something goes wrong with decoding the JSON. My WiFi network uses a password which contains spaces and and special characters, could that be an issue?

proddy, 2019-04-30:
good point. I did test with all strange characters in the password it works on my system. could you find that line in MyESP and change it to
` myDebug_P(PSTR("[FS] Failed to read config file. Error %s"), error.c_str());`

and check the error against https://arduinojson.org/v6/api/misc/deserializationerror/

besmit, 2019-04-30:
Just recompiled and got the following error `[FS] Failed to read config file. Error NoMemory` checking the documentation this means 
```
DeserializationError::NoMemory
The JsonDocument is too small; you need to increase its capacity.
```

besmit, 2019-04-30:
Some more debugging info. The spiffs JSON doc would be allowed to be 600 of what in MyESP.h? 
```
// SPIFFS
#define SPIFFS_MAXSIZE 600 // https://arduinojson.org/v6/assistant/
```

I added an extra debug string in the file reading part which gives me the following after I have entered all the mqtt settings and reboot.

```
myDebug_P(PSTR("[FS] Failed to read config file. Size %i"), size);
 myDebug_P(PSTR("[FS] Failed to read config file. Error %s"), error.c_str());

[FS] Failed to read config file. Size 419
[FS] Failed to read config file. Error NoMemory
```
Which I guess is within the boundaries of the 600

proddy, 2019-04-30:
yup. One thing you could try is to print out the config file. In MyESP, cpp find the _fs_setup() and just before the _fs_loadConfig() is called insert 

_fs_printConfig();

besmit, 2019-04-30:
Just did what you suggested, before the loadConfig is issued the file has the required contents, but as soon as it is being converted to JSON it goes blank. All settings are set to their respective defaults.

The error seems to occur in the conversion to a JSON document. I have have entered the JSON data from the config file in the online size calculator at https://arduinojson.org/v6/assistant/ which gives me a size requirement of 617 and not 419 (actual file size). So I changed the SPIFFS_MAXSIZE to 1000, but still no luck.

```
 StaticJsonDocument<SPIFFS_MAXSIZE> doc;
    JsonObject                         json = doc.to<JsonObject>();

    // Deserialize the JSON document
    DeserializationError error = deserializeJson(doc, configFile);
```

proddy, 2019-05-01:
great, and good catch. I'll increase the size of the buffer in 1.7

closed 2019-05-01 by proddy