emsesp/EMS-ESP #114: ultrasonic sensors on the wemos?

GARLAGANDIS, 2019-05-22:
Is it possible to configure a few pins of the Wemos as inputs for sensors? I would like to build a level gauge for my oil tanks with a few ultrasonic sensors.

proddy, 2019-05-22:
yes, we could extend the project to support external custom code but you'll need to program the external calls yourself e.g. enabling the gpio in setup() and capturing the sending MQTT data in the loop() section. How many pins would you need?

susisstrolch, 2019-05-22:
Stupid question...
why not use the burner runtime (the nozzle usually have a throughput ~2.1l/hr) to calculate the consumption? I do it since 3 years - and it's much more precise than a ultrasonic sensor...

closed 2019-06-09 by proddy