emsesp/EMS-ESP #117: Unrecognized device found. DeviceID 0x09, ProductID 152, Version 03.01

rskallies, 2019-05-24:
Running the latest dev with EMS+ / TX fixes my devices are found:

Junkers Cerapur ZSB 14-5C boiler recognized by autodetect as 
Model Buderus Logamax U122 (DeviceID:0x08 ProductID:203 Version:02.04), most values seems to be ok but some are not. 

My thermostat a Bosch EasyControl CT200 is still a unknown device ->

0x09 -> me, type 0x02 telegram: 89 0B 02 00 98 03 01 (CRC=75) #data=3
Unrecognized device found. DeviceID 0x09, ProductID 152, Version 03.01

currently communication with the thermostat is impossible  because of XMPP cloud stuff, anyway it would be nice to add at least the proper device id to the project.

Thank you
René

proddy, 2019-05-26:
Hi @rskallies. Thanks for the info, I've added the device and the onboard controller which is type 0x09

Which values from the boiler don't seem to be correct? The Junkers heatronic3 protocol behaves a little differently but if you can provide examples we can work on the mapping.

rskallies, 2019-05-28:
Thank you for clarification about the device unknown device ID was the controller and of course for adding it.  Because TX was not working the last few days until now (latest dev with ems-junkers-1.8.0b7.patch and tx_mode 3) I didn't play further.

I just have finished a deep EMS scan and the results show two boilers, were only the last in the list is correct, besides the fact that I have only one installed ;-)

autodetect deep

These 3 EMS devices were detected:
 Buderus GB172/Nefit Trendline (DeviceID:0x08 ProductID:123 Version:06.03)
 Junkers Controller (DeviceID:0x09 ProductID:152 Version:03.01)
 Buderus Logamax U122/Junkers Cerapur (DeviceID:0x08 ProductID:203 Version:02.04)

proddy, 2019-05-28:
this is strange. In https://github.com/proddy/EMS-ESP/issues/45 there were also issues with the Bosch Easy. 

Could you try a few things
1. leave `log v` on and see if there are messages coming from the Thermostat. See if the Device id is 2 
2. with `log v` on do the `autodetect deep` again and capture the log. Sorry, it'll be huge but I'm interested in seeing what comes back. Best attach the file to the github issue.

You shouldn't be seeing 2 boilers so this is a bug that I need to squash. Thanks for helping out

rskallies, 2019-05-29:
> 1. leave `log v` on and see if there are messages coming from the Thermostat. See if the Device id is 2

Nothing is printed from the device ID 2. 

> 2. with `log v` on do the `autodetect deep` again and capture the log. Sorry, it'll be huge but I'm interested in seeing what comes back. Best attach the file to the github issue.

See attached log file.

> You shouldn't be seeing 2 boilers so this is a bug that I need to squash. Thanks for helping out

I'm really want to help here! 

[autodetect_deep.log](https://github.com/proddy/EMS-ESP/files/3231112/autodetect_deep.log)

proddy, 2019-06-08:
could you try again with the latest dev build? do a `set tx_mode to 2` and do an `autodetect deep` and see what it detects

rskallies, 2019-06-10:
Running latest EMS-ESP version: 1.8.0b12:

After a fresh reboot the  "simple" autodetect routine prints this:

autodetect
Started scan on EMS bus for known devices
Device found. Model Junkers Controller with DeviceID 0x09, ProductID 152, Version 03.01
Boiler found. Model Buderus Logamax U122/Junkers Cerapur (DeviceID:0x08 ProductID:203 Version:02.04)
Boiler found. Model Buderus GB172/Nefit Trendline (DeviceID:0x08 ProductID:123 Version:06.03)

using autodetect deep just fills up the queue and prints

Starting a deep EMS device scan. This can take up to 2 minutes. Please wait...
Finished the deep EMS device scan.

The same behavior is seen using tx_mode 3

Should I post a complete log for the autodetect step?

proddy, 2019-06-10:
yes please.
`set tx_mode 2`
`log v`
`autodetect`
leave it for about 10-15 seconds and then send over the log. I'm interested in all the yellow lines which have "-> me".

rskallies, 2019-06-10:
Here are the relevant lines, the full log is attached below.

(00:01:26.026) Boiler -> 0x18, type 0x33 telegram: 88 18 33 01 FF 38 FB 00 28 00 02 46 00 (CRC=B0) #data=9
(00:01:28.288) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 20 (CRC=B4)
(00:01:28.322) 0x09 -> me, type 0x02 telegram: 89 0B 02 00 98 03 01 (CRC=75) #data=3
<--- Version(0x02)
Device found. Model Junkers Controller with DeviceID 0x09, ProductID 152, Version 03.01
--
--
(00:01:44.790) Boiler -> all, type 0x2A telegram: 88 00 2A 00 00 00 00 00 03 00 00 00 BA 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=1D) #data=21
(00:01:45.563) Sending read of type 0x02 to 0x18: telegram: 0B 98 02 00 20 (CRC=3C)
(00:01:45.623) 0x18 -> me, type 0x02 telegram: 98 0B 02 00 CB 02 04 00 00 00 00 00 00 01 00 (CRC=29) #data=11
<--- Version(0x02)
Boiler found. Model Buderus Logamax U122/Junkers Cerapur (DeviceID:0x08 ProductID:203 Version:02.04)
--
--
(00:02:28.485) 0x18 -> Boiler, type 0x1A telegram: 98 08 1A 00 00 00 (CRC=F2) #data=2
(00:02:28.964) Sending raw: 8B 88 02 00 20 00
(00:02:29.004) Boiler -> me, type 0x02 telegram: 88 0B 02 00 7B 06 03 00 00 00 00 00 00 00 (CRC=A3) #data=10
<--- Version(0x02)
Boiler found. Model Buderus GB172/Nefit Trendline (DeviceID:0x08 ProductID:123 Version:06.03)

[log.txt](https://github.com/proddy/EMS-ESP/files/3271170/log.txt)

proddy, 2019-06-10:
ok, this was tricky. I made some changes - if you have time can you try again?

rskallies, 2019-06-10:
Yes, this worked! Thank you for digging into this 👍
[log.txt](https://github.com/proddy/EMS-ESP/files/3271735/log.txt)

```
(00:01:11.127) 0x09 -> me, type 0x02 telegram: 89 0B 02 00 98 03 01 (CRC=75) #data=3
<--- Version(0x02)
Device found: Junkers Controller with DeviceID 0x09, ProductID 152, Version 03.01
--
(00:01:30.503) Thermostat -> me, type 0x02 telegram: 98 0B 02 00 CB 02 04 00 00 00 00 00 00 01 00 (CRC=29) #data=11
<--- Version(0x02)
Thermostat found: Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.04)
--
(00:01:59.234) Boiler -> me, type 0x02 telegram: 88 0B 02 00 7B 06 03 00 00 00 00 00 00 00 (CRC=A3) #data=10
<--- Version(0x02)
Boiler found: Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:06.03)
```

proddy, 2019-06-10:
great!

closed 2019-06-10 by proddy