emsesp/EMS-ESP #120: Support for multiple heating circuits

KristofRobberechts, 2019-05-27:
Hi Proddy,

I have a set up with a Junker Cerapur, connected to a MM200 mixer and 2 Bosch CR 100 thermostats. I have two separate heating circuits I can control with the mixer. The MM200 has 2 channels. I've been experimenting a bit with giving different device ID's to my devices.

When I select circuit 1 in my thermostat, it uses device address `0x18`. However, when I select circuit 2, the device ID becomes `0x19`. I can go up until circuit 8, which gives me `0x1F`. On the MM200, I can select addresses from `0x20` to `0x27`.

Interesting as well is that the CR 100 can also be selected as a remote control for a C 400 / 800, in that case it uses addresses `0x38` to `0x3F`. The product ID stays the same.

Would it be possible to add support for having multiple thermostats/mixers? I wouldn't keep using the device ID for auto-detecting any devices, but instead, just look at the product ID. It looks like the product ID's are unique to a specific device.

From what I gathered in the manuals that came with my system, it looks like the multiple circuits is a EMS2 thing only though.

proddy, 2019-06-09:
this is possible to add, but it'll take some work and I think you would need to do most of the work as most of us don't have the setup to test with. Would you want multiple heating circuits on a single thermostat or the ability to separately manage multiple thermostats?

KristofRobberechts, 2019-06-13:
I'm okay with doing most of the work. I have multiple thermostats, one per circuit.

I'm not sure how the single thermostat that controls multiple circuit works, but I suspect it will answer to multiple addresses, same as my mixer.

proddy, 2019-07-27:
we can combine the implementation of this feature with https://github.com/proddy/EMS-ESP/issues/162

proddy, 2019-10-13:
This has been added to 1.9.2.

closed 2019-10-13 by proddy