emsesp/EMS-ESP #123: Can't build current dev branch 1.8.0

higgers, 2019-06-04:
Hi Proddy,

I just downloaded the dev build to try out the new transmit modes to see if any of them solve the issue I have with the boiler not receiving any telegrams.  However, I'm failing at the first hurdle, Visual Studio Code is outputting the following error when I hit the build button:

```
> Executing task: C:\Users\sh\.platformio\penv\Scripts\platformio.exe run <

Processing release ()
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Error: Please specify platform for 'release' environment
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

I've copied the platform.ini.example to platform.ini and adjust the upload_port:

```
;
; PlatformIO Project Configuration File for EMS-ESP
; Uses PlatformIO 4.0 (pio upgrade --dev)
;

[platformio]
env_default = release
;env_default = debug

[common]
; hard code if you prefer. Recommendation is to set from within the app when in Serial or AP mode
; wifi_settings = '-DWIFI_SSID="XXXX"' '-DWIFI_PASSWORD="XXXX"'
wifi_settings =

[env]
board = d1_mini
; board = nodemcuv2
; board = d1_mini_pro
framework = arduino
platform = espressif8266
lib_deps =
  CRC32
  CircularBuffer
  JustWifi
  AsyncMqttClient
  ArduinoJson
  OneWire
  EEPROM_rotate
upload_speed = 921600
monitor_speed = 115200
upload_protocol = espota
upload_port = COM6

[env:debug]
build_flags = -g -Wall -Wextra -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-variable -DCRASH -DTESTS ${common.wifi_settings}

[env:clean]
extra_scripts = pre:scripts/clean_fw.py

[env:release]
build_flags = -g -w
extra_scripts = pre:scripts/rename_fw.py

[env:checkcode]
build_flags = -g -Wall -Wextra -Werror -Wno-missing-field-initializers -Wno-unused-parameter -Wno-unused-variable -DCRASH -DTESTS
extra_scripts = scripts/checkcode.py
```

Any ideas?

proddy, 2019-06-04:
The ini file is built using platformio version 4. Try

pio upgrade --dev

higgers, 2019-06-04:
That fixed it, you're a legend!  Thanks!

closed 2019-06-05 by proddy