emsesp/EMS-ESP #126: Device mix (DeviceID:0x08 ProductID:64 Version:03.05)

Frederik-V, 2019-06-05:
I have a Buderus GB152-24T combi boiler (central heating + domestic home water) installed.

readout: DeviceID:0x08 ProductID:64 Version:03.05

TO BE ADDED: Buderus MC10/UBA 3

READOUT NOW: Sieger BK15 Boiler/Nefit Smartline

proddy, 2019-06-05:
I'm guessing the Sieger BK15 Boiler, Nefit Smartline and Buderus GB152 are all the same h/w internally. I'm not sure how to differentiate from the types. Perhaps there is more data from the Version telegram we could use. Could you try sometime to do a `boiler read 2` and see what data comes back and whether its more than just the product id and version.

Frederik-V, 2019-06-05:
_boiler read 2_
**Boiler found. Model Sieger BK15 Boiler/Nefit Smartline (DeviceID:0x08 ProductID:64 Version:03.05)**

proddy, 2019-06-05:
sorry I meant with verbose logging, so first do a `log v` . Also do a `thermostat read 2` 

2 is the telegram type ID for Version

Frederik-V, 2019-06-05:
**boiler read 2**
Requesting type Version(0x02) from dest 0x08
(04:01:11.871) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 20 (CRC=BC)
(04:01:11.908) Boiler -> me, type 0x02 telegram: 08 0B 02 00 40 03 05 (CRC=71) #data=3
<--- Version(0x02)
Boiler found. Model Sieger BK15 Boiler/Nefit Smartline (DeviceID:0x08 ProductID:64 Version:03.05)
(04:01:12.019) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=9B) #data=13
(04:01:12.875) Boiler -> all, type 0x18 telegram: 08 00 18 00 1B 01 28 64 00 01 01 20 60 01 B9 02 32 01 20 00 00 10 30 59 00 CC 00 00 00 (CRC=D9) #data=25
<--- UBAMonitorFast(0x18)
(04:01:13.191) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 02 32 02 32 20 00 00 03 00 00 64 E2 00 12 2B 00 (CRC=C6) #data=17
<--- UBAMonitorWWMessage(0x34)
(04:01:13.485) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=B7) #data=13
(04:01:14.449) 0x11 -> Boiler, type 0

**thermostat read 2**(03:58:39.274) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 01 (CRC=00) #data=2

Requesting type Version(0x02) from dest 0x10
(03:58:40.818) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 20 (CRC=7C)
(03:58:40.905) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 56 15 08 (CRC=5E) #data=3
<--- Version(0x02)
Thermostat found. Model RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
(03:58:40.936) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=9B) #data=13
(03:58:41.903) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 1B 64 00 00 (CRC=C0) #data=4
(03:58:42.031) Thermostat -> 0x21, type 0xAC telegram: 10 21 AC 00 16 64 03 (CRC=89) #data=3
(03:58:42.388) SM -> all, type 0x97 telegram: 30 00 97 00 00 00 00 7E 00 02 31 01 05 B1 48 00 0A (CRC=B1) #data=13

proddy, 2019-06-10:
this was added in the dev build.

closed 2019-06-10 by proddy