emsesp/EMS-ESP #127: Buderus RC35 deviceID issue

Frederik-V, 2019-06-05:
Via the 'devices' command, following EMS device is detected: 

"RC35 (DeviceID:0x08 ProductID:86 Version:21.08)"

Strange enough, when I take a look into the service menu of my Buderus RC35 thermostat, the RC35 should have address 16 (or 0x10) and the version is 21.08

What can be the reason for this mixup?

proddy, 2019-06-05:
that looks like a bug indeed. The code looks ok so not sure what is happening. When you start up what do you see when it does the autodetect, e.g. "Thermostat found. Model RC35.....". Is the information correct there too?

Frederik-V, 2019-06-05:
using the 'DEVICES' command:

_These 6 EMS devices were detected:
 SM10 Solar Module (DeviceID:0x30 ProductID:73 Version:02.00)
 Sieger BK15 Boiler/Nefit Smartline (DeviceID:0x08 ProductID:64 Version:03.05)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 RC35 (DeviceID:0x08 ProductID:86 Version:21.08)
 unknown? (DeviceID:0x11 ProductID:71 Version:02.00) -**> bug report is written**
 unknown? (DeviceID:0x21 ProductID:69 Version:02.01) **-> bug report is written_**

using the 'AUTODETECT' command:

_Started scan on EMS bus for known devices
Boiler found. Model Sieger BK15 Boiler/Nefit Smartline (DeviceID:0x08 ProductID:64 Version:03.05)
Device found. Model BC10/RFM20 Receiver with DeviceID 0x09, ProductID 68, Version 02.03
Unrecognized device found. DeviceID 0x11, ProductID 71, Version 02.00
Unrecognized device found. DeviceID 0x21, ProductID 69, Version 02.01
Device found. Model SM10 Solar Module with DeviceID 0x30, ProductID 73, Version 02.00
SM10 Solar Module support enabled._

With the 'AUTODETECT' command, the RC35 thermostat is apparently not recognized...

proddy, 2019-06-05:
ohh there's definitely something strange going on. Could you do a `log v` and then an `autodetect` again? I want to see the read commands to find out where the bug in the code is. It's mixing up the deviceids as you correctly pointed out earlier.

Frederik-V, 2019-06-05:
**autodetect**
Started scan on EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x20
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
Requesting type Version(0x02) from dest 0x48
(03:53:22.468) Sending read of type 0x02 to 0x02: telegram: 0B 82 02 00 20 (CRC=EC)
(03:53:22.786) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=9B) #data=13
(03:53:23.041) Sending read of type 0x02 to 0x08: telegram: 0B 88 02 00 20 (CRC=BC)
(03:53:23.074) Boiler -> me, type 0x02 telegram: 08 0B 02 00 40 03 05 (CRC=71) #data=3
<--- Version(0x02)
Boiler found. Model Sieger BK15 Boiler/Nefit Smartline (DeviceID:0x08 ProductID:64 Version:03.05)
(03:53:23.465) Sending read of type 0x02 to 0x09: telegram: 0B 89 02 00 20 (CRC=B4)
(03:53:23.495) 0x09 -> me, type 0x02 telegram: 09 0B 02 00 44 02 03 (CRC=25) #data=3
<--- Version(0x02)
Device found. Model BC10/RFM20 Receiver with DeviceID 0x09, ProductID 68, Version 02.03
Requesting type SM10Monitor(0x97) from dest 0x30
(03:53:23.893) Sending read of type 0x02 to 0x10: telegram: 0B 90 02 00 20 (CRC=7C)
(03:53:24.034) Thermostat -> me, type 0x02 telegram: 10 0B 02 00 56 15 08 (CRC=5E) #data=3
<--- Version(0x02)
Thermostat found. Model RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
(03:53:24.124) Boiler -> all, type 0x18 telegram: 08 00 18 00 1B 01 2A 64 00 01 01 20 60 01 B6 02 34 01 20 00 00 10 30 59 00 CC 00 00 00 (CRC=B1) #data=25
<--- UBAMonitorFast(0x18)
(03:53:24.386) Boiler -> all, type 0x34 telegram: 08 00 34 00 28 02 34 02 34 20 00 00 03 00 00 64 E2 00 12 2B 00 (CRC=AF) #data=17
<--- UBAMonitorWWMessage(0x34)
(03:53:24.811) Sending read of type 0x02 to 0x11: telegram: 0B 91 02 00 20 (CRC=74)
(03:53:24.924) 0x11 -> me, type 0x02 telegram: 11 0B 02 00 47 02 00 (CRC=7C) #data=3
<--- Version(0x02)
Unrecognized device found. DeviceID 0x11, ProductID 71, Version 02.00
(03:53:25.310) Sending read of type 0x02 to 0x17: telegram: 0B 97 02 00 20 (CRC=44)
(03:53:26.803) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=B7) #data=13
(03:53:27.110) 0x11 -> Boiler, type 0x1E telegram: 11 08 1E 00 01 1E (CRC=7E) #data=2
(03:53:28.561) Thermostat -> all, type 0xA3 telegram: 10 00 A3 00 10 00 00 00 EC 00 F2 00 F2 7D 00 7D 00 (CRC=23) #data=13
<--- RCOutdoorTempMessage(0xA3)
(03:53:28.850) Sending read of type 0x02 to 0x18: telegram: 0B 98 02 00 20 (CRC=3C)
(03:53:29.143) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=9B) #data=13
(03:53:29.523) Sending read of type 0x02 to 0x20: telegram: 0B A0 02 00 20 (CRC=E5)
(03:53:31.013) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=B7) #data=13
(03:53:31.792) Sending read of type 0x02 to 0x21: telegram: 0B A1 02 00 20 (CRC=ED)
(03:53:31.886) 0x21 -> me, type 0x02 telegram: 21 0B 02 00 45 02 01 (CRC=D9) #data=3
<--- Version(0x02)
Unrecognized device found. DeviceID 0x21, ProductID 69, Version 02.01
(03:53:31.935) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=9B) #data=13
(03:53:32.141) Sending read of type 0x02 to 0x30: telegram: 0B B0 02 00 20 (CRC=65)
(03:53:32.176) SM -> me, type 0x02 telegram: 30 0B 02 00 49 02 00 (CRC=CC) #data=3
<--- Version(0x02)
Device found. Model SM10 Solar Module with DeviceID 0x30, ProductID 73, Version 02.00
SM10 Solar Module support enabled.
Requesting type SM10Monitor(0x97) from dest 0x30
(03:53:32.570) Sending read of type 0x02 to 0x48: telegram: 0B C8 02 00 20 (CRC=8E)
(03:53:33.968) Boiler -> all, type 0x18 telegram: 08 00 18 00 1B 01 2A 64 00 01 01 20 60 01 B6 02 34 01 20 00 00 10 30 59 00 CC 00 00 00 (CRC=B1) #data=25
<--- UBAMonitorFast(0x18)
(03:53:34.186) Sending read of type 0x97 to 0x30: telegram: 0B B0 97 00 20 (CRC=03)
(03:53:34.251) SM -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 81 00 02 32 01 05 B1 48 00 0A (CRC=02) #data=13
<--- SM10Monitor(0x97)
(03:53:34.834) Sending read of type 0x97 to 0x30: telegram: 0B B0 97 00 20 (CRC=03)
(03:53:34.935) SM -> me, type 0x97 telegram: 30 0B 97 00 00 00 00 81 00 02 32 01 05 B1 48 00 0A (CRC=02) #data=13
<--- SM10Monitor(0x97)
(03:53:36.177) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 03 00 02 00 01 00 00 00 00 00 00 00 (CRC=B7) #data=13
(03:53:36.863) 0x11 -> Boiler, type 0x1E telegram: 11 08 1E 00 01 1E (CRC=7E) #data=2
(03:53:38.403) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 01 01 (CRC=00) #data=2

proddy, 2019-06-05:
and after that last autodetect command, doing a `devices` doesn't show the RC35 in the list?

Frederik-V, 2019-06-05:
It shows the RC35 as (DeviceID:0x08 ProductID:86 Version:21.08), but that should not be the case

proddy, 2019-06-05:
ok, then the bug is in the list and the way I sort it. It shouldn't affect the behavior of the code. I'll send you a modified version of the ems.cpp file with more debugging later this week so we can see where it is actually failing. Probably in the _addDevice() function.

proddy, 2019-06-10:
could you try again with the latest build?

proddy, 2019-06-30:
closing due to no activity. Please reopen if its still fails with the latest dev build

closed 2019-06-30 by proddy