emsesp/EMS-ESP #129: Setting OTA parameters (update process) via Telnet

Frederik-V, 2019-06-08:
Dear.

Is there a possibility to update (via OTA) the EMS-ESP Wemos D1 firmware via the Telnet ?
This by changing parameters via Telnet afterwards instead of before the first .BIN flash:
_; for OTA comment out these sections
;upload_protocol = espota
;upload_port = ems-esp.local
;upload_port = <or add here the IP of the ESP8266>_

Or is this already implemented? If that is the case, can some guide me how to initiate a firmware update process via Telnet?

Kindest regards

proddy, 2019-06-08:
I'm not sure I understand what you're trying to do. The normal way to
upload is via OTA using the command line or via Platformio as described in
the wiki. It is possible to upload via http too but that would mean
starting a web server on the esp which eats even more memory and don't
think anyone would use it.

frankhommers, 2019-06-18:
I think this firmware update procedure is a bit easier, that might help @Frederik-V 

```bash
wget https://github.com/proddy/EMS-ESP/releases/download/1.8/firmware_d1_mini.bin
wget https://raw.githubusercontent.com/esp8266/Arduino/master/tools/espota.py
python espota.py -i 10.0.2.140 -f firmware_d1_mini.bin
```

@proddy: You might put this on the wiki as well ;-)

proddy, 2019-06-18:
Thanls @frankhommers - added it to https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware

susisstrolch, 2019-06-18:
Don‘t forget to open up your firewall bcause the ESP will initiate the connection to your upload host. 
You can use a fixed port with the -p option...

Sent by mobile device

closed 2019-07-27 by proddy