emsesp/EMS-ESP #133: Two devices... Unable to set temp. Make a primary device selectable?

frankhommers, 2019-06-18:
When I enter devices:

```
These 2 EMS devices were detected:
 Bosch Easy (DeviceID:0x02 ProductID:206 Version:01.08)
 RC30/Nefit Moduline 400 (DeviceID:0x10 ProductID:78 Version:03.01)
```

When I change the temp by mqtt:
```
MQTT topic: thermostat temperature value 11.50
Write not supported for this model Thermostat
```

Is it possible to make one device primary and allow writing to it? That would solve my problem.

frankhommers, 2019-06-18:
Another idea would be to write to the one that allows writing. And if only none of those exist show that error.

proddy, 2019-06-18:
The code currently only supports a single thermostat and the device id is stored (see `thermostat_type` when typing `set`). I need to think about how logic here, something like
- during detection, store all the found thermostats in a list from 1...n
- have a new set command as you suggested that picks the main one to use 1...n (default is 1)
- all reads/writes from telnet or MQTT go to the main one selected unless specified in the json

frankhommers, 2019-06-19:
I have this setup:

[image]

I use this to force the Moduline 400 into EMS mode, and that works (without the EMS-ESP).

But EMS-ESP is confused about the thermostat I think, also eventually when reading. 
I am not sure if 1..n is the right method, in some rare cases the Moduline gets detected first.

```
Thermostat stats:
  Thermostat: Bosch Easy (ProductID:206 Version:01.08)
  Set room temperature: 0.14 C
  Current room temperature: 2.34 C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to ?

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is on
  Thermostat is enabled, Boiler is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=4, # CRC errors=1
  Tx: Last poll=2.46 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: 2.5 l/min
  Warm Water # starts: 0 times
  Warm Water active time: 0 days 0 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 0 C
  Current flow temperature: 26.0 C
  Return temperature: 26.0 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (0)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 14.5 C
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 0 times
  Total burner operating time: 0 days 0 hours 0 minutes
  Total heat operating time: 0 days 0 hours 0 minutes

Thermostat stats:
  Thermostat: Bosch Easy (ProductID:206 Version:01.08)
  Set room temperature: 0.14 C
  Current room temperature: 2.35 C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to ?
```
The thermostat temps are divided by 10. Everything else is correct. I do not use warm water/boiler so that's correct. 

I have put it in listen_mode because in my opinion writing to the wrong thermostat now is a bad idea.

I would love to be able to use EMS-ESP for this!

proddy, 2019-06-19:
Yes, the reason the temps are wrong is because the values of setpoint temp and current temp are overwritten by the RC35 and it renders the values thinking its an Easy. Question: so you have a Bosch or a Nefit Easy?

So not sure what to do here. What solution do you think works?

frankhommers, 2019-06-19:
This is my stuff, see pics below, on the label it's Nefit Easy Connect (made by Bosch :-) ).

[image]
[image]
[image]

I am not sure.  Is there a slave number on the bus of each device? Maybe a blacklist or whitelist could help if that is the case?

proddy, 2019-06-20:
ah, I think I know a part of the problem. Your device is an Easy Connect and not an Easy thermostat (like the [Moduline Easy](https://www.nefit.nl/consument/producten/product_detail/product_11776)). In the `ems_devices.h` I had it incorrectly classified as a Thermostat with
    `{EMS_MODEL_EASY, 206, 0x02, "Bosch Easy", EMS_THERMOSTAT_WRITE_NO}`

when it should be under the Other types like how the Nefit Easy Connect
    `{EMS_MODEL_OTHER, 205, 0x02, "Nefit Moduline Easy Connect"}`

I've made the change, not sure if its fixes anything on your side though!

frankhommers, 2019-06-20:
That could be a solution as well! Can you please build it and put it on release? That would help me a lot because I don't have the development stack installed.

proddy, 2019-06-20:
sure, added to https://github.com/proddy/EMS-ESP/releases/tag/1.8.1

Note if you want to quickly build and upload without having a full dev environment I wrote some instructions on the Wiki [here](https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware) which works with Ubuntu (under Windows ot native) and MacOSX.

frankhommers, 2019-06-20:
Awesome it works now!

I also had to make a change in the homeassistant config so that the auto/manual/off icons would work.

This is because e.g. the value was set to 'manual' and the mqtt response was 'day'.

`    mode_state_template: "{{ {'auto':'auto','day':'manual','night':'off','manual':'manual','off':'off'}[value_json.thermostat_mode] | default('unknown') }}"`

proddy, 2019-06-20:
great! I'll look at the HA config again. Thanks for pointing that out and helping test

proddy, 2019-06-20:
@frankhommers I looked into the thermostat MQTT messages (line 728 in ems-esp.cpp). For RC20's I Send low, manual and auto depending on the mode. All other thermostat models send night, day, auto. What should it be for your Moduline 400 so I can fix this in the code?

frankhommers, 2019-06-21:
It's actually "auto", "manual" and "frost protection" ;-)
So it's seems like that it also would need to be "auto", "manual", "low".

I am not sure "low" (for any thermostat) would be an appropriate value according to Home Assistant.

closed 2019-07-27 by proddy