emsesp/EMS-ESP #136: Issues with 1.8.1b2 (f29c5ea)

S-Przybylski, 2019-06-20:
Hi @proddy 
I just udgrade to the lastest dev version today. Here are three comments:

* Failed Read attempts SM100 (complete log see bejond)
````
Read failed. Retrying attempt 1/2...
(00:18:11.810) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
(00:18:12.098) Sending read of type 0x262 to 0x30: telegram: 0B B0 62 00 20 (CRC=FC)
Read failed. Giving up, removing from queue
(00:18:12.155) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
````
* Help corrupted (using putty)
see below

* Documentation issue: In the readme i saw the the RC300 is not restricted to read only
````
Thermostats:
...
RC300/RC310
````

* SM100 still working as expected - good (no issue!)
see below

Thanks for the good work

``````
These 3 EMS devices were detected:
 Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
 SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.04)
 RC3x0/Nefit Moduline 1010H (DeviceID:0x10 ProductID:158 Version:11.07)
EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=87, # CRC errors=1
  Tx: Last poll=3.964 seconds ago, # successful write requests=0
``````

`````
(00:18:05.845) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 F5 01 F5 21 00 01 03 00 05 B6 40 00 F4 4F 00 80 00 (CRC=71) #data=19
<--- UBAMonitorWWMessage(0x34)
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Publishing SM data via MQTT
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Requesting type (0x262) from dest 0x30
(00:18:07.714) SM -> Boiler, type 0x33 telegram: 30 88 33 02 07 (CRC=C0) #data=1
(00:18:07.750) Boiler -> SM, type 0x33 telegram: 08 30 33 02 33 FB 00 1E FF 06 46 (CRC=C9) #data=7
(00:18:07.776) SM -> Boiler, type 0x35 telegram: 30 08 35 02 00 (CRC=BB) #data=1
(00:18:07.904) SM -> all, type 0x0262 telegram: 30 00 FF 00 02 62 01 FC 01 AC 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 80 00 (CRC=1E) #data=24
<--- SM100Monitor(0x262)
Publishing SM data via MQTT
(00:18:08.351) SM -> all, type 0x0262 telegram: 30 00 FF 18 02 62 80 00 (CRC=AE) #data=2
<--- SM100Monitor(0x262)
(00:18:08.577) SM -> all, type 0x0263 telegram: 30 00 FF 00 02 63 80 00 80 00 00 00 80 00 80 00 80 00 00 (CRC=74) #data=13
(00:18:08.761) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C)
(00:18:08.839) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 06 0C 14 38 13 03 01 10 FF 00 (CRC=32) #data=11
<--- RCTime(0x06)
(00:18:08.892) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76) #data=13
(00:18:09.158) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4)
(00:18:09.231) Boiler -> me, type 0x18 telegram: 08 0B 18 00 05 02 98 00 00 00 00 00 40 80 00 01 F5 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=2F) #data=25
<--- UBAMonitorFast(0x18)
(00:18:09.464) SM -> all, type 0x0264 telegram: 30 00 FF 00 02 64 00 00 00 04 00 00 FF 00 00 1E 0A 0D 64 00 00 00 00 (CRC=AD) #data=17
<--- SM100Status(0x264)
(00:18:09.670) SM -> all, type 0x0266 telegram: 30 00 FF 00 02 66 01 62 00 12 (CRC=73) #data=4
(00:18:10.004) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0)
(00:18:10.089) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 F7 80 00 80 00 00 00 00 00 03 58 97 0C 7B 1F 00 00 00 06 C4 DF 02 64 48 80 00 (CRC=05) #data=27
<--- UBAMonitorSlow(0x19)
(00:18:10.192) SM -> all, type 0x0268 telegram: 30 00 FF 00 02 68 0C 00 (CRC=1E) #data=2
(00:18:10.410) SM -> all, type 0x026A telegram: 30 00 FF 00 02 6A 03 03 03 00 03 03 03 03 03 00 04 03 (CRC=EB) #data=12
<--- SM100Status2(0x26A)
(00:18:10.776) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78)
(00:18:10.827) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 33 FB 00 1E FF 06 46 00 FF FF 00 (CRC=15) #data=13
<--- UBAParameterWW(0x33)
(00:18:11.056) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC)
(00:18:11.113) Boiler -> me, type 0x16 telegram: 08 0B 16 00 00 46 64 00 02 FE 05 01 03 64 0A 04 00 00 00 00 00 00 00 00 00 00 00 00 23 00 23 (CRC=1D) #data=27
<--- UBAParametersMessage(0x16)
(00:18:11.425) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4)
(00:18:11.480) Boiler -> me, type 0x14 telegram: 08 0B 14 00 25 63 1D (CRC=5D) #data=3
<--- UBATotalUptimeMessage(0x14)
(00:18:11.772) Sending read of type 0x262 to 0x30: telegram: 0B B0 62 00 20 (CRC=FC)
Read failed. Retrying attempt 1/2...
(00:18:11.810) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
(00:18:12.098) Sending read of type 0x262 to 0x30: telegram: 0B B0 62 00 20 (CRC=FC)
Read failed. Giving up, removing from queue
(00:18:12.155) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
(00:18:13.121) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A) #data=13
`````
Solar SM100 values are correct:

`````
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 54.3 C
  Bottom temperature: 43.7 C
  Pump modulation: 30 %
  Pump active: on
  Energy Last Hour: 351.0 Wh
  Energy Today: 857 Wh
  Energy Total: 3209.1 kWH

MQTT message:
{"collectortemp":54.3,"bottomtemp":43.7,"pumpmodulation":30,"pump":"on","energylasthour":351,"energytoday":857,"energytotal":3209.1}
`````

I use putty to connect to the ems. The help (?) shows me the following:
`````
?

* Connected to: EMS-ESP version 1.8.1b2
* Hostname: ems-esp (10.10.0.235)
* WiFi SSID: imprsAtHome (signal 54%)
* MQTT connected (heartbeat disabled)
*
* Commands:
*  ?=help, CTRL-D/quit=exit telnet session
*  set, system, reboot
*  crash <dump | clear | test [n]>
*       gpio>
*       _type <device ID>
*       e <device ID>
*       e <on | off>
*       er <on | off>
*       rt <on | off>
*       rcuit <1 | 2>
*        | t | r | v>
*
*
*
*
*        [deep]
*       .
*        read <type ID>
*        temp <degrees>
*        mode <mode>
*        scan <type ID>
*       d <type ID>
*       emp <degrees>
*       water <on | off>
*       wtemp <degrees>
*       fort <hot | eco | intelligent>
`````

proddy, 2019-06-21:
the putty/telnet issue - this could be because the baud is too fast. It's set to 115200 which is a high-rate for full duplex. Does the problem happen all the time? Can you try changing TELNET_SERIAL_BAUD in MyESP.h to 9600 and trying again? Did you have this problem with 1.7?

I'm looking to the "Failed Read attempts SM100" issue. 

thanks

proddy, 2019-06-21:
@S-Przybylski I haven't implemented writing to EMS+ which is why you get the "Failed Read attempts SM100". But its fixable. Can you do a quick test with
`send 0B B0 FF 00 02 62 20` and see if anything comes back. It's a request from our device to the SM on ID 0x30 for 32 bytes of data from type id 0x262. No idea if it'll work!

S-Przybylski, 2019-06-26:
Dear @proddy i just tested the send command:
````
send 0B B0 FF 00 02 62 20
(04:51:38.272) Sending raw: 0B B0 FF 00 02 62 20 00
(04:51:38.331) SM -> me, type 0x6220 telegram: 30 0B FF 00 62 20 (CRC=B5)

(04:52:52.748) Sending raw: 0B B0 FF 00 02 62 20 0B B0 FF 00 02 62 20 00
(04:52:52.821) SM -> me, type 0x6220 telegram: 30 0B FF 00 62 20 (CRC=B5)

````

S-Przybylski, 2019-06-26:
Dear @proddy 
While I changed the baud rate to from 57600 to 9600 and vice versa if found a reoccuring method to get the output stumbled (therefore i assume that its not related to the baud rate):
If i restart the esp help works as expected. Then i copied the output of the help to the clipboard and accindently i put that output back to the console (and yes of course i get unknown command response). After that the help command shows the broken behaviour (this is independent if log v is set). On top of that i reconnected directly with telnet instead of putty; the help output is still broken, but then the esp reboots suddenly with the crash dump:
````
[CRASH] Last crash was 0 days 0 hours 5 minutes 19 seconds since boot time      [CRASH] Reason of restart: 2 - Fatal exception                                  [CRASH] Exception cause: 28                                                     [CRASH] epc1=0x40226f98 epc2=0x00000000 epc3=0x00000000                         [CRASH] excvaddr=0x00000000 depc=0x00000000                                     [CRASH] sp=0x3fffff30 end=0x3fffffc0                                            >>>stack>>>                                                                     3fffff30: 3fffdad0 3fff4dd8 0000000d 40217eda                                   3fffff40: 3fffdad0 3fff4dd8 3fff4dc4 3fff4e48                                   3fffff50: 0000000d 3fff4dc4 3fff4b70 40213821                                   3fffff60: 0000000d 3fff50a4 3ffe971c 3fff5048                                   3fffff70: 00000000 3fff314c 3fff4b70 3fff5048                                   3fffff80: 3fffdad0 3fff314c 3fff4b70 40213902                                   3fffff90: 3fffdad0 00000000 3fff314c 4020be94                                   3fffffa0: 3fffdad0 00000000 3fff5018 40215464                                   3fffffb0: feefeffe feefeffe 3ffe971c 40100b01                                   <<<stack<<<                                                                                                                                                     To clean this dump use the command: crash clear
````
````
?
* Connected to: EMS-ESP version 1.8.1b2
* Hostname: ems-esp (10.10.0.235)
* WiFi SSID: imprsAtHome (signal 58%)
* MQTT connected (heartbeat disabled)
*
* Commands:
*  ?=help, CTRL-D/quit=exit telnet session
*  set, system, reboot
*  crash <dump | clear | test [n]>
*  info                                         show current captured on the devices
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  test <n>                                     insert a test telegram on to the EMS bus
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  devices                                      list all supported and detected EMS devices and types IDs
*  queue                                        show current Tx queue
*  autodetect [deep]                            detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    probe thermostat on all type id responses
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
*  boiler flowtemp <degrees>                    set boiler flow temperature
*  boiler comfort <hot | eco | intelligent>     set boiler warm water comfort setting

Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
log v
System Logging set to Verbose
(00:03:01.321) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DF 00 DF 80 00 00 80 00 80 00 80 00 00 (CRC=1B) #data=21
(00:03:01.636) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 EF 01 EF 09 00 01 03 00 05 B8 5F 00 F4 92 00 80 00 (CRC=74) #data=19
<--- UBAMonitorWWMessage(0x34)
[... 17 lines cut ...]
*       gpio>
*       _type <device ID>
*       e <device ID>
*       e <on | off>
*       er <on | off>
*       rt <on | off>
*       rcuit <1 | 2>
*        | t | r | v>
*
*
*
*
*        [deep]
*       .
*        read <type ID>
*        temp <degrees>
*        mode <mode>
*        scan <type ID>
*       d <type ID>
*       emp <degrees>
*       water <on | off>
*       wtemp <degrees>
*       fort <hot | eco | intelligent>

````

proddy, 2019-06-26:
ok, I'll see if I can reproduce. To get the same setup are you using putty with standard settings (i.e. not changing things like backspace key etc..)

S-Przybylski, 2019-06-26:
nothing changed except telnet instead of ssh

proddy, 2019-06-26:
and wifi strength is good (> 60%) to ensure there is no packet loss. You see this in 'system'

S-Przybylski, 2019-06-26:
The wifi strength is currently around 62%. In my post you can see, that it was 58% while testing

proddy, 2019-06-26:
@S-Przybylski made a fix for that telnet error.

S-Przybylski, 2019-06-27:
Dear @proddy 
i have tested the current 1.8.1b5 version: The help is fine (i tested it several times) up to the point i insert the help into the command window itself. Then its corrupted ... - no change
````
?

* Connected to: EMS-ESP version 1.8.1b5
* Hostname: ems-esp (10.10.0.235)
* WiFi SSID: wifi (signal 54%)
* MQTT connected (heartbeat disabled)
*
* Commands:
*  ?=help, CTRL-D/quit=exit telnet session
*  set, system, reboot
*  crash <dump | clear | test [n]>
*  info                                         show current captured on the devices
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  test <n>                                     insert a test telegram on to the EMS bus
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  devices                                      list all supported and detected EMS devices and types IDs
*  queue                                        show current Tx queue
*  autodetect [deep]                            detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    probe thermostat on all type id responses
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
*  boiler flowtemp <degrees>                    set boiler flow temperature
*  boiler comfort <hot | eco | intelligent>     set boiler warm water comfort setting

Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
Unknown command. Use ? for help.
?

* Connected to: EMS-ESP version 1.8.1b5
* Hostname: ems-esp (10.10.0.235)
* WiFi SSID: wifi (signal 52%)
* MQTT connected (heartbeat disabled)
*
* Commands:
*  ?=help, CTRL-D/quit=exit telnet session
*  set, system, reboot
*  crash <dump | clear | test [n]>
*       gpio>
*       _type <device ID>
*       e <device ID>
*       e <on | off>
*       er <on | off>
*       rt <on | off>
*       rcuit <1 | 2>
*        | t | r | v>
*
*
*
*
*        [deep]
*       .
*        read <type ID>
*        temp <degrees>
*        mode <mode>
*        scan <type ID>
*       d <type ID>
*       emp <degrees>
*       water <on | off>
*       wtemp <degrees>
*       fort <hot | eco | intelligent>
````

proddy, 2019-06-27:
I'll double check. So to reproduce, you copy & pasted that help text block
into the telnet window?

S-Przybylski, 2019-06-27:
indeed thats what i have done

S-Przybylski, 2019-06-28:
Dear @proddy 
i just found out that the 1.8.1b5 doesn't extract the Outside temp (boiler message) any more, either in 'info' nor in MQTT message. I rececked 1.8.1b2 - here the temperature is shown:
Outside temperature: 21.9 C

proddy, 2019-06-28:
oops - let me check what changed

proddy, 2019-06-29:
@S-Przybylski I can't reproduce the missing outdoor temp issue. I took your 0x19 data and put into a test case to simulate the telegrams. then `test 48` to inject it and the outdoor temp comes up in `info` as well as the mqtt payload.

S-Przybylski, 2019-06-29:
Dear @proddy 
sorry for annoying you!
I just compiled my clone (date 27.06.2019) of the repository and put it onto the esp. But again the outdoor temperature isn't visible!
Btw. this build crashes very often. Especially if i enter 'system' into the console... 

````
* Connected to: EMS-ESP version 1.8.1b5
* Hostname: ems-esp (10.10.0.235)
* WiFi SSID: wifi (signal 56%)
* MQTT connected (heartbeat disabled)

info
EMS-ESP system stats:
  System logging set to None
  LED is off, Listen mode is off
  Thermostat is enabled, Boiler is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=19, # CRC errors=10
  Tx: Last poll=2.385 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB172/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:04.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 51 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 48.5 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 62638 times
  Warm Water active time: 260 days 12 hours 2 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 65.6 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
  Total UBA working time: 1710 days 11 hours 53 minutes

Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 40.3 C
  Bottom temperature: 31.0 C
  Pump modulation: 30 %
  Pump active: on
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH

Thermostat stats:
  Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.07)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 11:13:09 29/6/2019
  Mode is set to ?

MQTT messages:
Boiler: {"wWComfort":"Hot","wWSelTemp":51,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"wWCurTmp":48.1,"wWCurFlow":0,"curFlowTemp":65.2,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"off","ServiceCode":"0H","ServiceCodeNumber":203}

SM100: {"collectortemp":40.7,"bottomtemp":31.4,"pumpmodulation":30,"pump":"on"}
Thermostat: {"thermostat_hc":"1","thermostat_mode":"auto"}
````

proddy, 2019-06-29:
if you do `test 48` do you see it then?

S-Przybylski, 2019-06-29:
test 48
System Logging set to Verbose
Invalid test. Pick between 1 and 47

S-Przybylski, 2019-06-29:
Dear @proddy 
should i use the todays dev version to test?

S-Przybylski, 2019-06-29:
I just installed the 1.8.1b6: Its the same - no outdoor temp
{"wWComfort":"Hot","wWSelTemp":51,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"wWCurTmp":49.2,"wWCurFlow":0,"curFlowTemp":65.2,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"on","wWHeat":"off","ServiceCode":"0H","ServiceCodeNumber":203}

I enabled dtests in platformio.ini  and execute the command: test48
````
test 48
[030218] System Logging set to Verbose
[030219] [TEST 48] Injecting telegram 08 0B 19 00 00 F7 80 00 80 00 00 00 00 00 03 58 97 0C 7B 1F 00 00 00 06 C4 DF 02 64 48 80 00
(00:00:30.219) Boiler -> me, type 0x19 telegram: 08 0B 19 00 00 F7 80 00 80 00 00 00 00 00 03 58 97 0C 7B 1F 00 00 00 06 C4 DF 02 64 48 80 00 (CRC=05) #data=27
[030221] <--- UBAMonitorSlow(0x19)
(00:00:30.251) 0x09 -> SM, type 0x02 telegram: 09 B0 02 00 03 (CRC=66) #data=1
(00:00:30.292) SM -> 0x09, type 0x02 telegram: 30 09 02 00 A3 15 04 (CRC=25) #data=3
(00:00:30.339) SM -> 0x09, type 0x96 telegram: 30 09 96 00 FF 18 1E 0A 02 50 28 (CRC=B5) #data=7
(00:00:30.361) 0x09 -> SM, type 0xFF00 telegram: 09 B0 FF 00 FF 00 01 (CRC=0C) #data=1
(00:00:30.398) SM -> 0x09, type 0x0001 telegram: 30 09 FF 00 00 01 (CRC=70)
(00:00:30.524) Thermostat -> all, type 0x01AF telegram: 10 00 F7 00 FF 01 AF ED (CRC=26)
(00:00:30.831) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 02 81 00 00 00 00 00 40 80 00 01 DA 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=A0) #data=25
[030832] <--- UBAMonitorFast(0x18)
(00:00:31.502) Boiler -> all, type 0x1C telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B) #data=25
(00:00:31.706) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(00:00:31.965) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 DA 01 DA 21 00 01 03 00 05 B9 5B 00 F4 B0 00 80 00 (CRC=3F) #data=19
[031966] <--- UBAMonitorWWMessage(0x34)
(00:00:32.233) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A) #data=13
(00:00:32.539) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 02 81 00 00 00 00 00 40 80 00 01 DA 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=A0) #data=25
[032541] <--- UBAMonitorFast(0x18)
(00:00:33.789) SM -> all, type 0x0266 telegram: 30 00 FF 00 02 66 01 62 00 12 (CRC=73) #data=4
(00:00:36.401) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 02 79 01 83 (CRC=15) #data=4
[036403] <--- RCPLUSStatusMessage(0x1A5)
(00:00:40.858) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(00:00:41.078) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 01 DA 01 DA 21 00 01 03 00 05 B9 5B 00 F4 B0 00 80 00 (CRC=3F) #data=19
[041079] <--- UBAMonitorWWMessage(0x34)
(00:00:41.338) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 02 81 00 00 00 00 00 40 80 00 01 DA 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=A0) #data=25
[041340] <--- UBAMonitorFast(0x18)
````

S-Przybylski, 2019-06-29:
Interessting: after entering the test 48 the outdoor temperature is visible again! I tested it several times. I also saw that at least the pump modulation wasn't initilisied before test 48...

````
[000130]
         * EMS-ESP version 1.8.1b6
[000140] [WIFI] Connecting to SSID: wifi
[003948] [WIFI] SSID  wifi
[003949] [WIFI] CH    1
[003949] [WIFI] RSSI  -73
[003950] [WIFI] IP    10.10.0.235
[003950] [WIFI] MAC   CC:50:E3:45:51:CE
[003953] [WIFI] GW    10.10.0.1
[003956] [WIFI] MASK  255.255.255.0
[003959] [WIFI] DNS   10.10.0.1
[003962] [WIFI] HOST  ems-esp
[003967] [OTA] listening to ems-esp.local:8266
[003969] Disabling serial port communication.
[003984] [UART] Opened Rx/Tx connection
[003985] Starting auto discover of EMS devices...
[003986] [MQTT] Connecting to MQTT...
[004009] [MQTT] Connected
[004010] [MQTT] Subscribing to buderus/ems-esp/restart (PID 2)
[004011] [MQTT] Subscribing to buderus/ems-esp/start (PID 3)
[004011] [MQTT] Subscribing to buderus/ems-esp/thermostat_cmd_temp (PID 5)
[004011] [MQTT] Subscribing to buderus/ems-esp/thermostat_cmd_mode (PID 6)
[004012] [MQTT] Subscribing to buderus/ems-esp/wwactivated (PID 7)
[004012] [MQTT] Subscribing to buderus/ems-esp/boiler_cmd_wwtemp (PID 8)
[004012] [MQTT] Subscribing to buderus/ems-esp/boiler_cmd_comfort (PID 9)
[004012] [MQTT] Subscribing to buderus/ems-esp/shower_timer (PID 10)
[004013] [MQTT] Subscribing to buderus/ems-esp/shower_alert (PID 11)
[004013] [MQTT] Subscribing to buderus/ems-esp/shower_coldshot (PID 12)
[004013] [MQTT] Subscribing to buderus/ems-esp/thermostat_cmd_hc (PID 13)
[004013] [MQTT] Subscribing to buderus/ems-esp/thermostat_daytemp (PID 14)
[004014] [MQTT] Subscribing to buderus/ems-esp/thermostat_nighttemp (PID 15)
[004014] [MQTT] Subscribing to buderus/ems-esp/thermostat_holidayttemp (PID 16)
[004031] [MQTT] Received boottime: start
[004776] [MQTT] Received boottime: 12:39:18 29/Jun/2019
[008263] Boiler found: Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
[008264] * Setting Boiler to model Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
[009239] Solar Module found: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.04)
[009239] Solar Module support enabled.
[012332] * Setting Thermostat to RC3x0/Nefit Moduline 1010H (DeviceID:0x10 ProductID:158 Version:11.07)
[022672] [TELNET] Telnet connection established
[022673]
[022673] * Connected to: EMS-ESP version 1.8.1b6
[022674] * Hostname: ems-esp (10.10.0.235)
[022674] * WiFi SSID: wifi (signal 56%)
[022674] * MQTT connected (heartbeat disabled)
[022674] *
[022675] * Commands:
[022675] *  ?=help, CTRL-D/quit=exit telnet session
[022675] *  set, system, reboot
[022675] *  crash <dump | clear | test [n]>
*  info                                         show current captured on the devices
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  test <n>                                     insert a test telegram on to the EMS bus
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  devices                                      list all supported and detected EMS devices and types IDs
*  queue                                        show current Tx queue
*  autodetect [deep]                            detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    probe thermostat on all type id responses
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
[... 260 lines cut ...]
  Burner # starts: 219287 times
[132562]   Total burner operating time: 568 days 0 hours 31 minutes
[132562]   Total heat operating time: 308 days 1 hours 35 minutes
[132562]   Total UBA working time: 1710 days 13 hours 24 minutes
[132563]
[132563] Solar Module stats:
[132563]   Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 54.3 C
  Bottom temperature: 42.4 C
  Pump modulation: 30 %
  Pump active: on
[132564]   Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH
[132564]
[132564] Thermostat stats:
[132564]   Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.07)
  Setpoint room temperature: ? C
  Current room temperature: ? C
[132565]   Thermostat time is 12:43:48 29/6/2019
[132565]   Mode is set to ?
[132565]

````

S-Przybylski, 2019-06-29:
I restored the 1.8.1b2 - it works better. 
For me it seems that the outdoor value after entering test 48 in 1.8.1b6 wasn't updated...

proddy, 2019-06-30:
Could you attach the ems.cpp and ems-esp.cpp files from your working b2 so I can compare?

proddy, 2019-07-01:
think I found the culprit and fixed in https://github.com/proddy/EMS-ESP/issues/143. Could you try again?

S-Przybylski, 2019-07-02:
Dear @proddy 
I installed the todays version 1.8.1b8

The outdoor temp is visible again. 

The issue with the restart when i enter the system command is still present. 

If seems that the solar values aren't updated as expected. The Info command provides the current values after some time, which are updated every minute. But the mqtt server doens't reflect any update from the solar modul (other values like boiler are transmitted). Therefore HA still shows old values for Solar. Refresh and publish command doesn't send any Solar Modul MQTT message!

New in Version 1.8.1b2 (b8 doesn't show any value!): Today i saw that the Solar engery total value switched to a negative value some days before. Updated values are less negative if the solar modul works (the gardient is positive):
[image]

S-Przybylski, 2019-07-02:
update after 1 hour (using b8):
the energy values are updated internally - see below, but there is no sm_data mqtt message at all! The info command shows at least a positive value for energy total!
@proddy could you please check, why the solar modul doesn't send updates to the mqtt brocker?
````
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 31.3 C
  Bottom temperature: 30.0 C
  Pump modulation: 0 %
  Pump active: off
  Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: 39.0 Wh
  Energy Today: 3741 Wh
  Energy Total: 3237.2 kWH
````

S-Przybylski, 2019-07-02:
update regarding system command:
Now i could initiate the command without a crash! But i got a negative heap size (used) value:
-3704 bytes used (173057%)

````
system
ESP8266 System stats:

 [APP] EMS-ESP version: 1.8.1b8
 [APP] MyESP version: 1.1.20
 [APP] Build timestamp: 2019-07-02 19:40:03
 [APP] Boot time: 20:10:49 2/Jul/2019
 [APP] Uptime: 0 days 0 hours 53 minutes 59 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 10.10.0.235
 [WIFI] WiFi signal strength: 52%
 [WIFI] WiFi MAC: CC:50:E3:45:51:CE
 [MQTT] connected (heartbeat disabled)
 [EEPROM] EEPROM size: 4096
 [EEPROM] EEPROM Sector pool size is 4, and in use are: 1019 1018 1017 1016
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x4551CE
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Software/System restart
 [SYSTEM] Last reset info: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
 [SYSTEM] Restart count: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x164068
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 463344
 [MEM] Max OTA size: 2674688
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24816 bytes initially | -3704 bytes used (173057%) | 28520 bytes free (114%)
````

S-Przybylski, 2019-07-02:
Update to 1.8.1b2:
energy total in mqtt message negative!

info value ok (22:05)
````
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 19.0 C
  Bottom temperature: 23.5 C
  Pump modulation: 0 %
  Pump active: off
  Energy Last Hour: 0.0 Wh
  Energy Today: 3741 Wh
  Energy Total: 3237.2 kWH
````

but mqtt message wrong (negative):
21:54 {"collectortemp":19.4,"bottomtemp":23.8,"pumpmodulation":0,"pump":"off"}
21:55 {"collectortemp":19.4,"bottomtemp":23.8,"pumpmodulation":0,"pump":"off","energylasthour":0,"energytoday":3741,"energytotal":-3237.2}
22:05
{"collectortemp":18.9,"bottomtemp":23.4,"pumpmodulation":0,"pump":"off","energylasthour":0,"energytoday":3741,"energytotal":-3237.2}

proddy, 2019-07-02:
arghh. that problem again. I'm not sure why its failing. Can you add a line before #805 in `ems.cpp` like
```
myDebug("** got energy total value of %d", EMS_SolarModule.EnergyTotal);  // add this
if (EMS_SolarModule.EnergyTotal != EMS_VALUE_SHORT_NOTSET)
            rootSM[SM_ENERGYTOTAL] = (double)EMS_SolarModule.EnergyTotal / 10;
```

so I can see if the value is correct and its the arduinojson not handled signed ints correctly, or converting to floats.

proddy, 2019-07-02:
> Update to 1.8.1b2:
> energy total in mqtt message negative!
>
> info value ok (22:05)
>
> ```
> Solar Module stats:
>   Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
>   Collector temperature: [...]

arghh. that problem again. I'm not sure why its failing. Can you add a line before #805 in `ems.cpp` like
```
myDebug("** got energy total value of %d", EMS_SolarModule.EnergyTotal);  // add this
if (EMS_SolarModule.EnergyTotal != EMS_VALUE_SHORT_NOTSET)
            rootSM[SM_ENERGYTOTAL] = (double)EMS_SolarModule.EnergyTotal / 10;
```

so I can see if the value is correct and its the arduinojson not handled signed ints correctly, or converting to floats.

proddy, 2019-07-02:
> update after 1 hour (using b8):
> the energy values are updated internally - see below, but there is no sm_data mqtt message at all! The info command shows at least a positive value for energy total!
> @proddy could you please check, why the solar modul doesn't send updates to the mqtt broker?
> 
> ```
> Solar Module stats:
>   Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
>   Collector temperature: 31.3 C
>   Bottom temperature: 30.0 C
>   Pump modulation: 0 %
>   Pump active: off
>   Pump working time: 11650 days 20 hours 15 minutes
>   Energy Last Hour: 39.0 Wh
>   Energy Today: 3741 Wh
>   Energy Total: 3237.2 kWH
> ```

I have no idea. Can you debug yourself? Put some debug statements after #815 in `ems-esp` to see why it's not getting picked up. If you fix it and do a PR, even better! I can't test everything at home unfortunately

S-Przybylski, 2019-07-02:
Dear @proddy 
i added the line in ems-esp.cpp (line 791 - version 1.8.1b2 because 1.8.1b8 doesn't work at all for mqtt solar messages currently).
After the reboot i got mulitply messages:
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32768
** got energy total value of -32372

....
without having any update from the solar module itself - the value is send once per hour at xx:55 in my installation. After the value was send from the solar modul at 22:55 the debug statement shows a different wrong value -32372.

proddy, 2019-07-02:
a few things to note:
- please try with the latest build, I'm not working on fixes from previous builds. The solar work was modified by another user and I can't verify it myself, only check the code structure. 
- if you solar is only sent every 55mins then no need to do the CRC check, so ok to delete all that code between the serializeJson and myESP.mqttPublish
- a quick fix for the negative value is changing the line to `rootSM[SM_ENERGYTOTAL] = (double)(abs(EMS_SolarModule.EnergyTota) / 10);`

S-Przybylski, 2019-07-02:
I personally do not have C knowledge, perhaps the orignator of the changes could help instead?

Regarding your comment 2: the solar energy updates are send only once per hour, all other solar values are send once per minute; Should i delete the crc check for testing then?

I tried:
        myDebug("** got energy total value of %d", EMS_SolarModule.EnergyTotal);  // add this
````        
        if (abs(EMS_SolarModule.EnergyTotal) != EMS_VALUE_SHORT_NOTSET)
            //rootSM[SM_ENERGYTOTAL] = (double)EMS_SolarModule.EnergyTotal / 10;
            rootSM[SM_ENERGYTOTAL] = (double)(abs(EMS_SolarModule.EnergyTotal) / 10);
            myDebug("** abs energy total value of %d", rootSM[SM_ENERGYTOTAL]); 
````
and got: 
** got energy total value of -32768
** abs energy total value of 1073741648

proddy, 2019-07-02:
@Vuego123 is the solar working for you in the latest build? I made some changes to how shorts are rendered but still getting the odd negative values in the MQTT payload. Would you mind trying, since you made some changes with https://github.com/proddy/EMS-ESP/pull/135

Vuego123, 2019-07-02:
I will test it tomorrow, but I'm having issues with my installation:

My thermostat is constantly giving error 13 and my boiler denies that it has a heating function for warm water (which it had for the last 5 years). 

This happened after I was sending raw messages to the ems bus that were supposed to update the desired warm water temperature. :-(

proddy, 2019-07-03:
Oops. I had a similar problem when I was experimenting with changing the
3-valve settings with raw messages. Luckily doing a factory reset fixed it.

Vuego123, 2019-07-03:
@proddy : thx for your tip to factory reset my boiler, that fixed it!

Test for the MQTT values was OK with the latest build. I added a screenshot of the raw mqtt messages. No negative values.

[image]

proddy, 2019-07-03:
and regarding the telnet issue, I'm moving most of the code to the web now so won't focus too much on the telnet. If you're happy with the latest quirks in the b2 build please close this github issue. cheers

Vuego123, 2019-07-03:
> phew, glad it works. That's a nice tool you have for peeking at MQTT messages, which is it?

mqttlens : it's very light, but good enough to peek at mqtt messages.

S-Przybylski, 2019-07-03:
Dear @proddy 
regarding the total energy value
i inserted yesterday the abs function as proposed, but it doesn''t work at all:
OK the MQTT value is now positive! But
i checked the energy total on the boilers menu this evening: 3321,1 kwh
The info command shows me 3232,5 kwh !!!
Grafana tells me that the energy total falls over the whole day hour by hour! 

Therefore i believe that the energy total function doesn't work correctly.

I try to switch over to the current dev b9.
Yes we can close this issue. Tomorrow i check if the sm100 mqtt messages are send for my SM100 module!

[image]

closed 2019-07-03 by S-Przybylski