emsesp/EMS-ESP #140: WIFI credential size doesn‘t Match underlying OS Sites

susisstrolch, 2019-06-29:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [doc](https://github.com/proddy/EMS-ESP/blob/master/README.md)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2019-06-29:
true. I'll change that. I'm re-writing the complete web interface so will take this into account. I'm fed up with Telnet ;-)

susisstrolch, 2019-06-30:
I've made a fix for MAX_PWD_LEN and MAX_SSID_LEN - both telnet/serial and webbrowser.
Now I can at least connect again to WIFI (my pwd is 16char ;))
Shall I attach the diff?

proddy, 2019-06-30:
2 PR is easy, less work for me to do. Thanks for fixing though! 

On a different note you may have noticed I put back the old Tx code and tx_mode options. The Tx logic (tx_mode=2) causes frequent ESP resets. I haven't been able to debug it - perhaps you can take a look?

susisstrolch, 2019-06-30:
Jep - Susi gave the permission to temporary put a 20m wire from cellar through the whole house to the office 😁

proddy, 2019-06-30:
LOL! I did the same, but luckily my office is on the 2nd floor. It'll make debugging much easier.

proddy, 2019-06-30:
added. thanks

closed 2019-06-30 by proddy