emsesp/EMS-ESP #141: Can‘t configure anymore via serial interface

susisstrolch, 2019-06-29:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [doc](https://github.com/proddy/EMS-ESP/blob/master/README.md)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

proddy, 2019-06-29:
that's correc, it's by design. Serial is disabled on an initial install as most users will be noobs that purchase a circuit board with the firmware pre-loaded. I'm open to suggestions though...what do you recommend?

susisstrolch, 2019-06-29:
I‘d suggest to keep the serial enabled. Imho with webinterface it‘s even for noobs easy  to switch to EMS-mode. Showing a „connect to EMS“ button would hide the ‚set serial on‘ but keep freedom to the experianced ones.

bbqkees, 2019-07-08:
I would keep it off by default.
I think of two kinds of people who would use the firmware.
One are the experienced guys like you who know how to build it, and those who either buy a preprogrammed one or can barely load the bin file on the Wemos themselves.

It's easier for an experienced user to change a build flag in the ini file to set serial on and then build and load the firmware than it is for 'noobs' to have yet another setting to make.
Major thing is that setting serial on by default disconnects you from the EMS bus initially which prevents you from using the firmware for its intended purpose: communicating with the EMS bus.
We know its like that because the ESP8266 only has one UART. People new to all this likely don't.

When plugging in the device to the boiler for the very first time its already quite comforting to have the blinking LED go solid so at least you know it had found the EMS bus before making any further settings.

I also found that when in AP mode (at least until 1.8.1b2), changing the serial to off is not persistent.
So on next boot its still on.

susisstrolch, 2019-07-08:
> It's easier for an experienced user to change a build flag in the in

Sounds fair...

Which flag - NO_SERIAL (as written in CHANGELOG.md) or DEFAULT_NO_SERIAL 
(as implemented in MyESP.cpp)?

bbqkees, 2019-07-08:
As I understood it now its -DNO_SERIAL in the ini file.

proddy, 2019-07-08:
sorry guys, it's DEFAULT_NO_SERIAL. I keep changing my mind ;)

proddy, 2019-07-27:
actually changed again to `-DFORCE_SERIAL`. Think I got it right this time in 1.8.1b23

closed 2019-07-27 by proddy