emsesp/EMS-ESP #144: Wrong reading thermostat in telnet interface Moduline 3000 (1.8.1b8) ?

kroon040, 2019-07-02:
Hi,

When reading the value in version 1.8.1b8, it displays the wrong value in de telnet interface.

In MQTT I see this

{"thermostat_hc":"1","thermostat_seltemp":16.5,"thermostat_currtemp":22.1,"thermostat_mode":"auto"}

in the telnet interface I see this

  Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.10)
  Setpoint room temperature: 16.1 C
  Current room temperature: 22.1 C
  Thermostat time is 22:50:37 2/7/2019
  Mode is set to ?

logging gives this

<--- RCPLUSStatusMessage(0x1A5)

(00:48:21.750) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:48:21.967) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:48:23.366) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 BA 00 17 (CRC=BC) #data=4
<--- RCPLUSStatusMessage(0x1A5)
(00:48:28.889) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:48:29.161) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 BB 80 00 21 00 00 01 00 00 FF 89 00 84 EA 00 80 00 (CRC=A2) #data=19

proddy, 2019-07-02:
what should the values be?

kroon040, 2019-07-02:
only setpoint is wrong, and should be 16.5

proddy, 2019-07-02:
can you provide some more 0x1A5 data. Looking for ones with an offset of 0 so starting with
`10 00 FF 00 01 A5` i.e. a 00 after the FF

kroon040, 2019-07-03:
(02:43:19.296) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 A9 01 91 (CRC=7C) #data=4

like this. It looks like that its wrong in de telnet console only, mqtt is correct 16.5 and in hass also. the values like 17.0 are ok so only the .5 displays as .1 in the telnet console

proddy, 2019-07-03:
I really need some examples to reproduce and find the rounding error. Do you have any telegrams that start with 10 00 FF 00 01 A5 ?

kroon040, 2019-07-03:
```
> <--- UBAMonitorFast(0x18)
> (04:56:19.765) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
> (04:56:21.445) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
> (04:56:21.502) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
> in(04:56:21.752) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 00 01 A5 00 E2 21 2B 00 00 2B 27 00 24 05 01 03 00 24 02 16 00 00 11 01 03 08 D8 00 (CRC=52) #data=25
> <--- RCPLUSStatusMessage(0x1A5)
> (04:56:22.129) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 08 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=F8) #data=15
> <--- RCPLUSStatusMessage(0x1A5)
> (04:56:22.331) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
> 
> Thermostat stats:
>   Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.10)
>   Setpoint room temperature: 21.1 C
>   Current room temperature: 22.6 C
>   Thermostat time is 16:54:54 3/7/2019
>   Mode is set to ?

```

proddy, 2019-07-03:
ok, perfect
`10 00 FF 00 01 A5 00 E2 21 2B 00 00 2B 27 00 24 05 01 03 00 24 02 16 00 00 11 01 03 08 D8 00`

current temp = 2 bytes at offset 0 = 0x00E2 = 226, divided by 10 = 22.6 degrees
setpoint temp = 1 byte at offset 3 = 0x2B = 43, divided by 2 = 21.5 degrees

what does `info` and mqtt payload say?

kroon040, 2019-07-03:
```
mosquitto_sub -h 192.168.180.150 -t 'home/ems-esp/#'
online
{"wWComfort":"Intelligent","wWSelTemp":60,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"wWCurTmp":23.9,"wWCurFlow":0,"curFlowTemp":27.9,"retTemp":25.4,"sysPress":1.5,"boilTemp":29.3,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"on","ServiceCode":"0H","ServiceCodeNumber":203}
0
0
{"thermostat_hc":"1","thermostat_seltemp":21.5,"thermostat_currtemp":22.4,"thermostat_mode":"auto"}
```

```
Thermostat stats:
  Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.10)
  Setpoint room temperature: 21.1 C
  Current room temperature: 22.4 C
  Thermostat time is 17:04:10 3/7/2019
  Mode is set to ?
```

moustic999, 2019-07-03:
I never noticed this wrong value as I don't have decimal ( 21.0 and 18.0 ) 
setpoint are multiples of 0.5 so I should never end with .1 :-)

kroon040, 2019-07-03:
Yes, but it's only in the telnet inteface and 1.8.0 was ok, but de latest bin (1.8.1b8) it is not ok.

proddy, 2019-07-03:
was a horrible mistake in my code, thanks for pointing it out. fixed now hopefully.

kroon040, 2019-07-05:
I uploaded 1.8.1b11 and that fixed.

`  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: 19.5 C
  Current room temperature: 22.5 C
  Thermostat time is 21:58:52 5/7/2019
  Mode is set to ?`

The only thing is does the reading from the mode already work for the ems+ protocol?

I always get

Mode is set to ?

proddy, 2019-07-05:
reading the mode on ems+ thermostats isn't implemented yet, because I don't know which telegram to request. I think its byte 10 from 0x1A5 but would need you guys to verify it. Please send me some example 1A5 messages when the thermostat is in manual, auto and day mode

kroon040, 2019-07-06:
Here some samples, dit 3 times switch to manual and switch to auto, and grabbed all the 0x1A5 messages.

`auto

(00:26:38.695) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 07 01 A5 21 (CRC=5D) #data=1
(00:26:39.816) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 15 01 A5 03 (CRC=EF) #data=1

manual
(00:28:46.305) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 05 A0 (CRC=A2) #data=2
(00:28:47.584) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 02 (CRC=16) #data=1

auto
(00:29:57.630) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 00 9A (CRC=92) #data=2
(00:29:58.182) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 03 (CRC=17) #data=1
(00:30:00.285) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 07 01 A5 21 (CRC=5D) #data=1

manual
(00:31:47.049) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 05 A0 (CRC=A2) #data=2
(00:31:48.270) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 02 (CRC=16) #data=1

auto
(00:33:08.091) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 07 01 A5 21 00 96 03 (CRC=EC) #data=4
(00:33:08.820) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 15 01 A5 03 (CRC=EF) #data=1
(00:33:09.880) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 96 00 3C (CRC=2F) #data=4

manual
(00:34:23.311) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 05 A0 (CRC=A2) #data=2
(00:34:25.019) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0A 01 A5 02 (CRC=16) #data=1
(00:34:25.941) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 07 01 A5 27 (CRC=5B) #data=1
(00:34:26.804) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 15 01 A5 02 (CRC=EE) #data=1`

kroon040, 2019-07-09:
@proddy , do you need more, or other telegram messages?

proddy, 2019-07-09:
Still hard to find a pattern. how often is 01A5 sent when you don't adjust the thermostat manually?

proddy, 2019-07-12:
this bug is resolved. reading the mode and setting temps are in #145

closed 2019-07-12 by proddy