emsesp/EMS-ESP #145: Write supports EMS+ device Nefit Moduline 3000

kroon040, 2019-07-02:
Hi,

It Should be nice to have write support on the nefit moduline 3000 (RC300/RC310) for setting the temperature

The all use the EMS+ protocol. Hoping to add support soon.

proddy, 2019-07-02:
Yes, it's on the list. I was working with a @moustic999 on some tests a while back, and there is also a special RC300 branch we started. The wiki https://github.com/proddy/EMS-ESP/wiki/RC3xx-Thermostats talks about the approach. I think a good start to change the temp is via the 0x01B9 type, so sending something like this:

`send 0B 10 FF 00 01 B9 00`

to set it to 20.5 degrees (0x29)

kroon040, 2019-07-03:
Hi

The thermostaat went from auto to manual, but the temperatuur was not set to 20.5, just took the temp where is was on manual mode 17.0 (see below the logging)

I was trying to understand the wiki (https://github.com/proddy/EMS-ESP/wiki/Deciphering-EMS-Plus)

AUTO Mode, setting temporary temp to 21.5(0x2B)
(03:43:45.965) Gateway -> Thermostat, type 0x01B9 telegram: 48 10 FF 08 01 B9 2B (CRC=FA)
(03:43:46.445) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 08 01 B9 2B (CRC=17)

How is the src work, 0B, 10, 48 when to choose what?

regards Richard

-----------------------------------------------------------------------------------------------------
send 0B 10 FF 00 01 B9 00
(03:56:19.147) Sending raw: 0B 10 FF 00 01 B9 00 00
(03:56:19.491) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(03:56:19.766) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 63 00 00 80 00 40 40 01 06 80 00 01 2E 00 00 0E 30 48 00 CB 00 00 00 (CRC=06) #data=25
<--- UBAMonitorFast(0x18)
(03:56:20.001) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
(03:56:20.265) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 08 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=A3) #data=19
<--- UBAMonitorWWMessage(0x34)
(03:56:20.571) Thermostat -> all, type 0x01B9 telegram: 10 00 FF 00 01 B9 00 (CRC=7C) #data=1
<--- RCPLUSStatusHeating(0x1B9)
(03:56:20.766) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20) #data=2
(03:56:21.037) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(03:56:27.412) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 70 00 CA (CRC=62) #data=4
<--- RCPLUSStatusMessage(0x1A5)
(03:56:29.623) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 63 00 00 80 00 40 40 01 06 80 00 01 2E 00 00 0E 30 48 00 CB 00 00 00 (CRC=06) #data=25
<--- UBAMonitorFast(0x18)
(03:56:29.890) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
(03:56:30.122) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 06 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=98) #data=19
<--- UBAMonitorWWMessage(0x34)
(03:56:31.755) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(03:56:31.808) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(03:56:32.186) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(03:56:39.570) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 06 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=98) #data=19
<--- UBAMonitorWWMessage(0x34)
(03:56:39.800) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 63 00 00 80 00 40 40 01 06 80 00 01 2E 00 00 0E 30 48 00 CB 00 00 00 (CRC=06) #data=25
<--- UBAMonitorFast(0x18)

Thermostat stats:
  Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.10)
  Setpoint room temperature: 17.0 C
  Current room temperature: 21.9 C
  Thermostat time is 11:22:35 3/7/2019
  Mode is set to ?

proddy, 2019-07-03:
ah, didn't realize you had a Gateway too which is the master. So either try impersonating the gateway with `send 48 10 FF 08 01 B9 2B` or send the request to the gateway which will pass it on to the thermostat with `send 0B 48 FF 08 01 B9 2B`

kroon040, 2019-07-03:
No I have nog gateway, that I have from the example of the wiki.

send 0B 48 FF 08 01 B9 2B

gives this, no temp set

send 0B 48 FF 08 01 B9 2B
(02:52:43.216) Sending raw: 0B 48 FF 08 01 B9 2B 00
(02:52:43.595) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(02:52:44.316) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(02:52:46.114) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0E 30 48 00 CB 00 00 00 (CRC=E0) #data=25
<--- UBAMonitorFast(0x18)
(02:52:46.336) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
(02:52:46.589) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 00 F3 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=6F) #data=19
<--- UBAMonitorWWMessage(0x34)
(02:52:48.724) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20) #data=2
(02:52:51.018) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:52:53.019) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0E 30 48 00 CB 00 00 00 (CRC=E0) #data=25
<--- UBAMonitorFast(0x18)
(02:52:56.011) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:52:56.250) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0E 30 48 00 CB 00 00 00 (CRC=E0) #data=25
<--- UBAMonitorFast(0x18)
(02:52:56.508) Boiler -> all, type 0x19 telegram: 08 00 19 00 80 00 01 3D 80 00 00 00 00 00 00 E1 8B 04 84 E7 00 00 00 03 85 5C 00 5C 9B 80 00 (CRC=99) #data=27
<--- UBAMonitorSlow(0x19)
(02:52:56.843) Boiler -> all, type 0x1C telegram: 08 00 1C 00 92 07 01 0D 08 0D 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=A4) #data=25
(02:52:57.067) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
(02:52:57.366) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 00 F3 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=6F) #data=19
<--- UBAMonitorWWMessage(0x34)
(02:52:57.601) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 0E 03 32 29 02 01 10 FF 00 (CRC=B9) #data=11
<--- RCTime(0x06)
(02:52:57.813) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(02:53:00.895) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C)
(02:53:00.971) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 07 0E 03 32 2C 02 01 10 FF 00 (CRC=5C) #data=11
<--- RCTime(0x06)
(02:53:00.983) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4)
(02:53:01.388) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(02:53:01.587) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0)
(02:53:02.017) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:53:02.290) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78)
(02:53:02.334) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 3C FB 00 28 00 00 46 EC FF FF 00 (CRC=7B) #data=13
<--- UBAParameterWW(0x33)
Publishing boiler data via MQTT
(02:53:02.584) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC)
(02:53:02.616) Corrupt telegram: 88 16 00 20 EC
(02:53:02.668) Boiler -> me, type 0x16 telegram: 08 0B 16 00 FF 46 4B 00 06 FA 0A 01 01 46 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C (CRC=19) #data=27
<--- UBAParametersMessage(0x16)
(02:53:02.674) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4)
(02:53:02.728) Boiler -> me, type 0x14 telegram: 08 0B 14 00 29 FD 1E (CRC=4B) #data=3
<--- UBATotalUptimeMessage(0x14)
(02:53:03.104) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0E 30 48 00 CB 00 00 00 (CRC=E0) #data=25
<--- UBAMonitorFast(0x18)
(02:53:03.672) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(02:53:05.071) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:53:06.029) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:53:06.245) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 06 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=B6) #data=21
(02:53:06.579) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 00 F3 80 00 21 00 00 01 00 00 FF 8B 00 84 F0 00 80 00 (CRC=6F) #data=19
<--- UBAMonitorWWMessage(0x34)
[... cut ...]
(02:53:19.971) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)
(02:53:21.022) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 2A 00 00 80 00 40 40 00 F3 80 00 01 08 00 00 0F 30 48 00 CB 00 00 00 (CRC=60) #data=25
<--- UBAMonitorFast(0x18)

Thermostat stats:
  Thermostat: RC3x0/Nefit Moduline 1010H (ProductID:158 Version:11.10)
  Setpoint room temperature: 16.1 C
  Current room temperature: 22.4 C
  Thermostat time is 14:50:44 3/7/2019
  Mode is set to ?

proddy, 2019-07-03:
from the wiki, 8 and 10 are offsets for setting the temp in auto and manual. Try experimenting with

`send 0B 10 FF 08 01 B9 2B`
or
`send 0B 10 FF 0A 01 B9 2B`

to see if anything jumps to 21.5 degrees

kroon040, 2019-07-03:
Hi it works like this now

The thermostat must be in manual mode
send 0B 10 FF 0A 01 B9 2B , it goes to 21.5 and stays on manual 

The thermostat is in auto mode
send 0B 10 FF 08 01 B9 2B, is goes to 21.5 and stays auto

This combination only works so in automode (0A) doesn't work, and manual mode (08) also not working

proddy, 2019-07-03:
yes, that is expected behavior. It's how far @moustic999 got with his investigation. A sloppy way is to check the op mode and then send the temp in 2 Tx calls. You can get the mode with `send 0B 10 FF 01 01 B9', which will return a telegram with data length 1 and last byte is 0 for Manual or FF for Auto.

proddy, 2019-07-03:
actually thinking about it, it should switch to manual and always set the manual temp?

kroon040, 2019-07-03:
nope it doesn't

this works for setting mode

send 0B 10 FF 00 01 B9 00 (manual, and it takes the temp what I have set on my thermostat last time it was on manual)
send 0B 10 FF 00 01 B9 FF (auto, uses the temp from the program)

proddy, 2019-07-03:
No, I meant the logic is that when you change the temp in HA or somewhere else it should first force the thermostat to manual mode and then set the temp. 2 telegrams

kroon040, 2019-07-03:
The logic is that it stays on the mode (auto or manual)

For example , I have a default program (auto) the temp from 17:00 - 23:00 is default in the program 19.5 . If I have it some cold, I let the program running and change it to 21.0, then om 23:00 the temp goes down for the night (program keeps running in auto). So that's how I use the thermostat and also with HA . I use manual mode only for vacation (15.0, for x weeks).

moustic999, 2019-07-03:
@kroon040 you are right for the mode.
Mode does not change when changing temp.
in Auto, when you change temp, it is till the next setpoint. 
Pretty happy to know that the 01B9 commands works for others than me :-)
@Proddy I know you're busy with solving bug, Don't hesitate to mail when you will begin writing EMS+ Write. I had a look on it but that needs lot of change, and I don' feel confortable enough with your code to implement it. But I can help and test.

kroon040, 2019-07-05:
Same is for me, I can test.

proddy, 2019-07-05:
ok I need some help with the sequence of steps. When setting the temp I guess we'll need to first determine the mode (auto/manual) and then send the new temperature to the correct offset. Its a bit sloppy and I'm sure there is an easier way but this is all we have at the moment, agree? If you provide the commands I'll add the code and explain how it works

kroon040, 2019-07-06:
agree, to do it like this, read the mode (auto/manual), and set it with the right command the temperature

reading auto / manual, see other topic.

these we had also.

The thermostat must be in manual mode
send 0B 10 FF 0A 01 B9 2B , it goes to 21.5 and stays on manual

The thermostat is in auto mode
send 0B 10 FF 08 01 B9 2B, is goes to 21.5 and stays auto

proddy, 2019-07-08:
ok. I'll add that to do the todo list

proddy, 2019-07-12:
https://github.com/proddy/EMS-ESP/issues/144 has example messages to be used to implement this feature. Waiting for the broadcast messages on 1A5

moustic999, 2019-07-12:
Proddy,
the 01A5 is broadcasted every 120 seconds, even with you don't change anything.
this broadcasted message is in 2 part because of the lenght 
example : 
`(03:22:49.018) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 00 01 A5 80 00 01 2C 29 00 2C 20 01 B6 03 03 01 01 B6 02 28 00 00 11 01 03 FF FF 00 (CRC=A1), #data=24
<--- RCPLUSStatusMessage(0x1A5) received
(03:22:49.708) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 06 04 00 00 00 00 FF 64 4B 00 3C 01 FF 01 02 (CRC=93), #data=14
<--- RCPLUSStatusMessage(0x1A5) received`

if you change something then, the 01A5 is sent but only with modified offsets, not full lenght
example : 
`(02:53:19.068) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 9F 01 9B (CRC=AE) #data=4
<--- RCPLUSStatusMessage(0x1A5)`

proddy, 2019-07-12:
Ok. In the broadcast messages do you have any idea  which byte holds the
mode setting?

moustic999, 2019-07-12:
as specified in the wiki, it is offset 0x0A 
it is 1-byte to be read bitwise
bit 1 is the current mode : 0 manual / 1 automatic
bit 2 specify the level in automatic : day/night or comfort/eco 
eco = night = 0
comfort = day = 1

The bit you need is the bit 1

proddy, 2019-07-12:
ok thanks @moustic999

proddy, 2019-07-15:
@kroon040 added a check for mode in 1.8.1b16 so it shouldn't show as ? any more. Can you test? Next step would be to implement the writing of the temps.

kroon040, 2019-07-15:
@proddy , I can check, is there a bin on the release page of 1.8.1b16, of do in need to extract it from dev, and compile it myself?

proddy, 2019-07-15:
always easier to build yourself from source (it's only 2 commands you need
to type). I can also generate a .bin file if you prefer - I'm ok either way

kroon040, 2019-07-15:
If you could generate a bin file, it should be nice

proddy, 2019-07-15:
ok, it's in https://github.com/proddy/EMS-ESP/releases/tag/1.8.1-beta

kroon040, 2019-07-15:
Tested, but not weel, I lost reading temp and default no reading (manual/auto)

Flashing firmware and rebooted, after 10 min, still

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 22:24:21 15/7/2019
  Mode is set to ?

setting to manual

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 22:28:23 15/7/2019
  Mode is set to manual

back to auto

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 22:34:17 15/7/2019
  Mode is set to low

proddy, 2019-07-15:
strange, I thought those 1A5's would be broadcasted every 2 mins without any intervention?

the "low" I can fix in the code tomorrow

kroon040, 2019-07-16:
I noticed that read the current temp , works after a long time (when it changes) , setpoint reads nothing.

I checked, these values 1A5 message I see  when I do not touch the thermostat

```
(01:02:06.738) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 C1 00 79 (CRC=27) #data=4
(01:03:04.490) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
(01:03:05.306) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 C0 00 7A (CRC=20) #data=4
```

proddy, 2019-07-16:
I was counting on telegrams like 

`Thermostat -> all, type 0x01A5 telegram: 10 00 FF 00 01 A5 .....`

transmitted every 2 mins. Note the offset is 00 (after the FF). If that is not the case let me know and we need to go to plan B

kroon040, 2019-07-16:
Hi,

Did a long look for 10mins, for the messages of 0x01A5, displayed below. 

`(03:22:54.854) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0F 01 A5 01 46 (CRC=3C) #data=2
<--- RCPLUSStatusMessage(0x1A5)
(03:22:55.048) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 F4 (CRC=AC) #data=2
(03:23:22.933) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15

(03:24:54.233) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 F2 01 48 (CRC=D0) #data=4
(03:25:23.220) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15

(03:25:57.412) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 F1 01 49 (CRC=DD) #data=4

(00:04:18.644) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 EA 01 50 (CRC=A8) #data=4
(00:04:48.280) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
(00:05:18.191) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 E9 01 51 (CRC=A5) #data=4
(00:06:21.428) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 E8 01 52 (CRC=A2) #data=4
(00:07:20.938) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 E7 01 53 (CRC=9F) #data=4
(00:08:48.613) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
(00:09:20.264) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 00 E5 01 55 (CRC=91) #data=4`

kroon040, 2019-07-17:
Hi,

in 1.8b17, I see you have manual and auto. Only manual = auto, and auto = manual.

If I put the thermostat in auto, the ems-esp give manual in telnet console.

proddy, 2019-07-17:
think I used the wrong bit. Pls try again with 1.8.1b18 (https://github.com/proddy/EMS-ESP/releases/tag/1.8.1-beta)

kroon040, 2019-07-18:
Hi,

The mode auto/manual is working now perfect.

Now it should be fine to have back the current room temp and the current setpoint temp, that worked in the old 1.8b5 (two weeks ago). 

regards Richard

proddy, 2019-07-18:
I looked at the code again and can't see any differences between earlier versions. If the 1A5 is broadcasted it should pick up the thermostat temperatures. Looks like you're not getting those broadcast messages, 

One last thing you can try is going back to the stable 1.8.0 and see what happens. Otherwise I'll need someone else with an actual Moduline 3000/1010 and C programming knowledge to debug and fix. I'm just guessing in the dark now I'm afraid.

moustic999, 2019-07-18:
Seems pretty strange that message isn't broadcasted. it is the case here at home on a buderus rc300.
@kroon040 can you try to trigger a manual read : 
`send 0B 90 FF 00 19 01 A5` 
that will read all missing data.
Can you post the response ?

kroon040, 2019-07-18:
```
the send and waited for a 0x1a5 message

`send 0B 90 FF 00 19 01 A5
(00:00:40.058) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:00:41.583) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 00 00 00 (CRC=7F) #data=3
(00:00:44.878) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:00:45.124) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 8F 80 00 21 00 00 01 00 01 01 66 00 86 B1 00 80 00 (CRC=A4) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:45.343) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 DD 00 00 80 00 40 40 01 8F 80 00 01 CF 00 00 0F 30 48 00 CB 00 00 00 (CRC=E8) #data=25
<--- UBAMonitorFast(0x18)
(00:00:51.816) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 16 12 32 2E 03 01 10 FF 00 (CRC=40) #data=11
<--- RCTime(0x06)
(00:00:54.805) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 DD 00 00 80 00 40 40 01 8F 80 00 01 CF 00 00 0F 30 48 00 CB 00 00 00 (CRC=E8) #data=25
<--- UBAMonitorFast(0x18)
(00:00:55.062) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:00:55.338) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 8F 80 00 21 00 00 01 00 01 01 66 00 86 B1 00 80 00 (CRC=A4) #data=19
<--- UBAMonitorWWMessage(0x34)
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:01:00.285) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 01 (CRC=20) #data=2
(00:01:01.023) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C) #data=216
(00:01:01.115) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 07 16 12 32 37 03 01 10 FF 00 (CRC=0E) #data=11
<--- RCTime(0x06)
(00:01:01.155) Corrupt telegram: 0B 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 F6
(00:01:01.422) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4) #data=216
(00:01:01.454) Corrupt telegram: 88 18 00 20 D4
(00:01:01.506) Boiler -> me, type 0x18 telegram: 08 0B 18 00 05 01 DD 00 00 80 00 40 40 01 8F 80 00 01 CF 00 00 0F 30 48 00 CB 00 00 00 (CRC=5B) #data=25
<--- UBAMonitorFast(0x18)
(00:01:01.512) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0) #data=216
(00:01:02.616) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(00:01:04.291) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78) #data=216
(00:01:04.335) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 3C FB 00 28 00 00 46 EC FF FF 00 (CRC=7B) #data=13
<--- UBAParameterWW(0x33)
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
(00:01:04.460) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(00:01:04.584) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC) #data=216
(00:01:04.638) Corrupt telegram: 88 16 00 20 EC
(00:01:05.030) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 DD 00 00 80 00 40 40 01 8F 80 00 01 CF 00 00 0F 30 48 00 CB 00 00 00 (CRC=E8) #data=25
<--- UBAMonitorFast(0x18)
(00:01:05.247) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:05.492) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 8F 80 00 21 00 00 01 00 01 01 66 00 86 B1 00 80 00 (CRC=A4) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:05.757) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4) #data=216
(00:01:05.787) Boiler -> me, type 0x14 telegram: 08 0B 14 00 2A 53 B7 (CRC=AB) #data=3
<--- UBATotalUptimeMessage(0x14)
(00:01:06.515) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(00:01:09.308) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 BC 00 15 (CRC=A6) #data=4
<--- RCPLUSStatusMessage(0x1A5)
````

kroon040, 2019-07-18:
back to 1.8.0, the temps working, and I get also better data with the manual send code

System Logging set to Verbose
send 0B 90 FF 00 19 01 A5
(00:00:52.617) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:00:52.751) Thermostat -> me, type 0x01A5 telegram: 10 0B FF 00 01 A5 00 D8 21 21 00 00 21 23 01 B6 01 01 03 01 B6 00 1B 00 00 11 01 04 08 75 00 (CRC=2B) #data=25
<--- RCPLUSStatusMessage(0x1A5)
(00:00:56.326) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 86 80 00 21 00 00 01 00 01 01 66 00 86 B1 00 80 00 (CRC=0E) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:56.598) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 D5 00 00 80 00 40 40 01 86 80 00 01 C1 00 00 0F 30 48 00 CB 00 00 00 (CRC=AA) #data=25
<--- UBAMonitorFast(0x18)
(00:00:56.871) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:00:57.750) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 16 12 39 31 03 01 10 FF 00 (CRC=79) #data=11
<--- RCTime(0x06)
log n

proddy, 2019-07-18:
ahh, I think I see the problem. I suspect you have tx_mode set to 0 with the 1.8.1 beta versions because I see none of the Tx working well. In the master 1.8.0 it's fixed at tx_mode of 2.

Sorry, but could you go back to 1.8.1 beta again, `set tx_mode 2` and see if its any better?

kroon040, 2019-07-18:
that didn't work

set tx_mode 2
tx_mode changed.

Error! Unable to read the EMS bus.

proddy, 2019-07-18:
comparing 1.8.0 with 1.8.1 (and tx_mode 2) it should be identical logic so not sure why Tx is so bad.

kroon040, 2019-07-19:
I can now set in de b19, de tx_mode 2 . The problem was that serial was on.

I don't get the right values yey of the temps.

send 0B 90 FF 00 19 01 A5
(00:01:19.669) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:01:24.672) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 DC 1C 00 80 00 40 40 01 AB 80 00 01 CF 00 00 0F 30 41 01 31 00 00 00 (CRC=67) #data=25
<--- UBAMonitorFast(0x18)
(00:01:24.917) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:25.134) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 AB 80 00 31 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=0D) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:25.484) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:01:25.533) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:01:25.742) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:01:32.903) Thermostat -> Boiler, type 0x23 telegram: 10 08 23 00 00 00 00 (CRC=7F) #data=3
(00:01:34.717) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 AA 80 00 31 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=55) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:34.949) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 DC 1C 00 80 00 40 40 01 AA 80 00 01 CE 00 00 0F 30 41 01 31 00 00 00 (CRC=15) #data=25
<--- UBAMonitorFast(0x18)
(00:01:35.247) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:44.611) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:44.877) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 AA 80 00 31 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=55) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:45.108) Boiler -> all, type 0x18 telegram: 08 00 18 00 00 01 DA 1C 00 80 00 40 40 01 AA 80 00 01 CE 00 00 0F 30 41 01 31 00 00 00 (CRC=E5) #data=25
<--- UBAMonitorFast(0x18)
(00:01:49.544) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 FE 00 3C (CRC=9E) #data=4
<--- RCPLUSStatusMessage(0x1A5)
(00:01:49.750) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 08 01 A5 01 FE (CRC=F4) #data=2
<--- RCPLUSStatusMessage(0x1A5)

proddy, 2019-07-19:
and you're 100% sure this worked with 1.8.0? the temps were coming in?

proddy, 2019-07-19:
Another peculiar thing is that you don't get a response back from your `send 0B 90 FF 00 19 01 A5`. You should see something back like
Thermostat -> me, type 0x01A5 telegram: 10 08 FF 00 01 A5 ..... 

@moustic999 does this work for you?

proddy, 2019-07-19:
actually @kroon040  can you also try `send 0B 90 FF 00 01 A5' and see what comes back?
if nothing than try `send 0B 90 FF 00 01 A5 1A'

I'm trying to work out how to read from EMS+

kroon040, 2019-07-19:
I did on the d1 mini a full erase and installed 1.8b19. but also no luck.
here some output of the commands and info from 1.8.0 and 1.8b19. And you also see the temps ok in 1.8.0

**with 18b19 tx_mode 2**

(00:02:55.145) Sending raw: 0B 90 FF 00 01 A5 00
(00:02:55.191) Thermostat -> me, type 0xA5B8 telegram: 10 0B FF 00 A5 B8 (CRC=DE)
(00:02:55.753) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:02:55.810) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:02:56.495) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:02:56.704) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:02:57.072) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 09 13 16 2A 04 01 10 FF 00 (CRC=EF) #data=11
<--- RCTime(0x06)
(00:02:57.289) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:02:57.511) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 67 80 00 21 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=4E) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:02:57.784) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C3 00 00 80 00 40 40 01 67 80 00 01 A0 00 00 0F 30 48 00 CB 00 00 00 (CRC=F5) #data=25
<--- UBAMonitorFast(0x18)
Requesting scheduled EMS device data
Requesting type (0x1A5) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 09:23:49 19/7/2019
  Mode is set to ?

(00:04:35.850) Sending raw: 0B 90 FF 00 01 A5 1A 00
(00:04:35.898) Thermostat -> me, type 0xA51A telegram: 10 0B FF 00 A5 1A (CRC=7C)
(00:04:36.904) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:04:37.224) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 65 80 00 21 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=FE) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:04:37.444) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C2 00 00 80 00 40 40 01 65 80 00 01 9D 00 00 0F 30 48 00 CB 00 00 00 (CRC=DA) #data=25
<--- UBAMonitorFast(0x18)
(00:04:42.929) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C1 00 00 80 00 40 40 01 65 80 00 01 9D 00 00 0F 30 48 00 CB 00 00 00 (CRC=A2) #data=25
<--- UBAMonitorFast(0x18)
(00:04:43.934) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C2 00 00 80 00 40 40 01 65 80 00 01 9D 00 00 0F 30 48 00 CB 00 00 00 (CRC=DA) #data=25
<--- UBAMonitorFast(0x18)
(00:04:44.864) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C1 00 00 80 00 40 40 01 65 80 00 01 9D 00 00 0F 30 48 00 CB 00 00 00 (CRC=A2) #data=25
<--- UBAMonitorFast(0x18)
(00:04:46.865) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C1 00 00 80 00 40 40 01 65 80 00 01 9D 00 00 0F 30 48 00 CB 00 00 00 (CRC=A2) #data=25
<--- UBAMonitorFast(0x18)
(00:04:47.163) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:04:47.382) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 65 80 00 21 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=FE) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:04:56.021) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:04:56.077) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:04:56.704) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:04:56.910) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:04:57.110) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 64 80 00 21 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=A6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:04:57.339) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 C1 00 00 80 00 40 40 01 64 80 00 01 9C 00 00 0F 30 48 00 CB 00 00 00 (CRC=D0) #data=25
<--- UBAMonitorFast(0x18)
(00:04:57.671) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:04:58.063) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 09 13 18 2C 04 01 10 FF 00 (CRC=84) #data=11
<--- RCTime(0x06)
Requesting scheduled EMS device data
Requesting type (0x1A5) from dest 0x10
Requesting type RCTime(0x06) from

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 3000 (ProductID:158 Version:11.10)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 09:24:47 19/7/2019
[... 83 lines cut ...]
(00:02:48.933) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:02:52.847) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 09 13 1C 2F 04 01 10 FF 00 (CRC=FD) #data=11
<--- RCTime(0x06)
(00:02:57.175) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 5E 80 00 21 00 00 01 00 01 01 68 00 86 B4 00 80 00 (CRC=FA) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:02:57.457) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 BE 00 00 80 00 40 40 01 5E 80 00 01 97 00 00 0F 30 48 00 CB 00 00 00 (CRC=23) #data=25
<--- UBAMonitorFast(0x18)
[... cut ...]
(00:04:48.256) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:04:48.510) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 00 01 A5 00 D0 21 22 00 00 22 27 01 E0 01 01 03 01 E0 00 5A 00 00 11 01 04 08 1D 00 (CRC=FC) #data=25
<--- RCPLUSStatusMessage(0x1A5)
(00:04:48.890) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:04:49.092) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4

moustic999, 2019-07-19:
> Another peculiar thing is that you don't get a response back from your `send 0B 90 FF 00 19 01 A5`. You should see something back like
> Thermostat -> me, type 0x01A5 telegram: 10 08 FF 00 01 A5 .....
> 
> @moustic999 does this work for you?

Indeed, It works here, The version I use i an old version before tx changes. i use TX-mode 2 others doesn't work.

But I did not test with latest version.
I will try to test this evening.

if you don't see the ' sending ... sentence, it means that your tx does not work properly.
you can also check the queue ? If there is too many items in queue your manual sending is never processed. cleaning the queue can help.

moustic999, 2019-07-19:
> actually @kroon040 can you also try `send 0B 90 FF 00 01 A5` and see what comes back?
> if nothing than try `send 0B 90 FF 00 01 A5 1A`
> 
> I'm trying to work out how to read from EMS+

the send command I proposed is to read-request for 01 A5  
also this 01 A5 is send every 2 mins.

the write command is 01 B9

moustic999, 2019-07-19:
@kroon040 
I checked your logs, but please with the version which works, can you try this :
`send 0B 90 FF 00 19 01 A5`
i don't see this test. others proposed by @proddy are not correct for EMS+

this raw means : 
0B : sender
90 : target
FF : EMS + extended message
00 : Offset
19 : Lenght to retrieve
01 A5 : Buffer / register to query --> also knowns as command type ;-)

proddy, 2019-07-19:
> the send command I proposed is to read-request for 01 A5
> also this 01 A5 is send every 2 mins.
>
> the write command is 01 B9

@moustic999 do those raw send commands work on your thermostat? With @kroon040's Moduline 3000 it comes back with the wrong type each time.

moustic999, 2019-07-19:
only the one I proposed is supposed to work, others are not ok 
I can redo trials this evening, but I used these commands for all my investigations without issues.
I would like to have the complete 01a5 message from @kroon040, to confirm the value are correct.

kroon040, 2019-07-19:
Hi

Did the test with working version 1.8.0, and looks ok. Als did the test with 1.8b19 with tx mode 0 and 2, both not ok.

**1.8.0** 

send 0B 90 FF 00 19 01 A5
(00:02:21.350) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:02:21.486) Thermostat -> me, type 0x01A5 telegram: 10 0B FF 00 01 A5 00 D7 21 22 00 00 22 27 01 65 01 01 03 01 65 00 D5 00 00 11 01 04 08 6B 00 (CRC=05) #data=25

**1.8b19**

  serial=off
  heartbeat=off
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=10
  heating_circuit=1
  boiler_type=08
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120
  tx_mode=0

send 0B 90 FF 00 19 01 A5
(00:01:10.892) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:01:11.328) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(00:01:11.982) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(00:01:14.113) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 B5 00 00 80 00 40 40 01 44 80 00 01 74 00 00 0F 30 48 00 CB 00 00 00 (CRC=78) #data=25
<--- UBAMonitorFast(0x18)
(00:01:14.335) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:14.588) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 44 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=35) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:19.788) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:01:19.841) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:01:20.697) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:01:20.910) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:01:21.610) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 61 00 D9 (CRC=35) #data=4
<--- RCPLUSStatusMessage(0x1A5)
(00:01:24.086) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 44 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=35) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:24.315) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 B5 00 00 80 00 40 40 01 44 80 00 01 74 00 00 0F 30 48 00 CB 00 00 00 (CRC=78) #data=25
<--- UBAMonitorFast(0x18)
(00:01:24.633) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21

**1.8b19**

  serial=off
  heartbeat=off
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=10
  heating_circuit=1
  boiler_type=08
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120
  tx_mode=2

log v
System Logging set to Verbose
send 0B 90 FF 00 19 01 A5(00:00:45.481) Thermostat -> all, type 0xA2 telegram: 10 00 A2 00 00 00 00 00 00 (CRC=CA) #data=5

(00:00:47.105) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:00:51.130) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:00:51.375) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 41 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=14) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:51.595) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 B2 00 00 80 00 40 40 01 41 80 00 01 72 00 00 0F 30 48 00 CB 00 00 00 (CRC=58) #data=25
<--- UBAMonitorFast(0x18)
(00:00:56.416) Thermostat -> all, type 0xBF telegram: 10 00 BF 00 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=71) #data=24
(00:00:57.990) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 0D 01 A5 01 5D 00 DD (CRC=C1) #data=4
<--- RCPLUSStatusMessage(0x1A5)
(00:00:58.192) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:00:58.352) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:00:59.081) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 04 07 00 00 00 00 FF 00 4B 00 3C 01 FF 01 00 (CRC=10) #data=15
<--- RCPLUSStatusMessage(0x1A5)
(00:00:59.294) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
Requesting scheduled EMS device data
Requesting type (0x1A5) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
l(00:01:01.081) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 B2 00 00 80 00 40 40 01 41 80 00 01 72 00 00 0F 30 48 00 CB 00 00 00 (CRC=58) #data=25
<--- UBAMonitorFast(0x18)
(00:01:01.339) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
og n(00:01:01.615) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 41 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=14) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:02.020) Sending read of type 0x1A5 to 0x10: telegram: 0B 90 A5 00 20 (CRC=D2) #data=216
Read failed. Retrying attempt 1/2...
(00:01:02.062) Thermostat -> me, type 0xA5 telegram: 10 0B A5 00 (CRC=FF)

(00:01:02.141) Boiler -

proddy, 2019-07-19:
ok, that explains it. The Tx code in 1.8.1 doesn't work for EMS+ devices - something broke since 1.8.0. I'll wait for more people to run tests to see if its an edge case with the Moduline 3000 or something generic in the code. I'll also continue looking into the diffs to see what possibly could have changed.

kroon040, 2019-07-19:
May we know later this day
@moustic999 , you will test it later this day with last beta?

susisstrolch, 2019-07-19:
Stay tuned - a revamped "tx_mode 2" pull request is in the queue..
Found a bunch of reasons for WTD timeout and other issues...

proddy, 2019-07-19:
cool. in the meantime @kroon040 try `tx_mode 1` which was the first EMS+ implementation

kroon040, 2019-07-19:
tx_mode 1, also not.....

send 0B 90 FF 00 19 01 A5
(00:01:50.062) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 14 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=17) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:50.281) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 65 00 00 80 00 40 40 01 14 80 00 01 2F 00 00 0F 30 48 00 CB 00 00 00 (CRC=47) #data=25
<--- UBAMonitorFast(0x18)
(00:01:50.523) Sending raw: 0B 90 FF 00 19 01 A5 00
(00:01:50.545) Corrupt telegram: 0B 90 FF 00 19 A5 FD
(00:01:50.948) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:01:51.262) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(00:01:51.852) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(00:01:53.951) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(00:01:54.001) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(00:01:54.209) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(00:01:55.965) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 0E 13 32 1C 04 01 10 FF 00 (CRC=E7) #data=11
<--- RCTime(0x06)
(00:02:00.048) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 65 00 00 80 00 40 40 01 14 80 00 01 2F 00 00 0F 30 48 00 CB 00 00 00 (CRC=47) #data=25
<--- UBAMonitorFast(0x18)
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:02:00.349) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
(00:02:00.568) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 14 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=17) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:02:02.276) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C) #data=216
(00:02:02.293) Corrupt telegram: 0B 90 06 20 6C
(00:02:02.352) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 07 0E 13 32 22 04 01 10 FF 00 (CRC=2D) #data=11
<--- RCTime(0x06)
(00:02:02.358) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4) #data=216
Read failed. Retrying attempt 1/2...
(00:02:02.374) 0x00 -> me, type 0x88 telegram: 00 0B 88 18 00 20 (CRC=D4) #data=2
(00:02:02.379) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4) #data=216
(00:02:02.396) Corrupt telegram: 00 0B 88 20 D4
lo(00:02:09.985) Boiler -> all, type 0x34 telegram: 08 00 34 00 3E 01 14 80 00 21 00 00 01 00 01 01 68 00 86 BA 00 80 00 (CRC=17) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:02:10.258) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 65 00 00 80 00 40 40 01 14 80 00 01 2F 00 00 0F 30 48 00 CB 00 00 00 (CRC=47) #data=25
<--- UBAMonitorFast(0x18)
(00:02:10.480) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B0 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=BA) #data=21
n(00:02:12.759) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0) #data=216
(00:02:12.776) Corrupt telegram: 0B 88 19 20 D0
(00:02:13.124) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=13
(00:02:13.307) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78) #data=216
(00:02:13.417) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 3C FB 00 28 00 00 46 EC FF FF 00 (CRC=7B) #data=13
<--- UBAParameterWW(0x33)
(00:02:13.854) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC) #data=216

Unknown command. Use ? for help.
(00:02:14.560) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(00:02:16.395) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4) #data=216
(00:02:16.411) Corrupt telegram: 0B 88 14 20 E4
(00:02:16.438) Boiler -> me, type 0x14 telegram: 08 0B 14 00 2A 57 7A (CRC=6E) #data=3
<--- UBATotalUptimeMessage(0x14)
l(00:02:16.718) 0x00 -> Boiler, type 0x00 telegram: 00 08 00 07 00 0B 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=F6) #data=14
og (00:02:17.539) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13

proddy, 2019-07-19:
ok. back to the drawing board then.

Note in the latest b20 release you can now do `thermostat read 1a5` or any other type value to query it

moustic999, 2019-07-19:
I just compiled and uploaded latest DEV and : 

Error! Unable to read the EMS bus.

so not working for me

kroon040, 2019-07-19:
Is the serial off?

[75 more replies]

closed 2019-08-31 by kroon040