emsesp/EMS-ESP #146: SM100 - no MQTT messages send (sm_data) - 1.8.1b9

S-Przybylski, 2019-07-04:
Dear @proddy 
dear @Vuego123 
i have a problem with the mqtt messages of the SM100 of this version:
I updated from 1.8.1b2 to 1.8.1b9: The SM100 module doesn't send any message to the MQTT broker (test one day!)! You also doesn't see any MQTT message update indications in the log. 

I expect SM100 mqtt messages if a value changes like temperature of the collector or bottom temp... I also expect that the hourly updates of the solar eanrnings are send out (like older versions have send out!).

Please check the log below - there is no MQTT update for SM100 visible! At least when the x028E update comes, the new energy values needs to be updated (and yes i expect messages for every change of any sensor of the SM100 too) 
(17:53:39.936) SM -> all, type 0x028E telegram: 30 00 FF 00 02 8E 00 00 41 82 00 00 28 36 00 00 82 21 (CRC=6D) #data=12
<--- SM100Energy(0x28E)

Could it be that the routine of the SM100 is a different one from yours @Vuego123 for your configuration? Because i saw other mqtt values in your screentshot than mine in my last issue?

Attachment 
````
Requesting scheduled EMS device data
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
Requesting type (0x262) from dest 0x30
(17:53:02.367) Sending read of type 0x06 to 0x10: telegram: 0B 90 06 00 20 (CRC=6C)
(17:53:02.458) Thermostat -> me, type 0x06 telegram: 10 0B 06 00 13 07 11 04 3B 04 03 01 10 FF 00 (CRC=E6) #data=11
<--- RCTime(0x06)
(17:53:02.771) Sending read of type 0x18 to 0x08: telegram: 0B 88 18 00 20 (CRC=D4)
(17:53:02.822) Boiler -> me, type 0x18 telegram: 08 0B 18 00 05 01 74 00 00 00 00 80 40 80 00 02 52 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=27) #data=25
<--- UBAMonitorFast(0x18)
(17:53:02.828) Sending read of type 0x19 to 0x08: telegram: 0B 88 19 00 20 (CRC=D0)
(17:53:04.086) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A) #data=13
(17:53:04.370) SM -> all, type 0x0264 telegram: 30 00 FF 09 02 64 52 (CRC=01) #data=1
<--- SM100Status(0x264)
(17:53:04.957) Thermostat -> Boiler, type 0x35 telegram: 10 08 35 00 11 11 (CRC=30) #data=2
(17:53:05.809) Sending read of type 0x33 to 0x08: telegram: 0B 88 33 00 20 (CRC=78)
(17:53:05.846) Boiler -> me, type 0x33 telegram: 08 0B 33 00 08 FF 33 FB 00 1E FF 06 46 00 FF FF 00 (CRC=15) #data=13
<--- UBAParameterWW(0x33)
(17:53:05.962) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76) #data=13
(17:53:06.102) Sending read of type 0x16 to 0x08: telegram: 0B 88 16 00 20 (CRC=EC)
(17:53:07.323) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A) #data=13
(17:53:07.669) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 74 00 00 00 00 80 40 80 00 02 52 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=94) #data=25
<--- UBAMonitorFast(0x18)
(17:53:07.872) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(17:53:08.174) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 02 53 02 53 21 00 05 03 00 05 BB 5E 00 F4 F1 00 80 00 (CRC=BB) #data=19
<--- UBAMonitorWWMessage(0x34)
(17:53:09.463) Sending read of type 0x14 to 0x08: telegram: 0B 88 14 00 20 (CRC=E4)
(17:53:09.501) Corrupt telegram: 14 00 20 E4 08 0B 14 00 25 B3 5F A6
(17:53:09.812) Boiler -> all, type 0x07 telegram: 08 00 07 00 0B 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=76) #data=13
(17:53:10.236) Sending read of type 0x262 to 0x30: telegram: 0B B0 62 00 20 (CRC=FC)
Read failed. Retrying attempt 1/2...
(17:53:10.296) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
(17:53:11.680) Sending read of type 0x262 to 0x30: telegram: 0B B0 62 00 20 (CRC=FC)
(17:53:11.712) Corrupt telegram: B0 62 00 20 FC
Read failed. Giving up, removing from queue
(17:53:11.748) SM -> me, type 0x62 telegram: 30 0B 62 00 (CRC=71)
(17:53:12.295) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=12) #data=13
(17:53:13.612) Thermostat -> all, type 0x01AF telegram: 10 00 F7 00 FF 01 AF E5 (CRC=2E)
(17:53:13.830) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=DA) #data=13
(17:53:17.627) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 74 00 00 00 00 80 40 80 00 02 52 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=94) #data=25
<--- UBAMonitorFast(0x18)
(17:53:17.911) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 00 01 A5 80 00 10 30 00 00 30 28 01 2D 03 03 01 01 2D 02 CF 00 00 11 01 01 FF FF 00 (CRC=E2) #data=25
<--- RCPLUSStatusMessage(0x1A5)
(17:53:18.130) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(17:53:18.352) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 02 53 02 52 21 00 05 03 00 05 BB 5E 00 F4 F1 00 80 00 (CRC=AD) #data=19
<--- UBAMonitorWWMessage(0x34)
(17:53:18.620) Thermostat -> all, type 0x01A5 telegram: 10 00 FF 19 01 A5 06 07 00 00 00 00 FF 00 37 00 3C 01 FF 01 (CRC=19) #data=14
<--- RCPLUSStatusMessage(0x1A5)
(17:53:18.830) Thermostat -> all, type 0x021D telegram: 10 00 FF 00 02 1D 00 00 0A 04 (CRC=01) #data=4
(17:53:19.260) Thermostat -> Boiler, type 0x1A telegram: 10 08 1A 00 00 00 00 (CRC=C4) #data=3
(17:53:19.306) Thermostat -> all, type 0x0167 telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
[... 26 lines cut ...]
<--- UBAMonitorWWMessage(0x34)
(17:53:28.947) Boiler -> all, type 0x07 telegram: 08 00 07 00 03 01 00 00 00 01 00 00 00 00 00 00 00 (CRC=5A) #data=13
(17:53:36.391) SM -> all, type 0x0264 telegram: 30 00 FF 09 02 64 50 38 2C (CRC=50) #data=3
<--- SM100Status(0x264)
Publishing boiler data via MQTT
(17:53:37.498) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 74 00 00 00 00 80 40 80 00 02 52 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=94) #data=25
<--- UBAMonitorFast(0x18)
(17:53:37.700) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(17:53:38.034) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 02 52 02 52 21 00 05 03 00 05 BB 5E 00 F4 F1 00 80 00 (CRC=F5) #data=19
<--- UBAMonitorWWMessage(0x34)
(17:53:39.936) SM -> all, type 0x028E telegram: 30 00 FF 00 02 8E 00 00 41 82 00 00 28 36 00 00 82 21 (CRC=6D) #data=12
<--- SM100Energy(0x28E)
(17:53:47.442) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 74 00 00 00 00 80 40 80 00 02 52 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=94) #data=25
<--- UBAMonitorFast(0x18)
(17:53:47.699) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(17:53:48.019) Boiler -> all, type 0x34 telegram: 08 00 34 00 33 02 52 02 52 21 00 05 03 00 05 BB 5E 00 F4 F1 00 80 00 (CRC=F5) #data=19
<--- UBAMonitorWWMessage(0x34)
(17:53:52.490) Boiler -> all, type 0x18 telegram: 08 00 18 00 05 01 74 00 00 00 00 00 40 80 00 02 53 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=4A) #data=25
<--- UBAMonitorFast(0x18)
(17:53:52.816) SM -> all, type 0x0266 telegram: 30 00 FF 00 02 66 01 62 00 12 (CRC=73) #data=4
(17:53:57.409) Boiler -> all, type 0x2A telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 DC 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=49) #data=21
(17:53:57.704) Thermostat -> all, type 0x06 telegram: 10 00 06 00 13 07 11 04 3B 3B 03 01 10 FF 00 (CRC=0C) #data=11
<--- RCTime(0x06)

````

proddy, 2019-07-04:
definitely a bug somewhere. When you type 'info' do you see 'Solar Module enabled' at the top ?

proddy, 2019-07-04:
also do you see the message "Solar Module support enabled." when starting up or after a `autodetect` ?

S-Przybylski, 2019-07-04:
Short: yes it appeared

Longer:
After Reboot  - ESP start messages (energy message aren't send out by hardware after the reboot - less than 1 h):

````
* EMS-ESP version 1.8.1b9
Boiler found: Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
* Setting Boiler to model Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 
Version:04.09)
Solar Module found: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.04)
Solar Module support enabled.
* Setting Thermostat to RC300/RC310/Nefit Moduline 1010/3000 (DeviceID:0x10 ProductID:158 Version:11.07)

info
EMS-ESP system stats:
  System logging set to None
  LED is off, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=47, # CRC errors=7
  Tx: Last poll=2.390 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB172/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:04.09)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 51 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 56.4 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 62705 times
  Warm Water active time: 260 days 20 hours 46 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 32.6 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Outside temperature: 23.1 C
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 219449 times
  Total burner operating time: 568 days 22 hours 21 minutes
  Total heat operating time: 308 days 1 hours 35 minutes
  Total UBA working time: 1715 days 21 hours 37 minutes

Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 40.2 C
  Bottom temperature: 52.3 C
  Pump modulation: 0 %
  Pump active: off
  Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.07)
  Setpoint room temperature: 24.0 C
  Current room temperature: ? C
  Thermostat time is 20:24:56 4/7/2019
  Mode is set to ?

autodetect
Started scan on EMS bus for known devices
Boiler found: Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
Device found: BC25 Base Controller (DeviceID:0x09 ProductID:125 Version:01.06)
Solar Module found: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.04)
Solar Module support enabled.

````

proddy, 2019-07-04:
the MQTT is only updated when the values change, so with the SM module if they are constant nothing will be sent. I did this to save on internet traffic to the MQTT broker. If you think its not needed and we should always publish the values I can remove that check.

S-Przybylski, 2019-07-04:
Dear @proddy 
this i a good behaviour i think to throttle the solar module messages to the minimum possible. In this is not enabled the number of messages are at least 2-4 per minute...
In the past (version 1.7.0) this check worked well. 

To test if the routine is the reason why nothing appears in mqtt - i can test it with your help - please provide me the necessary steps to do...
I can imagine that the problem is located in another area (believe). I wonder why the solar module of Vuego123 shows mqtt messages and mine none. When you have a closer look into the screenshot of his mqtt browser from yesterday, you also see a different set of values. For me it seems that the program has different routines for these modules. Am i correct? Could it be that the error is located here? I also noticed that a new value 'total run time of the solar pump' is introduced in that new version. I'am not sure how this was evaluated, but the value seems to be much to high (i do not know to get the value out of the display). Could this lead to problems sending out the values?

proddy, 2019-07-04:
I refactored the code a little to separate devices like solar modules from heatpumps. I don't expect it to fix your problem but it'll be easier to trace. The messages come in on the same type so the logic hasn't changed there. Can you test anyway and let me know if you see the SM values in `info` and whether they are sent via MQTT?

S-Przybylski, 2019-07-04:
first impression shows the first 4 values are updated accordingly, the energy values are unknown (once per hour), but i do not get any sm_data topic message via mqtt! All others are send out...

Does perhaps the new (for me unknown value) 'Pump working time: 11650 days 20 hours 15 minutes' - the value cannot be true, because the heating was installed less then 5 years ago and it never changes - responsible for the issue?

info
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 16.1 C
  Bottom temperature: 46.0 C
  Pump modulation: 0 %
  Pump active: off
  Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH

S-Przybylski, 2019-07-04:
update: energy values are updated and again - wrong total energy and no sm_data at all

Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 15.1 C
  Bottom temperature: 45.1 C
  Pump modulation: 0 %
  Pump active: off
  Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: 0.0 Wh
  Energy Today: 10820 Wh
  Energy Total: 3221.7 kWH

proddy, 2019-07-05:
energy total is taken from bytes 11+12 of type 0x28E and divided by 10, so in your example
`(17:53:39.936) SM -> all, type 0x028E telegram: 30 00 FF 00 02 8E 00 00 41 82 00 00 28 36 00 00 82 21 (CRC=6D) #data=12`

0x8221 = 33313 = 3331.3 kWH. I think that looks correct? If so it's a rendering problem

proddy, 2019-07-05:
as for the pump working time, you should not be even seeing this as its a feature from a Junkers ISM1 solar module. I think the problem is the introduction of the new code for Junkers is confusing the old SM100 implementation that worked in 1.7. Can you leave it running for a while and see if you get any telegrams from type 0x0003 ?

S-Przybylski, 2019-07-05:
Regarding https://github.com/proddy/EMS-ESP/issues/146#issuecomment-508640604 --> issue https://github.com/proddy/EMS-ESP/issues/147
yes, the display and the telegram data and the formula is correct; it seems to be a rendering problem

S-Przybylski, 2019-07-05:
i scanned the last 5 minutes - nothing with 0x0003
i continue to scan the log

S-Przybylski, 2019-07-05:
After more than 1 hour - no telegram with 0x0003
in the past i also never saw that

proddy, 2019-07-05:
ok. I have no idea why you're getting a value for pump working time as your SM100 doesnt support that yet. I'll dig into the code again and run some tests

proddy, 2019-07-05:
found the issue and fixed it. can you check again?

S-Przybylski, 2019-07-05:
Installed: 
The pump working time is not present any longer. 
So i have to wait for the energy update within the next hour...
Still no sm_data mqtt message at all

Output after some minutes:
````
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 53.1 C
  Bottom temperature: 42.9 C
  Pump modulation: 30 %
  Pump active: on
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH
````

proddy, 2019-07-05:
like I said there is no pump working time for SM100. That was left over code for the Junkers that Vuego added. Still no sm_data? ok, let me look into that next

S-Przybylski, 2019-07-05:
Total energy seems to be correct by now - cool:
````
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 54.4 C
  Bottom temperature: 44.0 C
  Pump modulation: 30 %
  Pump active: on
  Energy Last Hour: 487.5 Wh
  Energy Today: 935 Wh
  Energy Total: 3332.8 kWH
````

proddy, 2019-07-05:
mqtt is working for me. Are you sure it says "enabled" in `system`. Is anything received when you do a `publish` ?

S-Przybylski, 2019-07-05:
i entered 'system' then the esp crashed....
here is the result:
````
system
ESP8266 System stats:

 [APP] EMS-ESP version: 1.8.1b11
 [APP] MyESP version: 1.1.21
 [APP] Build timestamp: 2019-07-05 12:32:54
 [APP] Boot time: 13:02:29 5/Jul/2019
 [APP] Uptime: 0 days 0 hours 1 minutes 2 seconds
 [APP] System Load: 0%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 10.10.0.235
 [WIFI] WiFi signal strength: 58%
 [WIFI] WiFi MAC: CC:50:E3:45:51:CE
 [MQTT] connected (heartbeat disabled)
 [EEPROM] EEPROM size: 4096
 [EEPROM] EEPROM Sector pool size is 4, and in use are: 1019 1018 1017 1016
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x4551CE
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Software/System restart
 [SYSTEM] Last reset info: Fatal exception:0 flag:4 (SOFT_RESTART) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
 [SYSTEM] Restart count: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x164068
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DOUT
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 463984
 [MEM] Max OTA size: 2674688
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24272 bytes initially | 11688 bytes used (48%) | 12584 bytes free (51%)
````
Still get mqtt messages from other areas like boiler:
topic: buderus/ems-esp/boiler_data
{"wWComfort":"Hot","wWSelTemp":51,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"outdoorTemp":25.6,"wWCurTmp":46.7,"wWCurFlow":0,"curFlowTemp":58.1,"retTemp":3276.8,"boilTemp":3276.8,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"off","ServiceCode":"0H","ServiceCodeNumber":203}

The log show mqtt updates send out for:
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT

autodetect
Started scan on EMS bus for known devices
Device found: BC25 Base Controller (DeviceID:0x09 ProductID:125 Version:01.06)
Solar Module found: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.04)
Solar Module support enabled.

S-Przybylski, 2019-07-05:
entered publish -> seen mqtt topics:
buderus/ems-esp/boiler_data
buderus/ems-esp/tapwater_active
buderus/ems-esp/heating_active

but still no sm_data

proddy, 2019-07-05:
need to find if the crc checksum is failing. Can you add an extra debug line, in ems-esp.cpp after line #876 so
```
// send values via MQTT
myESP.mqttPublish(TOPIC_SM_DATA, data);
}

becomes

// send values via MQTT
myESP.mqttPublish(TOPIC_SM_DATA, data);
} else { myDebug("** SM data the same! %d %d %d", previousSMPublishCRC, fchecksum, sizeof(data)); }

```

S-Przybylski, 2019-07-05:
i doesn't see any debug message up to now; have i to enable something?

S-Przybylski, 2019-07-05:
I put an additional debug line into the file (line 830) and also nothing - and yes i saved the file and recompiled it!
````
    // For SM10 and SM100 Solar Modules
    if (ems_getSolarModuleEnabled()) {
        myDebugLog("Test if ems_getSolaraModuleEnabled routine executes");
````

proddy, 2019-07-05:
myDebugLog needs verbose mode. best to use myDebug() instead. but you should see something!

S-Przybylski, 2019-07-05:
changed and nothing! info shows that the solar modul is enabled but it seems that the routine doesn't run...
do you have an idea?

S-Przybylski, 2019-07-05:
i put some debug statements into the code
before the 'thermostat if case' i see that this routine is running, but after that no event appears ...

proddy, 2019-07-05:
the publish should run every 2 mins (do a `set` and look for publish_time which is default 120 seconds). I'm getting regular updates on mine for thermostat and boiler.

When you do a `publish` it should also force the MQTT publishes regardless of whether the values changed. Do you see anything after you issue that command?

S-Przybylski, 2019-07-05:
Could it be that a "return" in the thermostat routine is responsible that i do not get any update of solar?
To understand: My RC300 runs without a room temperature. It only uses the outdoor temp. to steer the heating!
line 756
````
if ((EMS_Thermostat.curr_roomTemp == EMS_VALUE_SHORT_NOTSET) || (EMS_Thermostat.setpoint_roomTemp == EMS_VALUE_SHORT_NOTSET))
            return;
````

S-Przybylski, 2019-07-05:
Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 1010/3000 (ProductID:158 Version:11.07)
  Setpoint room temperature: 24.0 C
  Current room temperature: ? C
  Thermostat time is 16:58:25 5/7/2019
  Mode is set to ?

S-Przybylski, 2019-07-05:
in 1.7.0 that part was solved differently:
        if ((EMS_Thermostat.curr_roomTemp <= 0) && (EMS_Thermostat.setpoint_roomTemp <= 0))
            return;
I think that this if case needs to be adapted without using a return.

S-Przybylski, 2019-07-05:
Dear @proddy 
i have deleted the return and moved the content of the thermostat code into an else path after the if cause as a test. Now it works!

Please adpat the if clause in the code. 
Does it make sense that the RC300 in my case needs to walk thru the if thermostat clauses too? The the logic needs to be adapted...

proddy, 2019-07-05:
OMG! you found the culprit. how silly. I'll fix it right away

S-Przybylski, 2019-07-06:
Dear @proddy 
a more or less cosmetic topic:
i just take a look to my HA dashboard and see that two values of the heating, which are not provided by my heating and therefore are not really interessing for me, show high values instead of '?'.
Perhaps this happend with your change of the rendering...

 Return temperature: 3276.8 C
 Boiler temperature: 3276.8 C

Firmware 181b11 (yesterday not with your last commit)
Later on i install your last dev version

S-Przybylski, 2019-07-06:
Dear @proddy 
by the way i always get warnings when i compile your current version 1.8.1bx:

````
Compiling .pio\build\debug\lib300\OneWire_ID1\OneWire.cpp.o
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp: In member function 'uint8_t OneWire::reset()':
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp:167:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp: In member function 'void OneWire::write_bit(uint8_t)':
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp:201:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
                        ^
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp: In member function 'uint8_t OneWire::read_bit()':
C:\users\xxx\.platformio\lib\OneWire_ID1\OneWire.cpp:229:24: warning: unused variable 'reg' [-Wunused-variable]
  volatile IO_REG_TYPE *reg IO_REG_BASE_ATTR = baseReg;
````

S-Przybylski, 2019-07-06:
Last but not least: Often i have to retry up to four times the upload of a image to the esp via ota. The connection can't be established. I notice that the esp reboot at least one time and also crash dumps...
I don't now why, but i have several crashs even when i enter commands like system (not every time but often)...

S-Przybylski, 2019-07-06:
@proddy 
Version 1.8.1b11 - sm_data is available again !

proddy, 2019-07-06:
1. the sm_data: glad it works, sorry for the stupid mistake
2. the OTA - yes I've noticed this too. I'll look into it again and find what is causing it not initialize
3. the crashes - I've seen this too after commands like 'system'. Haven't been able to find the root cause yet. It's annoying but not top of my list
4. your compile errors are strange. I'll update my install and see if I can reproduce.
5. the strange numbers in boiler and return temp is probably related to a recent change I made. I don't see it here. Does this happen in the telnet window or in MQTT?

S-Przybylski, 2019-07-06:
to 5.: in both MQTT and telnet

proddy, 2019-07-06:
think #5 is resolved now.

S-Przybylski, 2019-07-06:
Dear @proddy 
info shows '?'; thats quite good:
  Boiler temperature: ? C
  Return temperature: ? C

But the mqtt message seems to have wrong values:
  "retTemp":3276.8,"boilTemp":3276.8

''''
{"wWComfort":"Hot","wWSelTemp":51,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"outdoorTemp":31.3,"wWCurTmp":56.1,"wWCurFlow":0,"curFlowTemp":38.1,"retTemp":3276.8,"boilTemp":3276.8,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"off","ServiceCode":"0H","ServiceCodeNumber":203}
''''

S-Przybylski, 2019-07-06:
Perhaps you should disable them, when no value is present, like you have done for solrr energy...

proddy, 2019-07-06:
ok, done in latest build.

S-Przybylski, 2019-07-06:
Dear @proddy 
I think you have done a good job!

{"wWComfort":"Hot","wWSelTemp":51,"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"outdoorTemp":17.6,"wWCurTmp":53.4,"wWCurFlow":0,"curFlowTemp":70.5,"wWActivated":"on","burnGas":"off","heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"off","ServiceCode":"0H","ServiceCodeNumber":203}

S-Przybylski, 2019-07-06:
solved with 1.8.1b14

closed 2019-07-06 by S-Przybylski