emsesp/EMS-ESP #147: SM100 - total energy corrupted - 1.8.1b9

S-Przybylski, 2019-07-04:
Since several days i have a corruption of the total energy value!
The info shows wrong values:
Info  Energy Total: 3222.3 kWH <-> Boiler screen shows 3321,3 kWH
In addition to that the values decreases from hour to hour!

First time this happend when i reinstalled the running 1.8.1b2 some days before after testing newer dev versions. In the b2 version the mqtt message was also negative while the info value was positve (but wrong)...
Could it be, that newer libraries in vscode/platform.io are one of the reasons?

Info details:
Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.04)
  Collector temperature: 63.0 C
  Bottom temperature: 56.6 C
  Pump modulation: 30 %
  Pump active: on
  Pump working time: 11650 days 20 hours 15 minutes
  Energy Last Hour: 1677.0 Wh
  Energy Today: 10294 Wh
  Energy Total: 3222.3 kWH

S-Przybylski, 2019-07-04:
update:
(17:53:39.936) SM -> all, type 0x028E telegram: 30 00 FF 00 02 8E 00 00 41 82 00 00 28 36 00 00 82 21 (CRC=6D) #data=12
<--- SM100Energy(0x28E)

Calculation byte 10 and 11: 82 21 --> 33313 / 10 = 3331,3 - is the same i saw on the display!

S-Przybylski, 2019-07-04:
update: 
I reinstalled version 1.7.0 with the current platform.io 4 / vscode version and current libraries:
The mqtt message - total energy is also corrupted (wrong value and negative) - first time that the value is calculated:
{"collectortemp":56.9,"bottomtemp":53.4,"pumpmodulation":0,"pump":"off","energylasthour":97.5,"energytoday":10820,"energytotal":-3221.7}
It should be 3331,9 (display) instead. 
I think its related to library updates, because i am quite sure that this version function well when it was released ...

proddy, 2019-07-04:
ok, are you saying the data from the telegrams is correct but the way its parsed into a readable number is incorrect, since version 1.7?

S-Przybylski, 2019-07-04:
yes and no:
I  remember two months ago: This value was evaluated correctly incl. sending the correct value via mqtt ~ version 1.7.0 i think.

Now with the version 1.8.1b9 the value in the info section is wrong. Today i have recompiled and installed the version 1.7.0: now this value is also corrupted (negative mqtt and also wrong value).
In addition to that i saw in 1.7.0 (and also in 1.8.1b2; b9 not tested so far) that the value decreases over time when the telegram was send (once per hour). Strange
I could imagine, that a library update is responsible for that, or that my value rises over a point where the program gets into trouble with the figure ... (only ideas why this happens)...

S-Przybylski, 2019-07-04:
and yes the telegram represents the correct values as expected

S-Przybylski, 2019-07-05:
update: https://github.com/proddy/EMS-ESP/issues/146#issuecomment-508721458
seems to be solved with 1.8.1b11 - testing

closed 2019-07-05 by proddy