emsesp/EMS-ESP #149: Home Assistant integration

Sonusss, 2019-07-13:
When trying to integrate into HA I got lot of error message when checking config like:
Invalid config for [climate.mqtt]: [state_topic] is an invalid option for [climate.mqtt]. Check: climate.mqtt->state_topic. (See ?, line ?). Please check the docs at https://home-assistant.io/components/climate.mqtt/
Invalid config for [climate.mqtt]: [state_topic] is an invalid option for [climate.mqtt]. Check: climate.mqtt->state_topic. (See ?, line ?). Please check the docs at https://home-assistant.io/components/climate.mqtt/
Invalid config for [climate.mqtt]: [state_topic] is an invalid option for [climate.mqtt]. Check: climate.mqtt->state_topic. (See ?, line ?). Please check the docs at https://home-assistant.io/components/climate.mqtt/
Invalid config for [climate.mqtt]: [state_topic] is an invalid option for [climate.mqtt]. Check: climate.mqtt->state_topic. (See ?, line ?). Please check the docs at https://home-assistant.io/components/climate.mqtt/
Platform not found: climate.template

proddy, 2019-07-15:
couple of things to try
- check what the MQTT messages are coming into the broker
- check what HA is doing with the MQTT messages (by adding HA logging for MQTT debug)
- what does your climate.yaml look like?
- which model thermostat are you using? Does it have auto, manual and off modes?

Sonusss, 2019-07-21:
I finally figured out how to do it

closed 2019-07-21 by Sonusss