emsesp/EMS-ESP #150: SM50 not recognized in  1.8.1b15

DarthMob, 2019-07-14:
Hi guys,

i have the SM50 which was recognized as Solar Module in 1.7.0.
(except the negative value for total energy [https://github.com/proddy/EMS-ESP/issues/96#issuecomment-500239534](url) everything was fine).

Now with 1.8.1b15 the Solar Module is not showing up any more.
system is telling me:

> EMS-ESP system stats:
> Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled
> 

Any change the SM50 can be integrated again?
[https://www.buderus.de/de/produkte/catalogue/alle-produkte/7606_solarmodul-logamatic-sm50](url)

Thank you!

proddy, 2019-07-14:
does it show after an autodetect ?

DarthMob, 2019-07-16:
Sadly not (also not with „deep“ flag)

proddy, 2019-07-17:
oh, looking at the code during the start-up process it sends a version request to the SM (line 1808 in ems.cpp) and that hasn't changed so it could be that Tx is not working and no requests are being sent out? Make sure tx_mode is 0 and try doing a send command (`thermostat read` or `boiler read`). Or `send 0b 70 02 20`

DarthMob, 2019-07-20:
Hi Proddy,

i have deactivated Tx (`set listen_mode on`) by default (also on 1.7 and there it was working).

however if i `set listen_mode off` and execute `thermostat read` or `boiler read` or `send 0b 70 02 20` it changes nothing :(

proddy, 2019-07-20:
Tx is needed for ems-esp to detect the devices as it sends out version requests. Weird that it worked previously with listen_mode on! If youre sure Tx worked in 1.8.0 then I just have fudged something in 1.8.1. Oops

proddy, 2019-07-23:
set tx_mode to, serial off, listen_mode off
then reboot
and send me what  `info` says. It should detect your devices. If it doesn't we'll start there and figure out what is not working.

proddy, 2019-08-10:
did you manage to reproduce this?

DarthMob, 2019-08-11:
Hey, i tried now with 1.8.2 but when i "set listen_mode off" i get frequent reboots...
so ems-esp does not work for me with TX on.
I'm on nodemcu2

`ESP8266 System stats:

 [APP] EMS-ESP version: 1.8.2
 [APP] MyESP version: 1.1.24
 [APP] Build timestamp: 2019-08-11 19:10:25
 [APP] Uptime: 0 days 0 hours 36 minutes 39 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.2.110
 [WIFI] WiFi signal strength: 54%
 [WIFI] WiFi MAC: 84:F3:EB:5A:XX:XX
 [MQTT] is disconnected
 [SYSTEM] Board: PLATFORMIO_NODEMCUV2
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x5A7509
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: External System
 [SYSTEM] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
 [SYSTEM] Restart count: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 0
 [FLASH] Flash chip ID: 0x164068
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 461520
 [MEM] Max OTA size: 2678784
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 28984 bytes initially | 7760 bytes used (26%) | 21224 bytes free (73%)
`

proddy, 2019-08-11:
that's strange. What does 'info' and 'set' say?

DarthMob, 2019-08-12:
i know..

info:
```

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is on
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower T                                                                                                                                                              imer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=8
  Tx: Last poll=1.386 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 41.2 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 8151 times
  Warm Water active time: 61 days 13 hours 9 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 28.4 C
  Return temperature: 27.9 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.1 uA
  System pressure: 1.6 bar
  System service code: 0H (0)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 16.0 C
  Boiler temperature: 28.4 C
  Pump modulation: 0 %
  Burner # starts: 151233 times
  Total burner operating time: 586 days 4 hours 17 minutes
  Total heat operating time: 579 days 12 hours 55 minutes
  Total UBA working time: 2119 days 9 hours 30 minutes
```

set:

```
The following set commands are available:

  set erase
  set <wifi_ssid | wifi_password> [value]
  set <mqtt_host | mqtt_username | mqtt_password> [value]
  set serial <on | off>
  set led <on | off>                  toggle status LED on/off
  set led_gpio <gpio>                 set the LED pin. Default is the onboard LED 2. For external D1 use 5
  set dallas_gpio <gpio>              set the external Dallas temperature sensors pin. Default is 14 for D5
  set dallas_parasite <on | off>      set to on if powering Dallas sesnsors via parasite power
  set thermostat_type <device ID>     set the thermostat type ID (e.g. 10 for 0x10)
  set boiler_type <device ID>         set the boiler type ID (e.g. 8 for 0x08)
  set listen_mode <on | off>          when set to on all automatic Tx are disabled
  set shower_timer <on | off>         send MQTT notification on all shower durations
  set shower_alert <on | off>         stop hot water to send 3 cold burst warnings after max shower time is exceeded
  set publish_time <seconds>          set frequency for publishing data to MQTT (0=off)
  set heating_circuit <1 | 2>         set the main thermostat HC to work with (if using multiple heating circuits)
  set tx_mode <n>                     changes Tx logic. 0=ems 1.0, 1=ems+, 2=generic (experimental!), 3=HT3

Stored settings:

  wifi_ssid=Piratensender
  wifi_password=************
  mqtt_host=192.168.2.50
  mqtt_username=*****
  mqtt_password=*****
  serial=off
  heartbeat=off
  led=on
  led_gpio=16
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=<not set>
  heating_circuit=1
  boiler_type=<not set>
  listen_mode=on
  shower_timer=off
  shower_alert=off
  publish_time=120
  tx_mode=0
```

proddy, 2019-08-13:
you need listen_mode to off to be able to detect the devices. If the SM doesn't show I suggest wait for the dev 1.9 which I'll push out soon and try again. Because I'm not sure why its stopped working!

DarthMob, 2019-08-14:
sure think @proddy  i will try 1.9 then.

btw in serial monitoring mode the reboots (<3 sec) occur with the following error:

```
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

* EMS-ESP version 1.8.3
[FS] Settings loaded from SPIFFS
[WEB] Web server started
Note: Serial output will now be disabled. Please use Telnet.
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀
```

proddy, 2019-08-14:
1.9 on dev is the same as 1.8.3 master so don't bother with testing that.
I'm not sure what is causing the crashes. Perhaps compile with -DCRASH and
see if you can capture the stack?

proddy, 2019-09-21:
should work in 1.9.1. please check and re-open if it still is an issue

DarthMob, 2019-10-03:
Hi @proddy ,
installed version 1.9.1b10  on D1 now.
Sadly SM50 not recognized

[image]

proddy, 2019-10-03:
as mentioned in gitter, get onto telnet and check a few things like tx_mode and then do an `autodetect quick` with logging on

proddy, 2019-10-03:
actually I couldn't find anything wrong in the code, then I looked at the devices and none of yours are in the database
so device IDs 0x08, 0x09 and 0x30
can you tell me what they are? Perhaps look at 1.7.x and see what they say
the product IDs (169,170 etc) should be in the ems_devices.h file

proddy, 2019-10-05:
this should be working again now, so closing. Please re-open if its not in 1.9.1.

closed 2019-10-05 by proddy