emsesp/EMS-ESP #153: Home Assistant - breaking changes to climate integration

ReneHonig, 2019-07-18:
@proddy :

Home Assistant v 0.96 has changed the behaviour of the climate integration:
https://developers.home-assistant.io/blog/2019/07/03/climate-cleanup.html

This breaks the HA integration for the warm water control.
I have made some quick fixes and have extracted the relevant pieces of code (quicker than a pull request as my setup is somewhat different).

The approach I have taken is to add 'auto' as a synonym for 'on', as not to break existing configurations. Not sure if this is the best approach, but it should work.

HA file:
```

climate nefit_boiler:
  platform: mqtt
  name: boiler
  modes:
   - "auto"
   - "off"
  min_temp: 40
  max_temp: 60
  temp_step: 1
  current_temperature_topic: "home/ems-esp/boiler_data"
  temperature_state_topic: "home/ems-esp/boiler_data"
  temperature_command_topic: "home/ems-esp/boiler_cmd_wwtemp"
  current_temperature_template: "{{ value_json.wWCurTmp }}"
  temperature_state_template: "{{ value_json.wWSelTemp }}"
  mode_state_template: "{% if value_json.wWActivated == 'off' %} off {% else %} auto {% endif %}"
  mode_state_topic: "home/ems-esp/boiler_data"
  mode_command_topic: "home/ems-esp/wwactivated"

```
ems-esp.cpp:

        // wwActivated
        if (strcmp(topic, TOPIC_BOILER_WWACTIVATED) == 0) {
            if (message[0] == '1' || strcmp(message, "on") == 0) {
                ems_setWarmWaterActivated(true);
            } else if (message[0] == '1' || strcmp(message, "auto") == 0) {
                ems_setWarmWaterActivated(true);
            } else if (message[0] == '0' || strcmp(message, "off") == 0) {
                ems_setWarmWaterActivated(false);                
            }
        }

proddy, 2019-07-18:
thanks, I noticed HA 0.96 had some refactors to the climate component. I'll test on my system and see if we can make it work for all thermostats

proddy, 2019-07-18:
added to dev branch

closed 2019-07-18 by proddy

proddy, 2019-08-17:
@ReneHonig  any idea how to go about fixing https://github.com/proddy/EMS-ESP/issues/172 ?