emsesp/EMS-ESP #156: published temperatures factor 10 higher

balk77, 2019-07-21:
**Bug description**
Published values over MQTT for _thermostat_seltemp_ and _thermostat_currtemp_ are a factor 10 too high. I recently updated from 1.6 to 1.8 and I noticed very high room temperatures being published. Temperature shown under _info_ in the telnet environment is correct. I am using a "TC100/Nefit Easy". See screenshot.

**Steps to reproduce**
every time; went back to 1.6 and now reported temperatures are correct.

[image]

**Device information**  
Boiler type: Buderus GB172/Nefit Trendline (ProductID:123 Version:06.07)
Thermostat type: TC100/Nefit Easy (ProductID:202 Version:02.19)

proddy, 2019-07-22:
Thanks for reporting. This is in version 1.8.0 I believe correct? Can you send me some example telegrams of type 0x0A (EasyStatusMessage) so I can add to the tests. It's hard to keep track of the nuances between devices when I don't actually own any of them ;)

balk77, 2019-07-22:
yep, this is with 1.8.0
Is this sufficient?
```
(00:00:18.836) Sending read of type 0x0A to 0x18: telegram: 0B 98 0A 00 20 (CRC=1C)
(00:00:18.890) Corrupt telegram: 98 0A 00 20 1C
(00:00:19.013) Thermostat -> me, type 0x0A telegram: 18 0B 0A 00 00 00 00 00 00 00 00 00 09 8D 05 DC 00 00 00 00 13 1A 00 00 00 00 00 00 00 00 (CRC=2B) #data=26
<--- EasyStatusMessage(0x0A)
Requesting type EasyStatusMessage(0x0A) from dest 0x18
(00:01:05.470) Sending read of type 0x0A to 0x18: telegram: 0B 98 0A 00 20 (CRC=1C)
(00:01:05.592) Thermostat -> me, type 0x0A telegram: 18 0B 0A 00 00 00 00 00 00 00 00 00 09 8D 05 DC 00 00 00 00 13 1A 00 00 00 00 00 00 00 00 (CRC=2B) #data=26
<--- EasyStatusMessage(0x0A)
Requesting type EasyStatusMessage(0x0A) from dest 0x18
(00:02:03.600) Sending read of type 0x0A to 0x18: telegram: 0B 98 0A 00 20 (CRC=1C)
(00:02:03.740) Thermostat -> me, type 0x0A telegram: 18 0B 0A 00 00 00 00 00 00 00 00 00 09 8D 05 DC 00 00 00 00 13 1A 00 00 00 00 00 00 00 00 (CRC=2B) #data=26
<--- EasyStatusMessage(0x0A)
Requesting type EasyStatusMessage(0x0A) from dest 0x18
(00:02:09.522) Sending read of type 0x0A to 0x18: telegram: 0B 98 0A 00 20 (CRC=1C)
Requesting type EasyStatusMessage(0x0A) from dest 0x18
(00:02:25.492) Sending read of type 0x0A to 0x18: telegram: 0B 98 0A 00 20 (CRC=1C)
<--- UBAMonitorFast(0x18)
```
Temperature is approx
```
Thermostat stats:
  Thermostat: TC100/Nefit Easy (ProductID:202 Version:02.19)
  Set room temperature: 15.0 C
  Current room temperature: 24.45 C
```
(may have changed in the past minute or two)

proddy, 2019-07-22:
Yup. Perfect. Found the boo-boo and making a fix now.

proddy, 2019-07-26:
if this is still not working, please re-open

closed 2019-07-26 by proddy