emsesp/EMS-ESP #157: EMS bus no longer recognized

balk77, 2019-07-21:
**Bug description**
I have been happily using EMS-ESP for several months. Today tried 1.8.1b15 and now I get this error. There is not much else to show (`log v` is empty)
`Error! Unable to read the EMS bus.`
Went back to 1.6. It was recognized correctly in 1.8.0 but there I had another issue.

**Steps to reproduce**
uploaded sketch with 
```
#define EMSESP_BOILER_TYPE EMS_ID_NONE
#define EMSESP_THERMOSTAT_TYPE 0x18
```
and with
```
#define EMSESP_BOILER_TYPE EMS_ID_NONE
#define EMSESP_THERMOSTAT_TYPE EMS_ID_NONE
```
but that made no difference.

**Device information**
Boiler type: Buderus GB172/Nefit Trendline (ProductID:123 Version:06.07)
Thermostat type: TC100/Nefit Easy (ProductID:202 Version:02.19)

proddy, 2019-07-21:
We're experimenting with improving the Tx logic in the 1.8.1 beta. What was your 'tx_mode' value set too? Also would you mind trying the stable 1.8.0 from the master branch and seeing if that works.

balk77, 2019-07-21:
`set tx_mode 0` or `1,2,3` does not make a difference. 1.8.0 is connecting properly to the EMS bus.

susisstrolch, 2019-07-21:
Stupid question: did. you do a „set serial off“?

balk77, 2019-07-21:
> Stupid question: did. you do a „set serial off“? Von meinem iPad gesendet
> […](#)

nope; `set serial on` did not help either. Is there anything I can do/test to help finding the issue?

susisstrolch, 2019-07-21:
Hmm, the Code ist still in to show it in init command. 
Please post a 5min trace so we can investigate...

proddy, 2019-07-21:
@balk77 just a thought, did it work with 1.8.0 ? There have been some many tweaks and its hard to verify and validate every scenario of devices. Perhaps this is the point to start writing some unit tests ;)

balk77, 2019-07-21:
> @balk77 just a thought, did it work with 1.8.0 ? There have been some many tweaks and its hard to verify and validate every scenario of devices. Perhaps this is the point to start writing some unit tests ;)

yes, connecting to the EMS bus does work:
```
Boiler stats:
  Boiler type: Buderus GB172/Nefit Trendline (ProductID:123 Version:06.07)
Thermostat stats:
  Thermostat type: TC100/Nefit Easy (ProductID:202 Version:02.19)
``` 
(but I do have an issue with 1.8.0, so I am back to 1.6.... #156 )
unit tests.... no clue how to do that but I am happy to help somehow. I am not a hw or sw dev but I know how to compile something for an ESP, and make minor tweaks in code.

balk77, 2019-07-21:
> Hmm, the Code ist still in to show it in init command. Please post a 5min trace so we can investigate... Von meinem iPad gesendet
> […](#)

how do I do this?

susisstrolch, 2019-07-21:
Telnet to Ems-esp, enter ‚log r‘ and send the result.
Use putty or whatever you use to issue your ‚info‘ command.
I assume you have Windows, so it‘s no help to Tell you about netcat...

balk77, 2019-07-22:
> Telnet to Ems-esp, enter ‚log r‘ and send the result. Use putty or whatever you use to issue your ‚info‘ command. I assume you have Windows, so it‘s no help to Tell you about netcat... 
I am running OS X and have access to a Linux box. `log r` does not provide any usefull info:
```
log r
System Logging set to Raw mode
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
```

susisstrolch, 2019-07-22:
But you did Enter ‚set serial off‘ in your telnet Session?

balk77, 2019-07-22:
now it is giving me something :)
Is this what could help you solving the issue?
```
System Logging set to None
log r
System Logging set to Raw mode
(00:00:25.246) 08 00 34 00 3E 01 4F 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 4D
(00:00:25.481) 18 88 18 00 18 C5
(00:00:25.550) 08 18 18 00 05 01 C6 00 00 00 00 40 40 01 4F 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 2D
(00:00:25.620) 08 00 18 00 05 01 C6 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 2A
(00:00:25.949) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:00:32.353) 18 88 16 01 02 E5
(00:00:32.399) 08 18 16 01 41 40 F6
(00:00:32.666) 18 08 06 00 13 07 15 16 1D 2A 00 01 BF
(00:00:32.983) 18 88 1C 00 08 C5
(00:00:33.018) 08 18 1C 00 92 0B 0D 1E 11 00 00 00 BC
(00:00:33.281) 18 08 1A 00 00 00 7B
(00:00:33.482) 18 08 35 00 11 1C
(00:00:33.827) 18 88 33 02 01 74
(00:00:33.865) 08 18 33 02 3C B4
(00:00:35.283) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:00:35.505) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:00:35.812) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:00:45.310) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7D 00 00 0F 30 48 00 CB 00 00 00 36
(00:00:45.529) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:00:45.889) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:00:55.235) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:00:55.456) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:00:55.714) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:01.841) 18 0B 0A 00 00 00 00 00 00 00 00 00 09 65 05 DC 00 00 00 00 12 CB 00 00 00 00 00 00 00 00 DA
(00:01:02.057) 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 5A
(00:01:03.588) 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 5A
(00:01:04.028) 0B 88 19 00 20 D0
(00:01:04.083) 08 0B 19 00 80 00 01 ED 80 00 00 00 00 00 00 6C B1 03 0C 4D 00 00 00 02 B9 23 00 2D EB 00 00 BC
(00:01:04.415) 08 0B 33 00 08 FF 3C FB 00 28 00 00 46 EC FF FF 00 7B
(00:01:05.212) 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 5A
(00:01:05.434) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:05.738) 08 0B 14 00 18 EF 28 9D
(00:01:05.949) 08 00 19 00 80 00 01 ED 80 00 00 00 00 00 00 6C B1 03 0C 4D 00 00 00 02 B9 23 00 2D EB 00 00 42
(00:01:06.126) 08 00 1C 00 92 0B 0D 1E 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 38
(00:01:06.347) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:06.681) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:07.578) 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 76
(00:01:15.364) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:15.632) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:15.883) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:25.324) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:25.545) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:25.803) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:32.437) 18 88 16 01 02 E5
(00:01:32.466) 08 18 16 01 41 40 F6
(00:01:32.720) 18 08 06 00 13 07 15 16 1E 2A 00 01 A7
(00:01:33.036) 18 88 1C 00 08 C5
(00:01:33.072) 08 18 1C 00 92 0B 0D 1E 11 00 00 00 BC
(00:01:33.290) 18 08 1A 00 00 00 7B
(00:01:33.611) 18 08 35 00 11 1C
(00:01:33.907) 18 88 33 02 01 74
(00:01:33.951) 08 18 33 02 3C B4
(00:01:35.236) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:35.459) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:35.732) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:45.329) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:45.533) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:01:45.833) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:55.292) 08 00 34 00 3E 01 4D 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 FD
(00:01:55.560) 08 00 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 52
(00:01:55.779) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:02:01.248) 08 0B 18 00 05 01 C5 00 00 00 00 40 40 01 4D 80 00 01 7C 00 00 0F 30 48 00 CB 00 00 00 E1
(00:02:01.600) 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 5A
(00:02:02.008) 0B 88 33 00 20 78
[... 118 lines cut ...]
(00:05:08.556) 18 88 19 00 02 DB
(00:05:08.602) 08 18 19 00 80 00 51
(00:05:08.659) 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 5A
(00:05:08.917) 0B 88 14 00 20 E4
(00:05:08.944) 08 0B 14 00 18 EF 2C 99
(00:05:10.117) 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 76
(00:05:15.280) 08 00 18 00 05 01 C3 00 00 00 00 40 40 01 4B 80 00 01 7A 00 00 0F 30 48 00 CB 00 00 00 97
(00:05:15.484) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:05:15.781) 08 00 34 00 3E 01 4B 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 34
[... cut ...]
(00:05:32.922) 18 88 1C 00 08 C5
(00:05:32.974) 08 18 1C 00 92 0B 0D 1E 11 00 00 00 BC
(00:05:33.130) 18 08 1A 00 00 00 7B
(00:05:33.472) 18 08 35 00 11 1C
(00:05:33.751) 18 88 33 02 01 74
(00:05:33.780) 08 18 33 02 3C B4
(00:05:35.390) 08 00 2A 00 00 00 00 00 00 00 00 00 AA 00 00 80 00 00 80 00 80 00 80 00 00 C5
(00:05:35.599) 08 00 34 00 3E 01 4B 80 00 21 00 00 01 00 00 53 2A 00 3E C6 00 80 00 34
(00:05:35.873) 08 00 18 00 05 01 C3 00 00 00 00 40 40 01 4B 80 00 01 7A 00 00 0F 30 48 00 CB 00 00 00 97
```

proddy, 2019-07-22:
ok, so it should all be working right? Data is coming and and hopefully Tx is working (you can see with the ` info` command).

bbqkees, 2019-07-25:
I would like to add here that I noticed that in the latest beta the -DNO_SERIAL flag is not set.
So serial is on again by default.

Perhaps this feature needs its own Wiki entry so its more clear what the impact of this flag is.

proddy, 2019-07-25:
yeah, I made some changes. serial is always off by default unless compiled with -DFORCE_SERIAL

(just to confuse people even more)

proddy, 2019-07-26:
if this is still problematic in the beta, please re-open

closed 2019-07-26 by proddy