emsesp/EMS-ESP #159: Logamatic TC100/Nefit Moduline Easy

Sonusss, 2019-07-21:
I have this thermostat and I saw that ems-esp is not able to send commands to it.
Of course I'd like to do so.
What can I do in the  course to help you integrate this model ?

proddy, 2019-07-21:
It's tricky. I had an Easy for a few days to play with and the thing doesn't respond to the typical EMS+ bus calls. I think the only the way is to query via XMPP which is possible but requires some considerable code changes. I made a note of it [here](https://github.com/proddy/EMS-ESP/wiki/Wish-List). If anyone wants to give this a shot??

Sonusss, 2020-03-26:
Hi Proddy,
I finaly made a working setup and indeed the solution is through XMPP see here: https://community.home-assistant.io/t/buderus-tc100-junkers-ct100-thermostat/67992/5

closed 2020-03-26 by Sonusss

proddy, 2020-03-30:
That's awesome work @Sonusss ! Thank you for sharing.