emsesp/EMS-ESP #160: 1.8.1b21 External temp

Sonusss, 2019-07-21:
My boiler have external temp probe.
It is not reported anymore nor MQTT or telnet info
It was working in 1.8.0

Sonusss, 2019-07-21:
With 1.8.0 info reports:
Outside temperature: 26.3 C

Not present in 1.8.1b21...

susisstrolch, 2019-07-21:
Running the b21 I always get the outdoor temp.
Can you post a log - about 5min in length?

proddy, 2019-07-22:
I just need a few UBAMonitorSlow (type 0x19) to debug. The outdoor temp is in the first 2 bytes of the data part.

Sonusss, 2019-07-23:
My system have UBAMonitorFast(0x18)...
There are corrupt telegram see below

System Logging set to Verbose
(08:56:45.344) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(08:56:45.545) Thermostat -> Boiler, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(08:56:45.599) Boiler -> Thermostat, type 0x18, telegram: 08 18 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 (CRC=DB) #data=24
(08:56:45.657) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:56:55.202) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:56:55.542) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:56:56.114) Boiler -> all, type 0x1C, telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B) #data=25
(08:56:56.318) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(08:56:56.565) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
Requesting scheduled EMS device data
Requesting type EasyStatusMessage(0x0A) from dest 0x18
Requesting type RCTime(0x06) from dest 0x18
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(08:57:01.198) Sending read of type 0x0A to 0x18:telegram: 0B 98 0A 00 20 (CRC=1C) #data=96
(08:57:01.341) Thermostat -> me, type 0x0A, telegram: 18 0B 0A 00 00 00 00 00 00 00 00 00 09 1A 03 E8 00 00 00 00 12 35 00 00 00 00 00 00 00 00 (CRC=65) #data=26
<--- EasyStatusMessage(0x0A)
(08:57:01.347) Sending read of type 0x06 to 0x18:telegram: 0B 98 06 00 20 (CRC=2C)
Publishing hot water and heating states via MQTT
(08:57:01.428) Thermostat -> me, telegram: 18 0B 06 00 (CRC=E0)
(08:57:01.433) Sending read of type 0x18 to 0x08:telegram: 0B 88 18 00 20 (CRC=D4)
(08:57:01.500) Boiler -> me, type 0x18, telegram: 08 0B 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=40) #data=25
<--- UBAMonitorFast(0x18)
(08:57:02.770) Sending read of type 0x19 to 0x08:telegram: 0B 88 19 00 20 (CRC=D0) #data=96
(08:57:04.106) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(08:57:05.225) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:57:05.476) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(08:57:07.253) Sending read of type 0x33 to 0x08:telegram: 0B 88 33 00 20 (CRC=78) #data=96
(08:57:07.304) Boiler -> me, type 0x33, telegram: 08 0B 33 00 00 FF 3C FB 00 28 00 02 46 00 FF FF 00 (CRC=38) #data=13
<--- UBAParameterWW(0x33)
Publishing hot water and heating states via MQTT
** We missed the bus - Rx non-idle!
(08:57:07.403) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
(08:57:07.601) Sending read of type 0x16 to 0x08:telegram: 0B 88 16 00 20 (CRC=EC) #data=96
(08:57:08.831) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(08:57:10.588) Thermostat -> Boiler, type 0x16, telegram: 18 88 16 01 02 (CRC=E5) #data=1
(08:57:10.633) Boiler -> Thermostat, type 0x16, telegram: 08 18 16 01 46 64 (CRC=DC) #data=2
(08:57:10.962) Thermostat -> Boiler, type 0x06, telegram: 18 08 06 00 13 07 09 17 02 29 01 01 (CRC=F2) #data=8
(08:57:11.211) Thermostat -> Boiler, type 0x1C, telegram: 18 88 1C 00 08 (CRC=C5) #data=1
(08:57:11.262) Boiler -> Thermostat, type 0x1C, telegram: 08 18 1C 00 00 00 00 00 00 00 00 00 (CRC=41) #data=8
(08:57:11.414) Thermostat -> Boiler, type 0x1A, telegram: 18 08 1A 00 00 00 (CRC=7B) #data=2
(08:57:11.563) Sending read of type 0x14 to 0x08:telegram: 0B 88 14 00 20 (CRC=E4) #data=96
(08:57:11.588) Corrupt telegram: 0B 88 14 42 20 E4
** Write command failed from host
(08:57:11.664) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
(08:57:12.130) Thermostat -> Boiler, type 0x35, telegram: 18 08 35 00 11 (CRC=1C) #data=1
(08:57:12.482) Thermostat -> Boiler, type 0x33, telegram: 18 88 33 02 01 (CRC=74) #data=1
(08:57:12.525) Boiler -> Thermostat, type 0x33, telegram: 08 18 33 02 3C (CRC=B4) #data=1
(08:57:12.877) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(08:57:15.258) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:57:15.478) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
[... cut ...]
(08:58:24.971) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25
<--- UBAMonitorFast(0x18)
(08:58:25.229) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(08:58:34.992) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(08:58:35.236) Boiler -> all, type 0x18, telegram: 08 00 18 00 05 00 EE 00 00 00 00 00 40 80 00 80 00 80 00 00 00 FF 30 48 00 CB 00 00 00 (CRC=F3) #data=25

mf01, 2019-07-24:
Same here. Using 1.8.1b21 some other data are also missed in MQTT.

"boiler_data"
- pumpMod
- selFlowTemp
- outdoorTemp

"sm_data"
- energylasthour
- energytoday
- energytotal

proddy, 2019-07-26:
the problem was that 0x19 wasn't being picked up because of the buffer size was 32 and not 33 bytes. Anyway, should be fixed in 1.8.1b23

closed 2019-07-26 by proddy