emsesp/EMS-ESP #161: 1.8.1b21 data missing

zegerk, 2019-07-22:
Data based on "UBAParametersMessage" and "UBAMonitorSlow" is missing. It seems the requests are no longer send, when doing a refresh they do not shown in the verbose log.

Data was available in 1.8.0

  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times

susisstrolch, 2019-07-22:
```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=42, # CRC errors=35
  Tx: Last poll=1.408 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: DeviceID: 0x08 (ProductID:? Version:?)
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: off
  Selected flow temperature: 11 C
  Current flow temperature: 21.6 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (0)
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 100 %
  Outside temperature: 25.0 C
  Boiler temperature: 21.5 C
  Pump modulation: 0 %
  Burner # starts: 138298 times
  Total burner operating time: 361 days 5 hours 39 minutes
  Total heat operating time: 360 days 3 hours 35 minutes
  Total UBA working time: 2773 days 6 hours 46 minutes

Thermostat stats:
  Thermostat: RC35 (ProductID:86 Version:01.11)
  Setpoint room temperature: 0.0 C
  Current room temperature: ? C
  Thermostat time is 21:15:45 22/7/2019
  Mode is set to night

** We missed the bus - Rx non-idle!
```
Can‘t confirm

proddy, 2019-07-22:
@zegerk perhaps the serial on issue again which is preventing the EMS bus to active. What does `info` and `set` show you?

zegerk, 2019-07-22:
Info

`EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=120, # CRC errors=137
  Tx: Last poll=4.456 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB172/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 31.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 42398 times
  Warm Water active time: 11 days 2 hours 44 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 33.0 C
  Return temperature: 32.0 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.3 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
  Total UBA working time: 862 days 4 hours 43 minutes

Thermostat stats:
  Thermostat: DeviceID: 0x02 (ProductID:? Version:?)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 00:00:00 0/0/2000
  Mode is set to auto
`

`  wifi_ssid=het-galjoen 
  wifi_password=***************
  mqtt_host=<not set>
  mqtt_username=<not set>
  mqtt_password=<not set>
  serial=off
  heartbeat=off
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=02
  heating_circuit=1
  boiler_type=08
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120
  tx_mode=2
`

proddy, 2019-07-22:
ok, looks like data is coming in now from the boiler. Just not the Thermostat because it's not detected. Try `set tx_mode 0` reboot and see if the Thermostat is detected.

zeger-pon, 2019-07-23:
There is no thermostat attached - only using the EMS-OT convertor, or could that be the issue?

mf01, 2019-07-24:
Same here. Using 1.8.1b21 some other data are also missed in MQTT.

"boiler_data"
- pumpMod
- selFlowTemp
- outdoorTemp

"sm_data"
- energylasthour
- energytoday
- energytotal

**info**
```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is enabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=11, # CRC errors=3
  Tx: Last poll=2.568 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB172/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:04.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 59 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 61.2 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2898 times
  Warm Water active time: 31 days 12 hours 1 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 32.1 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
  Total UBA working time: 2047 days 18 hours 35 minutes

Solar Module stats:
  Solar Module: SM100 Solar Module (ProductID:163 Version:21.03)
  Collector temperature: 30.5 C
  Bottom temperature: 56.3 C
  Pump modulation: 0 %
  Pump active: ?
  Energy Last Hour: ? Wh
  Energy Today: ? Wh
  Energy Total: ? kWH

Thermostat stats:
  Thermostat: RC300/RC310/Nefit Moduline 3000 (ProductID:158 Version:11.04)
  Setpoint room temperature: ? C
  Current room temperature: ? C
  Thermostat time is 22:52:45 24/7/2019
  Mode is set to ?
```

**set**
```
[*****]

Stored settings:

  wifi_ssid=***** 
  wifi_password=*****
  mqtt_host=*****
  mqtt_username=*****
  mqtt_password=*****
  serial=off
  heartbeat=on
  led=on
  led_gpio=2
  dallas_gpio=14
  dallas_parasite=off
  thermostat_type=10
  heating_circuit=1
  boiler_type=08
  listen_mode=off
  shower_timer=off
  shower_alert=off
  publish_time=120
  tx_mode=0

** We missed the bus - Rx non-idle!
```

And of course ... the message _We missed the bus - Rx non-idle!_ is still there...

proddy, 2019-07-26:
Should be fixed in 1.8.1b23 now.

blacksocket, 2019-09-21:
Hi All,

I started using v. firmware_release_1_8_3.bin directly on ESP8266_12 and getting the below messages. However when flashed the same firmware on Wemos it works fine. Do you know what could be the issue ?

info
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  No connection can be made to the EMS bus

Boiler stats:
  Boiler: <not enabled>
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: ?
  Selected flow temperature: ? C
  Current flow temperature: ? C
  Return temperature: ? C
  Gas: ?
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: ? %
  Burner current power: ? %
  Flame current: ? uA
  System pressure: ? bar
  System service code: ?? (32768)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times

Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.

mf01, 2019-09-21:
> Error! Unable to read the EMS bus.
You have checked that serial communication is off?

blacksocket, 2019-09-21:
Hi, do you mean check if the physical serial communication is off or do you mean somewhere in the software?

mf01, 2019-09-21:
Make  a telnet connection, than enter 

`set serial off`

and reboot

blacksocket, 2019-09-21:
Indeed i forgot about that however still can't read EMS bus. When switch EMS bus to wemos it works fine. Both are in the same mode now. No idea maybe some physical connection issues?

System Logging set to Verbose
System is either not connect to the EMS bus or listen_mode is enabled
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.

blacksocket, 2019-09-21:
Finally I have separated the second PCB board with ESP8266_12. Seems Bus is connected:
  Rx: # successful read requests=0, # CRC errors=1
  Tx: Last poll=3.883 seconds ago, # successful write requests=0
However wne switch to "log v" I am still getting System is either not connect to the EMS bus or listen_mode is enabled
Error! Unable to read the EMS bus.
Strange that Boiler status is showing some data like temperature etc..

log v
System Logging set to Verbose
info
EMS-ESP system stats:
  System logging set to Verbose
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=1
  Tx: Last poll=3.883 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: <not enabled>
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 43.8 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1194 times
  Warm Water active time: 8 days 16 hours 11 minutes
  Warm Water 3-way valve: ?
  Selected flow temperature: ? C
  Current flow temperature: ? C
  Return temperature: ? C
  Gas: ?
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: ? %
  Burner current power: ? %
  Flame current: ? uA
  System pressure: ? bar
  System service code: ?? (32768)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times

proddy, 2019-09-21:
I don't think it should make a difference, but try compiling with a different board in PlatformIO instead of the default Wemos.

blacksocket, 2019-09-21:
Hi Proddy, I didn't build the binaries from source. I just upload the firmware_release_1_8_3.bin into the wemos (where it works fine) and then into my ESP8266_12 on the pcb (where I am having issues):

My new built PCB:
[image]

proddy, 2019-09-21:
that's nice! Are you not able to make a new .bin file? If not I'll make one for you.

blacksocket, 2019-09-21:
Not sure for what this release is targeted but I through that the only differences between wemos and simple ESP is that on wemos there are some additional features like USB-UART converter and that's it. If you will be so kind yes please! Hope this helps..  ;-)

proddy, 2019-09-21:
The code is compiled using the board spec [here](https://docs.platformio.org/en/latest/boards/espressif8266/d1_mini.html). I did try the [12e](https://docs.platformio.org/en/latest/boards/espressif8266/esp12e.html) once and noticed some of the pins were different.

proddy, 2019-09-21:
oh and you should be using the latest 1.9.1 from the dev branch for testing.

blacksocket, 2019-09-21:
On my board i used ESP8266 ESP-12F:

[image]

RST - Resetuje moduł
ADC - Konwerter A/C Napięcie w zakresie 0-1V konwertowane na wartość 0-1024
EN - Chip enable. HIGH - aktywny
GPIO16; WAKE - Używany do wybudzenia modułu z trybu DEEP SLEEP
GPIO14; HSPI_CLK
GPIO12; HSPI_MISO
GPIO13; HSPI_MOSI; UART0_CTS
VCC - zasilanie 3.3V
CS0 - Chip select
MISO
GPIO9
GPIO10
MOSI
SCLK - Zegar
GND
GPIO15; MTDO; HSPICS; UART0_RTS
GPIO2; UART1_TXD
GPIO0
GPIO4
GPIO5
RXD; UART0_RXD; GPIO3
TXD; UART0_TXD; GPIO1

blacksocket, 2019-09-21:
What additional I have found that pins determine what mode the chip starts up in.:
https://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html

[image]

blacksocket, 2019-09-21:
No idea, I tried many combinations and the best one is with GPIO15 connected to ground. Then Bus is connected and Boiler stats showing data but I am still getting the below messages :

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=0
  Tx: Last poll=1.969 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 39.4 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1194 times
  Warm Water active time: 8 days 16 hours 11 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 31.7 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times

Unable to read the EMS bus.

System Logging set to Verbose
System is either not connect to the EMS bus or listen_mode is enabled
Error! Unable to read the EMS bus.
Error! Unable to read the EMS bus.

proddy, 2019-09-21:
The rx and tx pins are different on this board. I also swap them in the code to d7 and d8. To get it working on the 12f I think you’ll need to use gpio21 and 22 and comment out the swap in the code in file emsuart.cpp

proddy, 2019-09-22:
@blacksocket could you create a new issue for supporting multiple ESP8266 board types? This issue will get lost under the current title which is about missing data with a very old version of EMS-ESP.

closed 2019-09-22 by proddy

blacksocket, 2019-09-22:
Hi Proddy, many thanks for you help! Sure I will create the new issue for this topic.