emsesp/EMS-ESP #162: Control both HC1 & HC2 at the same time in Home Assistant (MQTT)

Frederik-V, 2019-07-24:
I'm pretty new on this fora and starting to explore the possibilities of the EMS-ESP.
On top, I'm a Home Assistant (HA) user in order to control my home domotica.

I was very enthusiastic to read on the fora that it is possible to control the 2 heating circuits.
I'm using HC1 for radiators in the bedrooms (level 1), HC2 as floor heating (level 0).

Is there a possibility to control both Heating Circuits at the same time with MQTT?
Meaning, having two thermostat control systems in the Home Assistant GUI.
Now you need to switch all the time between the Heating Circuits in order to read out the temperature settings.

proddy, 2019-07-24:
Hi @Frederik-V, yes it's possible to extend the MQTT to support multiple Heating Circuits. Easiest is to have two sets of MQTT messages so you can add two climate components to HA.

It would need some additional code changes to fetch the thermostat values from the multiple HCs and store them. The difficulty is always testing these. Are you comfortable making code changes yourself to EMS-ESP ?

Frederik-V, 2019-07-25:
Hi @proddy, unfortunately I do not have any experience to make code changes into the EMS-ESP. 
On the other hand, support with testing of the changes and making some example's in HA, should be not an issue!

lmdecroos, 2019-09-07:
Hi proddy,

I have a Buderus with 2 heat circuits and 2 separate thermostats  an RC35 as main device and RMF20 for the first floor.
With version 1.8.x the gateway is randomly reading the of one of the thermostats. It doesn't take the Heat Circuit setting in account from the settings

[image]
  
As you can see in the screenshot the temperature is switching between the 2 locations

Hope you able to enhance the software that the HC setting is linked to a dedicated thermostat and will be able to read out both independently

Thanks in advance

proddy, 2019-09-07:
This a bug, I'll add some extra functionality to read multiple heating circuits to the dev build

proddy, 2019-09-07:
@lmdecroos I'm adding supporting for multiple Heating Circuits. Adding additional thermostats will be more work. Can you answer these questions
 - does EMS-ESP detect both your RC35 thermostats?
 - do each of the thermostats report value for both the 2 heating circuits?
 - can you do a 'log t' and send me some examples of telegrams 0x3E (HC1) and 0x48 (HC2) so I can build some test data. I don't have an RC35 so it'll be a lot of trial and error. be prepared to do a lot of testing!

lmdecroos, 2019-09-07:
Hi proddy,

I have one RC35 and one RFM20 ( RC20 wireless) 
The RC35 is used for programming the boiler settings and to control HC1 for heating on the ground floor
The RC20 is used to control HC2  for our heating on the fist floor.
They are both working independently .

When asking for "info” the gateway stays that i have an RC35
Thermostat stats:
 Thermostat: RC35 (ProductID:86 Version:21.08)

When requesting an autodetect
Started scan on EMS bus for known devices
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
Device found: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)

The gateway recognise also the switch for the pumps to control both HC’s

I had a look at the telegram messages in the EMS gateway and the MQTT received by Domoticz and it seems that the EMS gateway receives the temperature details from both heat circuit and send them both to the MQTT host.
The EMS software doesn’t filter the message based on the selected HC in the EMS configuration. Therefore the actual temperature in the MQTT message is switching between both HC’s ( thermostats) 

The MQTT messages includes the correct HC  (thermostat_hc) as set in the EMS gateway, but send always as current temperature the temperature from last received telegram message. 
I would assume that you have to build in a filter based on the selected HC to only send the temperature from the selected HC

Logs with HC1 selected:
System Logging set to Thermostat only
(03:49:30.725) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 12 07 14 2B 05 01 (CRC=A6) #data=8
(03:49:30.995) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 C5 00 00 00 00 00 00 00 00 11 05 04 (CRC=AD) #data=16
Publishing thermostat data via MQTT
(03:49:32.245) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 10 00 00 B4 00 00 00 00 00 00 00 00 11 05 04 (CRC=C8) #data=16
Publishing thermostat data via MQTT
(03:49:32.542) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 6E 11 00 00 C5 00 CB 00 CB 7D 00 7D 00 (CRC=D8) #data=13
(03:49:48.462) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
(03:49:57.581) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 11 01 (CRC=20) #data=2
(03:49:57.708) Thermostat -> 0x11, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Requesting scheduled EMS device data
(03:50:01.243) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 00 C5 00 00 00 00 00 00 00 00 11 05 04 (CRC=C5) #data=16
Publishing thermostat data via MQTT
(03:50:01.556) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=43) #data=27
Publishing thermostat data via MQTT
(03:50:01.869) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 12 07 15 0E 05 01 (CRC=9D) #data=8
Publishing boiler data via MQTT
(03:50:06.364) Thermostat -> all, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99) #data=14
(03:50:08.954) Thermostat -> Boiler, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(03:50:08.967) Boiler -> Thermostat, type 0x14, telegram: 08 10 14 00 32 81 BA (CRC=30) #data=3
(03:50:17.710) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
(03:50:31.225) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 12 07 15 2B 05 01 (CRC=AE) #data=8
(03:50:31.494) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 C5 00 00 00 00 00 00 00 00 11 05 04 (CRC=AD) #data=16
Publishing boiler data via MQTT
(03:50:32.744) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 10 00 00 B4 00 00 00 00 00 00 00 00 11 05 04 (CRC=C8) #data=16
Publishing thermostat data via MQTT
(03:50:33.039) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 6E 11 00 00 C5 00 CB 00 CB 7D 00 7D 00 (CRC=D8) #data=13

2019-09-07 18:22:11.643 (Buderus - System Pressure) Updating device from 1:'1' to have values 1:'1'.
2019-09-07 18:22:11.649 (Buderus) Current temp: 62.8
2019-09-07 18:22:11.649 (Buderus - Warmwater) Updating device from 1:'62.8' to have values 1:'62.8'.
2019-09-07 18:22:11.654 (Buderus) Current temp: 56.1
2019-09-07 18:22:11.654 (Buderus - Boiler) Updating device from 1:'56.1' to have values 1:'56.1'.
2019-09-07 18:22:11.659 (Buderus) Burner: off
2019-09-07 18:22:12.912 (Buderus) MQTT message: home/ems-esp/thermostat_data b'{"thermostat_hc":"1","thermostat_seltemp":0,"thermostat_currtemp":18,"thermostat_daytemp":19,"thermostat_nighttemp":15,"thermostat_holidayttemp":17,"thermostat_heatingtype":1,"thermostat_circuitcalctemp":5,"thermostat_mode":"night"}'
2019-09-07 18:22:12.912 (Buderus) Current temp: 18.0
2019-09-07 18:22:12.912 (Woonkamer) Updating device from 1:'19.7' to have values 1:'18.0'.
2019-09-07 18:22:12.918 (Buderus) Temp setting: 0
2019-09-07 18:22:12.918 (Buderus - Nefit) Updating device from 1:'0' to have values 1:'0'.
2019-09-07 18:22:19.335 (Buderus) Heartbeating...
2019-09-07 18:22:19.336 (Buderus) MqttClient::Ping
[... 14 lines cut ...]
2019-09-07 18:22:42.195 (Woonkamer) Updating device from 1:'18.0' to have values 1:'19.7'.
2019-09-07 18:22:42.200 (Buderus) Temp setting: 0
2019-09-07 18:22:42.200 (Buderus - Nefit) Updating device from 1:'0' to have values 1:'0'.
[... cut ...]
2019-09-07 18:43:22.544 (Buderus) MQTT message: home/ems-esp/thermostat_data b'{"thermostat_hc":"2","thermostat_seltemp":0,"thermostat_currtemp":17.7,"thermostat_daytemp":18,"thermostat_nighttemp":13.5,"thermostat_holidayttemp":17,"thermostat_heatingtype":1,"thermostat_circuitcalctemp":5,"thermostat_mode":"night"}'
2019-09-07 18:43:22.545 (Buderus) Current temp: 17.7
2019-09-07 18:43:22.545 (Woonkamer) Updating device from 1:'19.7' to have values 1:'17.7'.
2019-09-07 18:43:22.550 (Buderus) Temp setting: 0

Please let me know if you need more or other logs

Hope this helps 

Thanks in advance

proddy, 2019-09-07:
I spent some time today to implement handling of multiple HCs independently. It's checked into the dev branch 1.9.1 so give it whirl. It's far from perfect and I still need to decide on the mqtt format for driving multiple heating circuits.

lmdecroos, 2019-09-09:
I compiled the 1.9.1 dev branch and  have unfortunately issues to get the gateway stable (similar as with the release version)
The connection to be EMS bus is unstable and crashes ( the led is flashing)

When using the 1.9.x version, every time I’m doing a telnet to the gateway the gateway start scanning the EMS bus

proddy, 2019-09-10:
@lmdecroos When EMS-ESP starts it will auto-detect all known devices, this only happens once and should take a few seconds. If the Tx is not working this will fail, but shouldn't repeat. You can try different tx_modes (1 is default, 2 is EMS+ and 3 is Junkers). I've seen others report back that 1.9 keeps crashing but I cannot reproduce and I have no example logs, stacks etc to help pinpoint the issue. Is it really crashing (i.e. resetting)? See https://github.com/proddy/EMS-ESP/wiki/Troubleshooting

lmdecroos, 2019-09-10:
when using the default EMS protocol, the gateway crashes shortly and resets the serial value to on. the web UI seems to know that i have a RC35, but the console scans every time the EMS bus ( when starting a telnet session) and then crashes even before i’m try to change the log level

when using the EMS+ protocol, then the gateway is stable but reads only the boiler values, there is no information or update from the thermostat. in the web UI there is only info about the boiler. there is also no information about the thermostat in the temp & verbose logs

proddy, 2019-09-11:
@lmdecroos could you create a separate issue for this problem as this one is related to a feature request to support multiple heating circuits.

lmdecroos, 2019-09-11:
Yes, I can create a separate issue. But the EMS communication issues are introduced with version 1.9.x

With version 1.8.3 my RC35 thermostat was recognised by using the ems protocol ( tx_mode 1)
With version 1.9.0 and 1.9.1b the ems protocol chraches  the gateway :

System Logging set to None
System Logging set to Verbose
(00:00:05.704) Sending read of type 0x02 to 0x38, telegram: 0B B8 02 00 20 (CRC=25)
(00:00:06.476) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 72 02 72 21 00 00 03 00 00 98 A7 00 15 3C 00 80 00 (CRC=4A) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:06.706) Boiler -> all, type 0x18, telegram: 08 00 18 00 3D 02 89 72 00 00 00 64 60 02 67 02 72 02 82 00 00 0D 30 41 01 31 00 01 00 (CRC=A4) #data=25
<--- UBAMonitorFast(0x18)
(00:00:06.949) Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(00:00:08.112) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=13
(00:00:08.518) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(00:00:08.558) Boiler -> me, type 0x02, telegram: 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 (CRC=D7) #data=10
<--- Version(0x02)
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
(00:00:08.636) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=7F) #data=13
(00:00:08.892) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(00:00:08.930) 0x09 -> me, type 0x02, telegram: 09 0B 02 00 44 02 03 (CRC=25) #data=3
<--- Version(0x02)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
(00:00:09.240) Sending read of type 0x02 to 0x10, telegram: 0B 90 02 00 20 (CRC=7C)
(00:00:09.279) 0x10 -> me, type 0x02, telegram: 10 0B 02 00 56 15 08 (CRC=5E) #data=3
<--- Version(0x02)
Thermostat found: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
* Setting Thermostat to RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
Requesting type RC35StatusMessage_HC1(0x3E) from dest 0x10
Requesting type RC35Set_HC1(0x3D) from dest 0x10
Requesting type RC35StatusMessage_HC2(0x48) from dest 0x10
Requesting type RC35Set_HC2(0x47) from dest 0x10
Requesting type RC35StatusMessage_HC3(0x52) from dest 0x10
Requesting type RC35Set_HC2(0x51) from dest 0x10
Requesting type RC35StatusMessage_HC4(0x5C) from dest 0x10
Requesting type RC35Set_HC4(0x5B) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(00:00:09.539) Sending read of type 0x02 to 0x11, telegram: 0B 91 02 00 20 (CRC=74)
(00:00:09.584) 0x11 -> me, type 0x02, telegram: 11 0B 02 00 47 02 00 (CRC=7C) #data=3
<--- Version(0x02)
Device found: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
(00:00:09.813) Sending read of type 0x02 to 0x17, telegram: 0B 97 02 00 20 (CRC=44)
(00:00:10.417) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 02 88 00 00 00 00 40 60 02 69 02 73 02 81 00 00 0A 30 48 00 CB 00 00 00 (CRC=25) #data=25
<--- UBAMonitorFast(0x18)
(00:00:10.677) 0x11 -> Boiler, type 0x1E, telegram: 11 08 1E 00 00 CD (CRC=AF) #data=2
(00:00:10.984) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)

The gateway hangs every time on : Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C) and reboots itself afterwards

When using the EMS+ protocol (tx_mode 2) the gateway works stable, but doesn’t recognise my RC35 thermostat 
the EMS+ protocol recognises only my boiler ( same for version 1.8.3)

proddy, 2019-09-14:
see also https://github.com/proddy/EMS-ESP/wiki/MQTT on how to use MQTT to control multiple heating circuits.

lmdecroos, 2019-09-14:
Thanks for updating the MQTT wiki, but I'm still suffer to get my thermostat / HC recognised in version 1.9.x

With the standard EMS protocol, both my boiler and HC are recognised in version 1.8.3, but when using version 1.9.x the EMS gateway crashes at startup while polling the bus.
The gateway hangs every time on : Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C) and reboots itself afterwards, see my entry from 11th September.
I tested this messages in version 1.8.3, and then the 18.3 EMS gateway crashes also.

Can you advise where I have to look in the source code for the initiation and startup process, so that I can troubleshoot by check the differences between both versions.

susisstrolch, 2019-09-14:
I‘d suggest ‚set log j‘ to get an idea where it happens...

Sent by mobile device

lmdecroos, 2019-09-14:
please find the jabber log until the crash :

(00:00:08.141) ems_parseTelegram: 00 01: 8B 
(00:00:08.142) emsuart_tx_buffer: 00 01: 0B 82 02 00 20 EC 
(00:00:08.158) ems_parseTelegram: 00 02: 0B 82 02 00 20 EC 
(00:00:08.158) echo:telegram: 0B 82 02 00 20 (CRC=EC) #data=1
(00:00:08.390) ems_parseTelegram: 00 02: 90 
(00:00:08.395) ems_parseTelegram: 00 02: 10 
(00:00:08.415) ems_parseTelegram: 00 02: 89 
(00:00:08.425) ems_parseTelegram: 00 02: 09 
(00:00:08.440) ems_parseTelegram: 00 02: 8D 
(00:00:08.465) ems_parseTelegram: 00 02: 89 
(00:00:08.472) ems_parseTelegram: 00 02: 09 
(00:00:08.490) ems_parseTelegram: 00 02: 8E 
(00:00:08.515) ems_parseTelegram: 00 02: 91 
(00:00:08.519) ems_parseTelegram: 00 02: 11 
(00:00:08.540) ems_parseTelegram: 00 02: 89 
(00:00:08.549) ems_parseTelegram: 00 02: 09 
(00:00:08.565) ems_parseTelegram: 00 02: 99 
(00:00:08.569) ems_parseTelegram: 00 02: 19 
(00:00:08.590) ems_parseTelegram: 00 02: 8F 
(00:00:08.615) ems_parseTelegram: 00 02: A1 
(00:00:08.619) ems_parseTelegram: 00 02: 21 
(00:00:08.639) ems_parseTelegram: 00 02: 92 
(00:00:08.664) ems_parseTelegram: 00 02: 89 
(00:00:08.672) ems_parseTelegram: 00 02: 09 
(00:00:08.689) ems_parseTelegram: 00 02: 89 
(00:00:08.698) ems_parseTelegram: 00 02: 09 
(00:00:08.714) ems_parseTelegram: 00 02: 89 
(00:00:08.724) ems_parseTelegram: 00 02: 09 
(00:00:08.739) ems_parseTelegram: 00 02: 89 
(00:00:08.749) ems_parseTelegram: 00 02: 09 
(00:00:08.764) ems_parseTelegram: 00 02: 8B 
(00:00:08.789) ems_parseTelegram: 00 02: 90 
(00:00:08.794) ems_parseTelegram: 00 02: 10 
(00:00:08.814) ems_parseTelegram: 00 02: 89 
(00:00:08.821) ems_parseTelegram: 00 02: 09 
(00:00:08.839) ems_parseTelegram: 00 02: 93 
(00:00:08.864) ems_parseTelegram: 00 02: 89 
(00:00:08.873) ems_parseTelegram: 00 02: 09 
(00:00:08.888) ems_parseTelegram: 00 02: 94 
(00:00:08.913) ems_parseTelegram: 00 02: 91 
(00:00:08.918) ems_parseTelegram: 00 02: 11 
(00:00:08.938) ems_parseTelegram: 00 02: 89 
(00:00:08.945) ems_parseTelegram: 00 02: 09 
(00:00:08.964) ems_parseTelegram: 00 02: 99 
(00:00:08.968) ems_parseTelegram: 00 02: 19 
(00:00:08.988) ems_parseTelegram: 00 02: 95 
(00:00:09.013) ems_parseTelegram: 00 02: A1 
(00:00:09.017) ems_parseTelegram: 00 02: 21 
(00:00:09.038) ems_parseTelegram: 00 02: 96 
(00:00:09.063) ems_parseTelegram: 00 02: 89 
(00:00:09.073) ems_parseTelegram: 00 02: 09 
(00:00:09.088) ems_parseTelegram: 00 02: 89 
(00:00:09.094) ems_parseTelegram: 00 02: 09 
(00:00:09.113) ems_parseTelegram: 00 02: 89 
(00:00:09.120) ems_parseTelegram: 00 02: 09 
(00:00:09.138) ems_parseTelegram: 00 02: 89 
(00:00:09.145) ems_parseTelegram: 00 02: 09 
(00:00:09.163) ems_parseTelegram: 00 02: 8B 
(00:00:09.188) ems_parseTelegram: 00 02: 90 
(00:00:09.192) ems_parseTelegram: 00 02: 10 
(00:00:09.212) ems_parseTelegram: 00 02: 89 
(00:00:09.222) ems_parseTelegram: 00 02: 09 
(00:00:09.237) ems_parseTelegram: 00 02: 97 
(00:00:09.262) ems_parseTelegram: 00 02: 89 
(00:00:09.269) ems_parseTelegram: 00 02: 09 
(00:00:09.287) ems_parseTelegram: 00 02: 9C 
(00:00:09.312) ems_parseTelegram: 00 02: 91 
(00:00:09.316) ems_parseTelegram: 00 02: 11 
[... 221 lines cut ...]
(00:00:13.402) ems_parseTelegram: 00 01: 18 
(00:00:13.447) ems_parseTelegram: 00 01: 08 00 34 00 3D 02 45 02 45 21 00 00 03 00 00 98 EC 00 15 49 00 80 00 D4 
(00:00:13.448) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 45 02 45 21 00 00 03 00 00 98 EC 00 15 49 00 80 00 (CRC=D4) #data=19
(00:00:13.646) ems_parseTelegram: 00 01: B8 
(00:00:13.671) ems_parseTelegram: 00 01: 89 
(00:00:13.677) ems_parseTelegram: 00 01: 09 
(00:00:13.696) ems_parseTelegram: 00 01: C0 
(00:00:13.721) ems_parseTelegram: 00 01: 99 
(00:00:13.725) ems_parseTelegram: 00 01: 19 
(00:00:13.745) ems_parseTelegram: 00 01: 89 
(00:00:13.754) ems_parseTelegram: 00 01: 09 
(00:00:13.771) ems_parseTelegram: 00 01: A1 
(00:00:13.775) ems_parseTelegram: 00 01: 21 
(00:00:13.795) ems_parseTelegram: 00 01: C8 
(00:00:13.820) ems_parseTelegram: 00 01: 89 
(00:00:13.831) ems_parseTelegram: 00 01: 09 
(00:00:13.845) ems_parseTelegram: 00 01: D0 
(00:00:13.870) ems_parseTelegram: 00 01: 89 
(00:00:13.877) ems_parseTelegram: 00 01: 09 
(00:00:13.895) ems_parseTelegram: 00 01: 89 
(00:00:13.903) ems_parseTelegram: 00 01: 09 
(00:00:13.920) ems_parseTelegram: 00 01: 8B 
(00:00:13.920) emsuart_tx_buffer: 00 01: 0B 98 02 00 20 3C 
(00:00:13.935) ems_parseTelegram: 00 02: 0B 98 02 00 20 3C 
(00:00:13.936) echo:telegram: 0B 98 02 00 20 (CRC=3C) #data=1

proddy, 2019-09-15:
Please create a new GitHub issue for the reboot problems. From the log I see the devices are detected correctly so something else is causing the crash, either the MQTT or the Event Log.

majdzik84, 2019-09-17:
I have version 1.9.1b3.
I wanted to check if the RC35 temperature was in auto mode. (0.0C)
I see that it is no longer possible to switch heating circuits. By default, I see HC1 and want HC2 - how can I change it?

Additionally, I noticed that the web interface is very unstable (hangs)
[image]

proddy, 2019-09-17:
@majdzik84 all heating circuits should be detected and shown. What's strange is that you don't get any listed under the 'Thermostat stats' page?

The code picks up messages from telegram types 0x3E (HC1), 0x48 (HC2), 0x52 (HC3) and 0x5C (HC4)

I can't test when I make these changes and do my best to simulate what it would look like on the multitude of thermostat models so any test data would be helpful. Even better are code changes ;)

majdzik84, 2019-09-17:
below I am sending a log - maybe it will be helpful in something
```
log t
System Logging set to Thermostat only
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Requesting scheduled EMS device data
(04:06:02.034) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04 (CRC=ED) #data=16
(04:06:02.440) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2B 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=8F) #data=27
(04:06:02.908) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 00 00 E4 00 00 00 00 00 00 00 00 11 05 04 (CRC=45) #data=16
(04:06:03.315) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(04:06:03.784) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:06:04.191) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(04:06:04.659) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
Publishing boiler data via MQTT
(04:06:05.066) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
Publishing boiler data via MQTT
(04:06:05.514) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 11 12 34 34 02 01 (CRC=43) #data=8
(04:06:05.905) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 10 00 00 00 E4 00 EA 00 EA 7D 00 7D 00 (CRC=24) #data=13
(04:06:10.376) Thermostat -> 0x08, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:06:21.002) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 03 (CRC=0D) #data=3
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:06:28.869) Thermostat -> 0x08, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(04:06:28.888) 0x08 -> Thermostat, type 0x14, telegram: 08 10 14 00 17 7A 48 (CRC=B9) #data=3
(04:06:31.121) Thermostat -> 0x08, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(04:06:44.765) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 11 12 35 1F 02 01 (CRC=40) #data=8
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:06:49.659) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 00 00 E4 00 00 00 00 00 00 00 00 11 05 04 (CRC=2D) #data=16
Publishing boiler data via MQTT
(04:06:54.159) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04 (CRC=85) #data=16
Requesting scheduled EMS device data
(04:07:00.657) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04 (CRC=ED) #data=16
(04:07:01.189) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2B 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=8F) #data=27
(04:07:02.033) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 00 00 E4 00 00 00 00 00 00 00 00 11 05 04 (CRC=45) #data=16
(04:07:02.440) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(04:07:02.783) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
(04:07:03.315) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(04:07:03.783) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
Publishing boiler data via MQTT
(04:07:04.190) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
Publishing boiler data via MQTT
(04:07:04.514) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 11 12 35 33 02 01 (CRC=57) #data=8
(04:07:06.154) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 10 00 00 00 E4 00 EA 00 EA 7D 00 7D 00 (CRC=24) #data=13
(04:07:10.500) Thermostat -> 0x08, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
(04:07:21.126) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 03 (CRC=0D) #data=3
^C(04:07:31.621) Thermostat -> 0x08, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:07:45.264) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 11 12 36 1F 02 01 (CRC=58) #data=8
(04:07:49.908) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 00 00 E4 00 00 00 00 00 00 00 00 11 05 04 (CRC=2D) #data=16
Publishing boiler data via MQTT
(04:07:54.408) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04 (CRC=85) #data=16
Publishing boiler data via MQTT
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Requesting scheduled EMS device data
(04:08:01.781) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04 (CRC=ED) #data=16
[... cut ...]
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(04:08:21.500) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 03 (CRC=0D) #data=3
(04:08:29.243) Thermostat -> 0x08, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(04:08:29.265) 0x08 -> Thermostat, type 0x14, telegram: 08 10 14 00 17 7A 4A (CRC=BB) #data=3
(04:08:29.406) Thermostat -> all, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99) #data=14
(04:08:31.870) Thermostat -> 0x08, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
```

proddy, 2019-09-17:
what I can see is

 HC1 = 10 0B 3E 00   00 00 00 7D 00 00 00 00 00 00 00 00 00 11 05 04
 HC2 = 10 00 48 00   00 00 00 00 E4 00 00 00 00 00 00 00 00 11 05 04
 HC3 = 10 0B 52 00   00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00
 HC4 = 10 0B 5C 00   00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00

which tells me
HC1 has a setpoint of 0.0 and current temps is not defined
HC2 has a setpoint of 0.0 and a current temp is 22.8
HC3 and HC4 are both not used

Is this in line with what you expect?

majdzik84, 2019-09-17:
the thermostat controls the HC2 circuit (RC35) the current temperature is correct but the setpoint temp. should not show 0.0C. automatic mode was enabled and should be fixed in this version. 
HC3 and HC4 - not used

proddy, 2019-09-18:
What should the setpoint show then for HC1,2,3 and 4? All the telegrams that come in are telling EMS-ESP that the value is 0.0? I can change the logic if you can help me understand what it should be in 'auto' mode. I don't have an RC35 to test unfortunately.

majdzik84, 2019-09-18:
**Mode AUTO (day)**

```
(06:35:05.227) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00                                 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
Publishing boiler data via MQTT
(06:35:05.634) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00                                 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #dat                                a=27
Publishing boiler data via MQTT
(06:35:05.957) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 0D 13 3A                                 34 03 01 (CRC=8A) #data=8
(06:35:06.473) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 0D 00 00 00 E                                C 00 F2 00 F1 7D 00 7D 00 (CRC=F8) #data=13
(06:35:12.562) Thermostat -> 0x08, type 0x14, telegram: 10 88 14 00 03 (CRC=6E)                                 #data=1
(06:35:12.585) 0x08 -> Thermostat, type 0x14, telegram: 08 10 14 00 17 7E EA (CR                                C=13) #data=3
(06:35:31.194) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 0C 64 03 (CRC=E1) #data=3
(06:35:32.568) Thermostat -> 0x08, type 0x1A, telegram: 10 08 1A 00 2E 64 64 00 (CRC=B9) #data=4
(06:35:33.208) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 0D 13 3B 13 03 01 (CRC=B9) #data=8
(06:35:36.689) Thermostat -> 0x08, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(06:35:37.852) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 02 39 7D 00 00 00 00 00 00 00 00 64 11 2E 00 (CRC=DA) #data=16
Publishing boiler data via MQTT
(06:35:42.102) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 02 15 00 EC 00 00 00 00 00 00 00 64 11 0C 00 (CRC=2D) #data=16
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Requesting scheduled EMS device data
(06:36:00.850) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 02 39 7D 00 00 00 00 00 00 00 00 64 11 2E 00 (CRC=B2) #data=16
(06:36:01.382) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2B 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=8F) #data=27
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(06:36:01.726) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 02 15 00 EC 00 00 00 00 00 00 00 64 11 0C 00 (CRC=45) #data=16
(06:36:02.132) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(06:36:02.601) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
(06:36:03.008) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(06:36:03.476) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
Publishing boiler data via MQTT
(06:36:03.883) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
Publishing boiler data via MQTT
(06:36:04.331) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 0D 13 3B 32 03 01 (CRC=9A) #data=8
(06:36:06.847) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 0D 00 00 00 EC 00 F2 00 F2 7D 00 7D 00 (CRC=C8) #data=13
```

**When I physically changed the temperature on the RC35 controller knob (from automatic DAY: 10.5 C to target 15C I get such telegrams. Perhaps now it will report setpoint temperature 15C. Can you check?**

```
(06:45:01.250) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2B 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=8F) #data=27
(06:45:01.718) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 12 1E 00 EC 00 00 00 00 00 00 00 64 11 10 00 (CRC=41) #data=16
(06:45:02.126) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(06:45:02.468) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
(06:45:03.000) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(06:45:03.344) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
Publishing boiler data via MQTT
(06:45:03.750) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
Publishing boiler data via MQTT
** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(06:45:04.201) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 0E 13 08 32 03 01 (CRC=7B) #data=8
(06:45:09.840) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 0D 00 00 00 EC 00 F2 00 F2 7D 00 7D 00 (CRC=C8) #data=13
(06:45:13.929) Thermostat -> 0x08, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(06:45:13.948) 0x08 -> Thermostat, type 0x14, telegram: 08 10 14 00 17 7E F4 (CRC=0D) #data=3
(06:45:24.839) Thermostat -> all, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99) #data=14
(06:45:33.311) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 10 64 03 (CRC=91) #data=3
[... cut ...]
When auto-night mode is on, setpoint temperature shows night temperature**.

**When we change the temperature in auto-day or auto-night mode it does not permanently change the set day or nigt temperatures only temporarily until the next switching point.
This is how it looks on the thermostat.**

**When I have the day temp set to 10.5 C and in the auto mode I change to 15 C, it will be changed only until the evening when rc35 switch to the auto-night mode with the set temperature, e.g. 9C. The next morning, the Auto-day temperature will still be  10.5C instead of 15C**

proddy, 2019-09-18:
how I read those values:

Setpoint temp = bytes 2, multiplied by 2
Current room temps = bytes 3 + 4, is multiplied by 10 

Auto (day)
            

```
HC1 = 10 00 3E 00  04 02 39 7D 00 00 00 00 00 00 00 00 64 11 2E 00, setpoint=28.5 current=n/a
HC2 = 10 0B 48 00  04 02 15 00 EC 00 00 00 00 00 00 00 64 11 0C 00, setpoint=10.5, current=23.6
```

Manual, 15 C

```
HC1 = 10 00 3E 00  04 02 39 7D 00 00 00 00 00 00 00 00 64 11 2E 00, setpoint=28.5 current=n/a
HC2 = 10 0B 48 00  04 12 1E 00 EC 00 00 00 00 00 00 00 64 11 10 00, setpoint=15 current=23.6
```

It should display these.

majdzik84, 2019-09-18:
You wrote in an earlier post that it does not display by telegram. Since then, I have reset the stove and the gate. For now i  can't check in telnet or webpanel if it displays correctly because thermostat stats nothing shows

proddy, 2019-09-18:
it should work. can you try again with the latest build? The web only shows HC1. I didn't implement multiple HCs in the web front end

majdzik84, 2019-09-18:
**Version 1.9.1b4**

[image]

Unfortunately the thermostat stats still show nothing

proddy, 2019-09-18:
strange. I'll do some more simulation and may send over a build that has more debugging in it.

proddy, 2019-09-18:
oops, was a bug. I forgot to set the Heating Circuit to be 'active'. Fixed now in dev, still called b4.

majdzik84, 2019-09-18:
Heating circuits have now appeared.
Note that current temperature is missing.

I only use two heating circuits (HC1, HC2)
The HC4 temperature is interesting.
[image]

The default circuit is HC1. How can I change to HC2 to be able to control correctly via Home assistant. Currently, when I try to change the mode it changes in HC1 and not in HC2

```
MQTT topic: thermostat HC1 mode value off
Setting thermostat mode to 0 for heating circuit
MQTT topic: thermostat HC1 mode value auto
Setting thermostat mode to 2 for heating circuit
```

proddy, 2019-09-18:
looks like I need to make a few more fixes, it looks terrible!

to control a HC via Home Assistant look at https://github.com/proddy/EMS-ESP/wiki/home-assistant

majdzik84, 2019-09-18:
Good luck. Fingers crossed.

By the way Home assistant.
You have to change the 'manual' mode not supported by climate to 'heat'

majdzik84, 2019-09-18:
Probably the control with HA circuits does not fully work.
I have configured the HC2 heating circuit.
[image]

This is what I get in mqtt explore
[image]

lmdecroos, 2019-09-18:
mine 1.9.1b4 has a similar output 

Thermostat stats:
  Thermostat: RC35 (ProductID:86 Version:21.08)
  Thermostat time is 22:02:42 18/09/2019
  Heating Circuit 1
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 2
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 3
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 4
   Setpoint room temperature: 14724.0 C
   Current room temperature: 1361.6 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night

It’s missing the current temperature..

(00:08:00.854) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
(00:08:02.261) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 00 C6 00 00 00 00 00 00 00 00 00 05 04 (CRC=C0) #data=16
(00:08:02.919) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=43) #data=27
(00:08:03.637) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 00 00 B4 00 00 00 00 00 00 00 00 00 05 04 (CRC=9D) #data=16
(00:08:03.849) Thermostat -> 0x11, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(00:08:04.918) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
(00:08:06.387) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F2) #data=16
(00:08:07.043) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=8A) #data=27
Publishing boiler data via MQTT
(00:08:07.762) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
(00:08:07.993) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 16 12 09 2B 02 01 (CRC=89) #data=8
(00:08:08.918) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=F8) #data=27
(00:08:09.492) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 16 12 09 2D 02 01 (CRC=36) #data=8
(00:08:26.355) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
(00:08:27.762) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 C6 00 00 00 00 00 00 00 00 00 05 04 (CRC=A8) #data=16
(00:08:30.849) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(00:08:47.388) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 00 00 B4 00 00 00 00 00 00 00 00 00 05 04 (CRC=F5) #data=16
(00:08:47.632) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 6E 00 00 00 C6 00 CC 00 CC 7D 00 7D 00 (CRC=BE) #data=13
Requesting scheduled EMS device data
(00:09:00.978) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
(00:09:02.510) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 00 C6 00 00 00 00 00 00 00 00 00 05 04 (CRC=C0) #data=16
(00:09:02.917) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=43) #data=27
Publishing boiler data via MQTT
(00:09:03.635) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 00 00 B4 00 00 00 00 00 00 00 00 00 05 04 (CRC=9D) #data=16
(00:09:04.417) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
(00:09:04.473) Thermostat -> 0x11, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(00:09:05.761) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F2) #data=16
(00:09:06.792) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=8A) #data=27
Publishing boiler data via MQTT
(00:09:07.386) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
(00:09:08.042) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=F8) #data=27
(00:09:08.242) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 16 12 0A 2B 02 01 (CRC=91) #data=8
(00:09:08.867) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 16 12 0A 2C 02 01 (CRC=2A) #data=8
(00:09:20.346) Thermostat -> Boiler, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(00:09:20.379) Boiler -> Thermostat, type 0x14, telegram: 08 10 14 00 32 C0 79 (CRC=71) #data=3
(00:09:26.479) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
(00:09:28.136) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 C6 00 00 00 00 00 00 00 00 00 05 04 (CRC=A8) #data=16
(00:09:31.223) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(00:09:48.136) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 00 00 B4 00 00 00 00 00 00 00 00 00 05 04 (CRC=F5) #data=16
(00:09:48.381) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 6E 00 00 00 C6 00 CC

proddy, 2019-09-19:
Think I fixed the temp values that but will double check before committing to the dev branch.

proddy, 2019-09-19:
can you try the 1.9.1b5?

majdzik84, 2019-09-19:
current temperature is still missing
[image]

I also see that the thermostat_data topic is missing in mqtt
[image]

```
(00:14:23.734) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 04 00 00 00 F5 00 FB 00 FB 7D 00 7D 00 (CRC=D7) #data=13
(00:14:35.323) Thermostat -> Boiler, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(00:14:35.353) Boiler -> Thermostat, type 0x14, telegram: 08 10 14 00 17 86 F9 (CRC=E9) #data=3
(00:14:39.594) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 00 15 04 0F 05 01 (CRC=FD) #data=8
** Warning, we missed the bus - Rx non-idle!
(00:14:40.325) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(00:14:41.580) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 2C 64 64 00 (CRC=A9) #data=4
(00:14:41.955) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 64 03 (CRC=C5) #data=3
** Warning, we missed the bus - Rx non-idle!
(00:14:55.738) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 2D 7D 00 00 00 00 00 00 00 00 64 00 2C 00 (CRC=9A) #data=16
(00:14:55.989) Thermostat -> all, type 0x48, telegram: 10 00 48 00 44 00 13 00 F5 00 00 00 00 00 00 00 00 00 05 00 (CRC=77) #data=16
Publishing boiler data via MQTT
Requesting scheduled EMS device data
(00:15:01.986) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 2D 7D 00 00 00 00 00 00 00 00 64 00 2C 00 (CRC=F2) #data=16
(00:15:02.143) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 00 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=BD) #data=27
(00:15:02.862) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 44 00 13 00 F5 00 00 00 00 00 00 00 00 00 05 00 (CRC=1F) #data=16
(00:15:03.519) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(00:15:03.987) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F2) #data=16
** Warning, we missed the bus - Rx non-idle!
(00:15:04.393) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(00:15:04.862) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
(00:15:05.269) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
(00:15:05.717) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 00 15 04 29 05 01 (CRC=C2) #data=8
(00:15:09.324) Thermostat -> 0x09, type 0x29, telegram: 10 89 29 00 01 (CRC=90) #data=1
(00:15:09.344) 0x09 -> Thermostat, type 0x29, telegram: 09 10 29 00 7B (CRC=CF) #data=1
** Warning, we missed the bus - Rx non-idle!
(00:15:24.233) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 04 00 00 00 F5 00 FB 00 FB 7D 00 7D 00 (CRC=D7) #data=13
** Warning, we missed the bus - Rx non-idle!
(00:15:39.843) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 00 15 05 0F 05 01 (CRC=F5) #data=8
(00:15:40.575) Thermostat -> Boiler, type 0x35, telegram: 10 08 35 00 01 01 (CRC=00) #data=2
(00:15:41.704) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 2C 64 64 00 (CRC=A9) #data=4
(00:15:42.205) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 64 03 (CRC=C5) #data=3
(00:15:55.987) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 2D 7D 00 00 00 00 00 00 00 00 64 00 2C 00 (CRC=9A) #data=16
(00:15:56.238) Thermostat -> all, type 0x48, telegram: 10 00 48 00 44 00 13 00 F5 00 00 00 00 00 00 00 00 00 05 00 (CRC=77) #data=16
Publishing boiler data via MQTT
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Requesting scheduled EMS device data
(00:16:00.861) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 2D 7D 00 00 00 00 00 00 00 00 64 00 2C 00 (CRC=F2) #data=16
(00:16:01.393) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 00 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=BD) #data=27
(00:16:01.861) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 44 00 13 00 F5 00 00 00 00 00 00 00 00 00 05 00 (CRC=1F) #data=16
(00:16:02.268) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 13 15 26 0A 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=FD) #data=27
(00:16:03.111) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F2) #data=16
(00:16:03.519) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(00:16:03.986) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
(00:16:04.393) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
(00:16:04.842) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 00 15 05 28 05 01 (CRC=CE) #data=8
Publishing boiler data via MQTT
** Warning, we missed the bus - Rx non-idle!
(00:16:24.482) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 04 00 00 00 F5 00 FB 00 FB 7D 00 7D 00 (CRC=D7) #data=13
(00:16:34.731) Thermostat -> all, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99) #data=14
[... cut ...]
(00:18:03.141) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=1E) #data=27
(00:18:03.485) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
(00:18:04.025) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 13 15 26 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=6C) #data=27
(00:18:04.465) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 00 15 07 27 05 01 (CRC=E2) #data=8
```

majdzik84, 2019-09-23:
I also noticed that the auto temperature correction only works for auto-night. In auto-day shows 0.0 C

[image]

proddy, 2019-09-24:
I'm constantly making small changes. Is this with the latest dev build?

majdzik84, 2019-09-24:
I tested on the version 1.9.1b6.
When trying to install version 1.9.1b7 I have an error
```
Linking .pio\build\debug\EMS-ESP-1.9.1b7
c:/users/piotrek/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file local.eagle.app.v6.common.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\debug\EMS-ESP-1.9.1b7] Error 1
```

proddy, 2019-09-24:
the error was introduced yesterday when I made a change to the python script that renames the firmware filename - apparently, all the dots didn't go well with Linux. You should be able to build again.

The warning with OneWire you can ignore - it's not my library.

majdzik84, 2019-09-24:
After uploading version 1.9.1b7, the temperatures I wrote about earlier appeared correctly.
After some time, again in Auto-Day mode is (setpoint temp) 0.0C.

I tried to upload the firmware again, reset the stove, but still shows 0.0C

Current room temperature shows correctly

proddy, 2019-09-24:
can you send the telegrams when in this auto-day mode from types 3E and 48 so I can see what is coming in. You can use the `thermostat read` command

majdzik84, 2019-09-24:
I am sending an auto-night telegram because it is now also 0.0C.
In the morning I will post an auto-day.
[image]

```

(10:24:52.970) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 00 00 7D 0                          0 00 00 00 00 00 00 00 00 00 05 00 (CRC=75) #data=16
(10:25:01.968) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 00 00 7D 00                           00 00 00 00 00 00 00 00 00 05 00 (CRC=1D) #data=16
(10:25:02.844) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 00 10 00 E5                           00 00 00 00 00 00 00 00 00 05 00 (CRC=CD) #data=16
(10:25:06.720) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 00 10 00 E                          5 00 00 00 00 00 00 00 00 00 05 00 (CRC=A5) #data=16
(10:25:53.344) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 00 00 7D 0                          0 00 00 00 00 00 00 00 00 00 05 00 (CRC=75) #data=16
(10:26:00.718) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 00 00 7D 00                           00 00 00 00 00 00 00 00 00 05 00 (CRC=1D) #data=16
(10:26:07.344) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 00 10 00 E                          5 00 00 00 00 00 00 00 00 00 05 00 (CRC=A5) #data=16
(10:27:07.843) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 00 10 00 E                          5 00 00 00 00 00 00 00 00 00 05 00 (CRC=A5) #data=16
(10:27:54.217) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 00 00 7D 0                          0 00 00 00 00 00 00 00 00 00 05 00 (CRC=75) #data=16
(10:28:02.716) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 00 00 7D 00                           00 
(10:28:03.966) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 00 10 00 E5                           00 00 00 00 00 00 00 00 00 05 00 (CRC=CD) #data=16
(10:28:08.218) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 00 10 00 E5 00 00 00 00 00 00 00 00 00 05 00 (CRC=A5) #data=16

```

majdzik84, 2019-09-25:
unfortunately the bug has reappeared.
This time in NIGHT mode

[image]
**
```
(00:58:10.354) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
 (00:58:11.354) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=24) #data=16
(00:58:33.605) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=4C) #data=16
(00:58:28.480) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(00:59:28.729) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(01:01:29.478) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(01:01:34.353) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=4C) #data=16

```
After switching to DAY mode, setpoint temp shows temperature day.
Return to night mode does not change - the day temperature is still shown
[image]

```
(01:11:11.218) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(01:11:11.749) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27
(01:11:12.218) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=EC) #data=16
(01:11:12.625) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 10 16 26 0A 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=12) #data=27
(01:11:33.343) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
Publishing boiler data via MQTT
(01:11:37.719) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=84) #data=16
(01:12:12.092) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(01:12:13.092) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=EC) #data=16

```

proddy, 2019-09-25:
Ok - I have the data so I'll see what is happening.

proddy, 2019-09-27:
I see the problem - not sure how to fix it. Deciphering the telegrams is hard as some of the data is dependent on the state its in. I'll do a new build for you to test. If you get on gitter we can debug real time.

majdzik84, 2019-09-27:
OK

proddy, 2019-09-27:
I added some thoughts to the wiki [here](https://github.com/proddy/EMS-ESP/wiki/EMS-Telegram-Types#telegram-type--3e-48-52-5c). Once I know how to decipher these telegrams I can implement it correctly. Could you help?

proddy, 2019-09-27:
I think I found what is wrong. I'll work on a test build

majdzik84, 2019-09-27:
Great . I was preparing psychologically to decipher the telegrams :)

proddy, 2019-09-27:
my thinking is that I use the 3E monitor telegram for only reading the current room temperature from the thermostat. And use the 3D working mode telegram for bringing back the setpoint temps for night, day and holiday modes.

I'm just not sure about what happens when you manually override and set a temp. You'll need to try the latest build and show me some 3E/3D messages on HC1 so I can see.

majdzik84, 2019-09-27:
After uploading the latest compilation I can no longer see setpoint temperature.
I see "Program is set to summer mode.
[image]
```
I use HC2

(00:12:06.749) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 0                          0 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(00:14:09.996) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(00:14:10.904) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27
(00:14:11.371) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 03 00 00 E4 00 00 00 00 00 00 00 00 11 05 00 (CRC=CB) #data=16

(00:14:09.996) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(00:14:10.904) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27
(00:14:11.371) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 03 00 00 E4 00 00 00 00 00 00 00 00 11 05 00 (CRC=CB) #data=16

```
After setting the temperature manually no rc35 to 14C

```
(00:02:01.808) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 13 1C 00 E4 00 00 00 00 00 00 00 64 11 0F 00 (CRC=01) #data=16
(00:02:02.305) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(00:02:02.463) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27

(00:03:52.305) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16

(00:01:02.598) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
Publishing boiler data via MQTT
(00:01:02.879) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27
(00:01:03.473) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 13 1C 00 E5 00 00 00 00 00 00 00 64 11 0F 00 (CRC=A1) #data=16

```

proddy, 2019-09-27:
yes I extended to add the mode to summer, holiday etc. No idea if it works since I don't own a RC35 - does that make sense? Is program the correct naming for it?

As for the setpoint, I found no reliable way to determine this, and then thought since there are already 3 setpoint defined (day, night, vacation) just use them. Does this make sense too?

proddy, 2019-09-27:
can you tell me what the values for setpoint and current room temps should be for HC1 and HC2? Its hard to figure out your exact environment. What do you expect to see?

majdzik84, 2019-09-27:
It seems to me that the summer / winter operation mode will not work in this situation.

What I would like to get is for sure the Setpoint temperature value for Home Assistant to display it on the thermostat.

I'm only interested in the HC2 circuit because it supports the floor installation. HC1 supports standard heaters where I adjust the thermostatic heads.

On RC35 you can set the temperature:
Day, Night and in automatic mode it switches by itself.
The setpoint temperature at a given moment is Day or Night.
When we change the temperature in automatic mode (regardless of whether auto-day or auto-night) it only changes in setpoint temperature (not in day-temp / night-temp).
When switching to the next time of the auto mode (i.e. from auto-day to auto-night or vice versa), the time set by us is deleted and the day or night temperature is displayed.

Example - my HC2:
**Standard Auto-DAY**
Setpoint room temperature: 20 C
Current room temperature: 22 C
Day temperature: 20 C
Night temperature: 15 C

**When I change the temperature value to 25 C i get**

Setpoint room temperature: **25 C**
Current room temperature: 22 C
Day temperature: 20 C
Night temperature: 15 C

**Setpoint temperature will be valid until the next switching point to auto-night**

Setpoint room temperature: **15 C**
Current room temperature: 22 C
Day temperature: 20 C
Night temperature: 15 C

the pre-set temperature of 25C is deleted (it does not permanently change any set value such as day-temp, night-temp)

proddy, 2019-09-28:
its still a puzzle to m. I reverted the last changes and made it so it only responds to broadcasts.

Some of the telegrams you posted still confuse me, for example
`Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00` which basically says HC1 has a setpoint temp of 0.0C and no room temp available

and

`Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 03 00 00 E4 00 00 00 00 00 00 00 00 11 05 00` which says HC2 has a setpoint temp of 0.0 C and a room temp of 22.8 C

it's  baffling me!

majdzik84, 2019-09-28:
I don't think you should look at my HC1. It is not controlled by RC35 only by thermostats on the radiator (only the heating curve is set).So it is not surprising that you cannot see the current temperature.

I'm just thinking. Are the values ​​at the set temperature correct, even 0.0C in automatic mode?

Look:
- In the DAY manual mode we get the Setpoint temperature value, which is the set temperature for the day

- In the manual mode NIGHT we get the Setpoint temperature value, which is the set temperature for the night

- In AUTO mode, we get 0.0 C (because the auto mode simply switches day and night temperatures) and the Setpoint temperature parameter will change only when a different temperature is specified than in the (auto-day / auto-night) mode.

We have 3 variables at the moment.
1. Day temp.
2. Night temp.
3. Auto temp. (let's take that name)

` 0x48, telegram: 10 00 48 00 04 13 1C 00 E4 00 00 00 00 00 00 00 64 11 0F 00 (CRC=01) #data=16`
Does this telegram contain setpoint temperature 14C?

Could it be possible to make dependencies in the auto mode, e.g.
If setpoint temp. = 0.0C show temperature day or night.
If setpoint temp > 0.0C show auto temp.

In this case, the setpoint temperature would always have a value. In the home assistant on the thermostat the set temperature would be visible.

majdzik84, 2019-09-30:
[image]
**AUTO -DAY**
```
(00:03:14.981) Thermostat -> all, type 0x48, telegram: 10 00 48 00 04 03 00 00 EC 0       0 00 00 00 00 00 00 00 11 05 00 (CRC=B5) #data=16

(00:04:11.355) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 03 00 00 EC 00 00 00 00 00 00 00 00 11 05 00 (CRC=DD) #data=16

(00:06:10.259) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27

```

**After setting the temperature manually no rc35 to 14C** - 

[image]
```
(00:08:12.601) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 04 13 1C 00 EC 00 00 00 00 00 00 00 64 11 0F 00 (CRC=7F) #data=16
(00:08:12.007) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27

```

[46 more replies]

closed 2019-11-13 by proddy