emsesp/EMS-ESP #163: Wrong thermostat temperature

davidw3591, 2019-07-24:
Hi,

Using version 1.6.0 for a while nowe without any issues.

This week we are having a heatwave in Belgium. The thermostat (RC35) is now showing 26.2°C but EMS ESP is reading this as 0.6°C.
Last good reading in HomeAssistant was 25.3°C and then it dropped to 0°C with in the last 2 hours an incline to 0.6 and rising.

[image]

Any thoughts?

David

proddy, 2019-07-24:
oh, that's strange. I checked my RC20 and its showing 26.7°C but using the 1.8.1 dev build.

Could do a `log v` and then `thermostat read 3e` and `thermostat read 48` so I can see the values of the data coming back and turn this into a test case in the code.

reenx, 2019-07-24:
I have a similar problem. 
The thermostat display says 26.0 degrees.
The info page of ems-esp says 0.5 degrees.
Here is the contents of telegram 0x91. (that should provide the temperature data right?) 
Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 0E 01 05 00 00 00 00 00 05 08 01 05 00 (CRC=34), #data=14

(I am still on version 1.6.0)

davidw3591, 2019-07-24:
> oh, that's strange. I checked my RC20 and its showing 26.7°C but using the 1.8.1 dev build.
> 
> Could do a `log v` and then `thermostat read 3e` and `thermostat read 48` so I can see the values of the data coming back and turn this into a test case in the code.

Here is the output of the suggested commands:

```
(01:58:18.839) Sending read of type 0x3E to 0x10: telegram: 0B 90 3E 00 20 (CRC=8C), #data=1
(01:58:18.953) Thermostat -> me, type 0x3E telegram: 10 0B 3E 00 80 02 1E 01 08 00 00 00 00 00 00 00 00 11 05 00 (CRC=75), #data=16

(01:58:43.712) Sending read of type 0x48 to 0x10: telegram: 0B 90 48 00 20 (CRC=4D), #data=1
(01:58:43.828) Thermostat -> me, type 0x48 telegram: 10 0B 48 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=D1), #data=16
```

susisstrolch, 2019-07-24:
The 1.6 is a ‚little‘ bit outdated. The 1.8 (Not speaking about dev branch) shouldn‘t show those issues. What are the reasons you can‘t update?

reenx, 2019-07-24:
The reason for not updating was time. :)
I just updated to latest master and now it shows 0.6 degrees. (thermostat still says 26.0).
The telegram 0x91 is now: 
Thermostat -> me, type 0x91 telegram: 17 0B 91 00 80 0E 01 06 00 00 00 00 00 05 08 01 06 00 (CRC=9E) #data=14

reenx, 2019-07-24:
some further testing:
master build -> current temperature 0.6 degrees.
1.8.1-beta -> could not read EMS bus.
dev -> could not read EMS bus.

reenx, 2019-07-24:
Looking at the code I noticed the following macro:
#define _toShort(i) ((EMS_RxTelegram->data[i] << 8) + EMS_RxTelegram->data[i + 1])
This will leftshift the first byte 8 bits. But data[i] is a byte so it will always be 0 after the leftshift.
In effect all these macros end up with only the last 8 bits. 
I will see if i can find a better macro. But I think all toShort and toLong macros do not work as intended.

proddy, 2019-07-25:
in 1.8.1 all temps were changed to be 2-byte signed shorts as opposed to single bytes so this issue would be fixed. However its still in beta while we get the Tx working.

If you want to try 1.8.1 and get the 'EMS bus error' play around with the different tx_mode settings. A value of 0 is the old way it worked in 1.6.0/1.7.0 and 1 is for EMS+ and 3 for Junkers.

reenx, 2019-07-25:
I checked the dev branch. I tried all tree tx_mode settings. (rebooted the device after every setting). But all 4 (0,1,2,3) were not able to connect to the EMS bus.
I searched the other issues and saw something about set serial off. I tried that and now I can connect to the EMS bus. And the temperature is correct. (25.7 currently, but we expect new record temperatures this afternoon. :) )
Perhaps set serial default off?

proddy, 2019-07-25:
thanks for testing the dev build. I've noticed reports of serial being set to on when moving from earlier versions - I'll fix that.

jcmelati, 2019-07-25:
I can confirm the reading is not correct.

`Thermostat -> all, type 0xB1 telegram: 17 00 B1 00 04 20 01 52 00 00 01 52 (CRC=C6) #data=8
`
This is giving 8.2 but the correct is around 34.

proddy, 2019-07-25:
@jcmelati wow thats hot. I merged the fix into 1.8.0 and it was already solved in 1.8.1

jcmelati, 2019-07-25:
Hi!
I updated the firmware to 1.8.1b21 and now it can't seem to reach my heater.
Do you think it might be related to the updated calculation?

Should I downgrade it again?
Thanks! 

```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  No connection can be made to the EMS bus

Boiler stats:
  Boiler: DeviceID: 0x08 (ProductID:? Version:?) 
```

davidw3591, 2019-07-25:
Hi,

I updated to 1.8.1b21 as well. Had some issies with crashing during config, had to detach the ESP from the ESP board and send command "set serial off".

Now I'm receiving the right thermostat in HASS, but... I'm having a lot of these errors:
`** We missed the bus - Rx non-idle!`

And I'm unable to control the thermostat. MQTT command get received, it says it's changing temp or mode, but it doesn't happen.

I seem to be missing some parameters:
* pumpMod
* boilTemp
[image]

proddy, 2019-07-25:
Check the tx_mode setting. The '2' value is still experimental so if you're seeing ems bus issues go back to using 0 or 1 for ems+ and 3 for Junkers.

jcmelati, 2019-07-25:
I am running  1.8.1b21.
Setting serial=off seems to make it able to receive data.

tx_mode doesnt change the reading.
It is still reading wrong on my heater.

```
  Thermostat: RC10/Nefit Moduline 100 (ProductID:79 Version:01.02)
  Setpoint room temperature: 16.0 C
  Current room temperature: 7.5 C
```

proddy, 2019-07-26:
@jcmelati I think I might have a fix for this in 1.8.1b23 if you wouldn't mind testing?

jcmelati, 2019-07-26:
Updated to 1.8.1b23 and the temperature now is correct. 
I also noticed a message: _** We missed the bus - Rx non-idle!_ and the tx_mode was set to 1
It seems to be gone after changing it to 0 but I will keep watching.

Thanks again!

jcmelati, 2019-07-26:
Still getting the same message despite changing tx_mode
** We missed the bus - Rx non-idle!

Might be a different issue, tho.

susisstrolch, 2019-07-26:
That's a protocol handling problem - EMS bus master is sending data when we don't expect it.
Still investigating the reason...

proddy, 2019-07-27:
wrong temp issue is solved in b23 so closing this

closed 2019-07-27 by proddy