emsesp/EMS-ESP #165: Telnet session stops automatically v1.8.1

mf01, 2019-07-30:
Using telnet the session stops automatically after a short time. ~1min
Reconnect is possible directly. 

Any hint?

proddy, 2019-07-30:
Did the same problem happen in 1.8.0 do you know? Is this with verbose logging too? Most likely a memory leak somewhere.

mf01, 2019-07-30:
Not sure. I was using some beta versions of 1.8.1 before. If I remember right maybe 1.8.1b21 was affected by this issue, too. Because of beta I haven't reported this. Sorry for that. Maybe the issue started before b21. I don't know exactly.

mf01, 2019-07-30:
Maybe this can help you....

Tried to start telnet. It was not working. pressed Ctrl+C. Then again and it works.
All devices are new scanned and added. Maybe this looks like lost device configuration? 

```
markus@*******:~/projects/ems-esp$ !telnet
telnet ems-esp
Trying *******...

^C
markus@*******:~/projects/ems-esp$ telnet ems-esp
Trying *******...
Connected to ems-esp.fritz.box.
Escape character is '^]'.
S

* EMS-ESP version 1.8.1
[FS] Settings loaded from SPIFFS
[WEB] Web server started
[UART] Opened Rx/Tx connection
Starting auto discover of EMS devices...
[WIFI] Connecting to SSID: *******
[WIFI] SSID  *******
[WIFI] CH    11
[WIFI] RSSI  -39
[WIFI] IP    *******
[WIFI] MAC   *******
[WIFI] GW    *******
[WIFI] MASK  255.255.255.0
[WIFI] DNS   *******
[WIFI] HOST  ems-esp
[OTA] listening to ems-esp.local:8266
[SYSTEM] Disabling serial port communication.
[MQTT] Connecting to MQTT...
[MQTT] Connected
[MQTT] Subscribing to home/ems-esp/restart (PID 2)
[MQTT] Subscribing to home/ems-esp/start (PID 3)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_temp (PID 5)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_mode (PID 6)
[MQTT] Subscribing to home/ems-esp/wwactivated (PID 7)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_wwtemp (PID 8)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_comfort (PID 9)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_flowtemp (PID 10)
[MQTT] Subscribing to home/ems-esp/shower_timer (PID 11)
[MQTT] Subscribing to home/ems-esp/shower_alert (PID 12)
[MQTT] Subscribing to home/ems-esp/shower_coldshot (PID 13)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_hc (PID 14)
[MQTT] Subscribing to home/ems-esp/thermostat_daytemp (PID 15)
[MQTT] Subscribing to home/ems-esp/thermostat_nighttemp (PID 16)
[MQTT] Subscribing to home/ems-esp/thermostat_holidayttemp (PID 17)
[MQTT] Received boottime: start
[TELNET] Telnet connection established

* Connected to: EMS-ESP version 1.8.1
* Hostname: ems-esp (*******)
* WiFi SSID: ******* (signal 100%)
* MQTT connected (heartbeat enabled)
*
* Commands:
*  ?=help, CTRL-D/quit=exit telnet session
*  set, system, reboot
*  info                                         show current captured on the devices
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  devices                                      list all supported and detected EMS devices and types IDs
*  queue                                        show current Tx queue
*  autodetect [deep]                            detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    probe thermostat on all type id responses
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
*  boiler flowtemp <degrees>                    set boiler flow temperature
*  boiler comfort <hot | eco | intelligent>     set boiler warm water comfort setting

System Logging set to None
Boiler found: Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.08)
* Setting Boiler to model Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.08)
Solar Module found: SM100 Solar Module (DeviceID:0x30 ProductID:163 Version:21.03)
Solar Module support enabled.
* Setting Thermostat to RC300/RC310/Nefit Moduline 3000 (DeviceID:0x10 ProductID:158 Version:11.04)

[some time waiting. then pressing enter or any command .... ]

Connection closed by foreign host.

```

proddy, 2019-07-30:
Ok, I'll see if I can reproduce. The telnet will be replaced with a web interface in version 1.9. It's only really intended to set values (once) and take a look at values. It wasn't designed for heavy production use. The EMS-ESP should still work without it.

susisstrolch, 2019-07-30:
I really use it for debug/tortoure with a simple bash script:
```
Strolchi:~/Development/esp8266/EMS-ESP-Boiler$ cat monitor
#!/bin/bash
COUNT=0

while true; do
LOGFILE="emsbus-$(date +'%Y%m%d-%H%M%S').log"
touch $LOGFILE
ln -sf $LOGFILE emsbus.log
echo "monitor loop $COUNT - $LOGFILE"
cat <<EOC|nc -w 30 192.168.254.81 telnet >$LOGFILE
log j
EOC
stripansi -i $LOGFILE
let "COUNT++"
done
```

The uptime is between 1min and 8h…
```
emsbus-20190729-143637.log  emsbus-20190729-213059.log  emsbus-20190730-082217.log
emsbus-20190729-184341.log  emsbus-20190729-235407.log  emsbus-20190730-170345.log
emsbus-20190729-184411.log  emsbus-20190730-001012.log  emsbus.log
emsbus-20190729-211942.log  emsbus-20190730-061742.log  emsbus-20190730-170454.log
emsbus-20190729-212845.log  emsbus-20190730-074612.log
emsbus-20190729-213029.log  emsbus-20190730-080932.log
```
So I suspect the WIFI...

susisstrolch, 2019-07-30:
And about tortoure - here‘s an excerpt from the log…
```
000081] 

* EMS-ESP version 1.9.0
[000152] [FS] Settings loaded from SPIFFS
[000154] [WEB] Web server started
[000165] [UART] Opened Rx/Tx connection
[000165] Starting auto discover of EMS devices...
[000172] [WIFI] Connecting to SSID: SusiconStrolch
[003973] [WIFI] SSID  SusiconStrolch
[003973] [WIFI] CH    4
[003973] [WIFI] RSSI  -72
[003974] [WIFI] IP    192.168.254.81
[003974] [WIFI] MAC   18:FE:34:F3:BF:0F
[004016] [WIFI] GW    192.168.254.1
[004054] [WIFI] MASK  255.255.255.0
[004093] [WIFI] DNS   192.168.254.1
[004131] [WIFI] HOST  ems-esp
[004166] [OTA] listening to ems-esp.local:8266
[004214] [SYSTEM] Disabling serial port communication.
[004405] [MQTT] Connecting to MQTT...
[007154] [MQTT] Connected
[007155] [MQTT] Subscribing to home/ems-esp/restart (PID 2)
[007156] [MQTT] Subscribing to home/ems-esp/start (PID 3)
[007156] [MQTT] Subscribing to home/ems-esp/thermostat_cmd_temp (PID 5)
[007156] [MQTT] Subscribing to home/ems-esp/thermostat_cmd_mode (PID 6)
[007157] [MQTT] Subscribing to home/ems-esp/wwactivated (PID 7)
[007157] [MQTT] Subscribing to home/ems-esp/boiler_cmd_wwtemp (PID 8)
[007157] [MQTT] Subscribing to home/ems-esp/boiler_cmd_comfort (PID 9)
[007158] [MQTT] Subscribing to home/ems-esp/boiler_cmd_flowtemp (PID 10)
[007158] [MQTT] Subscribing to home/ems-esp/shower_timer (PID 11)
[007158] [MQTT] Subscribing to home/ems-esp/shower_alert (PID 12)
[007159] [MQTT] Subscribing to home/ems-esp/shower_coldshot (PID 13)
[007159] [MQTT] Subscribing to home/ems-esp/thermostat_cmd_hc (PID 14)
[007159] [MQTT] Subscribing to home/ems-esp/thermostat_daytemp (PID 15)
[007159] [MQTT] Subscribing to home/ems-esp/thermostat_nighttemp (PID 16)
[007160] [MQTT] Subscribing to home/ems-esp/thermostat_holidayttemp (PID 17)
[011822] Boiler found: MC10 Module (DeviceID:0x08 ProductID:72 Version:02.07)
[011822] * Setting Boiler to model MC10 Module (DeviceID:0x08 ProductID:72 Version:02.07)
[021119] * Setting Thermostat to RC35 (DeviceID:0x10 ProductID:86 Version:01.11)
[022985] [TELNET] Telnet connection established
[022986]
[022986] * Connected to: EMS-ESP version 1.9.0
[022987] * Hostname: ems-esp (192.168.254.81)
[022987] * WiFi SSID: SusiconStrolch (signal 42%)
[022988] * MQTT connected (heartbeat enabled)
[022988] *
[022988] * Commands:
[022988] *  ?=help, CTRL-D/quit=exit telnet session
[022988] *  set, system, reboot
[022988] *  crash <dump | clear | test [n]>
*  info                                         show current captured on the devices
*  log <n | b | t | r | v>                      set logging mode to none, basic, thermostat only, raw or verbose
*  publish                                      publish all values to MQTT
*  refresh                                      fetch values from the EMS devices
*  devices                                      list all supported and detected EMS devices and types IDs
*  queue                                        show current Tx queue
*  autodetect [deep]                            detect EMS devices and attempt to automatically set boiler and thermostat types
*  shower <timer | alert>                       toggle either timer or alert on/off
*  send XX ...                                  send raw telegram data as hex to EMS bus
*  thermostat read <type ID>                    send read request to the thermostat
*  thermostat temp <degrees>                    set current thermostat temperature
*  thermostat mode <mode>                       set mode (0=low/night, 1=manual/day, 2=auto)
*  thermostat scan <type ID>                    probe thermostat on all type id responses
*  boiler read <type ID>                        send read request to boiler
*  boiler wwtemp <degrees>                      set boiler warm water temperature
*  boiler tapwater <on | off>                   set boiler warm tap water on/off
*  boiler flowtemp <degrees>                    set boiler flow temperature
*  boiler comfort <hot | eco | intelligent>     set boiler warm water comfort setting
[... 8 lines cut ...]
(00:00:23.478) ems_parseTelegram: 00 01: 09
(00:00:23.483) ems_parseTelegram: 00 01: 8C
(00:00:23.513) ems_parseTelegram: 00 01: 89
(00:00:23.519) ems_parseTelegram: 00 01: 09
(00:00:23.523) ems_parseTelegram: 00 01: B5
(00:00:23.553) ems_parseTelegram: 00 01: 89
(00:00:23.560) ems_parseTelegram: 00 01: 09
(00:00:23.563) ems_parseTelegram: 00 01: 8B
(00:00:23.563) emsuart_tx_buffer: 00 01: 0B 90 3E 00 20 8C
(00:00:23.579) ems_parseTelegram: 00 02: 0B 90 3E 00 20 8C
(00:00:23.580) echo:telegram: 0B 90 3E 00 20 (CRC=8C) #data=1
(00:00:23.649) ems_parseTelegram: 00 02: 10 0B 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 00 05 00 AD
(00:00:23.650) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 00 7D 00 00 00 00 00 00 00 00 00 00 05 00 (CRC=AD) #data=16
(00:00:23.651) emsuart_tx_buffer: 00 01: 0B
(00:00:23.656) ems_parseTelegram: 00 01: 0B
Publishing hot water and heating states via MQTT
(00:00:23.825) ems_parseTelegram: 00 01: 8B
(00:00:23.826) emsuart_tx_buffer: 00 01: 0B 90 3D 00 20 80
(00:00:23.842) ems_parseTelegram: 00 02: 0B 90 3D 00 20 80
(00:00:23.842) echo:telegram: 0B 90 3D 00 20 (CRC=80) #data=1
(00:00:23.931) ems_parseTelegram: 00 02: 10 0B 3D 00 01 24 2C 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4D 01 00 3C FF 10 05 05 03 02 BF
(00:00:23.933) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 24 2C 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4D 01 00 3C FF 10 05 05 03 02 (CRC=BF) #data=27
(00:00:23.934) emsuart_tx_buffer: 00 01: 0B
(00:00:23.939) ems_parseTelegram: 00 01: 0B
```

Sonusss, 2019-08-05:
Hi,
Maybe another clue, starting a telnet session and typing system as the first command always creates a reboot other commands are fine.  
Running 1.8.1

proddy, 2019-08-17:
Can you test with 1.9 in dev?

proddy, 2019-08-27:
should not happen in 1.9. And less important as Telnet was designed for changing settings and not as a continuous monitor.

closed 2019-08-27 by proddy