emsesp/EMS-ESP #167: Led remains on after reboot

mrfixit1, 2019-08-09:
Issue: after every reboot, both soft and hard, the blue led on the wemos board remains on, even though both the info and set commands say that it is off.

Workaround: manual set led off command. 

Setup: v1.8.1 on wemos d1 mini, bbqkees interface board, Nefit Topline Compact boiler, RC20 thermostat.

proddy, 2019-08-09:
Thanks. I’ll look into that. Is the value of led_gpio set to 2 ? (which is LED_BULLETIN  when Pio uses the board d1_mini)

mrfixit1, 2019-08-09:
Yes. Btw stating the obvious: the issue has zero functional impact, so fixing it isn't urgent.

proddy, 2019-08-10:
fixed, in 1.8.2 and the upcoming 1.9. Thanks!

closed 2019-08-10 by proddy

mrfixit1, 2019-08-11:
Thanks Paul: updated to 1.8.2 and fix verified, very quick work!