emsesp/EMS-ESP #169: Logamatic TC100/Nefit not recognized in 1.8.2

Sonusss, 2019-08-11:
Hi,
I'm testing release 1.8.2 and my TC100 thermostat is not recognized anymore while it was working with 1.8.1.  

With 1.8.2 it is stable now (do not reboot constantly every 10 minutes as with 1.8.1)

Might all this be related to TC100 ?

proddy, 2019-08-11:
Was it detected in 1.8.1?

proddy, 2019-08-11:
Actually just read that it worked in 1.8.1. Sry. I’ll take a look at what could cause this. So auto detect doesn’t find the TC100?

Sonusss, 2019-08-11:
This is what i got from log...maybe relevant
Seems that telegram from Thermostat are well seen
But I'm having this also:
<--- UBAMonitorSlow(0x19)
(01:02:10.561) Sending read of type 0x33 to 0x08:telegram: 0B 88 33 00 20 (CRC=78) #data=25
(01:02:10.571) **Corrupt telegram: 0B CC 33 00 20 78**
(01:02:11.700) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(01:02:14.408) Sending read of type 0x16 to 0x08:telegram: 0B 88 16 00 20 (CRC=EC) #data=164
(01:02:14.425) **Corrupt telegram: 0B 88 16 00 20**

<--- UBAMonitorFast(0x18)
(01:01:49.012) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 01 44 00 00 80 00 00 80 00 80 00 80 00 04 (CRC=C0) #data=21
(01:01:53.472) Thermostat -> Boiler, type 0x16, telegram: 18 88 16 01 02 (CRC=E5) #data=1
(01:01:53.488) Boiler -> Thermostat, type 0x16, telegram: 08 18 16 01 46 64 (CRC=DC) #data=2
(01:01:53.668) Thermostat -> Boiler, type 0x06, telegram: 18 08 06 00 13 08 0D 0B 05 0A 06 01 (CRC=FA) #data=8
(01:01:54.004) Thermostat -> Boiler, type 0x1C, telegram: 18 88 1C 00 08 (CRC=C5) #data=1
(01:01:54.039) Boiler -> Thermostat, type 0x1C, telegram: 08 18 1C 00 00 00 00 00 00 00 00 00 (CRC=41) #data=8
(01:01:54.252) Thermostat -> Boiler, type 0x1A, telegram: 18 08 1A 00 00 00 (CRC=7B) #data=2
(01:01:54.597) Thermostat -> Boiler, type 0x35, telegram: 18 08 35 00 11 (CRC=1C) #data=1
(01:01:54.871) Thermostat -> Boiler, type 0x33, telegram: 18 88 33 02 01 (CRC=74) #data=1
(01:01:54.915) Boiler -> Thermostat, type 0x33, telegram: 08 18 33 02 3C (CRC=B4) #data=1

Sonusss, 2019-08-11:
I'm not sure it is relevant but:
I tried to downgrade to 1.8.1 with OTA, didn't worked, no answer from Wemos
After that TC100 appeared !
I rebooted Wemos still with 1.8.2 and TC100 still there

Sonusss, 2019-08-11:
Now have different corrupt telegram:
Corrupt telegram: 98 0B 88 18 00 20 D4
Corrupt telegram: 0B 88 14 00 20 E4 08 0B 14 00 03 73 B1 49

proddy, 2019-08-11:
There are no big differences between 1.8.1 and 1.8.2 other than turning the LED off and including Junkers FW100 in the thermostat values. Just two lines changed. What I think is happening is sometimes the ESP needs a hard reset (power on/off) specially after an OTA update. 

Release 1.9 is almost ready with some improved Tx handling. A few corrupted telegrams isn't a problem.

proddy, 2019-08-17:
Can you try and reproduce with 1.9 in dev?

proddy, 2019-08-27:
Please try with 1.9. If it still fails to detect we'll re-open and take a deeper dive into the root cause.

closed 2019-08-27 by proddy