emsesp/EMS-ESP #170: Support Milton/Nefit HR 11

mrwee, 2019-08-15:
I have this boiler and a ModuleLine 400, but the boiler doesn't seem to be supported. I've got a gateway connected, but EMS-ESP doesn't seems to receive anything.
A autodetect shows:

> autodetect
> Started scan on EMS bus for known devices
> Requesting type Version(0x02) from dest 0x02
> Requesting type Version(0x02) from dest 0x08
> Requesting type Version(0x02) from dest 0x09
> Requesting type Version(0x02) from dest 0x10
> Requesting type Version(0x02) from dest 0x11
> Requesting type Version(0x02) from dest 0x17
> Requesting type Version(0x02) from dest 0x18
> Requesting type Version(0x02) from dest 0x20
> Requesting type Version(0x02) from dest 0x21
> Requesting type Version(0x02) from dest 0x30
> Requesting type Version(0x02) from dest 0x38
> Requesting type Version(0x02) from dest 0x48
> 

deep scan doesn't return anything.

Is is possible to implement support for this boiler?

proddy, 2019-08-15:
Hi, Milton is a different brand right? If it supports EMS or EMS+ we could add it like we did with Junkers but would need the details. You could try a `autodetect deep` to do a scan on all address types on the EMS bus and see if anything triggers.

bbqkees, 2019-08-15:
The name 'EcomLine' suggests that its an old boiler.
And indeed, if I lookup the installation manual of the Milton EcomLine HR11 boiler it says it needs an iRT thermostat.
Unfortunately this means you don't have an EMS bus on your boiler and the Gateway/EMS-ESP firmware is not compatible.

> Digitalt modulerande iRT rumstermostat
Den digitalt modulerande iRT rumstermostaten anslutes på kläm-radens anslutning 3 – 4. Anslutningen är fasneutral.

moustic999, 2019-08-15:
Can you post some pictures of your heater and thermostat, your gateway too...

-------- Message d'origine --------

bbqkees, 2019-08-15:
I've seen the ModuLine 300 thermostat connected to an old Nefit EcomLine boiler once before. So perhaps a late Ecomline support the EMS bus?

Can you do a `log v` and check whether it can read anything?
If not, its not EMS or wired incorrectly and if so, it is EMS.

mrwee, 2019-08-15:
Yes, Milton is an Danish version of Nefit . It's an old boiler but the ModuleLine 400 is new and seems to work fine. I'm using a brand new (Couple of hours old :) ) EMS gateway from Kees. No power from the service port (3.5mm jack), so I'm powering it via USB.
[image]

I've upgraded to 1.8.3 and connecting provides:
```
(00:00:34.817) Corrupt telegram: FF 02 FF 02 FF
(00:00:36.698) Corrupt telegram: FF 03 F5 02 DB
(00:00:37.618) Corrupt telegram: FF 17 EF 13 FF 02 FD 1B FB
(00:00:38.458) Corrupt telegram: FF 03 FB 02 7B
(00:00:38.678) Corrupt telegram: FF 02 FF 03 7D
(00:00:39.378) Corrupt telegram: FF 03 FB 03 EE
(00:00:39.798) Corrupt telegram: 03 FD 13 DB 02 FD 03 BD
(00:00:40.138) Corrupt telegram: FF 02 FF 02 FB
(00:00:44.377) Corrupt telegram: FE 17 BD 12 DA 02 FF 33 FB
(00:00:44.917) Corrupt telegram: FF 03 FB 01 FE 02 FB 01 FC 07 FB
(00:00:45.177) Corrupt telegram: FF 02 EB 02 6D
System is either not connect to the EMS bus or listen_mode is enabled
Error! Unable to read the EMS bus.
```

Doing an autodetect:
```
Started scan on EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x20
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
Requesting type Version(0x02) from dest 0x38
Requesting type Version(0x02) from dest 0x48
(00:02:28.492) Corrupt telegram: FF 02 FD 02 FF
(00:02:29.392) Corrupt telegram: FF 02 FC 02 F6
(00:02:30.012) Corrupt telegram: 02 FB 06 6C 02 FC 03 FB
Error! Unable to read the EMS bus.
(00:02:30.333) Corrupt telegram: FF 02 FB 13 6D
(00:02:30.693) Corrupt telegram: FF 03 ED 02 DB
(00:02:31.333) Corrupt telegram: FF 06 DB 02 BB 02 F8
(00:02:31.753) Corrupt telegram: FF 02 FB 02 FF 02 FD 13 BD
(00:02:31.893) Corrupt telegram: FF 13 B5 02 DB
(00:02:32.073) Corrupt telegram: FF 03 FD 03 FB
(00:02:32.293) Corrupt telegram: 06 F7 13 DB 02 FC 02 FB
(00:02:32.613) Corrupt telegram: FF 02 FB 12 5B
(00:02:33.014) Corrupt telegram: F7 02 FC 02 FB
(00:02:33.334) Corrupt telegram: FF 02 DB 06 DF
(00:02:33.534) Corrupt telegram: 02 DF 02 B6 03 FD
(00:02:33.754) Corrupt telegram: FF 03 FB 02 FF 02 FD
(00:02:33.894) 0x7F -> 0x03, telegram: FF 03 BD 03 (CRC=DB)
(00:02:34.115) Corrupt telegram: FF 02 DB 06 FB 02 FE 02
(00:02:35.155) Corrupt telegram: FF 03 FB 12 DB
(00:02:35.335) Corrupt telegram: FF 02 FB 02 D9
(00:02:35.735) 0x06 -> 0x7B, type 0x07, telegram: 06 FB 07 FB 03 FE 02 (CRC=FB) #data=3
(00:02:35.916) Corrupt telegram: 13 FB 02 B6 02 FA 12 FB
(00:02:36.055) Corrupt telegram: FF 03 FB 02 DB
(00:02:36.276) Corrupt telegram: FF 02 FC 02 FB 02 FF 03 FB
(00:02:36.436) Corrupt telegram: 03 FB 02 FB 03 FD
(00:02:36.676) Corrupt telegram: FF 02 DB 03 BB
(00:02:37.076) Corrupt telegram: 02 FB 02 6C 02 FE
(00:02:37.737) Corrupt telegram: FF 02 EB 02 DB
(00:02:37.937) Corrupt telegram: 06 FE 02 FC 03 FB
(00:02:38.117) Corrupt telegram: FF 02 FB 03 FB 01 F9 02 FD
(00:02:38.297) Corrupt telegram: 02 FB 02 DB 02 FF 02 FD
(00:02:38.937) Corrupt telegram: FF 06 FB 02 ED
(00:02:39.138) Corrupt telegram: 02 FB 02 FD 13 FB
(00:02:39.598) Corrupt telegram: FF 02 FD 03 FB
(00:02:40.499) Corrupt telegram: FF 02 FB 03 FB
[... 56 lines cut ...]
(00:03:40.495) Corrupt telegram: FF 02 FB 17 FB
(00:03:40.575) Corrupt telegram: 13 6D 02 FF 02 FB
(00:03:41.076) Corrupt telegram: FF 13 FB 03 FD 03 EF
(00:03:41.417) Corrupt telegram: FF 03 FB 13 DB 03 FC 13 FF 02 FF 03 ED
(00:03:41.657) Corrupt telegram: FF 02 DB 03 DB
(00:03:41.918) Corrupt telegram: FF 12 E9 03 DB
(00:03:41.998) Corrupt telegram: FF 13 FB 02 FE 12 BE
(00:03:42.439) Corrupt telegram: 02 FD 02 FE 06 FB
(00:03:42.620) Corrupt telegram: FF 02 FB 13 FB
(00:03:42.960) Corrupt telegram: 13 FB 02 FD 02 BD
(00:03:43.561) Corrupt telegram: 17 FF 37 EC 02 FE 03 DB 02 FE 12 EF
(00:03:44.002) Corrupt telegram: FF 03 B4 1B FB 01 FF 03 FB
(00:03:44.263) Corrupt telegram: FF 13 FB 13 DB 01 FF 13 EB 03 FF 02 FB
(00:03:44.623) Corrupt telegram: FF 02 FF 17 FB 02 FF 03 FF
(00:03:45.485) Corrupt telegram: FF 03 ED 02 ED
(00:03:45.785) Corrupt telegram: FF 02 B4 1B ED 02 FF 13 DB 03 FE 13 FB
(00:03:46.266) Corrupt telegram: 1B FB 02 FE 03 FB
(00:03:47.207) Corrupt telegram: FF 02 AD 02 FB
(00:03:47.808) Corrupt telegram: 02 FD 02 FF 13 FB
(00:03:48.069) Corrupt telegram: 13 FA 02 FF 02 FB
(00:03:48.409) Corrupt telegram: 13 FB 02 FD 12 FB
(00:03:48.509) Corrupt telegram: FF 13 EE 13 FB
(00:03:48.770) Corrupt telegram: FE 13 FB 02 FF 07 FB
(00:03:48.910) Corrupt telegram: FF 02 FF 03 FF 02 FF 02 FB 02 FF 02 E9
```

mrwee, 2019-08-15:
Running a deepscan seems to stop the verbose logging. At least nothing is shown :(. 
After 5 mins of running, still no screen-update.

bbqkees, 2019-08-15:
It appears the RC35/ModuLine 300/400 is multi-protocol (EMS and iRT).
Your boiler has the UBA1.5 which is iRT only. So it won't work unfortunately.

I've seen others using an iRT to OT adapter and then an OT to EMS adapter but this will set you back 100+ Euro and likely won't work in this case.

mrwee, 2019-08-15:
Damn. Just thought I'd finally be able to get data out of the boiler :(. Well, now I have one gateway too much then..

closed 2019-10-08 by proddy