emsesp/EMS-ESP #171: EMS-ESP HTTP landing page

ViNYLeq, 2019-08-16:
Hello Everyone,

First of all, I'd like to thank you for this project, I thought only possibility to control my GB072 would be KM200 :) In the beginning of my way I'd like to ask about one thing I think can be a bug.

Initial configuration is done, wifi_ssid, wifi_password and serial off, then reboot. Successfully connected to my own wifi

```
* Connected to: EMS-ESP version 1.8.1
* Hostname: ems-esp (10.30.60.6)
* WiFi SSID: Fillory (signal 90%)
* MQTT disconnected
```

but HTTP landing page is the same as before.

```
EMS-ESP version 1.8.1
System stats:
Connected to wireless network ems-esp with signal strength 88% MQTT is disconnected 
System uptime: 0 days 0 hours 27 minutes 20 seconds

EMS devices found:
Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:04.09)
BC25 Base Controller (DeviceID:0x09 ProductID:125 Version:01.06)
RC300/RC310/Nefit Moduline 3000 (DeviceID:0x10 ProductID:158 Version:11.07)

Go here to connect the System to your wireless network.
```

I can't find whether it's bug or a feture, but I'd say I should see something else now.

```
[APP] EMS-ESP version: 1.8.1
[APP] MyESP version: 1.1.24
[APP] Build timestamp: 2019-08-06 18:38:59
```

Best regards,
Pawel

Sonusss, 2019-08-16:
Hi,
May I suggest you to start upgrading to the latest release.
You can find binaries here: https://github.com/proddy/EMS-ESP/releases/
Rgds,

proddy, 2019-08-16:
Hi @ViNYLeq , it could well be a bug. However all that web code has been stripped and re-written in the 1.9 which I'm planning to push to dev this weekend. The new build will probably open a new can of worms & bugs but I'm pretty sure this particular issue you won't see anymore.

ViNYLeq, 2019-08-16:
Hello Both,

Thanks. This I can check, but one question: the only way to proceed is to flash by cable? There's something about OTA in documentation but it's not like in espeasy or tasmota just to upload the file via web page.

> For upgrading via OTA (Over The Air):
> wget https://raw.githubusercontent.com/esp8266/Arduino/master/tools/espota.py
> wget https://github.com/proddy/EMS-ESP/releases/latest/download/firmware_d1_mini.bin
> python espota.py -i ems-esp -f firmware_d1_mini.bin
> 

Best regards,
Pawel

proddy, 2019-08-16:
there are a couple of ways to flash
 - via platform after building the firmware from source files, You can choose OTA or via USB depending on the setting in the platformio.ini. This is mainly for developers.
 - using the command line as described in the wiki https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware
 - From version 1.9, which is still in development, you can do this from the web interface. It will automatically upgrade to the latest stable version.

Sonusss, 2019-08-16:
BTW I checked on my 1.8.2 and have the same issue.

Web report hostname instead of WiFi SSID
EMS-ESP version 1.8.2
System stats:
Connected to wireless network **ems-esp** with signal strength 94%

While Telnet:
 EMS-ESP version 1.8.2
[FS] Settings loaded from SPIFFS
[WEB] Web server started
[UART] Opened Rx/Tx connection
Starting auto discover of EMS devices...
[WIFI] Connecting to SSID: VOO-013821
[WIFI] SSID  VOO-013821

ViNYLeq, 2019-08-16:
Flashed 1.8.3 using python script, also the same. Will wait for 1.9.0 then :)

proddy, 2019-08-17:
1.9 dev is there. Can you check?

ViNYLeq, 2019-08-17:
Interface bricked so far :) After flashing it's not getting IP from DHCP anymore. Investigating now.

C:\Python27>python.exe espota.py -i ems-esp -f c:\Users\Pawel\Downloads\ems-esp\firmware_debug_1_9_0b1.bin
Uploading..............................................................................................................................................................................................................................................................................................................................................................................................................................................................................

C:\Python27>

ViNYLeq, 2019-08-17:
My fault, false alarm. I've not checked whether module is back to factory defaults. But I can see HTTP interface is different now. Thanks.

proddy, 2019-08-27:
great!

closed 2019-08-27 by proddy