emsesp/EMS-ESP #172: Home Assistant - Thermostat - No manual mode

majdzik84, 2019-08-16:
I have a problem with the thermostat in the Home assistant.
- I have the option of "Auto" and "OFF", **No Manual mode** in Lovelace.
- When the auto mode is set and the temperature is changed, the day or night temperature changes but 0 degrees is shown on the thermostat. It's normal?.  [Screen](https://www.dropbox.com/s/cea9w8gg9b2ghtb/1.png?dl=0)
- When I try to call "Manual" I have an error :

`2019-08-16 10:49:02 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection.1799655120] value is not allowed for dictionary value @ data['hvac_mode']
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 128, in handle_call_service
    connection.context(msg),
  File "/usr/src/homeassistant/homeassistant/core.py", line 1213, in async_call
    processed_data = handler.schema(service_data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 267, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 589, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 427, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: value is not allowed for dictionary value @ data['hvac_mode']`

I have:
`EMS-ESP version 1.8.1
System stats:
Connected to wireless network ems-esp with signal strength 64%
MQTT is connected 
System uptime: 0 days 15 hours 40 minutes 41 seconds

EMS devices found:
Nefit Topline Compact/**Buderus GB162** (DeviceID:0x08 ProductID:115 Version:03.06)
**RC35** (DeviceID:0x10 ProductID:86 Version:31.07)
`

proddy, 2019-08-17:
Hi, HA made some changes to the climate component (see https://github.com/proddy/EMS-ESP/issues/153) after 0.96 and the climate.yaml file needed modifying as different thermostats respond to different commands. Can you check your .yaml file? I'll also try and reproduce here.

majdzik84, 2019-08-17:
Hi. I think I have the correct configuration from your climate.yaml

```
  - platform: mqtt
    name: Thermostat
    modes:
      - "auto"
      - "manual"
      - "off"

    mode_state_topic: "home/ems-esp/thermostat_data"
    current_temperature_topic: "home/ems-esp/thermostat_data"
    temperature_state_topic: "home/ems-esp/thermostat_data"

    temperature_command_topic: "home/ems-esp/thermostat_cmd_temp"
    mode_command_topic: "home/ems-esp/thermostat_cmd_mode"

    mode_state_template: "{{ value_json.thermostat_mode }}"
    current_temperature_template: "{{ value_json.thermostat_currtemp }}"
    temperature_state_template: "{{ value_json.thermostat_seltemp }}"

    temp_step: 0.5
```

https://github.com/proddy/EMS-ESP/blob/master/doc/home_assistant/climate.yaml

proddy, 2019-08-17:
The HA climate component doesn't support a 'manual' mode as such. See https://www.home-assistant.io/components/climate.mqtt, The modes have to be a subset of “auto”, “off”, “cool”, “heat”, “dry”, “fan_only”.

About the zero value when switching the thermostat to OFF, you should look in to the values been sent on the MQTT payload from EMS-ESP. If there is a zero it's a bug. My guess is that HA is not picking it up. You can simulate this yourself without EMS-ESP by using a program like http://mqtt-explorer.com/ and sending:

topic : `home/ems-esp/thermostat_data`
payload : `{"thermostat_hc":"1","thermostat_seltemp":20.5,"thermostat_currtemp":23.1}`

majdzik84, 2019-08-17:
You're right. After adding the "heat, cool, ..." subset, none of them switches to Day mode. The thermostat in auto mode does not change Setpoint temperature (it is always 0.0 C) - it changes the Day or Night value depending on the time set on the thermostat. 

[Picture 1](https://www.dropbox.com/s/9zldhbm4ay1v1ay/1.jpg?dl=0)
[Picture 2](https://www.dropbox.com/s/uf8usphd4slc35g/2.jpg?dl=0)
[Picture 3](https://www.dropbox.com/s/euoxrwcozqiaqgn/3.jpg?dl=0)

proddy, 2019-08-17:
ok, do you know what the correct logic is so I can implement it in the code, I don't have an RC35 and they all behave slightly differently.

majdzik84, 2019-08-18:
In the case of auto mode, it has two temperatures (DAY and NIGHT) and they are automatically switched at times defined in the RC35 controller (switch point). [RC35 Instruction](https://www.manualslib.com/manual/745802/Buderus-Rc35.html?page=36#manual)

In the auto mode, when the temperature on the controller is changed, it changes until the next mode change. 
(example: Day temp: 22C, Night temp: 19C.
Changing the temperature during the day to 23C will change it until switching to night mode) e.g. at 22:00) [RC35 Instruction](https://www.manualslib.com/manual/745802/Buderus-Rc35.html?page=31#manual).

Currently, it works in such a way that Setpoint room temperature: 0.0 C
[image]
 (manual change of temperature with the knob on RC35 causes that Setpoint room temperature shows the changed room temperature in force until the next switching point).
[image]

It would be great if the Setpoint room temperature value showed the temperature of the current mode or changed during its duration so that a temperature was always given.

----------------------------------------------------------
Returning to the inability to switch to Day mode in HA.
Will changing manual to heat solve the switching problem?
[image]

I will add that I am not a programmer - I am guided only by my logic. If I think wrong, I'm sorry;)

ReneHonig, 2019-08-22:
@majdzik84, I am not familiar with the RC35 thermostat, but a couple of comments that may move you forward:

The rationale for changing the Climate set-up in home assistant is explained [here](https://developers.home-assistant.io/blog/2019/07/03/climate-cleanup.html).

From what I know of the RC35, the relevant HVAC modes would be off, auto and heat, where auto would switch between day and night mode automatically and heat is equivalent to the prior 'manual' mode (as per majdzik84's comment).

As for the zero degrees C temperature, I am not sure what the cause is, it might be an initialisation problem as you probably restart HA many times during debugging. Maybe try `retain: true`, so MQTT keeps the most recent temperature value. Also remember that the MQTT climate component does not have any logic, it is merely a representation of a 'climate controller' (thermostat in your case), so all the logic needs to be somewhere else (e.g. EMS-ESP, HA automation or Node-red). 

Your proposed changes above are probably a good starting point, but i do not know that part of the code well, as I use a Opentherm controller.

proddy, 2019-09-22:
Changed in 1.9.1.

closed 2019-09-22 by proddy