emsesp/EMS-ESP #173: Error compiling 1.9.0b2 - gulp related?

higgers, 2019-08-18:
Hi again, it's higgers from issue #23 ;)

I've been following some of the issues around tx_mode2 and the revised code to read the EMS bus and have been hoping that the changes would allow telegrams to be sent to my Worcester-Bosch boiler.  I've just been trying to compile 1.9.0b2 and after installing nodejs and running the commands listed in the compilation instructions have hit this compilation error:

```
Compiling .pio\build\release\src\emsuart.cpp.o
In file included from src\MyESP.cpp:9:0:
src\MyESP.h:46:55: fatal error: webh/glyphicons-halflings-regular.woff.gz.h: No such file or directory
 #include "webh/glyphicons-halflings-regular.woff.gz.h"
                                                       ^
compilation terminated.
*** [.pio\build\release\src\MyESP.cpp.o] Error 1
In file included from src\ems-esp.cpp:16:0:
src/MyESP.h:46:55: fatal error: webh/glyphicons-halflings-regular.woff.gz.h: No such file or directory
 #include "webh/glyphicons-halflings-regular.woff.gz.h"
                                                       ^
compilation terminated.
*** [.pio\build\release\src\ems-esp.cpp.o] Error 1
In file included from src\ems.cpp:13:0:
src/MyESP.h:46:55: fatal error: webh/glyphicons-halflings-regular.woff.gz.h: No such file or directory
 #include "webh/glyphicons-halflings-regular.woff.gz.h"
                                                       ^
compilation terminated.
*** [.pio\build\release\src\ems.cpp.o] Error 1
====================================================================================== [ERROR] Took 3.52 seconds ====================================================================================== 

============================================================================================== [SUMMARY] ============================================================================================== 
Environment buildweb    [IGNORED]
Environment test        [IGNORED]
Environment crash       [IGNORED]
Environment debug       [IGNORED]
Environment clean       [IGNORED]
Environment release     [FAILED]
Environment checkcode   [IGNORED]
================================================================================ 1 failed, 0 succeeded in 3.53 seconds ================================================================================ 
The terminal process terminated with exit code: 1
```

Does anyone have any suggestions on how to resolve this?

proddy, 2019-08-18:
hi @higgers, it's been a while! The problem is you're building the `release` target from platformio.ini. It needs to be `debug` to build the web as described in the ChangeLog. If it still doesn't work let me know!

Yosh01, 2019-08-18:
I had the same Problem.
Node is missing. You have to install node.js to Build the missing Parts during build.

proddy, 2019-08-18:
guys, I explained it all in https://github.com/proddy/EMS-ESP/blob/dev/CHANGELOG.md.

higgers, 2019-08-18:
I'd installed nodejs but hadn't read the first line that states your have to build the debug build.  I clearly can't read properly!  Thanks for your patience, Proddy.

I've successfully built the code, flashed it to the wemos but I don't see the EMS-AP access point.  When I power up the board there's a brief single flash of the LED then nothing.  I've tried holding down the reset pin and then applying power and when I let go of the pin I see the same brief flash and then nothing else.  I vaguely remember there's a command to erase the SPIFFS and effectively start from scratch but I can't remember how to do it.  I'll go back and check issue #23 because I think the command was mentioned in there.

higgers, 2019-08-18:
"pio run -t erase"?

proddy, 2019-08-18:
thats the one.

Which version did you upgrade from? I did test going from 1.8.1 to 1.9.0 and it seemed to work.

higgers, 2019-08-18:
I upgraded from 1.8.3.

I tried running "pio run -t erase" and then flashed the firmware via a USB cable.  The flash process seems to run fine, I can see the ASCII progress bars count up to 100% but I just can't see the EMS-AP access point.  I've tried scanning for it on my laptop and my phone.

Should the LED on the wemos be flashing constantly when it's in access point mode?  Or should it do the single flash that I'm seeing?

proddy, 2019-08-18:
the LED will flash quickly on reboot and then either a slow pulse if serial is enabled or the EMS bus can't be reached or solid if it has a connection to the BUS. I noticed sometimes after a firmware the soft reset (RST) platformio does is not always enough to get it going. Try tapping the small reset button on the ESP and see if that helps. You should see an AP called ems-esp. 

If that doesn't work, go to platformio.ini and use `general_flags = -DFORCE_SERIAL`  which will force Serial mode. Then when attached to USB open the monitor and you'll be able to see a trace of what's happening.

higgers, 2019-08-18:
Great advice, thanks, I can what's happening now.  The board keeps restarting every 3 seconds.  THis is the output in the monitor:

```
* EMS-ESP version 1.9.0b3
[FS] Serial is forced
[FS] System settings loaded
[FS] Custom config loaded
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   CE:50:E3:44:EE:8E
[SYSTEM] Serial port communication is enabled.
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[TELNET] in AP mode
[TELNET] Telnet server started

Panic C:\Users\sh\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_esp8266_main.cpp:103 __yield

>>>stack>>>

ctx: sys
sp: 3ffffdc0 end: 3fffffb0 offset: 01b0
3fffff70:  3fffdad0 00000000 3fff200c 4023c2dd
3fffff80:  3fffdad0 3fff465c 3fff200c 402172f8
3fffff90:  3fffdad0 00000000 3fff2a08 40220ea6
3fffffa0:  feefeffe 00000000 3ffe8598 4023c29e
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld

* EMS-ESP version 1.9.0b3
[FS] Serial is forced
<and repeat every 3 seconds>
```

higgers, 2019-08-18:
This appears to be related to the watchdog reset.

proddy, 2019-08-18:
oh dear. Someone else reported this problem too. I'll see if I can reproduce it.

Perhaps do another 'pio run -t erase' and try again. just to be sure...

higgers, 2019-08-18:
I've tried that a few times with the same result.

I must be doing something wrong/differently to you.  I've just unzipped the source to a new folder in order to start from scratch and I'm getting those compilation errors for the files in the webh directory again.

I have node.js installed, and it's on the path.  I've tested this by opening a command prompt and typing "node".

I then navigated to the webfilesbuilder directory, ran "npm install --global gulp-cli" and "npm install gulp" which both ran successfully, though there 7 warnings from the "npm install gulp" command.  I then run "gulp" and receive a "The app you're trying to install isn't a Microsoft-verified app" pop.  I click on "Install anyway" and then get a Windows Script Host error popup saying there's a syntax error in gulp.js on line 7 character 16.  Line 7 character 16 is a right angle bracket:

```
process.env.NODE_PATH = (process.env.NODE_PATH || '').split(path.delimiter)
  .filter((p) => p).concat(__dirname + '/node_modules').join(path.delimiter);
```

There's also a very helpful error code: 899A03EA.

Any ideas?

proddy, 2019-08-18:
and if you just take the [binary](https://github.com/proddy/EMS-ESP/releases/download/1.9.0b3/firmware_debug_1_9_0b3.bin) and upload from the command line as described [here](https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware) ? That will rule out any environment issues.

higgers, 2019-08-18:
Well, that worked:

```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=0
  Tx: Last poll=1.249 seconds ago, # successful write requests=4

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 23.4 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1147 times
  Warm Water active time: 80 days 13 hours 33 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 40 C
  Current flow temperature: 29.0 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 76 %
  Burner current power: 0 %
  Flame current: 0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 3898 times
  Total burner operating time: 420 days 3 hours 33 minutes
  Total heat operating time: 339 days 14 hours 0 minutes
```

And look at that!  4 successful write requests!!!

Is there any way to see which write requests were successful?  That number doesn't seem to be increasing above 4.  What commands would you recommend I run?  I tried "autodetect deep" but it didn't seem to find anything.  The boiler is a Worcester-Bosch Highflow 550CDi and I've also installed a Bosch Easy Connect CT 200 thermostat.

higgers, 2019-08-18:
BTW, that web frontend is absolutely brilliant, that's some good work!

proddy, 2019-08-18:
For building the web if node can't find the gulp module just type `node .\node_modules\gulp\bin\gulp.js` in the webfilesbuilder directory and it should be it. The errors you see when installing gulp is normal and because I've used an older version of the gulp syntax, but you can safely ignore them.

As for Tx not working, that sucks as I was hoping the new txmode2 code would work on all boiler types. Back with 1.8.3 which tx_mode setting where you using?

susisstrolch, 2019-08-18:
What does ‚log j‘ show?

Sent by mobile device

higgers, 2019-08-18:
an excerpt from log j from 1.9.0b2:

```
(01:01:10.285) ems_parseTelegram: 00 01: 10
(01:01:10.316) ems_parseTelegram: 00 01: 18
(01:01:10.321) ems_parseTelegram: 00 01: 98
(01:01:10.348) ems_parseTelegram: 00 01: 11
(01:01:10.379) ems_parseTelegram: 00 01: 12
(01:01:10.410) ems_parseTelegram: 00 01: 18
(01:01:10.415) ems_parseTelegram: 00 01: 98
(01:01:10.441) ems_parseTelegram: 00 01: 13
(01:01:10.479) ems_parseTelegram: 00 01: 18
(01:01:10.484) ems_parseTelegram: 00 01: 98
(01:01:10.503) ems_parseTelegram: 00 01: 14
(01:01:10.535) ems_parseTelegram: 00 01: 15
(01:01:10.566) ems_parseTelegram: 00 01: 18
(01:01:10.571) ems_parseTelegram: 00 01: 98
(01:01:10.597) ems_parseTelegram: 00 01: 16
(01:01:10.629) ems_parseTelegram: 00 01: 17
(01:01:10.659) ems_parseTelegram: 00 01: 18
(01:01:10.665) ems_parseTelegram: 00 01: 98
(01:01:10.691) ems_parseTelegram: 00 01: 20
(01:01:10.728) ems_parseTelegram: 00 01: 18
(01:01:10.734) ems_parseTelegram: 00 01: 98
(01:01:10.753) ems_parseTelegram: 00 01: 28
(01:01:10.785) ems_parseTelegram: 00 01: 30
(01:01:10.816) ems_parseTelegram: 00 01: 18
(01:01:10.821) ems_parseTelegram: 00 01: 98
(01:01:10.847) ems_parseTelegram: 00 01: 38
(01:01:10.878) ems_parseTelegram: 00 01: 40
(01:01:10.909) ems_parseTelegram: 00 01: 18
(01:01:10.915) ems_parseTelegram: 00 01: 98
(01:01:10.941) ems_parseTelegram: 00 01: 48
(01:01:10.979) ems_parseTelegram: 00 01: 50
(01:01:11.034) ems_parseTelegram: 00 01: 18
(01:01:11.040) ems_parseTelegram: 00 01: 98
(01:01:11.071) ems_parseTelegram: 00 01: 88 13 05 22 00 80
(01:01:11.072) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
l(01:01:11.380) ems_parseTelegram: 00 01: 88 00 18 00 33 02 36 64 23 0A 31 65 40 01 FD 80 00 80 00 FF FF FF 00 00 00 00 2D 02 38 91
(01:01:11.381) 0x08 -> all, type 0x18, telegram: 88 00 18 00 33 02 36 64 23 0A 31 65 40 01 FD 80 00 80 00 FF FF FF 00 00 00 00 2D 02 38 (CRC=91) #data=25
og(01:01:11.620) ems_parseTelegram: 00 01: 88 00 34 00 33 01 FD 80 00 A8 00 00 01 2B 01 C5 3F 00 04 7D 00 2E
(01:01:11.621) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 FD 80 00 A8 00 00 01 2B 01 C5 3F 00 04 7D 00 (CRC=2E) #data=17
 (01:01:11.815) ems_parseTelegram: 00 01: 58
(01:01:11.846) ems_parseTelegram: 00 01: 18
(01:01:11.852) ems_parseTelegram: 00 01: 98
n(01:01:11.877) ems_parseTelegram: 00 01: 60
(01:01:11.909) ems_parseTelegram: 00 01: 68
(01:01:11.939) ems_parseTelegram: 00 01: 18
(01:01:11.945) ems_parseTelegram: 00 01: 98
(01:01:11.978) ems_parseTelegram: 00 01: 0A
(01:01:12.034) ems_parseTelegram: 00 01: 0B
(01:01:12.064) ems_parseTelegram: 00 01: 18
(01:01:12.070) ems_parseTelegram: 00 01: 98
(01:01:12.096) ems_parseTelegram: 00 01: 0C
(01:01:12.127) ems_parseTelegram: 00 01: 18
(01:01:12.133) ems_parseTelegram: 00 01: 98
(01:01:12.158) ems_parseTelegram: 00 01: 0D
(01:01:12.190) ems_parseTelegram: 00 01: 0E
(01:01:12.227) ems_parseTelegram: 00 01: 18
(01:01:12.233) ems_parseTelegram: 00 01: 98
(01:01:12.252) ems_parseTelegram: 00 01: 0F
(01:01:12.283) ems_parseTelegram: 00 01: 10
(01:01:12.314) ems_parseTelegram: 00 01: 18
(01:01:12.320) ems_parseTelegram: 00 01: 98
(01:01:12.346) ems_parseTelegram: 00 01: 11
(01:01:12.377) ems_parseTelegram: 00 01: 12
(01:01:12.408) ems_parseTelegram: 00 01: 18
(01:01:12.413) ems_parseTelegram: 00 01: 98
(01:01:12.439) ems_parseTelegram: 00 01: 13
```

Is that enough or would you like to see more?  Will try 1.8.3 tx_mode shortly.

susisstrolch, 2019-08-18:
An autodetect or some write command would be interresting to see whats going on...

Sent by mobile device

higgers, 2019-08-18:
I wasn't sure exactly what sort of logging you want to see during the autodetect.  I initially set log to raw and then set log to verbose.  Let me know if you want me to run some more commands...

```
autodetect deep
System Logging set to None
Starting a deep EMS device scan. This can take up to 2 minutes. Please wait...
log r
System Logging set to Raw mode
(01:17:36.940) 98 08 35 00 11 D4
(01:17:37.031) 8B 02 02 00 20 40
(01:17:37.315) 98 88 33 01 09 B2
(01:17:37.349) 88 18 33 01 FF 33 00 00 00 00 02 46 00 23
(01:17:38.560) 8B 03 02 00 20 48
(01:17:40.143) 88 13 05 22 00 80
(01:17:40.440) 88 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 6F
(01:17:40.670) 88 00 18 00 2A 01 E1 4C 00 00 21 40 40 01 47 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 52
(01:17:40.911) 88 00 34 00 33 01 47 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 7F
(01:17:41.158) 98 08 1A 00 2A 4C EA
(01:17:41.234) 88 00 19 00 80 00 80 00 80 00 FF FF 00 00 00 0F 3C 09 3B 6A 00 00 00 07 76 28 00 0A BF 80 00 9D
(01:17:41.936) 98 08 23 00 2A B7
(01:17:42.713) 8B 04 02 00 20 70
(01:17:50.162) 88 00 18 00 2A 01 E0 4C 00 00 21 40 40 01 49 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 49
(01:17:50.384) 88 13 05 22 00 80
(01:17:50.683) 88 00 34 00 33 01 49 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 BB
(01:17:51.337) 8B 05 02 00 20 78
log v
System Logging set to Verbose
Requesting type Version(0x02) from dest 0x3A
Requesting type Version(0x02) from dest 0x3B
Requesting type Version(0x02) from dest 0x3C
Requesting type Version(0x02) from dest 0x3D
Requesting type Version(0x02) from dest 0x3E
Requesting type Version(0x02) from dest 0x3F
Requesting type Version(0x02) from dest 0x40
Requesting type Version(0x02) from dest 0x41
Requesting type Version(0x02) from dest 0x42
Requesting type Version(0x02) from dest 0x43
Requesting type Version(0x02) from dest 0x44
Requesting type Version(0x02) from dest 0x45
Requesting type Version(0x02) from dest 0x46
Requesting type Version(0x02) from dest 0x47
Requesting type Version(0x02) from dest 0x48
(01:18:00.125) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
Requesting type Version(0x02) from dest 0x49
(01:18:00.434) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2A 01 E0 4C 00 00 21 40 40 01 4E 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=2B) #data=25
<--- UBAMonitorFast(0x18)
Requesting type Version(0x02) from dest 0x4A
(01:18:00.674) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 4E 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=D9) #data=17
<--- UBAMonitorWWMessage(0x34)
Requesting type Version(0x02) from dest 0x4B
Requesting type Version(0x02) from dest 0x4C
(01:18:01.431) Sending read of type 0x02 to 0x06:telegram: 8B 06 02 00 20 (CRC=60) #data=48
Requesting type Version(0x02) from dest 0x4D
Requesting type Version(0x02) from dest 0x4E
Requesting type Version(0x02) from dest 0x4F
Requesting type Version(0x02) from dest 0x50
Requesting type Version(0x02) from dest 0x51
Requesting type Version(0x02) from dest 0x52
Requesting type Version(0x02) from dest 0x53
Requesting type Version(0x02) from dest 0x54
Requesting type Version(0x02) from dest 0x55
Requesting type Version(0x02) from dest 0x56
Requesting type Version(0x02) from dest 0x57
Requesting type Version(0x02) from dest 0x58
Requesting type Version(0x02) from dest 0x59
Requesting type Version(0x02) from dest 0x5A
Requesting type Version(0x02) from dest 0x5B
Requesting type Version(0x02) from dest 0x5C
Requesting type Version(0x02) from dest 0x5D
Requesting type Version(0x02) from dest 0x5E
[... 188 lines cut ...]
Requesting type Version(0x02) from dest 0xF0
Requesting type Version(0x02) from dest 0xF1
Requesting type Version(0x02) from dest 0xF2
Requesting type Version(0x02) from dest 0xF3
(01:19:00.072) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
Requesting type Version(0x02) from dest 0xF4
(01:19:00.380) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2A 01 DE 4C 00 00 21 40 40 01 5A 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=9B) #data=25
<--- UBAMonitorFast(0x18)
Requesting type Version(0x02) from dest 0xF5
(01:19:00.621) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 5A 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=F8) #data=17
<--- UBAMonitorWWMessage(0x34)
Requesting type Version(0x02) from dest 0xF6
Requesting type Version(0x02) from dest 0xF7
Requesting type Version(0x02) from dest 0xF8
(01:19:01.783) Sending read of type 0x02 to 0x15:telegram: 8B 95 02 00 20 (CRC=9C) #data=48
Requesting type Version(0x02) from dest 0xF9
Requesting type Version(0x02) from dest 0xFA
Requesting type Version(0x02) from dest 0xFB
Requesting type Version(0x02) from dest 0xFC
Requesting type Version(0x02) from dest 0xFD
Requesting type Version(0x02) from dest 0xFE
Requesting type Version(0x02) from dest 0xFF
Finished the deep EMS device scan.
System Logging set to None
```

proddy, 2019-08-18:
don't think Tx is working. Looks like there are polls. Would be good if you could find out which tx_mode setting worked for you in 1.8.x

higgers, 2019-08-18:
Wow!!! I flashed 1.8.4 and as soon as I set the tx_mode to 2 it detected the thermostat:

```
set tx_mode 2
tx_mode changed.

(00:01:07.675) Corrupt telegram: 14 00 18 00 98
(00:01:13.257) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:01:13.565) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2C 01 9D 4C 00 00 21 40 40 01 04 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=45) #data=25
<--- UBAMonitorFast(0x18)
(00:01:13.806) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 04 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=B8) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:14.345) Sending read of type 0x02 to 0x18:telegram: 0B 98 02 00 20 (CRC=3C) #data=20
(00:01:14.406) 0x18 -> me, type 0x02, telegram: 98 0B 02 00 CB 02 06 00 00 00 00 00 00 01 00 (CRC=1B) #data=11
<--- Version(0x02)
Thermostat found: Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.06)
* Setting Thermostat to Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.06)
Requesting type EasyStatusMessage(0x0A) from dest 0x18
Requesting type RCTime(0x06) from dest 0x18
(00:01:15.999) Sending read of type 0x02 to 0x20:telegram: 0B A0 02 00 20 (CRC=E5) #data=156
(00:01:23.280) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:01:23.588) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2C 01 9D 4C 00 00 21 40 40 01 04 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=45) #data=25
<--- UBAMonitorFast(0x18)
(00:01:23.829) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 04 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=B8) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:24.311) Thermostat -> all, type 0x06, telegram: 98 00 06 00 13 08 16 12 19 32 06 A5 (CRC=EE) #data=8
<--- RCTime(0x06)
(00:01:24.898) Sending read of type 0x02 to 0x21:telegram: 0B A1 02 00 20 (CRC=ED) #data=20
(00:01:33.240) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:01:33.537) 0x08 -> all, type 0x07, telegram: 88 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=6F) #data=15
(00:01:33.767) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2C 01 9B 4C 00 00 21 40 40 01 04 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=B5) #data=25
<--- UBAMonitorFast(0x18)
(00:01:34.008) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 04 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=B8) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:34.252) Thermostat -> 0x08, type 0x35, telegram: 98 08 35 00 11 (CRC=D4) #data=1
(00:01:34.331) 0x08 -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 00 00 0F 3C 09 3B 6A 00 00 00 07 76 28 00 0A BF 80 00 (CRC=9D) #data=27
<--- UBAMonitorSlow(0x19)
(00:01:34.627) Thermostat -> 0x08, type 0x33, telegram: 98 88 33 01 09 (CRC=B2) #data=1
(00:01:34.654) 0x08 -> Thermostat, type 0x33, telegram: 88 18 33 01 FF 33 00 00 00 00 02 46 00 (CRC=23) #data=9
(00:01:35.452) Sending read of type 0x02 to 0x30:telegram: 0B B0 02 00 20 (CRC=65)
(00:01:37.284) Thermostat -> 0x08, type 0x1A, telegram: 98 08 1A 00 2C 4C (CRC=E6) #data=2
(00:01:38.231) Sending read of type 0x02 to 0x38:telegram: 0B B8 02 00 20 (CRC=25) #data=74
(00:01:38.530) Thermostat -> 0x08, type 0x23, telegram: 98 08 23 00 2A (CRC=B7) #data=1
(00:01:39.792) Sending read of type 0x02 to 0x48:telegram: 0B C8 02 00 20 (CRC=8E) #data=74
(00:01:43.291) 0x08 -> all, type 0x18, telegram: 88 00 18 00 2C 01 9D 4C 00 00 21 40 40 01 03 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=27) #data=25
<--- UBAMonitorFast(0x18)
(00:01:43.512) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:01:43.812) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 04 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=B8) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:01:44.631) Sending raw: 8B 88 02 00 20 00
(00:01:44.684) 0x08 -> me, type 0x02, telegram: 88 0B 02 00 5F 19 02 00 00 00 00 00 00 00 (CRC=F3) #data=10
<--- Version(0x02)
Boiler found: Bosch Condens 2500/Junkers Heatronics3 (DeviceID:0x08 ProductID:95 Version:25.02)
* Setting Boiler to model Bosch Condens 2500/Junkers Heatronics3 (DeviceID:0x08 ProductID:95 Version:25.02)
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:01:46.192) Sending read of type 0x0A to 0x18:telegram: 8B 98 0A 00 20 (CRC=D4)
(00:01:46.294) Thermostat -> me, type 0x0A, telegram: 98 0B 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=9A) #data=26
<--- EasyStatusMessage(0x0A)
(00:01:47.848) Sending read of type 0x06 to 0x18:telegram: 8B 98 06 00 20 (CRC=E4) #data=20
(00:01:47.909) Thermostat -> me, type 0x06, telegram: 98 0B 06 00 13 08 16 12 1A 0D 06 11 18 FF 00 (CRC=4F) #data=11
<--- RCTime(0x06)
(00:01:49.440) Sending read of type 0x18 to 0x08:telegram: 8B 88 18 00 20 (CRC=1C) #data=20
(00:01:49.504) Boiler -> me, type 0x18, telegram: 88 0B 18 00 2C 01 9B 4C 00 00 21 40 40 01 04 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=06) #data=25
<--- UBAMonitorFast(0x18)
(00:01:49.555) Boiler -> all, type 0x07, telegram: 88 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=DF) #data=15
(00:01:49.783) Sending read of type 0x19 to 0x08:telegram: 8B 88 19 00 20 (CRC=18)
[... 35 lines cut ...]
<--- UBAMonitorWWMessage(0x34)
(00:02:05.176) Sending read of type 0x18 to 0x08:telegram: 8B 88 18 00 20 (CRC=1C) #data=74
[... cut ...]
(00:02:06.541) Boiler -> all, type 0x07, telegram: 88 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=6F) #data=15
ex(00:02:13.234) Boiler -> all, type 0x18, telegram: 88 00 18 00 2C 01 9B 4C 00 00 21 40 40 01 03 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=D7) #data=25
<--- UBAMonitorFast(0x18)
(00:02:13.456) Boiler -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
it(00:02:13.756) Boiler -> all, type 0x34, telegram: 88 00 34 00 33 01 03 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=DA) #data=17
<--- UBAMonitorWWMessage(0x34)

```

proddy, 2019-08-18:
ok, that's good news. the new Tx code in 1.9.x is basically @susisstrolch's "tx_mode 2" implementation from 1.8.x but super charged. I'm surprised it doesn't work. Need to think more about it tomorrow. Thanks for testing though.

higgers, 2019-08-18:
No problem, just let me know if you want me to test anything else.

I've been having a play around and have set up MQTT again and noticed that the thermostat temperature data on the info section of the telnet connection is wrong:

```
Thermostat stats:
  Thermostat: Bosch EasyControl CT200 (ProductID:203 Version:02.06)
  Set room temperature: 0.0 C
  Current room temperature: 0.0 C

```

The Version is correct, the version number I see on the Bosch Easy Connect app on the "Main firmware version" is 02.06.00.  I can't see a ProductID field anywhere on the app.  The current room temperature on the app is 24 degrees and the set point is 29 degrees but they're both 0.0C in the telnet info data.

Here's a set of the requested data telegrams:

```
Requesting scheduled EMS device data
Requesting type EasyStatusMessage(0x0A) from dest 0x18
Requesting type RCTime(0x06) from dest 0x18
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(00:08:01.112) Sending read of type 0x0A to 0x18:telegram: 8B 98 0A 00 20 (CRC=D4) #data=14
(00:08:01.214) Thermostat -> me, type 0x0A, telegram: 98 0B 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=9A) #data=26
<--- EasyStatusMessage(0x0A)
(00:08:02.829) Sending read of type 0x06 to 0x18:telegram: 8B 98 06 00 20 (CRC=E4)
(00:08:02.891) Thermostat -> me, type 0x06, telegram: 98 0B 06 00 13 08 17 12 1A 2C 06 F0 18 FF 00 (CRC=55) #data=11
<--- RCTime(0x06)
(00:08:04.021) Boiler -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:08:04.329) Boiler -> all, type 0x18, telegram: 88 00 18 00 36 01 0A 4C 00 00 21 00 40 00 EB 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=A9) #data=25
<--- UBAMonitorFast(0x18)
(00:08:04.570) Boiler -> all, type 0x34, telegram: 88 00 34 00 33 00 EB 80 00 80 00 00 01 00 01 C5 42 00 04 7D 00 (CRC=AA) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:08:05.171) Sending read of type 0x18 to 0x08:telegram: 8B 88 18 00 20 (CRC=1C) #data=20
(00:08:05.241) Boiler -> me, type 0x18, telegram: 88 0B 18 00 36 01 09 4C 00 00 21 00 40 00 EA 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=74) #data=25
<--- UBAMonitorFast(0x18)
(00:08:05.286) Boiler -> all, type 0x07, telegram: 88 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=DF) #data=15
(00:08:05.483) Sending read of type 0x19 to 0x08:telegram: 8B 88 19 00 20 (CRC=18) #data=20
(00:08:05.549) Boiler -> me, type 0x19, telegram: 88 0B 19 00 80 00 80 00 80 00 FF FF 00 00 00 0F 3E 09 3B 6C 00 00 00 07 76 29 00 0A C1 80 00 (CRC=0C) #data=27
<--- UBAMonitorSlow(0x19)
(00:08:05.707) Sending read of type 0x33 to 0x08:telegram: 8B 88 33 00 20 (CRC=B0) #data=20
(00:08:05.751) Boiler -> me, type 0x33, telegram: 88 0B 33 00 08 FF 33 00 00 00 00 02 46 00 FF FF (CRC=63) #data=12
<--- UBAParameterWW(0x33)
Publishing boiler data via MQTT
(00:08:06.014) Sending read of type 0x16 to 0x08:telegram: 8B 88 16 00 20 (CRC=24) #data=156
(00:08:06.063) Boiler -> me, type 0x16, telegram: 88 0B 16 00 FF 36 4C 00 00 F6 0A 01 03 64 0A 04 (CRC=2E) #data=12
<--- UBAParametersMessage(0x16)
(00:08:06.207) Sending read of type 0x14 to 0x08:telegram: 8B 88 14 00 20 (CRC=2C) #data=20
(00:08:06.240) Boiler -> me, type 0x14, telegram: 88 0B 14 00 34 87 D1 (CRC=0F) #data=3
<--- UBATotalUptimeMessage(0x14)
(00:08:06.753) Boiler -> all, type 0x07, telegram: 88 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=6F) #data=15
(00:08:08.330) Thermostat -> all, type 0x06, telegram: 98 00 06 00 13 08 17 12 1A 32 06 A5 (CRC=D6) #data=8
```

Should the thermostat current temperature and setpoint be somewhere amongst those telegrams?

higgers, 2019-08-19:
Actually, now that I've recovered from the excitement of seeing tx_mode 2 working I've remembered that I don't need to write values to the thermostat.

My goal is to be able to set the boiler flow temperature in order to achieve weather and load compensation.  I.e., when it's very cold outside or there's a large load on the heating system I want to be able to set a high flow temperature to compensate for this.

I have wireless thermostat radiator valve controllers on (nearly) all the radiators in the house that report the percentage open of the valves so my home automation system (openHAB) can track the load on the heating system.  What I need to do now is being able to set the flow temperature.  It would be nice to be able to read the return temperature too but that's not crucial, I could glue a temperature probe onto the return pipe to deal with that.

So, in summary, and I realise I'm way off the original topic of this issue, do you know of anyone who has successfully set the flow temperature on a "Bosch Condens 2500/Junkers Heatronics3 (ProductID:95 Version:25.02)"?  I've set up MQTT on my openHAB system to set the flow temp and I can see success messages in the MQTT log but I don't see the flow temperature being updated in the telnet session.  I haven't had time to investigate whether the MQTT send is fire and forget or if the ESP sends a success/fail confirmation back to openHAB and I'm going away for a week now.  Arrrrgh! :)

proddy, 2019-08-19:
MQTT is fire and forget. The set flow temp works for Buderus and it should be possible to make it work for Junkers too if someone wants to reverse engineer that other python HT3 project. There are a few people on the forum that understand this better than I do. Perhaps open a separate github issue and I'll connect with the right contributors.

As for reading the temps from your Easy thermostat, we do a request to telegram 0x0A and the current temperatures is byte 8+9 of the data block and the setpoint are bytes 10+11. There are 2-byte short integers multiplied by 10.

But in your case all the values are zero. No idea why I'm afraid.

```
Sending read of type 0x0A to 0x18:telegram: 8B 98 0A 00 20 (CRC=D4) #data=14
Thermostat -> me, type 0x0A, telegram: 98 0B 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=9A) #data=26
<--- EasyStatusMessage(0x0A)
```

enjoy your holiday!

proddy, 2019-08-19:
One more thing

> Actually, now that I've recovered from the excitement of seeing tx_mode 2 working I've remembered that I don't need to write values to the thermostat.

You need Tx working also for Read operations. When EMS-ESP boots up it sends out a few Tx commands to figure out what EMS devices are attached.

higgers, 2019-08-19:
What is this other python HT3 project? Sounds interesting.

One of the features of the Bosch Easy Control is that as you turn up the room set point it raises the boiler flow temperature. Hopefully, if I spend some time changing the room set point on the Easy Control and capture the log I'll be able to see which telegram the thermostat sends to the boiler to change the flow temp.

I'll open a feature request.

Thanks for all your help again! :)

proddy, 2019-08-19:
https://github.com/norberts1/hometop_HT3

proddy, 2019-08-19:
And I recall now there were issues with the Bosch Easy. See https://github.com/proddy/EMS-ESP/issues/45. Perhaps we can pick-up off that topic and work on it together.

DarthMob, 2019-08-22:
> and if you just take the [binary](https://github.com/proddy/EMS-ESP/releases/download/1.9.0b3/firmware_debug_1_9_0b3.bin) and upload from the command line as described [here](https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware) ? That will rule out any environment issues.

Hi @proddy , can you provide a compiled version for ESP12 (nodemcuv2) as well? maybe this will solve the issues on my end as well?

Thanks!

proddy, 2019-08-22:
@DarthMob sure, but the should be same right? If I look at PIO its the same eagle config. I'll add one to the Release page anyway. Remind me, what issues are you having again?

DarthMob, 2019-08-22:
ah ok haven't tried the binary yet - will do and report.
 
i have the same issue as higgers (reboots on Tx enabled - since 1.8)

proddy, 2019-08-22:
I’m actually away for a while so wont be able to upload the firmware until
mid next week. unless you want to have a go and trying to build yourself?

DarthMob, 2019-08-23:
no worries, last night i tested different builds and configurations.
in summary only one working for me without reboots (every ~3s) and SM50 Module beeing recognized is 1.7.0 

all other pre- or selfmade builds from 1.8 to 1.9.0b4 either go into reboot loop or with "listen mode on" do not recognize my thermostat and SM50 Module (also tried with the different tx_mode configurations).

sorry i cannot narrow it down currently what seems to be the issue...

proddy, 2019-08-27:
And if you take a prebuilt firmware of 1.9beta from the github releases page does it work ? That will rule out and environmental side effects. You could compare the sizes of the two .bin files.

DarthMob, 2019-08-28:
I testet also all the pre-built firmwares inkl. 1.9 beta (size is same as my own build) - no luck  :(

proddy, 2019-08-28:
that is strange. Try:
 - wipe the ESP again with `pio run -t erase`
 - compile the firmware using the compile flags -DCRASH -DFORCE_SERIAL (adding to `general_flags`)
 - upload the firmware via USB
 - fire it up, connect the serial monitor and see where it crashes
 - if there is a stack dump, copy&paste it into `scripts/stackdmp.txt` and use `analyze_stackdmp.py` to find the line where it crashed

bbqkees, 2019-08-28:
@DarthMob : Which specific ESP module is on you board?
The issues you describe look a bit like some problems I had a while ago with a bad batch of ESP's from Do-It.am.
Some firmware versions worked, but others didn't.
(Like 1.7.0 worked fine, 1.7.1 never showed the AP etc)

Try out a different ESP/dev board to see if its not a hardware problem.
Also lots of cheap ESP dev boards/clones have a relatively small LDO regulator to power the ESP. In theory this could give some issues at boot as well.

DarthMob, 2019-08-30:
Thats good advice @bbqkees - i actually used a Do-It,´.am oard for the tests.
never thought that could be the root cause since 1.7.0 is working.

will try on a different board and report back!

bbqkees, 2019-09-06:
So I tested 1.9.0 with a spare DO-IT.am module and although the upload is succesful, the web interface doesn't work at all or only extremely slow. And constant reboots.
The same build on the general Wemos clone with the Espressif logo on it works fine.

NOT OK:
[image]

OK:
[image]

proddy, 2019-09-06:
would be interesting in comparing both with -DFORCE_SERIAL which disables
the UART

mellbo, 2019-11-13:
@higgers  can you make for me one log (raw)  when central is in heating mode ? i want to see what CT200 send to boiler when heating is in progress ? 

Thanks in advance