emsesp/EMS-ESP #174: Junkers / Bosch CW400 //EMS2

lsw271, 2019-08-20:
Hi, I've got Junkers Ceraptur Smart (boiler) and CW400 termostat (EMS2). I saw on github that your software is compatibile with CW100 which is very similar with my termostat. Can you confirm that your software will be compatibile with my elements? Or you can add to the list my termostat if its necesarry? I asked bbqkees about board compatibility and he said that the "EMS thermostats are electrically compatible with the boards, it depends on the firmware whether all functions are implemented"  so i hope it will be ok but it will be great if you could confirm that. 
I have'nt got the board yet, first i would like to know if that will be useful for me.
Regards
Luke

proddy, 2019-08-20:
There is basic support for reading from a Junkers Cerapur boiler and also reading from a CW100 thermostat that has been confirmed to work. But I don't know how much additional effort, if any, is needed to for your boiler and CW400. Note writing settings to the CW100 (i.e. setting a temperature) hasn't been implemented but I believe others are looking into how to do it. The EMS2 protocols differ between Buderus and Junkers/HT3. It's open source after all so if you're comfortable writing code it would be wonderful if you contribute to testing and extend the library.

lsw271, 2019-08-20:
Unfortunatelly i'm not comfortable writing code :( privately i'm an analyst not programmer, I only know sql, pl/sql. I would like to participate in project development and I could be your tester. I'm a little worried about my boiler - I don't want to screw up it so first maybe we can check to read parameters from my elements. What do you think about that?

proddy, 2019-08-20:
it should be pretty safe as long as you use the write commands. There's an option in the web interface called "Listen Mode" which turns of all Tx so basically it's just listening to bus traffic and parsing the data. I'm pretty sure it'll work. But....we haven't implemented writing to the Junkers yet so if you're planning to set boiler temperatures or thermostat settings it won't work yet.

lsw271, 2019-08-20:
Now i'm mainly interested about to read parameters, but in the future i think it would be great to writting but first i need to change thinking about the possibility of spoiling the boiler.

Ok, so I need to order the board from bbqkees and i will be contact you later.

lsw271, 2019-08-21:
proddy, it's possible to read gas consumption with yours software, maybe there are some hidden parameters, which i can't see in my boiler/thermostat?

proddy, 2019-08-21:
not yet, we know how to, just haven't gotten around to implementing it yet. See https://github.com/proddy/EMS-ESP/issues/91

proddy, 2019-09-22:
writing to Junkers thermostats may need some work and testing from those that can look at the code and have the HT3 devices. We're using this issue https://github.com/proddy/EMS-ESP/issues/138 to track it.

closed 2019-09-22 by proddy