emsesp/EMS-ESP #175: 1.9.0b4 ESP crashes when MQTT base is changed via web interface

bbqkees, 2019-08-27:
While testing 1.9.0b4 I noticed the following:

When changing the MQTT Base in the web interface via Settings->MQTT Settings->Base, the ESP crashes after clicking Save (and confirming the save).
After a reboot you can see this setting has not been saved.
However, if I  change the MQTT base via Telnet with `set mqtt_base` everything is fine.

If I look with the `system` command it says: Last reset reason: Restart from terminal.

proddy, 2019-08-27:
I can't reproduce. Does it happen all the time?

SpaceTeddy, 2019-08-28:
Hi,
I tried version  1.9.0b**5** yesterday and in this version the mqtt base was saved. OK, it is not b4, but probably a hint to find it. 
I used the pre-build binary, becasue I can not build the 1.9.0 becasue of node/gulp trouble with macOS. I followed your description, but unfortunatly it stops always with errors.

kr,
Chris

bbqkees, 2019-08-28:
I did some more tests yesterday. Not sure about why it was only this specific setting, but I made some hardware changes and now it ran overnight without problems and changing the Base setting from the web interface does not cause crashes anymore.
The last two weeks I have been testing the minimum capacitor size for the buck input in parasitic mode.
Because only the Base setting caused crashes, I assumed it was something in the code.
But now I have upped the cap size a bit I can't reproduce it anymore. So this glitch was probably caused by 'underpowered' hardware.

closed 2019-08-28 by bbqkees

proddy, 2019-08-28:
@SpaceTeddy I build on MacOSX too. If you open a new github issue I can help you setup the build environment for OSX and also use this to fine tune the build firmware instructions.